arendst/Tasmota #12571: 9.5.0(minimal) flashed, trying to upgrade, to no avail

corbolais, 2021-07-05:
### PROBLEM DESCRIPTION
ascillato2's github issue handling
@ascillato2

I'd appreciate you not immediately closing all issues (by referring to a discord service)
I'm happy to chat on IRC
it is counterproductive forcing users to use a service enforcing registration
no one will find any discussion and solutions to problems behind discord walls
What is your issue with github issues anyway?

You're turning away users, crippling github issues, preventing bug fixes and others from finding solutions.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): esp8266
- [ X] Tasmota binary firmware version number used: 9.5.0_____
  - [ C] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _esptool____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
not applicable

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
not applicable

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
not applicable

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
not applicable

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
not applicable

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
not applicable

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
not applicable

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
@ascillato2

I'd appreciate you not immediately closing all issues (by referring to a discord service)
I'm happy to chat on IRC
it is counterproductive forcing users to use a service enforcing registration
no one will find any discussion and solutions to problems behind discord walls
What is your issue with github issues anyway?

You're turning away users, crippling github issues, preventing bug fixes and others from finding solutions.

**(Please, remember to close the issue when the problem has been addressed)**

corbolais, 2021-07-05:
You're censoring this?

You're pathetic.
OMG

ascillato2, 2021-07-05:
Hi,

Sorry that you felt that way. No one is censoring anything. There is no point on that.

This is not a Tasmota issue. If you want to discuss other matters we can chat.

GitHub issues are meant for Tasmota bug reports with the template complete. For Tasmota usage is way more better to address this in a interactive chat.

How to upgrade has been addressed more countless times in older issues and also explained in the docs. 

That is why, I kindly ask you to join the chat for support.

closed 2021-07-05 by ascillato2

ascillato2, 2021-07-05:
Please, use the gz.zip files for upgrade to have more space while upgrading.

If you need further assistance, please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

If you don't want to use Discord, there is no problem, you can open a new Discussion Thread for support.

This issue is being closed due to it is duplicated and has been addressed several times in older issues and in the docs. Sorry.

Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
See [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md).

ascillato2, 2021-07-06:
https://github.com/arendst/Tasmota/discussions/12579#discussioncomment-970413