arendst/Tasmota #12575: 9.5.0(minimal) flashed, trying to upgrade, to no avail

corbolais, 2021-07-06:
### PROBLEM DESCRIPTION
Please, if you would for once refrain from constantly closing issues one could reply and provide more information at hand.

Of course, I ***already did try to use a tasmota-minimal.gz, it is not working***. I already had other esp8266s with the same issue, the only way to fix it was reflashing via uart. This ist however not a solution that is always feasible. And as this issue should never arise at all in the first place, I open bug reports about it.

Please, understand that constantly and immediately closing issues is frustrating for users who give their time to write an issue in the first place. They are motivated and want to help improve tasmota. Otherwise why would they bother? So just closing them away even prives them of a way to further communicate. And that is what really is a bad thing.

--

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ X Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ X Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ X Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ X Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ X]X Device used (e.g., Sonoff Basic): _esp8266____
- [ X]X Tasmota binary firmware version number used: ___9.5.0__
  - [ X] Pre-compiled
  - [ X] Self-compiled
- [ X] Flashing tools used: ____esptool_
- [ X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

not again
```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

not again
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

not again
```

### TO REPRODUCE
_Steps to reproduce the behavior:_

not again
### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

not again
### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

not again
### ADDITIONAL CONTEXT
_Add any other context about the problem here._

not again
**(Please, remember to close the issue when the problem has been addressed)**

This is getting tedious. Please  stop closing these bugs reports . Users need to be able to reply. They also need to find the issue and possible outcomes.

sfromis, 2021-07-06:
You should just not raise issues when you are not not able to make something work. That's not the way of doing it, wastes time for everyone. Better ask questions in Discord or *maybe* in Discussions.

corbolais, 2021-07-06:
Oh, I notice with tasmota mods like to assume failures on behalf of the user, PEBKAC. Fair enough. So, while this might be the case, it does not seem so to me, in this case. Anyway, it *is* an issue with tasmota upgrade until proven otherwise.

Thanks for not closing *this* issue, for a change.

It can be continued as a Discussion, I'm fine with that. It is helpful to keep the actual issue open so far and once it is solved it can be closed, though. 

Meanwhile, we can discuss it in Discussions if you prefer.

ascillato2, 2021-07-06:
Closing in favour of https://github.com/arendst/Tasmota/discussions/12579

closed 2021-07-06 by ascillato2

ascillato2, 2021-07-06:
https://github.com/arendst/Tasmota/discussions/12579#discussioncomment-970413