arendst/Tasmota #12577: Problem in KNX telegram <- the fourth value is missing

nitinuae, 2021-07-06:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

Hello
I noticed that in 9.5.0 (dont know about previous version or its because DPT9 to DPT14 change) I cannot send more than 3 values in one telegram.

I have setup KNX to send multiple values (PowerFactor, Voltage, Watt etc). But everything seems to be truncated after 3 values. Eg - If I have asked the module to update 7/0/1, 7/0/2, 7/0/3, 7/0/4, the telegram is sent correctly in console but my Gira server sees only 3 values. the fourth one doesnot show on bus.

Is some limitation of telegram sizes is being encountered?

I repeated by swapping ie, 7/0/1 7/0/2 7/0/4 7/0/1 <- here 7/0/1 goes missing on bus.
Thank You
### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): Gosund A23
- [ x] Tasmota binary firmware version number used: 9.5.0
  - [ x] Pre-compiled
  - [ ] Self-compiled
- [ x] Flashing tools used:Tasmotizer_
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
- [ ] 
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Can be reproduced by swapping the slots

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

nitinuae, 2021-07-06:
Update - Removing the "communication enhancement" solved this issue. 
So is my ABB IP/R S 2 creating this issue?

ascillato, 2021-07-06:
Yes, seems that the buffer of your receiving device got full and can not process more telegrams.
The "communication enhancement" is just repeating 3 times each telegram. So seems that the telegram buffer of your ABB device is 9 per second? 3 telegrams by 3 times.

Now you have found a solution. In the case of sending periodical data, you can avoid the "communication enhancement" but in the case of cheap Wi-Fi routers and the need of fast response to turn on and off a plug, the enhancement is needed because some of these routers don't retransmits all multicast messages

ascillato2, 2021-07-06:
I have found that your device has a configurable buffer of 1000, so, the limitation should be elsewhere. May be a misconfiguration in the ABB IP/R S 2 or in the router?

ascillato2, 2021-07-06:
If you want, please, can yoy try the following version that has a small delay between telegrams so as to not overrun your network?

[tasmota-knx.bin.gz](https://github.com/arendst/Tasmota/files/6770048/tasmota-knx.bin.gz)

Thanks.

nitinuae, 2021-07-06:
Thank you Adrian
The patch is working.
My IP router has NEVER missed any telegram (I have around 80+ KNX devices, both KNX TP and around 10 Tasmota)

This issue was discovered when for the first time I set teleperiod as 10 AND this is the first tasmota sensor on my network.

Regards
Nitin

ascillato, 2021-07-06:
> The patch is working.

Great. Thanks for testing.

closed 2021-07-06 by arendst