arendst/Tasmota #12611: Random heap corruption in tasmota32 leading to a reboot

augustozanellato, 2021-07-09:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
After random amounts of time my esp32 based tasmota devices reboots, I tried adding another tasmota device used as a wireless uart bridge in order to get information from the esp exception logger and it looks like the issue is due to sporadic heap corruptions

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff Dual R3 but should happen with every esp32 build
- [x] Tasmota binary firmware version number used: 9.5.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: ESP-flasher
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
23:07:29.872 MQT: stat/tasmota_266774/RESULT = {"NAME":"Sonoff Dual R3","GPIO":[32,0,0,0,0,0,0,0,0,576,225,0,0,0,0,0,0,0,0,0,0,7296,7328,224,0,0,0,0,160,161,0,0,0,0,0,0],"FLAG":0,"BASE":1}
23:07:30.105 MQT: stat/tasmota_266774/RESULT = {"Module":{"0":"Sonoff Dual R3"}}
23:07:30.368 MQT: stat/tasmota_266774/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"576":"LedLink_i"},"GPIO14":{"225":"Relay2"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"7296":"CSE7761 Tx"},"GPIO26":{"7328":"CSE7761 Rx"},"GPIO27":{"224":"Relay1"},"GPIO28":{"0":"None"},"GPIO29":{"0":"None"},"GPIO30":{"0":"None"},"GPIO31":{"0":"None"},"GPIO32":{"160":"Switch1"},"GPIO33":{"161":"Switch2"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
```
- [x] Provide the output of this command: `Status 0`:
```lua
23:08:41.637 MQT: stat/tasmota_266774/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota",""],"Topic":"tasmota_266774","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
23:08:41.690 MQT: stat/tasmota_266774/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"1T00:24:35","StartupUTC":"2021-07-08T21:44:06","Sleep":50,"CfgHolder":4617,"BootCount":22,"BCResetTime":"2021-06-10T18:04:35","SaveCount":42}}
23:08:41.726 MQT: stat/tasmota_266774/STATUS2 = {"StatusFWR":{"Version":"9.5.0(tasmota)","BuildDateTime":"2021-06-17T08:29:40","Core":"1_0_6","SDK":"v3.3.5-1-g85c43024c","CpuFrequency":80,"Hardware":"ESP32-D0WD-V3","CR":"419/699"}}
23:08:41.752 MQT: stat/tasmota_266774/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["TISCALI_9E0B9D",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000A8009","2805C8000100060000005A0A000000000000","00000280","00006000","00000000"]}}
23:08:41.798 MQT: stat/tasmota_266774/STATUS4 = {"StatusMEM":{"ProgramSize":1457,"Free":1856,"Heap":211,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":2,"Features":["00000809","9FDAC787","00152001","B7F7BFCF","05DA9BC0","60360DC7","00084052","20200000","0000002C"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,17,20,21,24,26,27,29,34,50,52","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,87"}}
23:08:41.848 MQT: stat/tasmota_266774/STATUS5 = {"StatusNET":{"Hostname":"tasmota_266774-1908","IPAddress":"*snip*","Gateway":"*snip*","Subnetmask":"255.255.255.0","DNSServer":"*snip*","Mac":"*snip*","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
23:08:41.878 MQT: stat/tasmota_266774/STATUS6 = {"StatusMQT":{"MqttHost":"*snip*","MqttPort":1883,"MqttClientMask":"*snip*","MqttClient":"*snip*","MqttUser":"*snip*","MqttCount":2,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
23:08:41.904 MQT: stat/tasmota_266774/STATUS7 = {"StatusTIM":{"UTC":"2021-07-09T22:08:41","Local":"2021-07-09T23:08:41","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"04:57","Sunset":"20:53"}}
23:08:41.930 MQT: stat/tasmota_266774/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
23:08:41.958 MQT: stat/tasmota_266774/STATUS10 = {"StatusSNS":{"Time":"2021-07-09T23:08:41","Switch1":"ON","Switch2":"ON","ENERGY":{"TotalStartTime":"2021-06-10T18:04:35","Total":0.620,"Yesterday":0.041,"Today":0.000,"Power":[0,0],"ApparentPower":[0,0],"ReactivePower":[0,0],"Factor":[0.00,0.00],"Voltage":0,"Current":[0.000,0.000]},"ESP32":{"Temperature":46.7},"TempUnit":"C"}}
23:08:42.005 MQT: stat/tasmota_266774/STATUS11 = {"StatusSTS":{"Time":"2021-07-09T23:08:41","Uptime":"1T00:24:35","UptimeSec":87875,"Heap":211,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":2,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"*snip*","BSSId":"*snip*","Channel":1,"Mode":"11n","RSSI":100,"Signal":-46,"LinkCount":1,"Downtime":"0T00:00:07"}}}
```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log: 
**logs from the other tasmota device which was collecting the uart data from the affected one**
```lua
22:44:06.459 RSL: RESULT = {"SerialReceived":"CORRUPT HEAP: Bad head at 0x3ffd4c30. Expected 0xabba1234 got 0x3ffd4c64\r\nabort() was called at PC 0x40089001 on core 1\r\n\r\nELF file SHA256: 0000000000000000\r\n\r\nBacktrace: 0x4008ae54:0x3ffb16d0 0x4008b0d1:0x3ffb16f0 0x40089001:0x3ffb1710 0x4008912d:0x3ffb1740 0x40161a73:0x3ffb1760 0x4015da4d:0x3ffb1a20 0x4015d9bd:0x3ffb1a70 0x40090371:0x3ffb1aa0 0x40083abe:0x3ffb1ac0 0x40088ef9:0x3ffb1ae0 0x4000bec7:0x3ffb1b00 0x40139c5e:0x3ffb1b20 0x40139cba:0x3ffb1b90 0x400de235:0x3ffb1bc0 0x400de31b:0x3ffb1c00 0x400de852:0x3ffb1"}
22:44:06.507 RSL: RESULT = {"SerialReceived":"c50 0x40110321:0x3ffb1c80 0x40110353:0x3ffb1cb0 0x400f6d63:0x3ffb1cd0 0x4010891f:0x3ffb1cf0 0x40108c10:0x3ffb1d60 0x401c08b1:0x3ffb1df0 0x4011c362:0x3ffb1e10 0x4011c659:0x3ffb1e30 0x4011c3aa:0x3ffb1e60 0x4011c52d:0x3ffb1eb0 0x400ddd73:0x3ffb1f00 0x400fd38b:0x3ffb1f20 0x400eadff:0x3ffb1f50 0x40115422:0x3ffb1f70 0x401154d3:0x3ffb1f90 0x40141fe5:0x3ffb1fb0 0x4008cb6a:0x3ffb1fd0\r\n\r\nRebooting...\r\nets Jul 29 2019 12:21:46\r\n\r\nrst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)\r\nconfigsip: 0, SPIWP:0xee\r\nclk_drv:0x00,q"}
22:44:06.664 RSL: RESULT = {"SerialReceived":"_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00\r\nmode:DIO, clock div:2\r\nload:0x3fff0018,len:4\r\nload:0x3fff001c,len:1044\r\nload:0x40078000,len:10124\r\nload:0x40080400,len:5828\r\nentry 0x400806a8\r\n"}
22:44:07.543 RSL: RESULT = {"SerialReceived":"\r\n00:00:00.002 HDW: ESP32-D0WD-V3\r\n00:00:00.057 UFS: FlashFS mounted with 304 kB free\r\n00:00:00.144 CFG: Loaded from File, Count 35\r\n00:00:00.168 QPC: Count 1\r\n00:00:00.192 BRY: Berry initialized, RAM used=1760\r\n00:00:00.205 BRY: no 'preinit.be'\r\n00:00:00.210 Project tasmota Tasmota Version 9.5.0(tasmota)-1_0_6(2021-06-17T08:29:40)\r\n00:00:00.226 BRY: no 'autoexec.be'\r\n"}
```

### TO REPRODUCE
N/D

### EXPECTED BEHAVIOUR
Device stays booted as long as it has power without random resets

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2021-07-10:
Can you share your map file? I would like to see what is the stack trace

augustozanellato, 2021-07-10:
> Can you share your map file? I would like to see what is the stack trace

I did not compile that binary, maybe it was the precompiled tasmota32.bin version 9.5.0

ascillato, 2021-07-19:
Hi, please, can you try latest Tasmota version from http://ota.tasmota.com/tasmota32/tasmota32.bin

You can do an OTA update to the address http://ota.tasmota.com/tasmota32/tasmota32.bin

Thanks.

augustozanellato, 2021-07-19:
> Hi, please, can you try latest Tasmota version from http://ota.tasmota.com/tasmota32/tasmota32.bin
> 
> You can do an OTA update to the address http://ota.tasmota.com/tasmota32/tasmota32.bin
> 
> Thanks.

Will try that and keep you updated!

augustozanellato, 2021-07-26:
@ascillato looks like it never happened in a week so looks like it's fixed now

ascillato2, 2021-07-26:
Great to know that it is working fine for you.

Thanks for reporting!

closed 2021-07-26 by ascillato2