arendst/Tasmota #12630: On ESP32, global temperature breaks aHumidity calculation with SGP30

sfromis, 2021-07-12:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

Seems that having `ESP32.Temperature` breaks the SGP30 sensor driver results for absolute humidity (aHumidity). Since `xsns_87_esp32_sensors.ino` use the CpuTemperature function in `support_esp.ino`, which again calls the ConvertTemp function in `support.ino`, which saves temperature in `TasmotaGlobal.temperature_celsius` (as other temperature sensors will also do), and `xsns_21_sgp30.ino` is using both `TasmotaGlobal.temperature_celsius` and `TasmotaGlobal.humidity` to calculate absolute humidity for the Adafruit_SGP30 library (and also report aHumidity in the JSON), the ESP32 internal temperature, which is much different from ambient temperature, will badly influence the values used in the SGP30 driver.

This explains me seeing wildly jumping aHumidity values (like between 13 and 71), and may help explain odd variations in the eCO2 and TVOC values too (depending on how the humidity/temperature may influence that calculation). I'd say that it does not make much sense to have the ESP32 CPU temperature update the global temperature, which should only be from sensors expected to be for ambient temperature. Ideally, a "trusted" temp sensor could be designated, in case of a multisensor board.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP32 dev board, similar to NodeMCU
- [x] Tasmota binary firmware version number used: 9.5.0.2
  - [x] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: esptool.py with recommended options
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

15:42:12.280 MQT: xxxxxxxx/stat/TEMPLATE = {"NAME":"Merged","GPIO":[160,1,1,1,1,1,1,1,1,1,1,1888,1,1,1856,1312,0,640,608,1216,0,1,1,1,0,0,0,0,1,1,1,1,7584,0,0,7585],"FLAG":0,"BASE":1}
15:42:12.525 MQT: xxxxxxxx/stat/MODULE = {"Module":{"0":"Merged"}}
15:42:12.790 MQT: xxxxxxxx/stat/GPIO = {"GPIO0":{"160":"Switch1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"1888":"SR04 Ech/RX"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"1856":"SR04 Tri/TX"},"GPIO19":{"1312":"DS18x20"},"GPIO20":{"0":"None"},"GPIO21":{"640":"I2C SDA1"},"GPIO22":{"608":"I2C SCL1"},"GPIO23":{"1216":"AM2301"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO28":{"0":"None"},"GPIO29":{"0":"None"},"GPIO30":{"0":"None"},"GPIO31":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"7584":"HallEffect1"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"7585":"HallEffect2"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
(no rules)

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

15:43:22.269 MQT: xxxxxxxx/stat/STATUS = {"Status":{"Module":0,"DeviceName":"ESP32 test 2","FriendlyName":["Unit j32sens"],"Topic":"xxxxxxxx","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":0,"LedMask":"FFFF","SaveData":1,"SaveState":0,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
15:43:22.286 MQT: xxxxxxxx/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://xxxxxxxx:8081/otaserver+j32sens+tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"1T18:52:21","StartupUTC":"2021-07-10T18:51:01","Sleep":50,"CfgHolder":4617,"BootCount":67,"BCResetTime":"2021-05-16T16:35:35","SaveCount":174}}
15:43:22.297 MQT: xxxxxxxx/stat/STATUS2 = {"StatusFWR":{"Version":"9.5.0.2(j32sens)","BuildDateTime":"2021-07-10T18:37:19","Core":"1_0_7_1","SDK":"v3.3.5-138-g714f27f56","CpuFrequency":80,"Hardware":"ESP32-D0WDQ6","CR":"425/699"}}
15:43:22.313 MQT: xxxxxxxx/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["xxxxxxxx",""],"TelePeriod":300,"Resolution":"A98180F0","SetOption":["00008018","2805C8000200350000005A0A000000000000","010080CC","00046002","00000000"]}}
15:43:22.336 MQT: xxxxxxxx/stat/STATUS4 = {"StatusMEM":{"ProgramSize":1257,"Free":1984,"Heap":182,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":3,"Features":["00000809","9FAA858F","000127A1","FFFFBFCF","05DA9BC8","307E0C47","020A70FA","20200000","0000000D"],"Drivers":"1,2,3,4,5,8,9,10,11,12,13,14,17,24,29,34,38,41,50,52","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,24,26,27,28,31,32,34,37,39,40,42,43,45,51,52,55,56,57,58,59,60,63,64,65,66,67,71,72,74,77,85,87"}}
15:43:22.351 MQT: xxxxxxxx/stat/STATUS5 = {"StatusNET":{"Hostname":"xxxxxxxx","IPAddress":"xxxxxxxx","Gateway":"xxxxxxxx","Subnetmask":"255.255.255.0","DNSServer":"xxxxxxxx","Mac":"xxxxxxxx","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
15:43:22.362 MQT: xxxxxxxx/stat/STATUS6 = {"StatusMQT":{"MqttHost":"xxxxxxxx","MqttPort":1883,"MqttClientMask":"xxxxxxxxc","MqttClient":"xxxxxxxxc","MqttUser":"","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
15:43:22.376 MQT: xxxxxxxx/stat/STATUS7 = {"StatusTIM":{"UTC":"2021-07-12T13:43:22","Local":"2021-07-12T15:43:22","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":99,"Sunrise":"04:52","Sunset":"21:42"}}
15:43:22.434 MQT: xxxxxxxx/stat/STATUS10 = {"StatusSNS":{"Time":"2021-07-12T15:43:22.381+02:00","Switch1":"ON","DS18B20":{"Id":"030B97942847","Temperature":25.00},"AM2301":{"Temperature":25.40,"Humidity":66.90,"DewPoint":18.79},"BME280":{"Temperature":25.63,"Humidity":57.99,"DewPoint":16.73,"Pressure":1014.28,"SeaPressure":1016.06},"BH1750":{"Illuminance":85},"SGP30":{"eCO2":401,"TVOC":1,"aHumidity":15.2476},"SR04":{"Distance":19.579},"HDC1080":{"Temperature":26.15,"Humidity":67.10,"DewPoint":19.55},"MCP9808":{"Temperature":25.62},"VL53L1X":{"Distance":27},"ESP32":{"Temperature":63.33,"HallEffect":0},"PressureUnit":"hPa","TempUnit":"C"}}
15:43:22.452 MQT: xxxxxxxx/stat/STATUS11 = {"StatusSTS":{"Time":"2021-07-12T15:43:22.440+02:00","Uptime":"1T18:52:21","UptimeSec":154341,"Heap":184,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":30,"MqttCount":1,"Wifi":{"AP":1,"SSId":"xxxxxxxx","BSSId":"xxxxxxxx","Channel":11,"Mode":"11n","RSSI":88,"Signal":-56,"LinkCount":1,"Downtime":"0T00:00:12"}}}
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

Nothing interesting. Multisensor board, long lines:
xxxxxxxx/tele/SENSOR = {"Time":"2021-07-09T18:49:23.682+02:00","Switch1":"ON","DS18B20":{"Id":"030B97942847","Temperature":24.94},"AM2301":{"Temperature":26.00,"Humidity":56.20,"DewPoint":16.59},"BME280":{"Temperature":25.67,"Humidity":48.95,"DewPoint":14.12,"Pressure":1016.29,"SeaPressure":1018.11},"BH1750":{"Illuminance":39},"SGP30":{"eCO2":422,"TVOC":15,"aHumidity":13.1246},"SR04":{"Distance":58.544},"MCP9808":{"Temperature":25.56},"ESP32":{"Temperature":62.78,"HallEffect":0},"PressureUnit":"hPa","TempUnit":"C"}
xxxxxxxx/tele/SENSOR = {"Time":"2021-07-09T18:44:23.294+02:00","Switch1":"ON","DS18B20":{"Id":"030B97942847","Temperature":24.94},"AM2301":{"Temperature":26.00,"Humidity":56.50,"DewPoint":16.67},"BME280":{"Temperature":25.64,"Humidity":49.95,"DewPoint":14.41,"Pressure":1016.41,"SeaPressure":1018.24},"BH1750":{"Illuminance":13},"SGP30":{"eCO2":401,"TVOC":1,"aHumidity":71.3359},"SR04":{"Distance":420.912},"MCP9808":{"Temperature":25.69},"ESP32":{"Temperature":62.78,"HallEffect":0},"PressureUnit":"hPa","TempUnit":"C"}
[... cut ...]
Only external temperature sensor readings used to calculate absolute humidity. ESP32 internal temperature sensor should not update global temperature.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

Nothing there over what's available in MQTT+JSON

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

Started a long time ago, when the ESP32 internal sensors became treated as "just another" sensor device. Did not notice it at first, as I'm not looking a lot at that specific sensor on a test board.

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2021-07-12:
I understand the issue. A certain esp32 based sonoff loves the esp32 temp sensor though.

I'll add code to not use the esp32 temp sensor as global sensor if another temp sensor is present.

sfromis, 2021-07-12:
Sounds good.

ascillato2, 2021-07-13:
Fixed by Theo. Thanks!

closed 2021-07-13 by ascillato2

sfromis, 2021-07-13:
Great. Upgraded my sensor test boards to a build based on latest source. Will report if I get more bad values.

sfromis, 2021-07-14:
Upgraded the test boards to 9.5.0.3 to include the most recent ESP32 temperature sensor improvements, putting the internal sensor last in the expanded sensor range.