arendst/Tasmota #12635: With XTP2046 touchscreen only reads zeroes after core32 1.0.7.1  1.0.7.0 works fine.

mikep1998, 2021-07-12:
### PROBLEM DESCRIPTION

After this change
[core32]
platform_packages           = framework-arduinoespressif32 @ https://github.com/tasmota/arduino-esp32/releases/download/1.0.7.1/tasmota-arduinoespressif32-release_v3.3.5.tar.gz

The XTP2046 touchscreen only reads zeros from the chip. 

With 1.0.7.0 the XTP2046 touchscreen works correctly

I have traced it down to the SPI tansfer16 calls but can not find the exact issue. 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): _____
- [x] Tasmota binary firmware version number used: 9.5_____
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: _____
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```l12:59:22.385 MQT: stat/messagebox/RESULT = {"NAME":"messagebox","GPIO":[6210,1,800,1,1024,1,0,0,1,0,0,768,7264,992,736,672,0,640,608,704,0,0,0,0,0,0,0,0,0,0,0,1,0,0,0,4704],"FLAG":0,"BASE":1}
12:59:22.592 MQT: stat/messagebox/RESULT = {"Module":{"0":"messagebox"}}
12:59:22.807 MQT: stat/messagebox/RESULT = {"GPIO0":{"6210":"Option A3"},"GPIO1":{"0":"None"},"GPIO2":{"800":"SPI DC1"},"GPIO3":{"0":"None"},"GPIO4":{"1024":"Display Rst"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"768":"SPI CS1"},"GPIO16":{"7264":"XPT2046 CS"},"GPIO17":{"992":"Backlight"},"GPIO18":{"736":"SPI CLK1"},"GPIO19":{"672":"SPI MISO1"},"GPIO20":{"0":"None"},"GPIO21":{"640":"I2C SDA1"},"GPIO22":{"608":"I2C SCL1"},"GPIO23":{"704":"SPI MOSI1"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO28":{"0":"None"},"GPIO29":{"0":"None"},"GPIO30":{"0":"None"},"GPIO31":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"4704":"ADC Input1"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
13:00:49.022 CMD: Status 0
13:00:49.030 MQT: stat/messagebox/STATUS = {"Status":{"Module":0,"DeviceName":"messagebox","FriendlyName":["messagebox"],"Topic":"messagebox","ButtonTopic":"0","Power":1,"PowerOnState":0,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":0,"SwitchTopic":"messagebox_switch","SwitchMode":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
13:00:49.036 MQT: stat/messagebox/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32-minimal.bin","RestartReason":"Vbat power on reset","Uptime":"0T00:25:49","StartupUTC":"2021-07-12T19:35:00","Sleep":250,"CfgHolder":9889,"BootCount":3,"BCResetTime":"2021-07-12T12:22:43","SaveCount":21}}
13:00:49.041 MQT: stat/messagebox/STATUS2 = {"StatusFWR":{"Version":"9.5.0.1(MessageBox)","BuildDateTime":"2021-07-12T12:33:51","Core":"1_0_7_1","SDK":"v3.3.5-138-g714f27f56","CpuFrequency":240,"Hardware":"ESP32-D0WDQ6","CR":"458/699"}}
13:00:49.046 MQT: stat/messagebox/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["mike1.1.3","mike1.1.1"],"TelePeriod":30,"Resolution":"558180C0","SetOption":["000A0108","2805C8000100060000005A00000000000000","000002C0","00006000","00000000"]}}
13:00:49.055 MQT: stat/messagebox/STATUS4 = {"StatusMEM":{"ProgramSize":1336,"Free":1984,"Heap":157,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":2,"Features":["00000809","0F60858E","00000001","00000443","00000000","00000000","00000000","00200000","00000000"],"Drivers":"1,2,4,9,10,12,13,50,54,55,91","Sensors":"1,2,5,9,87,92,94"}}
13:00:49.063 MQT: stat/messagebox/STATUS5 = {"StatusNET":{"Hostname":"messagebox","IPAddress":"192.168.1.245","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"30:AE:A4:03:22:20","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
13:00:49.067 MQT: stat/messagebox/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.250","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_032220","MqttUser":"pi","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
13:00:49.071 MQT: stat/messagebox/STATUS7 = {"StatusTIM":{"UTC":"2021-07-12T20:00:49","Local":"2021-07-12T13:00:49","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"-07:00","Sunrise":"05:58","Sunset":"20:31"}}
13:00:49.113 MQT: stat/messagebox/STATUS10 = {"StatusSNS":{"Time":"2021-07-12T13:00:49","ANALOG":{"A1":2889},"BME280":{"Temperature":72.6,"Humidity":63.6,"DewPoint":59.6,"Pressure":998.7},"ESP32":{"Temperature":117.0},"messagebox":{"currentTemp":73.000000,"currentHumidity":64.000000,"messageCount":0,"deviceLocation":"Master Bath","motion":0},"PressureUnit":"hPa","TempUnit":"F"}}
13:00:49.120 MQT: stat/messagebox/STATUS11 = {"StatusSTS":{"Time":"2021-07-12T13:00:49","Uptime":"0T00:25:49","UptimeSec":1549,"Heap":157,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":106,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"mike1.1.3","BSSId":"44:94:FC:35:CC:8B","Channel":11,"Mode":"11n","RSSI":88,"Signal":-56,"LinkCount":1,"Downtime":"0T00:00:12"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2021-07-13:
Interesting. The only change from 1.0.7 to 1.0.7.1 is in IDF config `sdkconfig`
In sdkconfig that entrys are changed/added to build the needed Arduino libs
```
CONFIG_SPIRAM_BOOT_INIT=y
CONFIG_SPIRAM_IGNORE_NOTFOUND=y
CONFIG_SPIRAM_MEMTEST=y
CONFIG_SPIRAM_USE_MALLOC=y
```
This should only affect PSRAM. Does the board you use have PSRAM?
If yes please compile with adding to `build_flags` in `platformio_override.ini`
So the entry has to be
```
build_flags             = ${env:tasmota32_base.build_flags} -DBOARD_HAS_PSRAM -mfix-esp32-psram-cache-issue -lc-psram-workaround -lm-psram-workaround
```

Jason2866, 2021-07-13:
Please use latest development version to compile

mikep1998, 2021-07-13:
When I use the latest code from development with 1.0.7.1 
I get this error about psram 'bailing out'

``` Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:160
ho 0 tail 12 room 4
load:0x40078000,len:10492
load:0x40080400,len:5664
entry 0x40080678
E (534) psram: PSRAM ID read error: 0xffffffff
E (534) spiram: SPI RAM enabled but initialization failed. Bailing out.
ÿ
00:00:00.002 HDW: ESP32-D0WDQ6
00:00:00.023 UFS: FlashFS mounted with 48 kB free
00:00:00.054 CFG: Loaded from File, Count 57
00:00:00.060 QPC: Count 1
00:00:00.062 SPI: Hardware using GPIO18(CLK), GPIO23(MOSI) and GPIO19(MISO)
00:00:00.069 DSP: Flash descriptor used

```
the hardware I'm using is 'DOIT DEV KIT VER 1 30 pin  clones. the boot output is HDW: ESP32-D0WDQ6
I don't know if it has PSRAM. But by the error message I would say no.

When I use the exact same code to build but I change 1.0.7.1 to 1.0.7 it compiles and the touchscreen works.
```

[core32]
platform                    = espressif32 @ 3.3.0
platform_packages           = framework-arduinoespressif32 @ https://github.com/tasmota/arduino-esp32/releases/download/1.0.7.1/tasmota-arduinoespressif32-release_v3.3.5.tar.gz
                              platformio/tool-mklittlefs @ ~1.203.200522
build_unflags               = ${esp32_defaults.build_unflags}
build_flags                 = ${esp32_defaults.build_flags}
```

When I diff the changes from 1.0.7.1 and 1.0.7 in file \arduino-esp32-1.0.7.1\libraries\SPI\src\SPI.cpp  I see two changes

```
#if CONFIG_IDF_TARGET_ESP32S2
        _sck = (_spi_num == FSPI) ? SCK : -1;
        _miso = (_spi_num == FSPI) ? MISO : -1;
        _mosi = (_spi_num == FSPI) ? MOSI : -1;
        _ss = (_spi_num == FSPI) ? SS : -1;
#elif CONFIG_IDF_TARGET_ESP32C3
        _sck = SCK;
        _miso = MISO;
        _mosi = MOSI;
        _ss = SS;
#else
        _sck = (_spi_num == VSPI) ? SCK : 14;
        _miso = (_spi_num == VSPI) ? MISO : 12;
        _mosi = (_spi_num == VSPI) ? MOSI : 13;
        _ss = (_spi_num == VSPI) ? SS : 15;
#endif
```

and 

```
#if CONFIG_IDF_TARGET_ESP32
SPIClass SPI(VSPI);
#else
SPIClass SPI(FSPI);
#endif
```

I would say it's in these changes that are causing the failure because this is the file I have worked down too. The SPI is reading zeros from the chip.

Some debug output when it fails
```
z=4095  ::  z1=0,  z2=0, csPin: 16, _pspi._miso: 19, _pspi._mosi: 23, _pspi._sck: 18, _pspi._div: 4984833,  _pspi._freq: 2000000, _pspi._use_hw_ss: 0, _pspi._spi_num: 3,  
p=4095,  0,0  0,0  0,0    0,0
```
All the pins in SPI are the correct pins. But the data values are all zeros.

Here is when 1.0.7 is used 
```
z=2950  ::  z1=1607,  z2=2752, csPin: 16, _pspi._miso: 19, _pspi._mosi: 23, _pspi._sck: 18, _pspi._div: 4984833,  _pspi._freq: 2000000, _pspi._use_hw_ss: 0, _pspi._spi_num: 3,  
p=2950,  3192,2563  3177,2571  3171,2550    3174,2567
```

Same pins but valid data read.

Let me know what to do next.

Jason2866, 2021-07-13:
It seems your platformio setup is screwed. The changes you have are not in file 1.0.7.1\libraries\SPI\src\SPI.cpp. Just checked. Download the 1.0.7.1 tarball and look yourself in the file.
To fix your setup. Delete the whole hidden folder `.platformio`. Platformio will rebuild

Jason2866, 2021-07-13:
Your board has no PSRAM. Core 1.0.7.1 did work as intended.
```
E (534) psram: PSRAM ID read error: 0xffffffff
E (534) spiram: SPI RAM enabled but initialization failed. Bailing out.
```
With core 1.0.7 you get bootloops when enabling PSRAM and there is none found

mikep1998, 2021-07-13:
Interesting... I did download the code from https://github.com/tasmota/arduino-esp32/releases  but I downloaded 1.0.7.1 source code (zip).  https://github.com/tasmota/arduino-esp32/archive/refs/tags/1.0.7.1.zip  this has the changes in it. So does https://github.com/tasmota/arduino-esp32/archive/refs/tags/1.0.7.1.tar.gz. 
And https://github.com/tasmota/arduino-esp32/releases/download/1.0.7.1/tasmota-arduinoespressif32-release_v3.3.5.tar.gz does not have the changes.  Why is that?

Jason2866, 2021-07-13:
Ahh, release error. Accidently released on `master` branch. It isnt. It is branch `release/v1.0`
So the wrong source files.

mikep1998, 2021-07-13:
I am using the configured settings. I just downloaded the link to see the source files. did not know where to find them on my windows system.

In .platformio I do not see the changes. I try deleting .platformio but PIO will not rebuild all that is needed lots of compile errors and the loss my projects when trying to add libraries back. 
Good thing I old renamed .platformio so I restore it.

So I was looking at the wrong source files. No changes in SPI.cpp then. But if I build with 1.0.7 it works and 1.0.7.1 does not work.

What do you suggest next?

Jason2866, 2021-07-13:
If platformio does not correctly rebuilds the folder `.platformio` something is badly broken.
Which OS do you use? If it is Windows, you can use the [Portable VSC/Platformio build](https://github.com/Jason2866/Portable_VSC_PlatformIO)
Have you tried my suggestion to use precompiled Tasmota32-lvgl?
If the touch is working, it is your build environment.

mikep1998, 2021-07-13:
Windows 10 OS. 

I will try  to use precompiled Tasmota32-lvgl. and the Portable VSC/Platformio build.

Thanks

mikep1998, 2021-07-13:
@Jason2866 I do not see a Tasmota32-lvgl build. I do see a Tasmota32-display build does that have lvgl in it?

sfromis, 2021-07-13:
You can download tasmota32-lvgl.bin here:
https://github.com/arendst/Tasmota/tree/firmware/firmware/tasmota32

mikep1998, 2021-07-13:
Well I have installed the tasmota32-lvgl.bin from the link above.
Then uploaded to tasmota file system display.ini that works with my hardware.
Then copy and pasted the berry script from Adventures with LVGL 
nothing on the display. Here is the output from the serial log:
```
22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:160
ho 0 tail 12 room 4
load:0x40078000,len:10492
load:0x40080400,len:5664
entry 0x40080678
E (596) psram: PSRAM ID read error: 0xffffffff
E (596) spiram: SPI RAM enabled but initialization failed. Bailing out.

00:00:00.002 HDW: ESP32-D0WDQ6
00:00:00.024 UFS: FlashFS mounted with 48 kB free
00:00:00.072 CFG: Loaded from File, Count 25
00:00:00.080 QPC: Count 1
00:00:00.084 SPI: Hardware using GPIO18(CLK) and GPIO19(MISO)
00:00:00.106 BRY: Berry initialized, RAM used=4953
00:00:00.115 BRY: no 'preinit.be'
00:00:00.145 DSP: File descriptor used
xs : 480
ys : 320
bpp: 16
Nr. : 1
CS  : 15
CLK : 18
MOSI: 23
DC  : 2
BPAN: 17
RES : 4
MISO: 19
SPED: 40000000
Pixels: 18
SaMode: 16
opts: 2c,ff,ff
SetAddr : 2a,2b,2c
Rot 0: 36,28 - 0 - 0
00:00:00.154 TS: XPT2046
cmd, args e0, 15 00 03 09 08 16 0a 3f 78 4c 09 0a 08 16 1a 0f 
cmd, args e1, 15 00 16 19 03 0f 05 32 45 46 04 0e 0d 35 37 0f 
cmd, args c0, 2 17 15 
cmd, args c1, 1 41 
cmd, args c5, 3 00 12 80 
cmd, args 36, 1 48 
cmd, args 3a, 1 66 
cmd, args b0, 1 80 
cmd, args b1, 1 a0 
cmd, args b4, 1 02 
cmd, args b6, 2 02 02 
cmd, args e9, 1 00 
cmd, args f7, 4 a9 51 2c 82 
cmd, args 11, 0 
delay 150 ms
cmd, args 29, 0 
Dsp Init complete 
00:00:01.125 DSP: ILI9488!
00:00:01.738 Project tasmota Tasmota Version 9.5.0.2(lvgl)-1_0_7_1(2021-07-13T15:58:01)
00:00:01.792 I2C: BME280 found at 0x77
00:00:01.803 BRY: no 'autoexec.be'
00:00:02.548 WIF: Connecting to AP1 mike1.1.3 Channel 11 BSSId 44:94:FC:35:CC:8B in mode 11n as tasmota_032220-0544...
00:00:07.616 WIF: Connected
00:00:07.874 HTP: Web server active on tasmota_032220-0544 with IP address 192.168.1.245
00:00:07.913 QPC: Reset
21:21:48.708 RSL: INFO1 = {"Info1":{"Module":"ESP32-DevKit","Version":"9.5.0.2(lvgl)","FallbackTopic":"cmnd/DVES_032220_fb/","GroupTopic":"cmnd/tasmotas/"}}
21:21:48.714 RSL: INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota_032220-0544","IPAddress":"192.168.1.245"}}
21:21:48.724 RSL: INFO3 = {"Info3":{"RestartReason":"RTC Watch dog reset digital core and rtc module"}}
21:21:48.733 RSL: RESULT = {"POWER":"OFF"}
21:21:48.736 RSL: POWER = OFF
21:21:52.926 RSL: STATE = {"Time":"2021-07-13T21:21:52","Uptime":"0T00:00:12","UptimeSec":12,"Heap":192,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":87,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"mike1.1.3","BSSId":"44:94:FC:35:CC:8B","Channel":11,"Mode":"11n","RSSI":96,"Signal":-52,"LinkCount":1,"Downtime":"0T00:00:07"}}
21:21:52.984 RSL: SENSOR = {"Time":"2021-07-13T21:21:52","ANALOG":{"A1":2439},"BME280":{"Temperature":22.9,"Humidity":68.7,"DewPoint":16.9,"Pressure":998.9},"ESP32":{"Temperature":53.3},"Global":{"Temperature":22.9,"Humidity":68.7,"DewPoint":16.9,"Pressure":998.9,"SeaPressure":999.4},"PressureUnit":"hPa","TempUnit":"C"}
21:22:07.483 CMD: seriallog 4
21:22:07.487 RSL: RESULT = {"SerialLog":{"4":{"Active":"4"}}}
21:22:08.720 CFG: Saved, Count 27, Bytes 4096
21:22:08.759 WIF: Checking connection...
21:22:18.912 HTP: Consoles
21:22:20.771 HTP: Berry Console
21:22:28.678 WIF: Checking connection...
21:22:48.672 WIF: Checking connection...
21:23:08.642 WIF: Checking connection...
21:23:09.344 BRY: GC from 10045 to 6799 bytes (in 2 ms)
21:23:09.385 LVG: LVGL initialized
21:23:19.929 BRY: GC from 13609 to 6027 bytes (in 3 ms)
21:23:28.892 WIF: Checking connection...

```

mikep1998, 2021-07-13:
Another unusual thing is with the precompiled code the code hangs after power on boot.  SPI RAM enabled but initialization failed. Bailing out.  It may take 1-5 boot to get past this hang.

Serial output:

Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:160
ho 0 tail 12 room 4
load:0x40078000,len:10492
load:0x40080400,len:5664
entry 0x40080678
E (596) psram: PSRAM ID read error: 0xffffffff
E (596) spiram: SPI RAM enabled but initialization failed. Bailing out.

Jason2866, 2021-07-14:
You have to erase flash before. Please flash via [ESP_Flasher](https://github.com/Jason2866/ESP_Flasher) or [Tasmota WebInstaller](https://arendst.github.io/Tasmota/)

mikep1998, 2021-07-14:
Thanks for the info. I used  ESP_Flasher, then installed the display.ini in FS. Ran Berry code. No graphics on the screen. I can see the backlight is working. 

```
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
E (613) psram: PSRAM ID read error: 0xffffffff
E (614) spiram: SPI RAM enabled but initialization failed. Bailing out.
ÿ
00:00:00.002 HDW: ESP32-D0WDQ6
00:00:00.013 UFS: FlashFS mounted with 304 kB free
00:00:00.038 CFG: Loaded from File, Count 20
00:00:00.046 QPC: Count 2
00:00:00.047 CFG: CR 366/699, Busy 0
00:00:00.049 SPI: Hardware using GPIO18(CLK), GPIO23(MOSI) and GPIO19(MISO)
00:00:00.052 CFG: CR 366/699, Busy 0
00:00:00.057 BRY: GC from 1900 to 1900 bytes (in 0 ms)
00:00:00.062 BRY: GC from 3816 to 3716 bytes (in 0 ms)
00:00:00.073 BRY: GC from 6538 to 4953 bytes (in 1 ms)
00:00:00.073 BRY: Berry initialized, RAM used=4953
00:00:00.080 BRY: no 'preinit.be'
00:00:00.101 DSP: File descriptor used
xs : 480
ys : 320
bpp: 16
Nr. : 1
CS  : 15
CLK : 18
MOSI: 23
DC  : 2
BPAN: 17
RES : 4
MISO: 19
SPED: 40000000
Pixels: 18
SaMode: 16
opts: 2c,ff,ff
SetAddr : 2a,2b,2c
Rot 0: 36,28 - 0 - 0
00:00:00.111 TS: XPT2046
cmd, args e0, 15 00 03 09 08 16 0a 3f 78 4c 09 0a 08 16 1a 0f 
cmd, args e1, 15 00 16 19 03 0f 05 32 45 46 04 0e 0d 35 37 0f 
cmd, args c0, 2 17 15 
cmd, args c1, 1 41 
cmd, args c5, 3 00 12 80 
cmd, args 36, 1 48 
cmd, args 3a, 1 66 
cmd, args b0, 1 80 
cmd, args b1, 1 a0 
cmd, args b4, 1 02 
cmd, args b6, 2 02 02 
cmd, args e9, 1 00 
cmd, args f7, 4 a9 51 2c 82 
cmd, args 11, 0 
delay 150 ms
cmd, args 29, 0 
Dsp Init complete 
00:00:01.116 DSP: ILI9488!
00:00:01.723 SRC: Restart
00:00:01.725 Project tasmota Tasmota Version 9.5.0.2(lvgl)-1_0_7_1(2021-07-13T15:58:01)
00:00:01.738 SHT: Sensor did not ACK command
00:00:01.781 I2C: BME280 found at 0x77
00:00:01.788 BRY: no 'autoexec.be'
00:00:02.057 WIF: Checking connection...
00:00:02.057 WIF: Attempting connection...
00:00:02.533 WIF: Connecting to AP1 mike1.1.3 Channel 11 BSSId 44:94:FC:35:CC:8B in mode 11n as tasmota_032220-0544...
00:00:03.624 WIF: Checking connection...
00:00:03.624 WIF: Attempting connection...
00:00:04.623 WIF: Checking connection...
00:00:04.623 WIF: Attempting connection...
00:00:04.601 WIF: Checking connection...
00:00:04.601 WIF: Attempting connection...
00:00:06.601 WIF: Checking connection...
00:00:06.601 WIF: Connected
00:00:06.859 HTP: Web server active on tasmota_032220-0544 with IP address 192.168.1.245
00:00:07.193 RTC: UTC 2021-07-14T11:58:57, DST 2021-03-28T02:00:00, STD 2021-10-31T03:00:00
12:58:57.659 RSL: INFO1 = {"Info1":{"Module":"ESP32-DevKit","Version":"9.5.0.2(lvgl)","FallbackTopic":"cmnd/DVES_032220_fb/","GroupTopic":"cmnd/tasmotas/"}}
12:58:57.665 RSL: INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota_032220-0544","IPAddress":"192.168.1.245"}}
12:58:57.675 RSL: INFO3 = {"Info3":{"RestartReason":"RTC Watch dog reset digital core and rtc module"}}
12:58:57.684 RSL: RESULT = {"POWER":"OFF"}
12:58:57.687 RSL: POWER = OFF
12:58:57.693 TFS: File 'autoexec.bat' not found
12:58:57.733 QPC: Reset
12:58:59.934 APP: Boot Count 9
12:59:00.570 CFG: Saved, Count 21, Bytes 4096
12:59:02.269 RSL: STATE = {"Time":"2021-07-14T12:59:02","Uptime":"0T00:00:12","UptimeSec":12,"Heap":192,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":162,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"mike1.1.3","BSSId":"44:94:FC:35:CC:8B","Channel":11,"Mode":"11n","RSSI":100,"Signal":-45,"LinkCount":1,"Downtime":"0T00:00:06"}}
12:59:02.328 RSL: SENSOR = {"Time":"2021-07-14T12:59:02","ANALOG":{"A1":2311},"BME280":{"Temperature":21.6,"Humidity":63.5,"DewPoint":14.4,"Pressure":997.5},"ESP32":{"Temperature":53.3},"Global":{"Temperature":21.6,"Humidity":63.5,"DewPoint":14.4,"Pressure":997.5,"SeaPressure":997.9},"PressureUnit":"hPa","TempUnit":"C"}
12:59:17.775 WIF: Checking connection...
12:59:37.739 WIF: Checking connection...
12:59:57.703 BRY: GC from 9961 to 7163 bytes (in 2 ms)
12:59:57.744 LVG: LVGL initialized
12:59:58.012 WIF: Checking connection...
13:00:17.990 WIF: Checking connection...
13:00:37.982 WIF: Checking connection...
```

mikep1998, 2021-07-14:
Does anyone have a version of tasmota32-lvgl.bin with 1.0.7 ? I would like to try that to see if it works.

Jason2866, 2021-07-14:
> Does anyone have a version of tasmota32-lvgl.bin with 1.0.7 ? I would like to try that to see if it works.

[tasmota32-lvgl.bin.zip](https://github.com/arendst/Tasmota/files/6818512/tasmota32-lvgl.bin.zip)

mikep1998, 2021-07-14:
I tried the tasmota32-lvgl.bin.zip build but that does the same thing. Does not seem to work with my hardware. On power on I can see the screen flash white and draws to black. So I think the code is writing to screen but when I run the berry script it only lights up the backlight. No button or text displayed.

mikep1998, 2021-07-14:
Well though trial and error I have narrowed it down to this file: 
framework-arduinoespressif32\tools\sdk\lib\libesp32.a
in 1.0.7.1 as the cause. How do I see source code for this file?

Jason2866, 2021-07-15:
The libs are compiled via [Arduino lib builder](https://github.com/Jason2866/esp32-arduino-lib-builder/tree/tasmota/v3.3) source from [IDF3.3.5](https://github.com/tasmota/esp-idf/tree/release/v3.3)

Jason2866, 2021-07-15:
I cant follow your logic. You said the build with core 1.0.7 doesnt work either.
So why you are thinking there is a problem in the libesp32.a from core 1.0.7.1 ?

mikep1998, 2021-07-15:
Sorry for the confusion.
I have never tried tasmota32-lvgl so I did not know if it worked.
tasmota32-lvgl built with either 1.0.7 or 1.0.7.1 does not work for my hardware. No TFT and no touchscreen.

tasmota with my driver and lvgl and no berry work fine with 1.0.7. The touchscreen does not work with 1.0.7.1 but the TFT works fine.

mikep1998, 2021-07-15:
Through trial and error I isolated the libesp32.a as the cause of the touchscreen failure. I have tried it both directions using 1.0.7.1 libesp32.a  in 1.0.7 touchscreen fails. Using 1.0.7 libesp32.a  in 1.0.7.1 touchscreen works fine. 

I have looked at the changes in IDF3.3.5 and I can not find any indication of what is in 1.0.7.1 and 1.0.7 Going by the build dates it looks like 3 changes are in 1.0.7.1 one large change for ble_mesh. 

At this point I have hit my limit as to trouble shooting this problem. Not sure where to go from here.
Any help is appreciated.

s-hadinger, 2021-07-15:
Jason, did we change anything in sdkconfig between these two versions?

Jason2866, 2021-07-15:
Yes, see https://github.com/arendst/Tasmota/issues/12635#issuecomment-878973481

s-hadinger, 2021-07-15:
Ah thanks. So it looks like PSRAM failed initialization causes troubles with the touchscreen. @mikep1998 what are the gpio used for the touchscreen?

mikep1998, 2021-07-15:
Yes that is correct.  The FTF and touch use the same SPI. 
in tasmota I have 
```
    pin 0   = AGPIO(GPIO_OPTION_A) +2; 
    pin 2   = AGPIO(GPIO_SPI_DC);                   TFT SPI DC
    pin 4   = AGPIO(GPIO_OLED_RESET);          TFT RESET 
    pin 15 = AGPIO(GPIO_SPI_CS);                    TFT_CS 
    pin 16 = AGPIO(GPIO_XPT2046_CS);           TOUCH_CS
    pin 17 = AGPIO(GPIO_BACKLIGHT);            BACKLIGHTPIN 
    pin 18 = AGPIO(GPIO_SPI_CLK);                  TFT_SCLK          
    pin 19 = AGPIO(GPIO_SPI_MISO);               TFT_MISO
    pin 23 = AGPIO(GPIO_SPI_MOSI);               TFT_MOSI
```

In display.ini (desc) I have

```
CS  : 15
CLK : 18
MOSI: 23
DC  : 2
BPAN: 17
RES : 4
MISO: 19
TS: 16

:H,ILI9488,480,320,16,SPI,1,15,18,23,2,17,4,19,40
:S,2,1,1,0,40,20
:I
E0,0F,00,03,09,08,16,0A,3F,78,4C,09,0A,08,16,1A,0F
E1,0F,00,16,19,03,0F,05,32,45,46,04,0E,0D,35,37,0F
C0,2,17,15
C1,1,41
C5,3,00,12,80
36,1,48
3A,1,66
B0,1,80
B1,1,A0
B4,1,02
B6,2,02,02
E9,1,00
F7,4,A9,51,2C,82
11,80
29,0
:o,28
:O,29
:A,2A,2B,2C,16
:R,36
:0,28,00,00,02
:1,48,00,00,01
:2,E8,00,00,00
:3,88,00,00,03
:P,18
:i,20,21
:TS,16
:B,20,0
:M,340,3900,300,3800
#
```

With 1.0.7 This is a trace of the SPI pins and values used-
Note the SPI pins are correct.
```
z=2950  ::  z1=1607,  z2=2752, csPin: 16, _pspi._miso: 19, _pspi._mosi: 23, _pspi._sck: 18, _pspi._div: 4984833,  _pspi._freq: 2000000, _pspi._use_hw_ss: 0, _pspi._spi_num: 3,  
p=2950,  3192,2563  3177,2571  3171,2550    3174,2567  (VALID DATA HERE)
```
With 1.0.7.1 the trace of the SPI pins and values used- 
Note the SPI pins are correct.
But the data read is ZERO
```
z=4095  ::  z1=0,  z2=0, csPin: 16, _pspi._miso: 19, _pspi._mosi: 23, _pspi._sck: 18, _pspi._div: 4984833,  _pspi._freq: 2000000, _pspi._use_hw_ss: 0, _pspi._spi_num: 3,  
p=4095,  0,0  0,0  0,0    0,0
```

Hope this helps

mikep1998, 2021-07-15:
The ESP32 board is ESP32 DOIT dev kit v1 clone and the TFT is 3.5" TFT SPI 480x320 V1.0

mikep1998, 2021-07-15:
I have couple of these hardware setups with the same esp32 and TFT. So I tried another physical setup with the same results 1.0.7 everything works. 1.0.7.1 touchscreen does not work.

The only thing I see different is in the serial log:

E (553) psram: PSRAM ID read error: 0xffffffff
E (553) spiram: SPI RAM enabled but initialization failed. Bailing out.
ÿ

So something may be left dirty in the SPI?
Since my esp32 does not have PSRAM why is it checking? How I stop the check?

mikep1998, 2021-07-16:
@Jason2866 I tried modifying platformio_override.ini with the suggest build_flags
```
build_flags             = ${env:tasmota32_base.build_flags} -DBOARD_HAS_PSRAM -mfix-esp32-psram-cache-issue -lc-psram-workaround -lm-psram-workaround
```
The touchscreen still does not work.
Since my board does not have PSRAM is there a way to disable the checking with build_flags?

Thanks

s-hadinger, 2021-07-16:
Try removing `-DBOARD_HAS_PSRAM` this will disable PSRAM completely. I suspect a conflict on GPIO17 that is used for PSRAM. It is also Touchscreen CS pin. I suppose that the PSRAM probe sends corrupt data to the touchscreen. Maybe there is a way to reinit completely the touchscreen. 

Can you confirm it is working without `-DBOARD_HAS_PSRAM`?

s-hadinger, 2021-07-16:
If your test is positive, I will see how we can disable PSRAM init code if GPIOs 17/18 are already used

Jason2866, 2021-07-16:
From your description i assume you wired the display. For testing do not use GPIO 17 for backlight use a other free GPIO. GPIO12 would be good.

s-hadinger, 2021-07-16:
Sorry for the confusion. The problem is Gpio 16 not 17.

mikep1998, 2021-07-16:
I searched the code and the only place that -DBOARD_HAS_PSRAM is defined is in [env:tasmota32-cam] by default -DBOARD_HAS_PSRAM is NOT used in my code. But the error in the log E (553) spiram: SPI RAM enabled but initialization failed. Bailing out. does happen when 1.0.7.1 is used. 
I'm using this env to build

```[env:tasmota32-messagebox]
extends                     = env:tasmota32_base
build_flags                 = ${env:tasmota32_base.build_flags}
                              -Duserconfig_messagebox

board                       = esp32doit-devkit-v1
board_build.partitions      = esp32_partition_app1984k_spiffs64k.csv

src_filter = 
    +<*>
    +<..\MikesFiles\analog\*.c>
```

Any idea why the code is checking SPRAM without it being defined?

PS: I will try to move GPIO16 to another pin and let you know.

mikep1998, 2021-07-16:
Which pins does PSRAM use?

Jason2866, 2021-07-16:
Gpio 16 and 17 as i wrote above

mikep1998, 2021-07-16:
Good news when I moved the touch CS pin from 16 to 27 everything works.
What's next?

Jason2866, 2021-07-17:
We are going to "red mark" this GPIOs as restricted. The behaviour is normal since a check is done if PSRAM is connected.

PR #12679 (2021-07-18): Mark GPIOs 16 and 17 as read on ESP32, as they are used for PSRAM
## Description:

Mark also GPIO 16/17 as red on ESP32. They are used by PSRAM and are probed at boot. If you don't have a PSRAM, you can you those GPIOs but please expect them to be used as outputs for a short period of time at start-up.

**Related issue (if applicable):** fixes #12635

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [ ] The code change is t [... cut]
closed 2021-07-17 by Jason2866