arendst/Tasmota #12643: Enhancement: Upload Self Compiled FW to Tasmota Webinstaller

helmar74, 2021-07-13:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Just a little Enhancement: MAybe it would be possible to add a "Upload a self compiled Firmware"-Button to the Tasmota Webinstaller?

In addition it would also be great to have a checkbox "Delete Flash before Uploading".

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2021-07-13:
Where should the self compiled binary come from???

Flash is erased before flashing when you select the checkbox about clearing the settings.

helmar74, 2021-07-13:
> Where should the self compiled binary come from???
> 
> Flash is erased before flashing when you select the checkbox about clearing the settings.
Self compile Tasmota and then 
from a "Upload" Input Field? (Stored in temporarly file...)
Similar to Tasmoadmin:
[image]

helmar74, 2021-07-13:
Was only a suggestion...

closed 2021-07-13 by helmar74

Jason2866, 2021-07-17:
Not possible by design of WebInstaller to provide a solution for from our side.
**ALL** files have to be on the same server.
If you want to this clone the branch `firmware` and setup a web server with your files.
If you do this by a public webserver everyone can flash your firmwares