arendst/Tasmota #12652: Restart 1 or 99 always put GPIO16 to low, does not preserve state, when SetOption0 0 is set

lmagyar, 2021-07-14:
### PROBLEM DESCRIPTION
`Restart 1` or `Restart 99` preserves state's of `Relay` or `Relay_i` outputs even when `SetOption0 0` is set, except on GPIO16, where it puts this line low, independent of previous state or `PowerOnStateX` configuration when `SetOption0 0` is set. Low means 0V, not the logical or inverted logical ON or OFF.

`SetOption0 0` is for minimizing flash writes and results in power state loss on power failure, it is OK. But `Restart 1` or `Restart 99` tries to save state, but it is not working if GPIO is 16 and `SetOption0 0` is set. Devices can restart silently on several errors, but GPIO16 is not "silent", `Relay_i` will turn on unnoticed, this is only matter of time.

Using `SetOption63 1` makes things worse, the line will go low, in case of `Releay_i` config it is `ON`, but the webserver shows `OFF` state, that is inconsistent, so I reverted to the default `SetOption63 0`. 

Investigation:
- Tested config that works (hard reset puts the line to the configured PowerOnState, Restart 99 preserves previous state, even when 'SetOption0 0' is set):
  - GPIO14: Relay 1 and PowerOnState1 0
  - GPIO14: Relay 1 and PowerOnState1 1
  - GPIO14: Relay_i 1 and PowerOnState1 0
  - GPIO14: Relay_i 1 and PowerOnState1 1
- Tested config that has the issue (hard reset puts the line to the configured poweronstate, Restart 99 does not preserve previous state when 'SetOption0 0' is set, puts line to low state):
  - GPIO16: Relay 1 and PowerOnState1 0
  - GPIO16: Relay 1 and PowerOnState1 1
  - GPIO16: Relay_i 1 and PowerOnState1 0
  - GPIO16: Relay_i 1 and PowerOnState1 1

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4) **!!! I've got "Invite Invalid" !!!**
- [x] Device used (e.g., Sonoff Basic): ESP-07 module and NodeMcu V2 ESP-12E
- [x] Tasmota binary firmware version number used: 9.5.0 and latest dev 9.5.0.2
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: Tasmotizer 1.2
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
11:41:30.028 RSL: RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
11:41:30.236 RSL: RESULT = {"Module":{"0":"Generic"}}
11:41:30.490 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"320":"Led_i1"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"256":"Relay_i1"},"GPIO17":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
```
- [x] Provide the output of this command: `Status 0`:
```lua
11:43:16.948 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_47D2CA","ButtonTopic":"0","Power":1,"PowerOnState":0,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":0,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
11:43:16.974 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:06:25","StartupUTC":"2021-07-14T10:36:51","Sleep":50,"CfgHolder":4617,"BootCount":68,"BCResetTime":"2021-07-14T10:27:22","SaveCount":109,"SaveAddress":"FB000"}}
11:43:16.007 RSL: STATUS2 = {"StatusFWR":{"Version":"9.5.0.2(tasmota)","BuildDateTime":"2021-07-14T08:59:27","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"355/699"}}
11:43:16.027 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["ICS",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008008","2805C8000100060000005A0A000000000000","00000080","00006000","00000000"]}}
11:43:16.056 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":608,"Free":392,"Heap":27,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640D8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
11:43:16.086 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota_47D2CA-4810","IPAddress":"192.168.1.114","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"79.121.0.2","Mac":"9C:9C:1F:47:D2:CA","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
11:43:16.108 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_47D2CA","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
11:43:16.129 RSL: STATUS7 = {"StatusTIM":{"UTC":"2021-07-14T10:43:16","Local":"2021-07-14T11:43:16","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"05:02","Sunset":"20:49"}}
11:43:16.147 RSL: STATUS10 = {"StatusSNS":{"Time":"2021-07-14T11:43:16"}}
11:43:16.154 RSL: STATUS11 = {"StatusSTS":{"Time":"2021-07-14T11:43:16","Uptime":"0T00:06:25","UptimeSec":385,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"ON","Wifi":{"AP":1,"SSId":"ICS","BSSId":"C0:56:27:51:F5:32","Channel":1,"Mode":"11n","RSSI":74,"Signal":-63,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
```

### TO REPRODUCE
- Module: Generic
- GPIO16: Relay_i 1
- \> PowerOnState1 0
- \> SetOption0 0
- Turn off relay if it is not off, GPIO16 line is high.
- \> Restart 99 or Restart 1 or change the config and make Tasmota to restart itself
- Issue: GPIO16 line goes low

### EXPECTED BEHAVIOUR
- GPIO16 line preserves it's previous state on Restart

### SCREENSHOTS

### ADDITIONAL CONTEXT

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2021-07-14:
As ESP8266 turns GPIO16 high at boot, it is not possible to silently preserve pin state. Thus, this is not a good pin for a relay or similar.

lmagyar, 2021-07-14:
Thanks! I understand now. Closing this.

closed 2021-07-14 by lmagyar