arendst/Tasmota #12667: When using some routers Tasmota from 9.3 up lags and is slow

maslyankov, 2021-07-16:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
When using some routers to connect Tasmota to, when using firmware from 9.3 and above, its web server interface lags, is slow or unresponsive. I tested this behavior using 3 Wi-Fi routers:
- TP-Link TL-WR542G - show and unresponsive web config
- Apple Airport Express (last gen - A1392) - show and unresponsive web config
- TP-Link Archer C7 with OpenWRT - no issues
Issue persists on:
- Tasmota 9.5 (Tested with a NodeMCU v3, Wemos D1 Mini)
- Tasmota 9.4 (Tested with a NodeMCU v3, Wemos D1 Mini)
Issue not happening on:
- Tasmota 9.1

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): NodeMCU (ESP8266)
- [x] Tasmota binary firmware version number used: tasmota-sensors.bin
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: tasmotizer
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
18:11:52 RSL: stat/tasmota_611B0A/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
18:11:52 RSL: stat/tasmota_611B0A/RESULT = {"Module":{"1":"Sonoff Basic"}}
18:11:53 RSL: stat/tasmota_611B0A/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
18:09:47 RSL: stat/tasmota_611B0A/STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_611B0A","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
18:09:47 RSL: stat/tasmota_611B0A/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:00:35","StartupUTC":"2021-07-16T17:09:12","Sleep":50,"CfgHolder":4617,"BootCount":4,"BCResetTime":"2021-07-16T18:09:29","SaveCount":14,"SaveAddress":"F6000"}}
18:09:47 RSL: stat/tasmota_611B0A/STATUS2 = {"StatusFWR":{"Version":"9.1.0(sensors)","BuildDateTime":"2020-11-07T11:59:31","Boot":31,"Core":"2_7_4_5","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"356/699"}}
18:09:47 RSL: stat/tasmota_611B0A/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["5G-Vivacom",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
18:09:47 RSL: stat/tasmota_611B0A/STATUS4 = {"StatusMEM":{"ProgramSize":632,"Free":368,"Heap":21,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16400E","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDA8787","0415A005","B7FFBFCF","01DA9BC4","64367CC7","00084052","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,51,52,55,56,58,59,62,64,66,67,74"}}
18:09:47 RSL: stat/tasmota_611B0A/STATUS5 = {"StatusNET":{"Hostname":"tasmota_611B0A-6922","IPAddress":"192.168.1.106","Gateway":"192.168.1.2","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.2","Mac":"DC:4F:22:61:1B:0A","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
18:09:47 RSL: stat/tasmota_611B0A/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_611B0A","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
18:09:47 RSL: stat/tasmota_611B0A/STATUS7 = {"StatusTIM":{"UTC":"2021-07-16T17:09:47","Local":"2021-07-16T18:09:47","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"05:04","Sunset":"20:47"}}
18:09:47 RSL: stat/tasmota_611B0A/STATUS10 = {"StatusSNS":{"Time":"2021-07-16T18:09:47"}}
18:09:47 RSL: stat/tasmota_611B0A/STATUS11 = {"StatusSTS":{"Time":"2021-07-16T18:09:47","Uptime":"0T00:00:35","UptimeSec":35,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"5G-Vivacom","BSSId":"9E:DA:C4:3F:2A:45","Channel":11,"RSSI":62,"Signal":-69,"LinkCount":1,"Downtime":"0T00:00:15"}}}

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
When experiencing the issue the device is hardly responsive

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Use 9.5.0 and connect to an old wifi router. Try to access the web interface.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Fast and quick web interface

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2021-07-17:
Sorry nothing can be done from Tasmota. It is known that some routers do not work nicly with ESP82xx devices. Tasmota uses Arduino core which hooks up on the espressif nonos-sdk  (closed source). This part is not changed since v.8.x

closed 2021-07-17 by Jason2866

s-hadinger, 2021-07-17:
Try the latest development version and use `SetOption127 1`. Does it solve it?

maslyankov, 2021-07-17:
> Try the latest development version and use `SetOption127 1`. Does it solve it?

Just tried it. No. It does not seem to make any difference.

maslyankov, 2021-07-18:
.

> Sorry nothing can be done from Tasmota. It is known that some routers do not work nicly with ESP82xx devices. Tasmota uses Arduino core which hooks up on the espressif nonos-sdk (closed source). This part is not changed since v.8.x

This does not seem to be the case as 9.1 works well with those routers I tested.

bsdice, 2021-11-21:
Found this issue through a web search. I have a Delock 11827 plug and the same issue with release 10, also tried -latest as of yesterday. Very slow and non-reacting web UI. Downgraded to release 9.1.0 + reset 3 + unplug/replug + reset 5 and plug works alot better now. I saved the calibration values VoltageCal/CurrentCal/PowerCal before and set them back after reset 5. Running an AR9390 Atheros PCIe as AP with hostapd on kernel 5.4.160 and a bunch of other devices on wifi, which do not exhibit this problem. Also got three Delock 11826 plugs which do not have this problem even on Release 10.0.0.3. So possibly something changed from 9.1 to 9.3 in the voltage/current sensing driver that hangs the system? Wild guess. Also tried SO127, did not help.

matthijskooijman, 2023-02-14:
I also had connectivity issues with a Delock 11827 and my Linksys WRT1900ACS (running a rather old OpenWRT 18.06), which for me were completely fixed by disabling sleep in Tasmota (`Sleep 0` command IIRC). I'm not sure this is the same issue as reported here (IIRC I had issues with all versions, from the Delock-customized 7.x version up to including the 12.1.1(lite) version I am running now. I only tried disabling sleep on that 12.1.1 version), but just in case it helps anyone, I wanted to mention this here.

Note that I also had issues connecting to the delock unit's own wifi network for setup (I have two units - one stayed online long enough to connect it to my house network, the other one is now not even showing up at all in wifi scans), so maybe there's also some issues with the units I have...