arendst/Tasmota #12668: Scripting, Subroutines. String parameter passes corrupted but float/integer passes fine

nikvostro, 2021-07-16:
### PROBLEM DESCRIPTION
Passing string parameter to Scripting subrouting doesn't work. But float/integer parameter passes well.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): tested on ESP-12E and ESP32CAM
- [ ] Tasmota binary firmware version number used: 
  - [ ] Pre-compiled
  - [ ] Self-compiled release 9.5.0 and development 9.5.0.3
- [ ] Flashing tools used: VS Code/Platmormio
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"224":"Relay1"},"GPIO5":{"1088":"IRrecv"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"1056":"IRsend"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```
- [ ] If using ~~rules~~ Scripting, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
>D
msg=""

>S

#AL(msg)
print mess: %msg%
```
- [ ] Provide the output of this command: `Status 0`:
```lua
20:55:44.895 RSL: STATUS = {"Status":{"Module":18,"DeviceName":"TasmotaDEV","FriendlyName":["TasmotaDEV"],"Topic":"tasmota_D3444F","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
20:55:44.925 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"https://ota.tasmota.com/tasmota32/tasmota32.bin","RestartReason":"External System","Uptime":"0T00:06:03","StartupUTC":"2021-07-16T17:49:41","Sleep":50,"CfgHolder":4617,"BootCount":62,"BCResetTime":"2021-07-08T00:01:14","SaveCount":204,"SaveAddress":"F7000"}}
20:55:44.955 RSL: STATUS2 = {"StatusFWR":{"Version":"9.5.0.3(tasmota)","BuildDateTime":"2021-07-16T19:52:28","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"381/699"}}
20:55:44.974 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["domuk",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008001","2805C8000100060000005A0A000000000000","00000080","00006000","00000000"]}}
20:55:44.003 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":624,"Free":376,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440E0","FlashFrequency":40,"FlashMode":3,"Features":["00000809","87DAC787","043E8001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
20:55:44.034 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota_D3444F-1103","IPAddress":"192.168.11.104","Gateway":"192.168.11.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.11.1","Mac":"18:FE:34:D3:44:4F","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
20:55:44.056 RSL: STATUS7 = {"StatusTIM":{"UTC":"2021-07-16T17:55:44","Local":"2021-07-16T20:55:44","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+03:00","Sunrise":"04:07","Sunset":"21:01"}}
20:55:44.074 RSL: STATUS10 = {"StatusSNS":{"Time":"2021-07-16T20:55:44"}}
20:55:44.082 RSL: STATUS11 = {"StatusSTS":{"Time":"2021-07-16T20:55:44","Uptime":"0T00:06:03","UptimeSec":363,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":42,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"domuk","BSSId":"F0:79:59:36:57:40","Channel":3,"Mode":"11n","RSSI":58,"Signal":-71,"LinkCount":1,"Downtime":"0T00:00:04"}}}
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
20:56:35.635 CMD: weblog 4
20:56:35.641 RSL: RESULT = {"WebLog":4}
20:56:35.038 CFG: Saved to flash at F6, Count 205, Bytes 4096
20:56:40.926 CMD: al testmessage
20:56:40.929 SRC: WebConsole from 192.168.11.233
20:56:40.931 CMD: Grp 0, Cmnd 'AL', Idx 1, Len 11, Data 'testmessage'
20:56:40.934 mess: ���?
20:56:40.936 RSL: RESULT = {"AL":"testmessage"}
```

### TO REPRODUCE
1. Add to user_config_override.h:
```lua
#ifndef USE_SCRIPT
#define USE_SCRIPT
#endif
#ifdef USE_RULES
#undef USE_RULES
#endif  

#define USE_SCRIPT_SUB_COMMAND
```
2. Build with env:tasmota or env:tasmota32-webcam and upload
3. Execute command in console
```lua
al testmessage
```
And you'll get your parameter in surbouting corrupted - see output with `weblog 4` above and screenshot below.

### EXPECTED BEHAVIOUR
Parameter should be passed to the certain subroutine correctly as for integer/float:
```lua
21:01:16.010 CMD: al 1234
21:01:16.012 SRC: WebConsole from 192.168.11.233
21:01:16.014 CMD: Grp 0, Cmnd 'AL', Idx 1, Len 4, Data '1234'
21:01:16.017 mess: 1234.000000
21:01:16.020 RSL: RESULT = {"AL":1234}
21:01:17.787 WIF: Checking connection...
21:01:20.563 CMD: al 11223344
21:01:20.565 SRC: WebConsole from 192.168.11.233
21:01:20.568 CMD: Grp 0, Cmnd 'AL', Idx 1, Len 8, Data '11223344'
21:01:20.571 mess: 11223344.000000
21:01:20.573 RSL: RESULT = {"AL":11223344}
```

### SCREENSHOTS
[image]

### ADDITIONAL CONTEXT

**(Please, remember to close the issue when the problem has been addressed)**

nikvostro, 2021-07-16:
ESP32CAM (some additional xdrvs are included in build):
```lua
00:14:59.317 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"TasmotaESP32CAM","FriendlyName":["TasmotaESP32CAM"],"Topic":"tasmota_7882E8","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":0,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
00:14:59.324 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T05:08:08","StartupUTC":"2021-07-16T16:06:51","Sleep":150,"CfgHolder":4617,"BootCount":149,"BCResetTime":"2021-06-17T12:47:22","SaveCount":957}}
00:14:59.329 RSL: STATUS2 = {"StatusFWR":{"Version":"9.5.0(webcam)","BuildDateTime":"2021-07-06T11:07:46","Core":"1_0_6","SDK":"v3.3.5-1-g85c43024c","CpuFrequency":240,"Hardware":"ESP32-D0WD","CR":"446/699"}}
00:14:59.333 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Vostrouter","domuk"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00000001","2805C8000100060000005A0A000000000000","00000080","00006008","00000000"]}}
00:14:59.341 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":1632,"Free":1856,"Heap":115,"PsrMax":4095,"PsrFree":3337,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":80,"FlashMode":2,"Features":["00000809","878AC78F","001C0001","000000CF","010013C0","40000981","8000C000","00200000","00000024"],"Drivers":"1,2,3,4,5,8,9,10,20,21,24,26,27,29,40,50,81","Sensors":"1,2,3,5,6,87"}}
00:14:59.347 RSL: STATUS5 = {"StatusNET":{"Hostname":"ESP32CAM","IPAddress":"192.168.11.89","Gateway":"192.168.11.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.11.1","Mac":"40:F5:20:78:82:E8","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
00:14:59.353 RSL: STATUS7 = {"StatusTIM":{"UTC":"2021-07-16T21:14:59","Local":"2021-07-17T00:14:59","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+03:00","Sunrise":"04:09","Sunset":"21:00"}}
00:14:59.359 RSL: STATUS10 = {"StatusSNS":{"Time":"2021-07-17T00:14:59","ESP32":{"Temperature":47.2},"TempUnit":"C"}}
00:14:59.372 RSL: STATUS11 = {"StatusSTS":{"Time":"2021-07-17T00:14:59","Uptime":"0T05:08:08","UptimeSec":18488,"Heap":115,"SleepMode":"Dynamic","Sleep":150,"LoadAvg":5,"MqttCount":0,"PWM":{"PWM1":0},"Wifi":{"AP":2,"SSId":"domuk","BSSId":"F0:79:59:36:57:40","Channel":3,"Mode":"11n","RSSI":76,"Signal":-62,"LinkCount":1,"Downtime":"0T00:00:05"}}}
```
[image]

gemu2015, 2021-07-17:
you did not pass a string.

try
al "testmessage"

nikvostro, 2021-07-18:
@gemu2015 OK, clear! Thank you!

Didn't mind of passing string constant with the double quotas as it works normally for most of apps... My bad.
By the way - I can't see any warnings or notices regarding this point in the Tasmota/Docs. Probably it could helped me to avoid such a beginner mistake...

So case can be closed if you're not going to add any remarks in the Docs regardng passing strings in double quotas.

ascillato2, 2021-07-18:
@nikvostro 

Please, help us and do the modifications in the docs.

Thanks.

closed 2021-07-18 by ascillato2