arendst/Tasmota #12676: Devices fails to reconnect to wifi after access point is restarted

ahmaddxb, 2021-07-18:
### PROBLEM DESCRIPTION
Devices fails to reconnect to wifi after access point is restarted. This happens randomy not alway the same device not reconnecting, I have 20+ tasmota devices. To resolve I need to manual cut the power to the device and then turn back on.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _9.5.0____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

13:51:24.148 MQT: stat/sonoff/mum_room/STATUS = {"Status":{"Module":29,"DeviceName":"Mums Room","FriendlyName":["Light 1","Light 2"],"Topic":"sonoff/mum_room/","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1,"InfoRetain":0,"StateRetain":0}}
13:51:24.181 MQT: stat/sonoff/mum_room/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"sonoffs","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Power On","Uptime":"0T00:07:45","StartupUTC":"2021-07-18T09:43:39","Sleep":50,"CfgHolder":4617,"BootCount":68,"BCResetTime":"2020-04-20T12:39:47","SaveCount":1461,"SaveAddress":"F9000"}}
13:51:24.218 MQT: stat/sonoff/mum_room/STATUS2 = {"StatusFWR":{"Version":"9.5.0(tasmota)","BuildDateTime":"2021-06-17T08:26:35","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"403/699"}}
13:51:24.236 MQT: stat/sonoff/mum_room/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["ASUS",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["400AA029","2805C8000100060000005A0A000000000000","000002C8","00006000","00000000"]}}
13:51:24.268 MQT: stat/sonoff/mum_room/STATUS4 = {"StatusMEM":{"ProgramSize":607,"Free":396,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
13:51:24.303 MQT: stat/sonoff/mum_room/STATUS5 = {"StatusNET":{"Hostname":"sonoff/mum_room/-0437","IPAddress":"192.168.1.191","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"DC:4F:22:82:A1:B5","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
13:51:24.324 MQT: stat/sonoff/mum_room/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.235","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_82A1B5","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
13:51:24.349 MQT: stat/sonoff/mum_room/STATUS7 = {"StatusTIM":{"UTC":"2021-07-18T09:51:24","Local":"2021-07-18T13:51:24","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+04:00","Sunrise":"08:06","Sunset":"23:45"}}
13:51:24.369 MQT: stat/sonoff/mum_room/STATUS10 = {"StatusSNS":{"Time":"2021-07-18T13:51:24"}}
13:51:24.378 MQT: stat/sonoff/mum_room/STATUS11 = {"StatusSTS":{"Time":"2021-07-18T13:51:24","Uptime":"0T00:07:45","UptimeSec":465,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"ASUS","BSSId":"B4:FB:E4:71:44:79","Channel":11,"Mode":"11n","RSSI":92,"Signal":-54,"LinkCount":1,"Downtime":"0T00:00:32"}}}
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Restart Wireless AP

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2021-07-18:
Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Tasmota/issues/new/choose) in order to have more information so as to properly help you?

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

ahmaddxb, 2021-07-18:
> Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Tasmota/issues/new/choose) in order to have more information so as to properly help you? [...]

Which part am I missing that you require

ascillato, 2021-07-18:
Mainly the output of the console where all the logs of what is happening are. Remember to set weblog to 4 before starting. Thanks.

ahmaddxb, 2021-07-18:
How can I get to the log when the device doesn't connect back to the WiFi. If I want to get the device back on the WiFi I need to cut the power to it and then turn it back on again. In which case the log is gone. Is there a way to make the log persist after a hard reboot?

ascillato, 2021-07-18:
With the serial log

ascillato2, 2021-07-20:
Hi, were you able to gather some logs?

digiblur, 2021-07-20:
Is SetOption56 disabled?  I noticed recently not sure if it was 9.5 or maybe a little earlier but I've had to turn off so56 now as when devices reboot they go into AP mode for a cycle or two before they connect.

ahmaddxb, 2021-07-20:
> 
> 
> Hi, were you able to gather some logs?

All my devices are plugged in to AC power so cant get logs as its not safe to do serial, when I can I will pull out a device and repeatly reboot my AP's untill I can make it fail to reconnect to wifi and check the serial output.

ahmaddxb, 2021-07-20:
> 
> 
> Is SetOption56 disabled? I noticed recently not sure if it was 9.5 or maybe a little earlier but I've had to turn off so56 now as when devices reboot they go into AP mode for a cycle or two before they connect.

I do have setoption56 enabled but I thought that only applies to a device after restart. I think the device gets stuck in a look on not connecting and the only way to bring it back it to hard reset it by cutting AC power. Also wificonfig is on 4 (detault) so i dont think that would bring up AP mode.

I was thinking to try wificonfig 5 (wait until selected AP is available again without rebooting) or if that still didnt work wificonfig 2 which would bring up AP mode then restart after 3 min and try connect to wifi.

Can you tell me what wificonfig you have set?

ascillato, 2021-07-20:
> Can you tell me what wificonfig you have set?

I have all by default. I have tested your parameters and couldn't reproduce your issue. That is why logs are needed to know what is happening in yours.

You said you have 20+ devices. Don't you have any nodemcu of sonoff basic that can be powered directly from a 3.3V power source? So as to gather some logs to have some information so as to be able to help you.

ahmaddxb, 2021-07-20:
22:55:55.766 WIF: Checking connection...
22:56:15.799 WIF: Checking connection...
so13
22:56:19.791 CMD: so13
22:56:19.793 SRC: Serial
22:56:19.795 CMD: Grp 0, Cmnd 'SO', Idx 13, Len 0, Data ''
22:56:19.800 MQT: stat/sonoff/d1mini/RESULT = {"SetOption13":"ON"}
22:56:35.755 WIF: Checking connection...
SetOption65
22:56:41.660 CMD: SetOption65
22:56:41.662 SRC: Serial
22:56:41.664 CMD: Grp 0, Cmnd 'SETOPTION', Idx 65, Len 0, Data ''
22:56:41.669 MQT: stat/sonoff/d1mini/RESULT = {"SetOption65":"ON"}
22:56:55.785 WIF: Checking connection...
22:57:15.773 WIF: Checking connection...
22:57:27.037 MQT: Attempting connection...
22:57:27.040 MQT: Connect failed to 192.168.1.235:1883, rc -2. Retry in 10 sec
22:57:35.793 WIF: Checking connection...
22:57:35.796 WIF: Connect failed as AP cannot be reached
22:57:36.584 CFG: Saved to flash at FA, Count 34, Bytes 4096
22:57:36.795 WIF: Checking connection...
22:57:36.797 WIF: Connect failed as AP cannot be reached
22:57:37.794 WIF: Checking connection...
22:57:37.796 WIF: Connect failed as AP cannot be reached
22:57:38.798 WIF: Checking connection...
22:57:38.800 WIF: Connect failed as AP cannot be reached
22:57:39.799 WIF: Checking connection...
22:57:39.801 WIF: Connect failed as AP cannot be reached
22:57:40.800 WIF: Checking connection...
22:57:40.802 WIF: Connect failed as AP cannot be reached
22:57:41.755 WIF: Checking connection...
22:57:41.757 WIF: Connect failed as AP cannot be reached
22:57:42.756 WIF: Checking connection...
22:57:42.758 WIF: Connect failed as AP cannot be reached
22:57:43.760 WIF: Checking connection...
22:57:43.763 WIF: Connect failed as AP cannot be reached
22:57:44.761 WIF: Checking connection...
22:57:44.763 WIF: Connect failed as AP cannot be reached
22:57:45.764 WIF: Checking connection...
22:57:45.766 WIF: Connect failed as AP cannot be reached
22:57:46.765 WIF: Checking connection...
22:57:46.768 WIF: Connect failed as AP cannot be reached
22:57:47.766 WIF: Checking connection...
22:57:47.768 WIF: Connect failed as AP cannot be reached
22:57:48.767 WIF: Checking connection...
22:57:48.770 WIF: Connect failed as AP cannot be reached
22:57:49.769 WIF: Checking connection...
22:57:49.771 WIF: Connect failed as AP cannot be reached
22:57:50.771 WIF: Checking connection...
22:57:50.773 WIF: Connect failed as AP cannot be reached
22:57:51.774 WIF: Checking connection...
22:57:51.776 WIF: Connect failed as AP cannot be reached
22:57:52.773 WIF: Checking connection...
22:57:52.775 WIF: Connect failed as AP cannot be reached
22:57:53.774 WIF: Checking connection...
22:57:53.777 WIF: Connect failed as AP cannot be reached
22:57:54.776 WIF: Checking connection...
22:57:54.778 WIF: Connect failed as AP cannot be reached
22:57:55.775 WIF: Checking connection...
22:57:55.777 WIF: Connect failed as AP cannot be reached
22:57:56.777 WIF: Checking connection...
22:57:56.779 WIF: Connect failed as AP cannot be reached
22:57:57.778 WIF: Checking connection...
22:57:57.779 WIF: Connect failed as AP cannot be reached
22:57:58.782 WIF: Checking connection...
22:57:58.784 WIF: Connect failed as AP cannot be reached
22:57:59.784 WIF: Checking connection...
22:57:59.787 WIF: Connect failed as AP cannot be reached
22:58:00.786 WIF: Checking connection...
22:58:00.788 WIF: Connect failed as AP cannot be reached
[... 379 lines cut ...]
23:01:07.775 WIF: Connect failed as AP cannot be reached
23:01:08.775 WIF: Checking connection...
23:01:08.777 WIF: Connect failed as AP cannot be reached
23:01:09.779 WIF: Checking connection...
23:01:09.781 WIF: Connect failed as AP cannot be reached
23:01:10.781 WIF: Checking connection...
23:01:10.783 WIF: Connect failed as AP cannot be reached
23:01:11.780 WIF: Checking connection...
23:01:11.782 WIF: Connect failed as AP cannot be reached
23:01:12.781 WIF: Checking connection...
23:01:12.783 WIF: Connect failed as AP cannot be reached
23:01:13.782 WIF: Checking connection...
23:01:13.784 WIF: Connect failed as AP cannot be reached
23:01:14.782 WIF: Checking connection...
23:01:14.784 WIF: Connect failed as AP cannot be reached
23:01:15.783 WIF: Checking connection...
23:01:15.785 WIF: Connect failed as AP cannot be reached
23:01:16.785 WIF: Checking connection...
23:01:16.787 WIF: Connect failed as AP cannot be reached
23:01:17.785 WIF: Checking connection...
23:01:17.787 WIF: Connect failed as AP cannot be reached
23:01:18.787 WIF: Checking connection...
23:01:18.789 WIF: Connect failed as AP cannot be reached
23:01:19.792 WIF: Checking connection...
23:01:19.794 WIF: Connect failed as AP cannot be reached

ahmaddxb, 2021-07-20:
That is how it look when it doesnt connect, other device have come back on wifi and this one is still trying

ahmaddxb, 2021-07-20:
Here I issue restart 1 and as you can see it connects to wifi

23:03:18.772 WIF: Checking connection...
23:03:18.775 WIF: Connect failed as AP cannot be reached
23:03:19.776 WIF: Checking connection...
23:03:19.778 WIF: Connect failed as AP cannot be reached
restart 1
23:03:20.167 CMD: restart 1
23:03:20.169 SRC: Serial
23:03:20.171 CMD: Grp 0, Cmnd 'RESTART', Idx 1, Len 1, Data '1'
23:03:20.175 RSL: RESULT = {"Restart":"Restarting"}
23:03:20.766 WIF: Checking connection...
23:03:20.768 WIF: Connect failed as AP cannot be reached
23:03:21.518 APP: Restarting

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v3969889e
~ld

00:00:00.001 HDW: ESP8266EX
00:00:00.050 CFG: Loaded from flash at FA, Count 34
00:00:00.055 CFG: CR 332/699, Busy 0
00:00:00.059 SRC: Restart
00:00:00.061 Project tasmota D1Mini Version 9.5.0(tasmota)-2_7_4_9(2021-06-17T08:26:35)
00:00:00.213 WIF: Checking connection...
00:00:00.214 WIF: Attempting connection...
00:00:01.164 WIF: Network (re)scan started...
00:00:01.165 WIF: Checking connection...
00:00:01.166 WIF: Attempting connection...
00:00:02.164 WIF: Checking connection...
00:00:02.165 WIF: Attempting connection...
00:00:03.167 WIF: Checking connection...
00:00:03.167 WIF: Attempting connection...
00:00:04.167 WIF: Network 0, AP1, SSId ASUS, Channel 11, BSSId B4:FB:E4:71:42:B0, RSSI -54, Encryption 1
00:00:04.505 WIF: Connecting to AP1 ASUS Channel 11 BSSId B4:FB:E4:71:42:B0 in mode 11n as d1mini-5563...
00:00:04.506 WIF: Checking connection...
00:00:04.507 WIF: Attempting connection...
00:00:05.801 WIF: Checking connection...
00:00:05.802 WIF: Connected
00:00:06.003 HTP: Web server active on d1mini-5563 with IP address 192.168.1.165
00:00:06.479 RTC: UTC 2021-07-20T19:03:28, DST 2021-03-28T02:00:00, STD 2021-10-31T03:00:00
23:03:28.061 CFG: Saved to flash at F9, Count 35, Bytes 4096
23:03:29.001 MQT: Attempting connection...
23:03:29.014 MQT: Connected
23:03:29.017 MQT: tele/sonoff/d1mini/LWT = Online (retained)
23:03:29.019 MQT: cmnd/sonoff/d1mini/POWER = 
23:03:29.020 MQT: Subscribe to cmnd/sonoff/d1mini/#
23:03:29.022 MQT: Subscribe to cmnd/sonoff/sonoffs/#
23:03:29.026 MQT: Subscribe to cmnd/DVES_4ED5BB_fb/#
23:03:29.031 MQT: Unsubscribe from homeassistant/status
23:03:29.038 MQT: tele/sonoff/d1mini/INFO1 = {"Info1":{"Module":"Generic","Version":"9.5.0(tasmota)","FallbackTopic":"cmnd/DVES_4ED5BB_fb/","GroupTopic":"cmnd/sonoff/sonoffs/"}}
23:03:29.053 MQT: tele/sonoff/d1mini/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"d1mini-5563","IPAddress":"192.168.1.165"}}
23:03:29.064 MQT: tele/sonoff/d1mini/INFO3 = {"Info3":{"RestartReason":"Software/System restart"}}
23:03:30.476 APP: Boot Count 15
23:03:30.563 CFG: Saved to flash at F8, Count 36, Bytes 4096
23:03:33.437 MQT: tele/sonoff/d1mini/STATE = {"Time":"2021-07-20T23:03:33","Uptime":"0T00:00:13","UptimeSec":13,"Heap":29,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"ASUS","BSSId":"B4:FB:E4:71:42:B0","Channel":11,"Mode":"11n","RSSI":84,"Signal":-58,"LinkCount":1,"Downtime":"0T00:00:07"}}

ahmaddxb, 2021-07-20:
Ok here is my findings so far

If I have so56 1 enabled i have the issue in question, when I have so56 0 disabled I was able to reboot my access point 10+ time and the D1 Mini always managed to reconnect.

Also the devices that are currenly deployed that didnt reconnect I have changed them to so56 0 and they are always reconnecting now.

I have not sent a grouptopic command to all my device to have so56 0 and reboot every access point i have multiple time and every device is reconnecting.

So it must be something with so56 enabled but from the log I didnt get much more info than AP cannot be reached even though it was 100% online as other device started to connect to it.

ascillato2, 2021-07-21:
Please, show us the output of the commands SSID1 and SSID2. Thanks

ahmaddxb, 2021-07-22:
10:51:30.027 HTP: Main Menu
10:51:31.582 HTP: Console
10:51:47.770 WIF: Checking connection...
10:52:07.752 WIF: Checking connection...
10:52:15.518 CMD: ssid1
10:52:15.520 SRC: WebConsole from 192.168.1.200
10:52:15.522 CMD: Grp 0, Cmnd 'SSID', Idx 1, Len 0, Data ''
10:52:15.528 MQT: stat/sonoff/d1mini/RESULT = {"SSId1":"ASUS"}
10:52:17.668 CMD: ssid2
10:52:17.670 SRC: WebConsole from 192.168.1.200
10:52:17.672 CMD: Grp 0, Cmnd 'SSID', Idx 2, Len 0, Data ''
10:52:17.678 MQT: stat/sonoff/d1mini/RESULT = {"SSId2":""}

NsinghP, 2021-07-23:
I think I am having the same issue, when AP is turned off for few minutes. All my devices are set to wificonfig 7, so wasn't expecting these devices to show Tasmota configuration access point, but they all did (5 devices).

MaceMoneta, 2021-07-24:
I have 75 devices (S20, S31, Basic, slampher, mini, IW101, TX T2, Wemos D1, esp-01s, others) running Tasmota 9.5.0, and have setoption 56 set to 1 on all. I use Ubiquiti access points (UAP-AC-M-US), and don't have any issue with devices recovering after an access point going offine for firmware upgrade (several minutes). Because of the varying experience, this sounds like an access point issue.

ahmaddxb, 2021-07-25:
I have those exact AP's too, 5 of them around the house / garden I have experienced this happen on most AP's that the esp's are connected to, It wasnt issolated to perhaps one failing AP.

Not sure if it is the settings ive configed on unifi, everything is default/auto except below.

2.4Ghz - On Seperate SSID different from 5Ghz
Set a manual channel
channel width VHT40
Disabled meshing, Airtime Fairness and Mininmum RSSI

Band Steering I have Prefer 5G

MaceMoneta, 2021-07-25:
I have similar settings, except vt20, no band steering.

ascillato, 2021-07-25:
Band Steering is very bad for all devices that are 2.4GHz. Please, try to disable it. Thanks.

About 40Mhz of channel width, that won't affect the devices directly but if you have neighbors in close channels it will interfere and make both networks slower. If in your 2.4Ghz you don't have video streaming, 40MHz is not needed. For MQTT, web browsing and audio streaming, 20MHz are more than enough.

MaceMoneta, 2021-07-28:
A 20MHz channel width on 2.4GHz will get you about 70Mb/s of bandwidth. A Netflix 4K stream uses about 25Mb/s or about 13Mb/s on YouTube, as a point of reference.

popy2k14, 2021-08-10:
Hey guys, exactly the same issue here with tasmota devices and an Unifi Network (2x UAP-AC-LR) and seperate 5Ghz SSID.
Some Tasmota devices just wont reconnect after an AP Update/Reboot of the APs.
I have to power cycle the tasmota devices.
All other Wifi network devices, like Alexa, Smartphones... have no issues at all.

@ahmaddxb 
Did you try the 40Mhz & band steering settings like @ascillato mentioned.

Will get back here in the evening when i can check/post my config.

ahmaddxb, 2021-08-10:
@popy2k14
Yes those are my current setting now but haven’t turned back on so56 to see if that part is resolved with 20Mhz & band steering off.

When I turned off so56 it always worked after AP update or restart even with 40mhz and band steering on so I kept so56 off.

Best Regards,

Ahmad Haghshenas
Mob: +971-50-4956331

popy2k14, 2021-08-10:
@ascillato2
I have different SSID's for 2,4Ghz and 5ghz, so should be no issue for me.

@ahmaddxb
On all my devices the so56 (scan to select strongest signal) and so57 (Wi-Fi network re-scan every 44 minutes with alternate to +10dB stronger signal if detected (only visible networks)) is DISABLED = 0.

On 9/10 i have wificonfig set to 5 = wait and on the other one -> 4 = retry.

So whats the best options (so56, so57 & wificonfig) for Unifi networks to keep the device reconnecting after AP reboot and select always the AP with the best signal?

I think:
so56 0    (so our issue should be solved)
so57 1    (so tasmota devices dont get stucked on an far away AP when the nearest will be updated and rebooted)
wificonfig 4 or 5, please an advice?

Any advice?

Thank you

ahmaddxb, 2021-08-10:
Not sure which is better but Im running with below setting and not having any issues currently. I think so56 0 might be enough to solve your issues. wificonfig 4/5 might do the same thing if you dont have second SSID setup in your config. If you only have one wifi SSID set in your config wificonfig 5 makes the most sense.

settings

so56 0 
so57 1 
wificonfig 4 (default)

popy2k14, 2021-08-10:
ok, thanks.
Will try that and come back here how it goes.

ascillato, 2021-08-10:
The recommended values are the defaults of Tasmota:

`so56 0`  (default)
`so57 1`  (default)
`wificonfig 4` (default)

ascillato2, 2021-08-10:
Moving this to Discussions.

closed 2021-08-10 by ascillato2