arendst/Tasmota #12677: ESP32: Berry Driver for TFMiniPlus: "Software reset CPU" several times during the night - sometimes it hangs completely and requires a power off/on

msperl, 2021-07-18:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): NodEmcu
- [ ] Tasmota binary firmware version number used: 9.5.0.3(tasmota) - 2021-07-17T13:25:49
  - [X] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: OTA
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
10:49:09.135 RSL: RESULT = {"NAME":"TF-MINI-PLUS","GPIO":[1,1792,0,1824,1,1,1,1,1,1,1,1,0,0,0,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
10:49:09.381 RSL: RESULT = {"Module":{"1":"ESP32-DevKit"}}
10:49:09.655 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO28":{"0":"None"},"GPIO29":{"0":"None"},"GPIO30":{"0":"None"},"GPIO31":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
not used
```
- [X] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
10:50:53.569 CMD: Status 0
10:50:53.578 RSL: STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_FC23B8","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":0,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
10:50:53.618 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T00:00:14","StartupUTC":"2021-07-18T09:50:39","Sleep":50,"CfgHolder":4617,"BootCount":129,"BCResetTime":"2021-07-02T14:51:21","SaveCount":178}}
10:50:53.654 RSL: STATUS2 = {"StatusFWR":{"Version":"9.5.0.3(tasmota)","BuildDateTime":"2021-07-17T13:25:49","Core":"1_0_7_1","SDK":"v3.3.5-138-g714f27f56","CpuFrequency":80,"Hardware":"ESP32-D0WDQ6","CR":"383/699"}}
10:50:53.689 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":2,"LogHost":"XXXX","LogPort":514,"SSId":["XXXXX",""],"TelePeriod":10,"Resolution":"558180C0","SetOption":["00C08009","2805C8000100060000005A0A000000000000","00000080","00006000","00000000"]}}
10:50:53.736 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":1242,"Free":1984,"Heap":182,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":3,"Features":["00000809","9FDAC787","00152001","B7F7BFCF","05DA9BC0","60360DC7","000840D2","20200000","0000002D"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,17,20,21,24,26,27,29,34,38,50,52","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,127"}}
10:50:53.776 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota_FCXXXX","IPAddress":"XXXX","Gateway":"XXXX","Subnetmask":"255.255.255.0","DNSServer":"XXXX","Mac":"XXXX","Webserver":2,"WifiConfig":0,"WifiPower":17.0}}
10:50:53.810 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_FC23B8","MqttUser":"tasmota","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
10:50:53.843 RSL: STATUS7 = {"StatusTIM":{"UTC":"2021-07-18T09:50:53","Local":"2021-07-18T10:50:53","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"05:06","Sunset":"20:45"}}
10:50:53.889 RSL: STATUS10 = {"StatusSNS":{"Time":"2021-07-18T10:50:53","ESP32":{"Temperature":53.3},"TFMiniPlus":{"Distance":1741,"Signal":3344,"Temperature":26.0},"Global":{"Temperature":53.3},"TempUnit":"C"}}
10:50:53.930 RSL: STATUS11 = {"StatusSTS":{"Time":"2021-07-18T10:50:53","Uptime":"0T00:00:14","UptimeSec":14,"Heap":173,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":99,"MqttCount":0,"Wifi":{"AP":1,"SSId":"XXX","BSSId"XXX","Channel":1,"Mode":"11n","RSSI":76,"Signal":-62,"LinkCount":1,"Downtime":"0T00:00:08"}}}
10:50:54.211 RSL: STATE = {"Time":"2021-07-18T10:50:54","Uptime":"0T00:00:15","UptimeSec":15,"Heap":182,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":99,"MqttCount":0,"Wifi":{"AP":1,"SSId":"XXX","BSSId":"XXX","Channel":1,"Mode":"11n","RSSI":76,"Signal":-62,"LinkCount":1,"Downtime":"0T00:00:08"}}
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
Console is lost in such cases...
```

### TO REPRODUCE
Keep the system running and eventually the system resets ( as long as the Berry Device Driver is loaded and running)

### EXPECTED BEHAVIOUR
Device should not restart or be stuck

### SCREENSHOTS
Here from the syslog server the restarts today:
```
Jul 18 05:42:26 tasmota_FC23B8-0952 ESP-RSL: INFO3 = {"Info3":{"RestartReason":"Vbat power on reset"}}
Jul 18 05:46:05 tasmota_FC23B8-0952 ESP-RSL: INFO3 = {"Info3":{"RestartReason":"Vbat power on reset"}}
Jul 18 05:51:21 tasmota_FC23B8-0952 ESP-RSL: INFO3 = {"Info3":{"RestartReason":"Vbat power on reset"}}
Jul 18 05:52:26 tasmota_FC23B8-0952 ESP-RSL: INFO3 = {"Info3":{"RestartReason":"Software reset CPU"}}
Jul 18 05:55:39 tasmota_FC23B8-0952 ESP-RSL: INFO3 = {"Info3":{"RestartReason":"Software reset CPU"}}
Jul 18 06:01:44 tasmota_FC23B8-0952 ESP-RSL: INFO3 = {"Info3":{"RestartReason":"Software reset CPU"}}
Jul 18 06:04:51 tasmota_FC23B8-0952 ESP-RSL: INFO3 = {"Info3":{"RestartReason":"Software reset CPU"}}
Jul 18 06:06:11 tasmota_FC23B8-0952 ESP-RSL: INFO3 = {"Info3":{"RestartReason":"Software reset CPU"}}
Jul 18 06:07:26 tasmota_FC23B8-0952 ESP-RSL: INFO3 = {"Info3":{"RestartReason":"Software reset CPU"}}
Jul 18 06:10:15 tasmota_FC23B8-0952 ESP-RSL: INFO3 = {"Info3":{"RestartReason":"Software reset CPU"}}
Jul 18 06:16:29 tasmota_FC23B8-0952 ESP-RSL: INFO3 = {"Info3":{"RestartReason":"Software reset CPU"}}
[... 62 lines cut ...]
Jul 18 06:16:35 tasmota_FC23B8-0952 ESP-RSL: RESULT = {"SerialReceived":"5959CA06460D200AFF"}
Jul 18 06:16:36 tasmota_FC23B8-0952 ESP-RSL: RESULT = {"SerialReceived":"5959CA06460D200AFF"}
```

### ADDITIONAL CONTEXT
The NodEmcu is connected directly to a 2.5A powersupply, so power should not be a problem.

The Lidar device (TFMiniPlus) is connected to:
* GND
* RX-Pin
* TX-Pin
* 5V

Note that the TFMiniPlus must have an on board 3V3 LDO as the levels on RX/TX are 3V3.

One observation is that quite ofter the last 10 messages are all SerialReceived messages (and the Sensor report interval is set to 10 seconds)

Attached the source for the berry driver: [tfminiplus.be.gz](https://github.com/arendst/Tasmota/files/6836031/tfminiplus.be.gz)

I admit it is not perfect, but for now it is working and reporting results as expected (except for the reboot/freeze)

Autoexec.be only includes `import tfminiplus` to start the driver on boot

One other note:
The mqtt server configured is connected via a WAN link (VPN) - as of now I have no mqtt server configured locally.

Jason2866, 2021-07-18:
How is the TFMiniPlus related to Berry? It is a standard driver.

msperl, 2021-07-18:
it is a Berry Driver - see the attachment - see also the discussion with regards to SerialBridge messages that I asked if there was a way to remove it.

Jason2866, 2021-07-18:
I will move to discussions

closed 2021-07-18 by Jason2866