arendst/Tasmota #12683: ENERGY_Total value suddenly jumps -2kwh

lolax80, 2021-07-18:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): Sonoff pow r2_
- [ x] Tasmota binary firmware version number used: __9.5.0
  - [ x] Pre-compiled
  - [ ] Self-compiled
- [ x] Flashing tools used: ESPflasher
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

8:58:50.164 MQT: stat/tasmota_9C1794/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
18:58:50.174 MQT: tele/tasmota_9C1794/STATE = {"Time":"2021-07-18T18:58:50","Uptime":"0T02:57:09","UptimeSec":10629,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":3,"POWER":"ON","Wifi":{"AP":2,"SSId":"das_ist_mein_Buero","BSSId":"C0:25:06:25:61:41","Channel":11,"RSSI":100,"Signal":-37,"LinkCount":1,"Downtime":"0T00:00:03"}}
18:58:50.182 MQT: tele/tasmota_9C1794/SENSOR = {"Time":"2021-07-18T18:58:50","ENERGY":{"TotalStartTime":"2021-07-17T00:00:00","Total":48.799,"Yesterday":2.059,"Today":3.640,"Period":5,"Power":296,"ApparentPower":296,"ReactivePower":0,"Factor":1.00,"Voltage":243,"Current":1.215}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

stat/tasmota_9C1794/STATUS = {"Status":{"Module":43,"DeviceName":"Balkonkraftwerk Garbenteich","FriendlyName":["Tasmota"],"Topic":"tasmota_9C1794","ButtonTopic":"0","Power":1,"PowerOnState":1,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2021-07-18:
Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Tasmota/issues/new/choose) in order to have more information so as to properly help you?

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

lolax80, 2021-07-18:
there are no further information
today and yesterday the value from ENERGY_Total dropped. today from 47 to 45 at 04pm. I can see on the mqtt client

ascillato, 2021-07-18:
we need the output of the Tasmota console to have some information in order to help you. As the previous issues where this exact same problem occurs, it was due to hardware failure or bad connection. We can troubleshoot that from the output of the console. Remember to do that with weblog set to 4 before attempting to reproduce the problem.

lolax80, 2021-07-19:
i have updated the status.
But until today I can't post you the weblog, because  when the drop comes im not online.
 i do not know when the problem comes

ascillato, 2021-07-19:
From your status 0, your device didn't have power 17 hours ago. Tasmota saves the energy today al midnight but if your device is not powered, the value is lost.

lolax80, 2021-07-19:
Ok i understand, but the total value dropped at 4pm

ascillato, 2021-07-19:
because your mains AC have a power loss, or because your device have restarted for any reason.

If you see in the STATUS 0 the RESTART REASON as POWER ON, that means that your device was left without power.

lolax80, 2021-07-19:
Ok, but if there was at 3pm a Power the Value total is the on power up the value from 0am?
Maybe tasmota can write the total value every hour in the Flash?

Jason2866, 2021-07-19:
Please use latest Tasmota version 9.5. There where changes/fixes

ascillato, 2021-07-19:
> Maybe tasmota can write the total value every hour in the Flash?

Tasmota is designed to make your device last as long as it can. If a firmware saves data very frequently to the flash, the flash wears out and eventually fails. That is why, those values are saved only at midnight.

Just a rough example calculation. The storage of this IoT devices are flash memories, not hard drives. Most of flash memories has 150.000 write cycles. (The more expensive memories have more, the cheapers have less).
If you have an event that writes data every 5 minutes, you will reach the 150.000 events in 1 year and a half. After that, your device will fail with random issues. To mitigate that, Tasmota has delayed savings (to gather more events and write them all at once) and also sequential settings blocks (so this data is saved in different parts of the memory to not wear the same part and distribute the usage). Tasmota as it is is very good. I have several devices running Tasmota since jan/2018 flawlessly.

If your power grid is not stable (like having microcuts or so), may be would be better to add an UPS to what your are measuring (depending on which are your use-case) Or just saving the reported data in your home automation software. All depends on what you need.

sfromis, 2021-07-19:
What I do to not lose energy monitoring data in case of a power outage is to save it in my backend server. Thus, I can correct data if i see a drop, only losing up to 5 minutes (teleperiod) of data. I could even detect this, and send commands to the device to restore last seen values upon power up.

lolax80, 2021-07-19:
Cool, how do you Monitor Irans How you send the last Right value.
I use mqtt with iobroker.

Jason2866, 2021-07-19:
Correct data: when you receive energy data every teleperiod time. If this packet (heartbeat) is missing you have a power drop.

sfromis, 2021-07-19:
The command EnergyReset can be used to set the values. Any command can be sent via MQTT. How to do in ioBroker is outside of what I've played with.

ascillato2, 2021-07-19:
Closing this issue as it has been answered.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2021-07-19 by ascillato2

katodev, 2022-06-01:
I have the same problem with tasmota version 9, 10, 11. If the load is greater 40 Watt the energy today value stops by 2000 kW/h and raise again with less 40 Watt load.
And the calculation is wrong too. With 30 W load the energy today value can't raise by 1200kW/h in 45 min.