arendst/Tasmota #12684: Upgradeing from 9.4 to 9.5 results in `RUL: No valid JSON` error

kquinsland, 2021-07-18:
### PROBLEM DESCRIPTION

(i believe this is a bug; there is no mention of needing to re-create rules due to format/parse change in release notes... that i can see. If this is not a bug and should be in the discussion/support form, let me know and i'll close / re-post there)

**Note:** while drafting this, I was able to resovle the issue by downgrading to 9.4.0. If this is _not_ a regression, let me  know...

-------

I rely on rules to [fill in some of the gaps](https://karlquinsland.com/2021/05/fixing-home-assistant-discovery-with-tasmota-on-the-treatlife-ds03/) in Home Assistant Discovery.

I recently upgraded the two devices in question to Tasmota 9.5.0. About a day later, I noticed that the two devices are listed in HA as `unavailable`. After quickly checking that the scheduled "announce yourself, please" message was being broadcast to all tasmotas over MQTT, I started looking at the rules on the device to see what's wrong.

Below is a snip from the console of one device. The same `No valid JSON` error is present on both devices in question.

```
09:17:39.044 HTP: Console
09:17:43.647 CMD: WebLog 4
09:17:43.649 SRC: WebConsole from 192.168.133.37
09:17:43.652 CMD: Grp 0, Cmnd 'WEBLOG', Idx 1, Len 1, Data '4'
09:17:43.657 MQT: stat/my_ceiling_fan/RESULT = {"WebLog":4}
09:17:46.366 WIF: Checking connection...
09:17:47.419 TYA: Send "55aa00000000ff"
09:17:47.422 {"TuyaReceived":{"Data":"55AA030000010104","Cmnd":0,"CmndData":"01"}}
09:17:47.442 TYA: Heartbeat
09:17:53.186 CMD: rule1
09:17:53.188 SRC: WebConsole from 192.168.133.37
09:17:53.191 CMD: Grp 0, Cmnd 'RULE', Idx 1, Len 0, Data ''
09:17:53.199 MQT: stat/my_ceiling_fan/RESULT = {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":344,"Free":167,"Rules":"on TuyaReceived#Data=55AA03070005030400010016 do publish2 stat/%topic%/speed 25 endon on TuyaReceived#Data=55AA03070005030400010117 do publish2 stat/%topic%/speed 50 endon on TuyaReceived#Data=55AA03070005030400010218 do publish2 stat/%topic%/speed 75 endon on TuyaReceived#Data=55AA03070005030400010319 do publish2 stat/%topic%/speed 100 endon"}}
09:17:57.200 SRC: WebConsole from 192.168.133.37
09:17:57.202 CMD: Grp 0, Cmnd 'RULE', Idx 2, Len 0, Data ''
09:17:57.210 MQT: stat/my_ceiling_fan/RESULT = {"Rule2":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":473,"Free":38,"Rules":"on SetOption19#Data=OFF do publish2 homeassistant/fan/%macaddr%/config {\"uniq_id\":\"DEADBE\",\"~\":\"%topic%/POWER1\",\"cmd_t\":\"cmnd/~\",\"pl_off\":\"OFF\",\"pl_on\":\"ON\",\"stat_t\":\"stat/~\",\"avty_t\":\"tele/%topic%/LWT\",\"pl_avail\":\"Online\",\"pl_not_avail\":\"Offline\",\"pct_cmd_t\":\"cmnd/%topic%/tuyasend4\",\"pct_cmd_tpl\":\"{%set v=value|int%}{%if v<=25%}3,0{%elif v<=50%}3,1{%elif v<=75%}3,2{%elif v<=100%}3,3{%endif%}\",\"pct_stat_t\":\"stat/%topic%/speed\",\"dev\":{\"cns\":[[\"mac\",\"%macaddr%\"]]}} endon"}}
09:17:57.221 RUL: No valid JSON ({"RULE2":{"STATE":"ON","ONCE":"OFF","STOPONERROR":"OFF","LENGTH":473,"FREE":38,"RULES":"ON SETOPTION19#DATA=OFF DO PUBLISH2 HOMEASSISTANT/FAN/%MACADDR%/CONFIG { "UNIQ_ID ": "DEADBE ", "~ ": "%TOPIC%/POWER1 ", "CMD_T ": "CMND/~ ", "PL_OFF ": "OFF ", "PL_ON ": "ON ", "STAT_T ": "STAT/~ ", "AVTY_T ": "TELE/%TOPIC%/LWT ", "PL_AVAIL ": "ONLINE ", "PL_NOT_AVAIL ": "OFFLINE ", "PCT_CMD_T ": "CMND/%TOPIC%/TUYASEND4 ", "PCT_CMD_TPL ": "{%SET V=VALUE|INT%}{%IF V<=25%}3,0{%ELIF V<=50%}3,1{%ELIF V<=75%}3,2{%ELIF V<=100%}3,3{%ENDIF%} ", "PCT_STAT_T ": "STAT/%TOPIC%/SPEED ", "DEV ":{ "CNS ":[[ "MAC ", "%MACADDR% "]]}} ENDON"}})
09:17:57.231 RUL: No valid JSON ({"RULE2":{"STATE":"ON","ONCE":"OFF","STOPONERROR":"OFF","LENGTH":473,"FREE":38,"RULES":"ON SETOPTION19#DATA=OFF DO PUBLISH2 HOMEASSISTANT/FAN/%MACADDR%/CONFIG { "UNIQ_ID ": "DEADBE ", "~ ": "%TOPIC%/POWER1 ", "CMD_T ": "CMND/~ ", "PL_OFF ": "OFF ", "PL_ON ": "ON ", "STAT_T ": "STAT/~ ", "AVTY_T ": "TELE/%TOPIC%/LWT ", "PL_AVAIL ": "ONLINE ", "PL_NOT_AVAIL ": "OFFLINE ", "PCT_CMD_T ": "CMND/%TOPIC%/TUYASEND4 ", "PCT_CMD_TPL ": "{%SET V=VALUE|INT%}{%IF V<=25%}3,0{%ELIF V<=50%}3,1{%ELIF V<=75%}3,2{%ELIF V<=100%}3,3{%ENDIF%} ", "PCT_STAT_T ": "STAT/%TOPIC%/SPEED ", "DEV ":{ "CNS ":[[ "MAC ", "%MACADDR% "]]}} ENDON"}})
09:17:57.240 RUL: No valid JSON ({"RULE2":{"STATE":"ON","ONCE":"OFF","STOPONERROR":"OFF","LENGTH":473,"FREE":38,"RULES":"ON SETOPTION19#DATA=OFF DO PUBLISH2 HOMEASSISTANT/FAN/%MACADDR%/CONFIG { "UNIQ_ID ": "DEADBE ", "~ ": "%TOPIC%/POWER1 ", "CMD_T ": "CMND/~ ", "PL_OFF ": "OFF ", "PL_ON ": "ON ", "STAT_T ": "STAT/~ ", "AVTY_T ": "TELE/%TOPIC%/LWT ", "PL_AVAIL ": "ONLINE ", "PL_NOT_AVAIL ": "OFFLINE ", "PCT_CMD_T ": "CMND/%TOPIC%/TUYASEND4 ", "PCT_CMD_TPL ": "{%SET V=VALUE|INT%}{%IF V<=25%}3,0{%ELIF V<=50%}3,1{%ELIF V<=75%}3,2{%ELIF V<=100%}3,3{%ENDIF%} ", "PCT_STAT_T ": "STAT/%TOPIC%/SPEED ", "DEV ":{ "CNS ":[[ "MAC ", "%MACADDR% "]]}} ENDON"}})
09:17:57.249 RUL: No valid JSON ({"RULE2":{"STATE":"ON","ONCE":"OFF","STOPONERROR":"OFF","LENGTH":473,"FREE":38,"RULES":"ON SETOPTION19#DATA=OFF DO PUBLISH2 HOMEASSISTANT/FAN/%MACADDR%/CONFIG { "UNIQ_ID ": "DEADBE ", "~ ": "%TOPIC%/POWER1 ", "CMD_T ": "CMND/~ ", "PL_OFF ": "OFF ", "PL_ON ": "ON ", "STAT_T ": "STAT/~ ", "AVTY_T ": "TELE/%TOPIC%/LWT ", "PL_AVAIL ": "ONLINE ", "PL_NOT_AVAIL ": "OFFLINE ", "PCT_CMD_T ": "CMND/%TOPIC%/TUYASEND4 ", "PCT_CMD_TPL ": "{%SET V=VALUE|INT%}{%IF V<=25%}3,0{%ELIF V<=50%}3,1{%ELIF V<=75%}3,2{%ELIF V<=100%}3,3{%ENDIF%} ", "PCT_STAT_T ": "STAT/%TOPIC%/SPEED ", "DEV ":{ "CNS ":[[ "MAC ", "%MACADDR% "]]}} ENDON"}})
09:17:57.261 RUL: No valid JSON ({"RULE2":{"STATE":"ON","ONCE":"OFF","STOPONERROR":"OFF","LENGTH":473,"FREE":38,"RULES":"ON SETOPTION19#DATA=OFF DO PUBLISH2 HOMEASSISTANT/FAN/%MACADDR%/CONFIG { "UNIQ_ID ": "DEADBE ", "~ ": "%TOPIC%/POWER1 ", "CMD_T ": "CMND/~ ", "PL_OFF ": "OFF ", "PL_ON ": "ON ", "STAT_T ": "STAT/~ ", "AVTY_T ": "TELE/%TOPIC%/LWT ", "PL_AVAIL ": "ONLINE ", "PL_NOT_AVAIL ": "OFFLINE ", "PCT_CMD_T ": "CMND/%TOPIC%/TUYASEND4 ", "PCT_CMD_TPL ": "{%SET V=VALUE|INT%}{%IF V<=25%}3,0{%ELIF V<=50%}3,1{%ELIF V<=75%}3,2{%ELIF V<=100%}3,3{%ENDIF%} ", "PCT_STAT_T ": "STAT/%TOPIC%/SPEED ", "DEV ":{ "CNS ":[[ "MAC ", "%MACADDR% "]]}} ENDON"}})
09:17:58.395 TYA: Send "55aa00000000ff"
09:17:58.398 {"TuyaReceived":{"Data":"55AA030000010104","Cmnd":0,"CmndData":"01"}}
09:17:58.417 TYA: Heartbeat
09:18:00.419 TYA: Send "55aa001c0008011507120912000774"
09:18:06.358 WIF: Checking connection...
09:18:09.412 TYA: Send "55aa00000000ff"
09:18:09.416 {"TuyaReceived":{"Data":"55AA030000010104","Cmnd":0,"CmndData":"01"}}
09:18:09.435 TYA: Heartbeat
09:18:18.980 CMD: rule3
09:18:18.982 SRC: WebConsole from 192.168.133.37
09:18:18.984 CMD: Grp 0, Cmnd 'RULE', Idx 3, Len 0, Data ''
09:18:18.990 MQT: stat/my_ceiling_fan/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
09:18:20.426 TYA: Send "55aa00000000ff"
09:18:20.429 {"TuyaReceived":{"Data":"55AA030000010104","Cmnd":0,"CmndData":"01"}}
09:18:20.448 TYA: Heartbeat
```

The original value for `rule2`:

```
rule2 on SetOption19#Data=OFF do publish2 homeassistant/fan/%macaddr%/config <.json_here.> endon
```

where `json_here` is a compact version of:

```json
{
	"uniq_id": "AABBCC",
	"~": "%topic%/POWER1",
	"cmd_t": "cmnd/~",
	"pl_off": "OFF",
	"pl_on": "ON",
	"stat_t": "stat/~",
	"avty_t": "tele/%topic%/LWT",
[... 111 lines cut ...]
09:52:45.820 MQT: homeassistant/light/EFBABE_LI_7/config =  (retained)
09:52:45.823 MQT: homeassistant/switch/EFBABE_RL_8/config =  (retained)
09:52:45.827 MQT: homeassistant/light/EFBABE_LI_8/config =  (retained)
09:52:45.830 MQT: homeassistant/sensor/EFBABE_status/config =  (retained)
09:52:45.833 LOG: Home Assistant MQTT Discovery disabled.
09:52:45.843 MQT: tasmota/discovery/DEADBEEFBABE/config = {"ip":"192.168.133.191","dn":"My Ceiling Fan","fn":["Fan","Light",null,null,null,null,null,null],"hn":"my_ceiling_fan-3274","mac":"DEADBEEFBABE","md":"Tuya MCU","ty":1,"if":0,"ofln":"Offline","onln":"Online","state":["OFF","ON","TOGGLE","HOLD"],"sw":"9.4.0","t":"my_ceiling_fan","ft":"%prefix%/%topic%/","tp":["cmnd","stat","tele"],"rl":[1,2,0,0,0,0,0,0],"swc":[-1,-1,-1,-1,-1,-1,-1,-1],"swn":[null,null,null,null,null,null,null,null],"btn":[0,0,0,0,0,0,0,0],"so":{"4":0,"11":0,"13":0,"17":1,"20":0,"30":0,"68":0,"73":0,"82":0,"114":0,"117":0},"lk":1,"lt_st":1,"sho":[0,0,0,0],"ver":1} (retained)
09:52:45.850 MQT: tasmota/discovery/DEADBEEFBABE/sensors = {"sn":{"Time":"2021-07-18T09:52:45"},"ver":1} (retained)
[... cut ...]
09:52:46.458 {"TuyaReceived":{"Data":"55AA030000010104","Cmnd":0,"CmndData":"01"}}
```

**Note:** I obtained the above console output by downgrading one of the affected devices to 9.4.0 **after** I had started drafting this post; I believe the is a confirmed regression, then!

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2021-07-18:
Fixed in development. See release notes.

kquinsland, 2021-07-18:
Closing as this is a dupe of https://github.com/arendst/Tasmota/issues/12440

closed 2021-07-18 by kquinsland