arendst/Tasmota #12716: Sonoff iFan03 interlocking problem

hassuser, 2021-07-21:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
If the fan is turn on in any speed and the light relay is then turn on, the speed relay is then turn off. {"Interlock":"OFF","Groups":"1,2,3,4"}

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____iFan03
- [ ] Tasmota binary firmware version number used: _____Tasmota 9.5.0
  - [ x] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
{"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
12:23:45.993 CMD: weblog 4
12:23:45.000 MQT: stat/bedroom_fan/RESULT = {"WebLog":4}
12:23:46.652 CFG: Saved to flash at FB, Count 4069, Bytes 4096
12:23:47.757 WIF: Checking connection...
12:24:04.715 SRC: MQTT
12:24:04.718 CMD: Grp 0, Cmnd 'FANSPEED', Idx 1, Len 1, Data '1'
12:24:04.725 MQT: stat/bedroom_fan/RESULT = {"FanSpeed":1}
12:24:05.622 CFG: Saved to flash at FA, Count 4070, Bytes 4096
12:24:07.778 WIF: Checking connection...
12:24:10.637 CFG: Saved to flash at F9, Count 4071, Bytes 4096
12:24:18.348 SRC: MQTT
12:24:18.351 CMD: Grp 0, Cmnd 'POWER', Idx 1, Len 2, Data 'ON'
12:24:18.357 MQT: stat/bedroom_fan/RESULT = {"POWER1":"ON"}
12:24:18.360 MQT: stat/bedroom_fan/POWER1 = ON
12:24:18.465 SRC: MQTT
12:24:18.467 CMD: Grp 0, Cmnd 'FANSPEED', Idx 1, Len 1, Data '0'
12:24:18.474 MQT: stat/bedroom_fan/RESULT = {"FanSpeed":0}
12:24:18.620 CFG: Saved to flash at F8, Count 4072, Bytes 4096
12:24:27.791 WIF: Checking connection...

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
turn on the ifan03 in any speed and then turn on the light relay, the speed relay is then turn off

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

hassuser, 2021-08-15:
The issue still there

arendst, 2021-08-16:
This is not a Tasmota issue as can be seen clearly in the log; All commands are requested over MQTT so your MQTT environment is at fault.

To assure you my ifan03 is working perfectly well.

hassuser, 2021-08-16:
I don’t understand how the MQTT environment is at fault when I can
reproduce it directly on the tasmota gui? My MQTT environment has been
working fine for a while now. Never mind.

closed 2021-08-21 by github-actions[bot]