arendst/Tasmota #12717: [ESP32] Buzzer in PWM-Mode (Setoption111 1) lets the ESP reboot

Schnup89, 2021-07-21:
### PROBLEM DESCRIPTION
ESP32-D0WDQ6
A Piezo is working as expected on gpio16 if it is declared as "PWM".
The Piezo is not working if i switch the GPIO from "PWM" zu "Buzzer" and enable "Setoption111 1".
If the Command "Buzzer" with a Parameter is executed the device will restart.

### REQUESTED INFORMATION
- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Tasmota-Unknown (LivyRingG2:  https://github.com/Schnup89/LIVY_RING_G2-Tasmota/blob/main/README.md)
- [x] Tasmota binary firmware version number used:9.5.0.3
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: ESP32-Flashing Tools
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
18:03:53.657 RSL: RESULT = {"NAME":"ESP32-LivyRingG2","GPIO":[1,1,7552,1,449,7936,0,0,609,226,641,0,480,8000,608,640,0,448,450,161,0,0,1,224,0,0,0,0,7584,257,0,160,0,0,0,0],"FLAG":0,"BASE":1}
18:03:53.880 RSL: RESULT = {"Module":{"1":"ESP32-DevKit"}}
18:03:54.143 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"416":"PWM1"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"256":"Relay_i1"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO28":{"0":"None"},"GPIO29":{"0":"None"},"GPIO30":{"0":"None"},"GPIO31":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}
```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
no rules

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
18:05:23.453 RSL: STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota","Tasmota2"],"Topic":"tasmota_0B50DC","ButtonTopic":"0","Power":0,"PowerOnState":1,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
18:05:23.490 BRY: GC from 3864 to 2261 bytes (in 1 ms)
18:05:23.496 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T00:06:23","StartupUTC":"2021-07-21T16:59:00","Sleep":50,"CfgHolder":4617,"BootCount":46,"BCResetTime":"2021-07-21T16:24:59","SaveCount":163}}
18:05:23.527 RSL: STATUS2 = {"StatusFWR":{"Version":"9.5.0.3(tasmota)","BuildDateTime":"2021.07.21 12:10:59","Core":"1_0_7_3","SDK":"v3.3.5-138-g714f27f56","CpuFrequency":80,"Hardware":"ESP32-D0WDQ6","CR":"388/699"}}
18:05:23.544 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["TDHome",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000080","20006000","00000001"]}}
18:05:23.579 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":1247,"Free":1856,"Heap":187,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":2,"Features":["00000407","9FDAC787","00152001","B7F7BFCF","05DA9BC0","60360DC7","000840D2","20200000","0000002D"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,17,20,21,24,26,27,29,34,38,50,52","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,127"}}
18:05:23.625 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota_0B50DC-4316","IPAddress":"192.168.10.104","Gateway":"192.168.10.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.10.1","Mac":"24:0A:C4:0B:50:DC","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
18:05:23.646 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0B50DC","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
18:05:23.666 RSL: STATUS7 = {"StatusTIM":{"UTC":"2021-07-21T17:05:23","Local":"2021-07-21T18:05:23","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"05:10","Sunset":"20:42"}}
18:05:23.688 RSL: STATUS10 = {"StatusSNS":{"Time":"2021-07-21T18:05:23","ESP32":{"Temperature":53.3},"Global":{"Temperature":53.3},"TempUnit":"C"}}
18:05:23.700 RSL: STATUS11 = {"StatusSTS":{"Time":"2021-07-21T18:05:23","Uptime":"0T00:06:23","UptimeSec":383,"Heap":186,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER1":"OFF","POWER2":"OFF","Dimmer":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"TDHome","BSSId":"B0:BE:76:F7:05:3F","Channel":8,"Mode":"11n","RSSI":100,"Signal":-45,"LinkCount":1,"Downtime":"0T00:00:07"}}}

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
18:06:53.975 CMD: buzzer 2
18:06:53.977 SRC: WebConsole from 192.168.10.110
18:06:53.979 CMD: Grp 0, Cmnd 'BUZZER', Idx 1, Len 1, Data '2'
18:06:53.982 BUZ: Count 2(4), Time 1/1, Tune 0x00000000(0x00000000), Size 0, Mode 1
18:06:53.985 RSL: RESULT = {"Buzzer":"Done"}
***then restart***
```

### TO REPRODUCE
On ESP32, enable setoption111 1, configure a pin to "buzzer", execute "buzzer 2"

### EXPECTED BEHAVIOUR
No reboot, Sound from Piezo (through PWM)

Jason2866, 2021-07-21:
Please try a other gpio than 16 or 17. This gpios have restrictions

Schnup89, 2021-07-21:
@Jason2866 Thanks for your answer, unfortunately it's not possible because the ESP is fixed on the PCB.

[image]

Is there an other thing I could try?

Jason2866, 2021-07-21:
@Schnup89  I answered in your github. Not really a Tasmota issue.

Jason2866, 2021-07-22:
Tried Buzzer on a DevKit on GPIO12. It is crashing
```
17:50:18.563 CMD: Buzzer 2

17:50:18.571 MQT: stat/sonoff-362330/RESULT = {"Buzzer":"Done"}

/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:1457 (xQueueGenericReceive)- assert failed!

abort() was called at PC 0x4008ba2d on core 1

ELF file SHA256: 0000000000000000

Backtrace: 0x4008a864:0x3ffb1e30 0x4008abc1:0x3ffb1e50 0x4008ba2d:0x3ffb1e70 0x4012023f:0x3ffb1eb0 0x400d3d17:0x3ffb1ed0 0x400e2f61:0x3ffb1ef0 0x400e364b:0x3ffb1f10 0x400f012c:0x3ffb1f30 0x400e5e87:0x3ffb1f50 0x400fc5b3:0x3ffb1f70 0x400fc61b:0x3ffb1f90 0x40121e59:0x3ffb1fb0 0x4008bcfe:0x3ffb1fd0

Rebooting...

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

configsip: 0, SPIWP:0xee

clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00

mode:DOUT, clock div:2

load:0x3fff0018,len:4

load:0x3fff001c,len:160

ho 0 tail 12 room 4

load:0x40078000,len:10492

load:0x40080400,len:5664

entry 0x40080678

E (452) psram: PSRAM ID read error: 0xffffffff

E (452) spiram: SPI RAM enabled but initialization failed. Bailing out.

00:00:00.002 HDW: ESP32-D0WDQ6

00:00:00.009 UFS: FlashFS mounted with 304 kB free

00:00:00.021 CFG: Loaded from File, Count 23

00:00:00.036 BRY: Berry initialized, RAM used=1712

00:00:00.041 BRY: no 'preinit.be'

00:00:00.043 Project sonoff - Tasmota Version 9.5.0.3(UNIVERSAL)-1_0_7_3(2021-07-22T17:38:08)
```

@arendst

Jason2866, 2021-07-28:
Buzzer is working now

closed 2021-07-28 by Jason2866