arendst/Tasmota #12724: Scripting. =>tmsend crashes esp8266 being executed in >E section or in subroutine

nikvostro, 2021-07-22:
### PROBLEM DESCRIPTION
The following command crashes esp8266 being executed in >E section or in subroutine (or manually in Console with "script >"):
`=>tmsend somesampletext`
But this problem doesn't occur on ESP32 devices - tested on ESP32CAM and Lillygo TTGO T-Display.

`tmsend [text]` command works fine in console and in other places (sections) of Script:
```lua
16:30:53.731 CMD: tmsend testmess
16:30:54.248 RSL: RESULT = {"TmSend":"Done"}
```

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): Sonoff ZbBridge and also crashes on clean ESP-12E. Works on ESP32CAM and Lillygo TTGO T-Display 
- [ ] Tasmota binary firmware version number used: 
  - [ ] Pre-compiled
  - [ ] Self-compiled: tested with dev 9.5.0.3 and master 9.5.0
- [ ] Flashing tools used: VSC/PlatformIO
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
14:15:00.351 RSL: RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
14:15:00.557 RSL: RESULT = {"Module":{"18":"Generic"}}
14:15:00.811 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"224":"Relay1"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```

- [ ] Sample Script used:
```lua
>D
msg=""
pwrstate=0

>E
pwrstate=Power1#State
if %pwrstate%==1 {
=>tmsend pwrstate1!
}

>S

#ALERT(msg)
print message=%msg%
=>tmsend tstmsg
```

- [ ] Provide the output of this command: `Status 0`:
```lua
14:15:38.222 RSL: STATUS = {"Status":{"Module":18,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_D3444F","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
14:15:38.248 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software Watchdog","Uptime":"0T00:03:08","StartupUTC":"2021-07-22T13:12:30","Sleep":50,"CfgHolder":4617,"BootCount":10,"BCResetTime":"2021-07-22T13:39:58","SaveCount":40,"SaveAddress":"F4000"}}
14:15:38.281 RSL: STATUS2 = {"StatusFWR":{"Version":"9.5.0.3(tasmota)","BuildDateTime":"2021-07-22T15:38:12","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"376/699"}}
14:15:38.300 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["domuk",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008001","2805C8000100060000005A0A000000000000","00000080","00006000","00000000"]}}
14:15:38.330 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":672,"Free":328,"Heap":17,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440E0","FlashFrequency":40,"FlashMode":3,"Features":["00000809","87DAC787","043E8001","000000CF","010013C0","C000F981","0000C004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,40,45","Sensors":"1,2,3,4,5,6"}}
14:15:38.361 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota_D3444F-1103","IPAddress":"192.168.11.104","Gateway":"192.168.11.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.11.1","Mac":"18:FE:34:D3:44:4F","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
14:15:38.384 RSL: STATUS7 = {"StatusTIM":{"UTC":"2021-07-22T13:15:38","Local":"2021-07-22T14:15:38","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"05:11","Sunset":"20:41"}}
14:15:38.402 RSL: STATUS10 = {"StatusSNS":{"Time":"2021-07-22T14:15:38"}}
14:15:38.409 RSL: STATUS11 = {"StatusSTS":{"Time":"2021-07-22T14:15:38","Uptime":"0T00:03:08","UptimeSec":188,"Heap":17,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":132,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"domuk","BSSId":"F0:79:59:36:57:40","Channel":3,"Mode":"11n","RSSI":52,"Signal":-74,"LinkCount":1,"Downtime":"0T00:00:03"}}}
14:15:38.445 RSL: STATUS12 = {"StatusSTK":{"Exception":4,"Reason":"Software Watchdog","EPC":["4024e8ae","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["4024e9aa","4024e9aa","4024375e","402645f0","4026459c","4026417a","4024387c","402439ba","401011e4","40243a6d","40252e80","40252e80","402164a1","40101b77","4021680b","40216889","40220c00","4028cfa4","4028cf90","4022b858","40217479","4022b42a","4029e8c6","402475fa","40254d00","401018f3","40101b77","40294e05","40294da6","40207e39","4022b61c"]}}
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
14:22:52.232 CMD: weblog 4
14:22:52.237 RSL: RESULT = {"WebLog":4}
14:22:52.241 Script: performs "tmsend pwrstate1!"

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Soft WDT reset

>>>stack>>>

ctx: cont
sp: 3ffff600 end: 5c5c5c5c offset: 01a0
3ffff7a0:  00000078 00000000 00000001 4024e9aa
3ffff7b0:  00000000 00000000 3fff581c 4024e9aa
3ffff7c0:  00000008 00067a6a 3fff5544 4024375e
3ffff7d0:  00000078 00000005 13cb2fa7 00000000
3ffff7e0:  402645f0 4026459c 3fff581c 4026417a
3ffff7f0:  3fff5d1c 3fff713c 3fff581c 3fff5544
3ffff800:  3fff713c 00000001 3fff5544 4024387c
3ffff810:  3fff713c 3fff905c 00000000 402439ba
..................................
```

### TO REPRODUCE
1. user_config_override.h
```lua
#define USE_TELEGRAM 

#undef USE_RULES
#ifndef USE_SCRIPT
#define USE_SCRIPT  // adds about 17k flash size, variable ram size
#endif
```

2. Edit script as below:
```lua
>D
msg=""
pwrstate=0

>E
pwrstate=Power1#State
if %pwrstate%==1 {
=>tmsend pwrstate1!
}

>S

#ALERT(msg)
print message=%msg%
=>tmsend tstmsg
```

3. Try executing one of the following commands:
3.1 alert test
3.2 power1 1
3.3 script>=>tmsend test

### EXPECTED BEHAVIOUR
Text message should be sent

### SCREENSHOTS
N/A

### ADDITIONAL CONTEXT
N/A

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2021-07-22:
It looks like it is running out of stack space when telegram is executed from script.

I see no easy solution.

nikvostro, 2021-07-22:
> It looks like it is running out of stack space when telegram is executed from script.
> 
> I see no easy solution.

Hello Theo! Glad to see you there here in my bug :)
OK, more complex solutions are also welcome! Do you have any ideas?

p.s. I'm making not smart but a bit clever home without any automation servers - decentralized ecosystem. It's quite important to have most of Tasmota features available for every device making them independant of each other. Indeed I understand the physical MCU limitations but nevertheless I beleive in magic of coding :)

sfromis, 2021-07-25:
Tasmota is specifically designed for operation via MQTT, and not for devices being maximally independent. It outside the concept to insist on multiple "big" features existing on the same ESP8266. Using ESP32 may provide more headroom, as you've observed.

nikvostro, 2021-07-25:
> Tasmota is specifically designed for operation via MQTT, and not for devices being maximally independent. It is outside the concept to insist on multiple "big" features existing on the same ESP8266. Using ESP32 may provide more headroom, as you've observed. 

Indeed I do understand the concept. For my case I'm don't want to have additional MQTT server/broker as my automation needs are potentially covered by exciting rules/scripting features. Sure MQTT setup is much more easy to control in one place and it's endless flexible. But it's rather redundant for me neither bringing value.
Telegram integration is really useful thing to bring control to end-user without any additional manipulations with LAN/WAN port forwarding, using VPN etc.
I'm trying to use factory modules (devices) maximally. Sure ESP32 would be much reasonable to be used here but esp8266 (and wb3s) are not pin compatible with esp32. It would be great to use more powerful esp32 MCU but this is not a case unfortunately because I'm using ZbBridge... for other wb3s devices it was not a problem for me to swap MCUs.

sfromis, 2021-07-25:
Using the new ESP32-C3 chip, you can actually get a module which is largely pin compatible with WB3S. See [this](https://www.youtube.com/channel/UC5ZdPKE2ckcBhljTc2R_qNA) video.

Do note that the Tasmota support for ESP32-C3 is in a very early stage (but largely compatible with the "classic" ESP32), meaning that while the basics seem to work, the risk of running into issues is very much there.

nikvostro, 2021-07-27:
> Using the new ESP32-C3 chip, you can actually get a module which is largely pin compatible with WB3S. See [this](https://www.youtube.com/channel/UC5ZdPKE2ckcBhljTc2R_qNA) video.
>
> Do note that the [...]

Will try C3 later on on closer to stable stage... resoldering...

Another one example of unstable work of subroutines. ESP8266.
Command IRHVAC executes correctly but RSL is returned corrupted each time.

```lua
#ACSET(par)
modenum=par%10
fan=((par%100)-modenum)/10
temp=(par-(fan*10)-modenum)/100
switch modenum
case 0
mode="Cool"
case 1
mode="Heat"
case 2
mode="Fan"
ends  
print %temp% %fan% %modenum% %mode%
->IRHVAC {"Vendor":"FUJITSU_AC","Model":1,"Mode":"%mode%","Power":"On","Temp":%temp%,"FanSpeed":"%fan%"}

#ACOFF
print acoff
temp=0
```

acset sub testing:
```lua
10:14:54.977 CMD: acset 2533
10:14:54.983 25.00 3.00 3.00 
10:14:54.138 RSL: RESULT = {"ACSET":"{"V��?@��?FUJITSU_AC"}
10:15:01.115 CMD: acoff
10:15:01.118 acoff
10:15:01.202 RSL: RESULT = {"ACOFF":"��?1��?���?��%@"}
10:15:14.503 CMD: acset 2400
10:15:14.509 24.00 0.00 0.00 Cool
10:15:14.663 RSL: RESULT = {"ACSET":"{"V��?@��?FUJITSU_AC"}
10:20:05.555 CMD: acoff
10:20:05.559 acoff
10:20:05.642 RSL: RESULT = {"ACOFF":"��?1��?���?��%@"}
```

arendst, 2021-07-27:
Just scanned my tasmota source code but I'm sorry to say I see nowhere that a command like ``acset`` is present.

Do I miss something?

nikvostro, 2021-07-27:
> Just scanned my tasmota source code but I'm sorry to say I see nowhere that a command like `acset` is present.
> 
> Do I miss something?

it's all about subroutines see 1 comment above please - `#ACOFF` sub

arendst, 2021-07-27:
Ah it's scripting. I guess you'll have to escape the quotes in your command line but as I'm not a script expert @gemu2015 may know better.

nikvostro, 2021-07-27:
> Ah it's scripting. I guess you'll have to escape the quotes in your command line but as I'm not a script expert @gemu2015 may know better.

I guess I can't avoid quotas for IRHVAC JSON, but let's try anyway:

```lua
#ACOFF
print acoff
temp=0
->IRHVAC {Vendor:FUJITSU_AC,Model:1,Power:Off}

11:24:28.603 script compressed to 751 bytes = 60 %
11:24:28.606 Script: nv=11, tv=2, vns=106, ram=289
11:24:44.191 CMD: acoff
11:24:44.195 acoff
11:24:44.200 RSL: RESULT = {"ACOFF":""}
11:24:54.240 CMD: acset 2533
11:24:54.245 25.00 3.00 3.00 
11:24:54.399 RSL: RESULT = {"ACSET":"{"V��?@��?FUJITSU_AC"}
11:24:58.445 CMD: acoff
11:24:58.448 acoff
11:24:58.454 RSL: RESULT = {"ACOFF":""}
```

nikvostro, 2021-07-27:
Removed quotas for `acset` sub:
```lua
#ACSET(par)
modenum=par%10
fan=((par%100)-modenum)/10
temp=(par-(fan*10)-modenum)/100
switch modenum
case 0
mode="Cool"
case 1
mode="Heat"
case 2
mode="Fan"
ends  
print %temp% %fan% %modenum% %mode%
+>IRHVAC {Vendor:FUJITSU_AC,Model:1,Mode:%mode%,Power:On,Temp:%temp%,FanSpeed:%fan%}

11:32:01.772 script compressed to 735 bytes = 60 %
11:32:01.775 Script: nv=11, tv=2, vns=106, ram=289
11:32:04.951 CMD: acset 2533
11:32:04.957 25.00 3.00 3.00 
11:32:04.962 RSL: RESULT = {"IRHVAC":"Invalid JSON"}
11:32:04.965 RSL: RESULT = {"ACSET":"{Vendor:FUJd�%@0��?!��?���?��%@ "}
```

gemu2015, 2021-07-27:
try this:

+>IRHVAC {Vendor:"FUJITSU_AC",Model:1,Mode:"%mode%",Power:On,Temp:%0temp%,FanSpeed:%0fan%}

by the way. i also replaced my iobroker raspberry setup with tasmota scripting.

i am using esp8266 devices for all sensors and actors. most of these devices have dedicated web pages with google graphs for local sensors. additionally all variables are made public by global variables.

then i have several esp32 devices with displays that collect all these data, show them on webui with extended graphs, store them on SD card and every week send the collected data as email attachments.

nikvostro, 2021-07-28:
```lua
print +>IRHVAC {Vendor:"FUJITSU_AC",Model:1,Mode:"%mode%",Power:On,Temp:%0temp%,FanSpeed:%0fan%}
+>IRHVAC {Vendor:"FUJITSU_AC",Model:1,Mode:"%mode%",Power:On,Temp:%0temp%,FanSpeed:%0fan%}

12:13:35.769 Script: nv=11, tv=2, vns=106, ram=289
12:13:38.443 24.00 0.00 0.00 Cool
12:13:38.445 +>IRHVAC {Vendor:"FUJITSU_AC",Model:1,Mode:"Cool",Power:On,Temp:24,FanSpeed:0}
12:13:38.450 RSL: RESULT = {"IRHVAC":"Invalid JSON"}
12:13:38.453 RSL: RESULT = {"ACSET":"{Vendor:"FUd�%@0��?!��?���?��%@ "}
```

Good approach with using esp32 as a core device! And global variables are in my plan for imlementation :)
In my case I've decided (not to dive into Android programming) to utilize [Tasker](https://tasker.joaoapps.com/). I'm don't need to collect and analyze any data (as for now:)).
Here are my 2 screens for controlling Light and ACs:
[image] [image]

gemu2015, 2021-07-28:
sorry other quotes missing:

+>IRHVAC {"Vendor":"FUJITSU_AC","Model":1,"Mode":"%mode%","Power":"On","Temp":%1temp%,"FanSpeed":%0fan%}

example from docs
IRhvac {"Vendor":"Mitsubishi_Heavy_152", "Power":"On","Mode":"Hot","FanSpeed":3,"Temp":22.5}

nikvostro, 2021-07-28:
I guess %1 forces 1/10 decimal precision level in parameter definition...
So the result is still same:
```lua
12:52:01.218 24.00 0.00 0.00 Cool
12:52:01.219 +>IRHVAC {"Vendor":"FUJITSU_AC","Model":1,"Mode":"Cool","Power":"On","Temp":24.0,"FanSpeed":0}
12:52:01.372 RSL: RESULT = {"IRHVAC":{"Vendor":"FUJITSU_AC","Model":1,"Mode":"Cool","Power":"On","Celsius":"On","Temp":24,"FanSpeed":"Auto","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}
12:52:01.375 RSL: RESULT = {"ACSET":"{"V��?@��?FUJITSU_AC"}
```

gemu2015, 2021-07-28:
yes is decimal precision 

no, now you don't get invalid json.
so it should send correctly.
the command ACSET however returns an invalid result.
i dont use irhvac, so i cant test directly.
why do you use +> and not -> ? this may cause weird effects by unwanted recursions.

nikvostro, 2021-07-28:
This worked initially actually - command was executed properly, but idea was to fix RSL JSON. My app is getting a bit confused by getting this response for HTTP request. Sure it doesn't make big sense here but it can be quite critical in other appliances.
> Command IRHVAC executes correctly but RSL is returned corrupted each time.

I faced with incorrect behaviour somewhere by using => (or ->) and replacing it with +> has solved the problem. To be honest I do understand recursion in programming but I don't understand what does mean "+> Execute recursion enabled" in Scripting. Nothing explained in Docs :(

gemu2015, 2021-07-29:
=> execute tasmota cmd with MQTT output enabled
-> execute tasmota cmd with MQTT output disabled
+> execute tasmota cmd with MQTT output enabled and recursion enabled.

exampe, if you execute a tasmota cmd in an >E section and this cmd itself executes >E you will get an infinite loop.
this is disabled normally and enabled by the +> in case you know what you are doing 

feel free to edit the tasmota documents to clarify things!

arendst, 2021-08-03:
Good news.

Starting with ArduinoEsp8266 core 3.x more user RAM is available. This allows more heap space too which solves the OP's issue with executing telegram from a script.

To test use the latest development environment and enable Core 3.0.2

closed 2021-08-04 by ascillato2