arendst/Tasmota #12729: Libraries

uviniciusc, 2021-07-22:
hi everyone i have this problem,

Arduino: 1.8.10 (Windows 10), Placa:"Generic ESP8285 Module, 80 MHz, Flash, Disabled (new aborts on oom), Disabled, All SSL ciphers (most compatible), 32KB cache + 32KB IRAM (balanced), Use pgm_read macros for IRAM/PROGMEM, dtr (aka nodemcu), 26 MHz, 1MB (FS:64KB OTA:~470KB), 2, nonos-sdk 2.2.1+100 (190703), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

Foram encontradas múltiplas bibliotecas para "TasmotaSerial.h"
Usado: C:\Users\vico_\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.1\libraries\TasmotaSerial-2.4.1
Não usado: C:\Users\vico_\Documents\Ambiente
Foram encontradas múltiplas bibliotecas para "ESP8266HTTPClient.h"
Usado: C:\Users\vico_\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.1\libraries\ESP8266HTTPClient
Foram encontradas múltiplas bibliotecas para "ArduinoJson.h"
Usado: C:\Users\vico_\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.1\libraries\ArduinoJson-5.13.4
Não usado: C:\Users\vico_\Documents\Ambiente
In file included from C:\Users\vico_\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.1\libraries\NeoPixelBus-2.5.0.09\src/NeoPixelBus.h:68,
Foram encontradas múltiplas bibliotecas para "ESP8266mDNS.h"
                 from C:\Users\vico_\Documents\Tasmota-7.2.0\tasmota\xlgt_01_ws2812.ino:48:
Usado: C:\Users\vico_\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.1\libraries\ESP8266mDNS
Foram encontradas múltiplas bibliotecas para "IRremoteESP8266.h"
Usado: C:\Users\vico_\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.1\libraries\IRremoteESP8266-2.7.1
C:\Users\vico_\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.1\libraries\NeoPixelBus-2.5.0.09\src/internal/NeoEsp8266DmaMethod.h:42:10: fatal error: i2s.h: No such file or directory
Não usado: C:\Users\vico_\Documents\Ambiente
Foram encontradas múltiplas bibliotecas para "ihx.h"
Usado: C:\Users\vico_\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.1\libraries\C2Programmer-1.0.0
Não usado: C:\Users\vico_\Documents\Ambiente
   42 | #include "i2s.h"
Foram encontradas múltiplas bibliotecas para "EEPROM.h"
Usado: C:\Users\vico_\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.1\libraries\EEPROM
Foram encontradas múltiplas bibliotecas para "ESP8266WebServer.h"
Usado: C:\Users\vico_\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.1\libraries\ESP8266WebServer
      |          ^~~~~~~
Foram encontradas múltiplas bibliotecas para "ESP8266httpUpdate.h"
Usado: C:\Users\vico_\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.1\libraries\ESP8266httpUpdate
compilation terminated.

Foram encontradas múltiplas bibliotecas para "Ticker.h"
Usado: C:\Users\vico_\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.1\libraries\Ticker
Foram encontradas múltiplas bibliotecas para "PubSubClient.h"
Usado: C:\Users\vico_\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.1\libraries\PubSubClient-EspEasy-2.7.12
Não usado: C:\Users\vico_\Documents\Ambiente
Foram encontradas múltiplas bibliotecas para "esp-knx-ip.h"
Usado: C:\Users\vico_\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.1\libraries\esp-knx-ip-0.5.2
Não usado: C:\Users\vico_\Documents\Ambiente
Foram encontradas múltiplas bibliotecas para "ESP8266WiFi.h"
Usado: C:\Users\vico_\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.1\libraries\ESP8266WiFi
Foram encontradas múltiplas bibliotecas para "Wire.h"
Usado: C:\Users\vico_\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.1\libraries\Wire
Foram encontradas múltiplas bibliotecas para "DNSServer.h"
Usado: C:\Users\vico_\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.1\libraries\DNSServer
Foram encontradas múltiplas bibliotecas para "NeoPixelBus.h"
Usado: C:\Users\vico_\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.1\libraries\NeoPixelBus-2.5.0.09
Não usado: C:\Users\vico_\Documents\Ambiente
exit status 1
Erro compilando para a placa Generic ESP8285 Module

Este relatório teria mais informações com
"Mostrar a saida detalhada durante a compilação"
opção pode ser ativada em "Arquivo -> Preferências"

My version code is 7.2.0

What is wrong?

### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

uviniciusc, 2021-07-22:
What the correct board version of chip 8266 on Arduino IDE to 7.2.0 tasmota version? my current version is 2.7.0 for board.

ascillato, 2021-07-22:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md)

Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2021-07-22 by ascillato2

Jason2866, 2021-07-23:
We do support very old Tasmota versions like 7.x. Actual Tasmota versions can only built with Visual Studio Code. Arduino IDE support has been dropped.
How to compile Tasmota is explained in our [docu](https://tasmota.github.io/docs/)