arendst/Tasmota #12731: issues with confif tasmota-knx

richardpub4136, 2021-07-22:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**
[image]
I updated a Sonoff basic R2 module to Tasmota KNX the module is functional but I am struggling to set up the KNX operation 
I have created a KNX virtual address in ETS5 and I am looking for the button that operates this address to trigger the relay action on the sonoff basic r2.
I have tried many methods but nothing activates this relay in KNX, however it works well with the interface of the IP address.
But I don't see how to integrate this button in my current KNX project.
I also use Home Adssistant.
Maybe there is a way to do this by going directly through the Tasmota KNX configuration
If someone has already done this it would help me a lot to understand the method to practice.
Thanks in advance

ascillato2, 2021-07-22:
From Tasmota side, your configuration is ok.

You just need to send from your KNX network the command power on or off to 1/1/3.

Remember that Tasmota uses KNX IP Multicast. You need a KNX to IP router and enable KNX IP Multicast on it. KNX tunnel is not supported.

richardpub4136, 2021-07-25:
I have KNX IP Multicast with a KNX to IP router. What do you mean by "KNX tunnel is not supported"? It means that I communicate with the KNX bus through the IP routing Multicast address 224.0.23.12???
I used the Tasmota interface to update the Tasmota KNX firmware, thinking that this is where my problem comes from, but since then I can't connect with the IP address.
I'm going to put back the 1st firmware to test again.

richardpub4136, 2021-07-25:
Could my problem be that my sonff basic r2 is configured like this?
[image]

ascillato2, 2021-07-26:
> What do you mean by "KNX tunnel is not supported"? It means that I communicate with the KNX bus through the IP routing Multicast address 224.0.23.12???

KNX IP standard has KNX IP Multicast and KNX IP Tunneling. Tasmota Supports KNX IP Multicast protocols. If you are using tunneling is not going to work.

> Could my problem be that my sonff basic r2 is configured like this?
No.

> I am struggling to set up the KNX operation

You can troubleshoot if your ETS configuration is really sending any commands to your Wireless Network by using Multicast Tester App for your phone. There, you can see if there are any telegrams.

richardpub4136, 2021-07-28:
Thanks for your answer, I will test the Multicast Tester application with my phone. And also, I will add a connector as a participant in ETS5, with the group address dedicated to my tasmota knx

richardpub4136, 2021-08-01:
The Multicast Tester application does not seem to exist for phone.
Can you recommend another method to test the IP multicast connection?
I have tried many methods but I have not yet managed to order my Sonoff in tasmota knx.

ascillato, 2021-08-01:
> The Multicast Tester application does not seem to exist for phone.

Sorry. It exists as I use it a lot. There are several Apps for multicast testing and I already tried some of them. I use the one called multicast tester.
Please, check again and try to see how your KNX is configured due to seems not to be an issue in Tasmota. Seems to be a misconfiguration in your network.

richardpub4136, 2021-08-02:
I think it is indeed a problem with the multicast configuration in ETS5.
Apparently the app for Iphone is not available from France.
I will find an android to test.
Did it work by pinging the Ethernet network? In my case, no ping result on the multicast address.
I really need to test this before I try to change the settings of my IP/KNX gateway (Albrecht Jung)
Thanks for your patience.

richardpub4136, 2021-08-02:
I just looked at the flows with wireshark, I have flows from my network in 192.168.1.203 to the multicast 224.0.23.12.
I will test with tasmota knx, but maybe it is inconsistent?

richardpub4136, 2021-08-02:
I think my concern is with KNXnet/IP tunnelling to multicast. With whireshark, I can see messages going to multicast, but not the other way around

My albrecht Jung IP router settings are set like this:
[image]
https://nsa40.casimages.com/img/2021/08/02/210802114937150285.gif
ideas ????

richardpub4136, 2021-08-02:
I think my concern is with KNXnet/IP tunnelling to multicast. With whireshark, I can see messages going to multicast, but not the other way around

My albrecht Jung IP router settings are set like this:

some ideas ????

https://nsa40.casimages.com/img/2021/08/02/21080211590075208.gif

https://nsa40.casimages.com/img/2021/08/03/210803120102737117.gif

richardpub4136, 2021-08-02:
[URL=https://www.casimages.com/i/210803120102737117.gif.html][IMG]https://nsa40.casimages.com/img/2021/08/03/210803120102737117.gif[/IMG][/URL]

richardpub4136, 2021-08-02:
[image]

ascillato, 2021-08-02:
KNXnet/IP tunnelling is not supported in Tasmota. Sorry. Only KNXnet/IP Multicast is supported.

richardpub4136, 2021-08-03:
[image]
In ETS5, I am configured like this: 
I thought I was multicasting???
I was wondering if my Cisco SG 500 28P switch could block multicast on my network???

ascillato2, 2021-08-04:
Multicast is not an IP you connect to. Multicast is an address devices broadcast. If you are connecting to an IP it means you are using tunnelling and that is not supported.

You should contact your KNX Router manufacturer to see if your device supports KNXnet/IP Multicast and how configure it. Sorry.

richardpub4136, 2021-08-04:
In my router I thought I had both possibilities and in tunneling, I have an IP address that corresponds to my ethernet network in 192.168.1.xxx, and the address 224.0.23.12 was the address corresponding to multicast which is the one reserved by default to KNX

on the other hand, my Cisco switch which is manageable, and is not in a position allowing multicast on any of its 28 ports. could this be the reason for my wifi network not being broadcast through the switch to the KNX bus connection

ascillato2, 2021-08-09:
Hi, any news on this?

Were you able to enable multicast in your router/switch? Were you able to check if the multicast messages are being published using any multicast tester app or software?

richardpub4136, 2021-08-09:
I was able to test the multicast with "multicast tester" on android. I see multicast from knx bus to IP, but not from Tasmota to KNX. I haven't taken the time to change the multicast settings in my Cisco switch yet because of the vacation season, but I should do it this week and I'll keep you posted.
Sincerely

Shiro94, 2021-08-12:
Hello, sorry for barging in the conversation. Like richardpub i am also doing an integaration of KNX with tasmota. My setup with richarpub is almost the same, the only thing differ is that i use KNX Switch Button for turning my LED on the tasmota side ON/OFF.

Using wireshark and the "multicast tester" app, i can see the multicast (KNXnet/IP) is being sent to the local network. But my tasmota is not pickup it up, thus the led is not turning ON/OFF.

Sorry for barging in and thank you.
Best Regards

ascillato, 2021-08-12:
@Shiro94 Hi, are you using Tasmota_KNX latest version? Have you enabled KNX in the webUI? Have you configured Tasmota group addresses? Have you tried Tasmota sending KNX telegrams and seeing them in Multicast Tester App?

Shiro94, 2021-08-12:
im using hte tasmota-knx.bin build from https://github.com/arendst/Tasmota/releases/. according to the web-ui it is 9.4.0.4. i have tried enabling it, but it still doesnt work.
i doubt its from the tasmota side. im gonna have to read up more on it from the ETS side.

ascillato, 2021-08-12:
the release version is 9.5.0 ( http://ota.tasmota.com/tasmota/release/ )
the latest Tasmota is 9.5.0.3 ( http://ota.tasmota.com/tasmota/ )

Better use latest. You can upgrade from the WebUI using `OTAurl http://ota.tasmota.com/tasmota/tasmota-knx.bin.gz` 

> i have tried enabling it, but it still doesnt work.

can you post screenshots?

ascillato2, 2021-08-12:
Glad to know that it is working fine for you. 👍

closed 2021-08-12 by ascillato2

richardpub4136, 2021-08-14:
I am sorry to continue this issues, because I still have this communication problem during a manual action I should see the trace of sending to the multicast of the current state of my sonoff, but I do not see any traces even with mulicast tester, nor with wireshark.
I only have the trace of the sending of the KNX bus on the multicast.
I will try to do a lean install with just my router, my knx ip gateway, and my tasmota.
I do not think that the presence of "home assistant" as an interface with the tasmota is the cause of my concern.
I have yet to try to change the settings of my Cisco switch and its multicast settings because my level of knowledge of this switch is too poor, but I think it is the cause of my concerns.

ascillato, 2021-08-14:
> I will try to do a lean install with just my router, my knx ip gateway, and my tasmota.

That is a very good idea.

> I do not think that the presence of "home assistant" as an interface with the tasmota is the cause of my concern.

No, It shouldn't. I use it with both MQTT and KNX and there is no issue with HA.

richardpub4136, 2021-08-15:
I will do this as soon as possible because here it is the holiday period there are a lot of people coming to the house thank you for your follow-up cordially

richardpub4136, 2021-08-15:
just a subsidiary question if you trigger the light of your tasmota from a command in mqtt does the change of state automatically modify the state of the KNX address sent by the KNX tasmota ????

ascillato, 2021-08-15:
Yes, instantly.

richardpub4136, 2021-08-15:
Before modifying my network to have something simpler I used wireshark to test the flows in my network
I have noticed that pressing the toggle sends the info of the group address "Data to Send to Group Addresses" 1/1/3 with the participant number 1.1.108 declared in the Tasmota KNX configuration to multicast IP. The material is therefore well recognized and recognized.
In the diagnostic menu of ETS5, I have the "Group Addresses to Receive Data from" 1/1/2 declared in the Tasmota KNX configuration that I go from ON to OFF and vice versa, and I see it go into the multicast stream in Wireshark. On the other hand, in this sense, the Group Address is indeed 1/1/2, but the participant number is 0.0.1 ???
Of course in ETS5 the participant number is 1.1.108 has not been declared ....
Should this be done ??? If so I don't know how to declare what appears to be a virtual participant.
Thank you again for your help.

ascillato, 2021-08-15:
In ETS you only need to configure your Group Address to send data and the Group Address to receive data. There is no need to set physical addresses. Just set the GA to turn on and off the switch and the GA to receive the status of the switch. Nothing more is required.

richardpub4136, 2021-08-15:
I have indeed since reversed the two groups of address the 1/1/2 to activate the tasmota and the 1/1/3 for the return of state.
I hadn't fully understood which way address groups work and this method is more in line with what I usually do in ETS5.
Following this confirmation, I will therefore make a mini network as planned, with the tasmota, its WIFI router, and the gateway between this network and the KNX. I hope it will work that way.

richardpub4136, 2021-08-21:
I have tried many methods by modifying all of my network settings to try to solve my problem.
Since I couldn't get any command output from ETS5, I chose to install the tasmota KNX.ino version of envy / esp-knx-ip.

With this version I was able to get a command from the diagnostics mode of ETS5. With this version the control address 1/1/2 and the status address 1/1/3 work.
Unfortunately, as I have no participant for these GAs, except that of the communicator, the Write action in the Diagnostics mode of ETS5 uses the physical address 0.0.1 by default. This is well recognized by the Sonoff Tasmota which turns on and off through the KNX bus? And the state return of the Sonoff Tasmota is done well with the physical address 1.1.108 that I declared in the parameters of the Sonoff Tasmota.
I will try to use the method from https://github.com/thelsing/CreateKnxProd to create an importable file.knxprod participant directly with this app.
But perhaps you would have thanks to your experience another method so that I can associate my 2 AG d in a parameterizable actuator participant directly in ETS5.
Kind regards

ascillato, 2021-08-21:
esp-knx-ip is a library. Tasmota uses that library. So, if you could make the examples of the library to work in your setup, the same configuration should work in Tasmota as it is the same internal code.

I don't understand what you meant about "tasmota KNX.ino". There is no such file neither in tasmota nor in envy's esp-knx-ip repository.

The code from the thelsing is incompatible with Tasmota.

I think you have chosen an overcomplicated path of using other libraries and I can't help you on that.

KNX is very simple. I don't know what you are configuring in ETS that you can't make it work.

KNX IP is just a multicast. All devices write and read to the same multicast IP address (that is a channel)
Then to identify to whom is the telegram, GA (group addresses are used). Group addresses has nothing to do with physical addresses and nothing to do with IP.

Forget about physical addresses.

Just send the `power on` telegram to the `group address to receive data from` in tasmota and that's it. No more configuration is needed.

ascillato, 2021-08-21:
Are you configuring Tasmota as a dummy device in ETS?
It is required to be configured as such in order to make it work. No need for anything else.

richardpub4136, 2021-08-21:
I want to be able to control my Sonoff using an on / off switch physically installed in my KNX bus;
Currently I manage to activate the Sonoff by writing the action directly in the diagnostic mode of ETS5.
Yes, I am trying to configure Tasmota as a dummy device in ETS.
I think I am not sure how to set up a dummy device, can you give me any advice for this ???
I block when downloading the dummy participant in ETS ...