arendst/Tasmota #12771: 5-7 Second delay between power toggle (Either web UI toggle button or MqTT) and power being applied

L0rdCha0s, 2021-07-28:
### PROBLEM DESCRIPTION
When I toggle power using any method (Web UI, MqTT, HTTP call), the device clearly receives the signal (as seen in the console log below), but only sends the command to TuyaMCU 5-7 seconds later (also seen in console).

1. As seen below, the request is made at 07:03:31 (Power On)
2. The TuyaMCU command is then sent at 07:03:38, and the light comes on (7 seconds later)

...
07:02:31.203 RSL: RESULT = {"POWER":"ON"}
07:02:31.205 RSL: POWER = ON
07:02:31.222 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
07:02:31.224 TYA: Heartbeat
07:02:31.750 CFG: Saved to flash at FB, Count 98, Bytes 4096
07:02:36.915 WIF: Checking connection...
07:02:38.447 TYA: Send "55aa00000000ff"
07:02:38.454 {"TuyaReceived":{"Data":"55AA0007000501010001010F","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
07:02:38.456 TYA: fnId=11 is set for dpId=1
....

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ X] Device used (e.g., Sonoff Basic): Generic ESP 8266 dimmer light switch_____
- [ X] Tasmota binary firmware version number used: _9.5.0.3____
  - [ X] Pre-compiled
  - [ ] Self-compiled
- [X ] Flashing tools used: _Web upgrade____
- [ X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

07:05:58.750 CMD: Backlog Template; Module; GPIO 255
07:05:58.752 SRC: WebConsole from 10.200.1.1
07:05:58.755 CMD: Grp 0, Cmnd 'BACKLOG', Idx 1, Len 26, Data 'Template; Module; GPIO 255'
07:05:58.820 SRC: Backlog
07:05:58.823 CMD: Grp 0, Cmnd 'TEMPLATE', Idx 1, Len 0, Data ''
07:05:58.827 RSL: RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
07:05:59.034 SRC: Backlog
07:05:59.036 CMD: Grp 0, Cmnd 'MODULE', Idx 1, Len 0, Data ''
07:05:59.040 RSL: RESULT = {"Module":{"54":"Tuya MCU"}}
07:05:59.283 SRC: Backlog
07:05:59.285 CMD: Grp 0, Cmnd 'GPIO', Idx 1, Len 3, Data '255'
07:05:59.293 RSL: RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"2272":"Tuya Tx"},"GPIO2":{"0":"None"},"GPIO3":{"2304":"Tuya Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"288":"Led1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
07:05:59.388 RSL: STATE = {"Time":"2021-07-28T07:05:59","Uptime":"0T22:45:09","UptimeSec":81909,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":0,"POWER":"ON","Wifi":{"AP":1,"SSId":"chaospace-home-automation","BSSId":"F2:9F:C2:FA:48:DF","Channel":6,"Mode":"11n","RSSI":88,"Signal":-56,"LinkCount":1,"Downtime":"0T00:00:03"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

07:12:02.611 CMD: status 0
07:12:02.613 SRC: WebConsole from 10.200.1.1
07:12:02.616 CMD: Grp 0, Cmnd 'STATUS', Idx 1, Len 1, Data '0'
07:12:02.621 RSL: STATUS = {"Status":{"Module":54,"DeviceName":"Tasmota-Upstairs-Living-Master","FriendlyName":["Tasmota-Upstairs-Living-Master"],"Topic":"tasmota_3673B2","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
07:12:02.628 RSL: STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T22:51:12","StartupUTC":"2021-07-27T07:20:50","Sleep":50,"CfgHolder":4617,"BootCount":9,"BCResetTime":"2021-07-27T07:43:40","SaveCount":104,"SaveAddress":"F5000"}}
07:12:02.632 RSL: STATUS2 = {"StatusFWR":{"Version":"9.5.0.3(tasmota)","BuildDateTime":"2021-07-26T15:16:37","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"403/699"}}
07:12:02.637 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["chaospace-home-automation",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000A009","2805C8000100060000005A0A000000000000","00000080","00006008","00000000"]}}
07:12:02.647 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":608,"Free":392,"Heap":25,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"154068","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
07:12:02.653 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-3673B2-5042","IPAddress":"10.200.2.19","Gateway":"10.200.2.254","Subnetmask":"255.255.255.0","DNSServer":"10.200.2.240","Mac":"40:F5:20:36:73:B2","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
07:12:02.658 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_3673B2","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
07:12:02.664 RSL: STATUS7 = {"StatusTIM":{"UTC":"2021-07-28T06:12:02","Local":"2021-07-28T07:12:02","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"05:18","Sunset":"20:34"}}
07:12:02.669 RSL: STATUS10 = {"StatusSNS":{"Time":"2021-07-28T07:12:02"}}
07:12:02.674 RSL: STATUS11 = {"StatusSTS":{"Time":"2021-07-28T07:12:02","Uptime":"0T22:51:12","UptimeSec":82272,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"ON","Wifi":{"AP":1,"SSId":"chaospace-home-automation","BSSId":"F2:9F:C2:FA:48:DF","Channel":6,"Mode":"11n","RSSI":86,"Signal":-57,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

As above in issue description
```

### TO REPRODUCE
_Steps to reproduce the behavior:_
1. turn on relay using either web console or mqtt
2. 5-7 second delay before relay is activated

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

Much faster response

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

This hasn't always happened - but I can't downgrade far back enough to confirm.  I can confirm this is happening for this hardware at least as far back as 8.5.1.  This is happening on several switches from the same manufacturer.

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2021-07-28:
Have you tried to switch of mqtt since you do not use it?
`so3 0`

L0rdCha0s, 2021-07-29:
Yes - unfortunately the behaviour is the same.  Of particular note is that prior versions did not do this with this hardware.

digiblur, 2021-07-29:
In the cases I've seen this happen it was due to people blocking some combination of DNS, DHCP, NTP, etc from the device.

L0rdCha0s, 2021-07-30:
Hmm - doesn't seem to be that.  Just confirmed DNS, DHCP is being picked up right, default route is ok.. NTP seems fine (and time is correct)

L0rdCha0s, 2021-07-30:
OK - looks like we can close this one.  Highly embarrassing - and it took a previous issue raised here to get me to try it.

I went a cut the power to the device and brought it back up, and now it's fine (as opposed to the reboots i'd been doing previously.   Highly odd..

closed 2021-07-30 by L0rdCha0s