arendst/Tasmota #12772: MQTT Subscribe command adds /# at the end of the topic

serifis84, 2021-07-28:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
The issue is that when i use the command SUBSCRIBE to an event with a specific topic, tasmota will add a /# at the end of the topic making impossible to subscribe to a specific topic. I am using ThingsBoard platform. On the device api documentation it is clearly written that the topic to be subscribed to is v1/devices/me/attributes. If someone would be subscribed to v1/devices/me/attributes/# it wouldn't get any data

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): NodeMCU ESP-12E
- [ ] Tasmota binary firmware version number used: 9.4.0.2(TasmoCompiler-esp8266generic)
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: TasmoCompiler docker
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua

08:37:21.562 RSL: RESULT = {"Module":{"0":"Generic"}}
08:37:21.805 SRC: Backlog
08:37:21.808 CMD: Group 0, Index 1, Command "GPIO", Data "255"
08:37:21.814 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"320":"Led_i1"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"256":"Relay_i1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua

08:39:50.590 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_57D6B2","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
08:39:50.602 RSL: STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T14:53:36","StartupUTC":"2021-07-27T16:46:14","Sleep":50,"CfgHolder":4617,"BootCount":77,"BCResetTime":"2021-06-24T10:49:31","SaveCount":203,"SaveAddress":"F6000"}}
08:39:50.611 RSL: STATUS2 = {"StatusFWR":{"Version":"9.4.0.2(TasmoCompiler-esp8266generic)","BuildDateTime":"2021-07-27T12:56:55","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"357/699"}}
08:39:50.619 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"MqttLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["comcoms",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00408001","2805C8000100060000005A0A000000000000","00000080","00006000","00000000"]}}
08:39:50.631 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":517,"Free":484,"Heap":30,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"16301C","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0F804582","04260001","00000003","00000000","C000E981","00000004","00001000","00000000"],"Drivers":"1,2,4,6,8,9,10,18,19,22,24,26,27,29,30,35,37,45","Sensors":"1,2"}}
08:39:50.642 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota_57D6B2-5810","IPAddress":"172.16.0.46","Gateway":"172.16.0.1","Subnetmask":"255.255.255.0","DNSServer":"172.16.0.1","Mac":"2C:F4:32:57:D6:B2","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
08:39:50.652 RSL: STATUS7 = {"StatusTIM":{"UTC":"2021-07-28T07:39:50","Local":"2021-07-28T08:39:50","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"05:18","Sunset":"20:34"}}
08:39:50.661 RSL: STATUS10 = {"StatusSNS":{"Time":"2021-07-28T08:39:50"}}
08:39:50.669 RSL: STATUS11 = {"StatusSTS":{"Time":"2021-07-28T08:39:50","Uptime":"0T14:53:36","UptimeSec":53616,"Heap":30,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"ON","Wifi":{"AP":1,"SSId":"comcoms","BSSId":"B0:4E:26:B0:5B:62","Channel":9,"RSSI":46,"Signal":-77,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua

08:42:48.406 CMD: Subscribe   DHWTEMPEVENT, v1/devices/me/attributes, DHWsetpoint
08:42:48.409 SRC: WebConsole from 172.16.0.23
08:42:48.412 CMD: Group 0, Index 1, Command "SUBSCRIBE", Data "DHWTEMPEVENT, v1/devices/me/attributes, DHWsetpoint"
08:42:48.416 MQT: Subscribe to v1/devices/me/attributes/#
08:42:48.419 RSL: RESULT = {"Subscribe":"DHWTEMPEVENT,v1/devices/me/attributes/#,DHWsetpoint"}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Compile with SUPPORT_MQTT_EVENT feature
Configure MQTT server host and credentials
Open Tasmota console
Execute Subscribe   DHWTEMPEVENT, v1/devices/me/attributes, DHWsetpoint

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
08:42:48.406 CMD: Subscribe   DHWTEMPEVENT, v1/devices/me/attributes, DHWsetpoint
08:42:48.409 SRC: WebConsole from 172.16.0.23
08:42:48.412 CMD: Group 0, Index 1, Command "SUBSCRIBE", Data "DHWTEMPEVENT, v1/devices/me/attributes, DHWsetpoint"
08:42:48.416 MQT: Subscribe to v1/devices/me/attributes
08:42:48.419 RSL: RESULT = {"Subscribe":"DHWTEMPEVENT,v1/devices/me/attributes,DHWsetpoint"}

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Elix-g, 2021-08-08:
[MQTT V3.1 Protocol Specification Appendix A Topic Wildcards ](http://public.dhe.ibm.com/software/dw/webservices/ws-mqtt/mqtt-v3r1.html#appendix-a)

> The number sign (#) is a wildcard character that matches any number of levels within a topic. For example, if you subscribe to finance/stock/ibm/#, you receive messages on these topics:
>
>      finance/stock/ibm
>      finance/stock/ibm/closingprice
>      finance/stock/ibm/currentprice
>
> The multi-level wildcard can represent zero or more levels. Therefore, finance/# can also match the singular finance, where # represents zero levels. The topic level separator is meaningless in this context, because there are no levels to separate.

The ThingsBoard definition you refer to, fits the definition of the Single-Level wildcard (+-sign).

From my point of view, the Tasmota implementation follows the standard definition.

serifis84, 2021-08-10:
@Elix-g thank you for your interest
I checked Thingsboard with mosquitto_sub and does not work with multilevel subscription. I also experimented subscribing to v1/devices/me/attributes/+ and tasmota also added # at the end of the topic resulting to v1/devices/me/attributes/+/#

sfromis, 2021-08-10:
According to the MQTT specifcation, what Tasmota does is not incorrect, and does not hinder you from subscribing to one specific topic, unless that topic is also a prefix of longer topic strings. Here's a specific example:

For example, if a Client subscribes to “sport/tennis/player1/#”, it would receive messages published using these topic names:
* “sport/tennis/player1”
* “sport/tennis/player1/ranking”
* “sport/tennis/player1/score/wimbledon”

If you have a broker not working this way, it looks like a flaw in how wildcards are implemented in that broker.

ascillato2, 2021-08-10:
Moving to Discussions as this issue is how to use MQTT subscriptions.

closed 2021-08-10 by ascillato2