arendst/Tasmota #12773: rule automatically disabled after a while (no bootloop)

Gifford47, 2021-07-28:
rule automatically disabled after a while (no bootloop)

### PROBLEM DESCRIPTION
A rule which reports the sensor status is automatically disabled from tasmota after a (unknown) while.
I need to enable the rule manually.
The is executed several times within ~24h, then it is disabled. The boot count already stays at 0.

### REQUESTED INFORMATION
- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Generic
- [X] Tasmota binary firmware version number used: 9.5.0.1
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: Tasmotizer
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
13:47:33.085 CMD: Backlog Template; Module; GPIO 255
13:47:33.148 MQT: stat/tasmota_7A8961/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
13:47:33.383 MQT: stat/tasmota_7A8961/RESULT = {"Module":{"18":"Generic"}}
13:47:33.687 MQT: stat/tasmota_7A8961/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"640":"I2C SDA"},"GPIO5":{"608":"I2C SCL"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"160":"Switch1"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
13:56:36.128 CMD: Backlog Rule1; Rule2; Rule3
13:56:36.130 SRC: WebConsole from 192.168.1.94
13:56:36.132 CMD: Grp 0, Cmnd 'BACKLOG', Idx 1, Len 19, Data 'Rule1; Rule2; Rule3'
13:56:36.189 SRC: Backlog
13:56:36.191 CMD: Grp 0, Cmnd 'RULE', Idx 1, Len 0, Data ''
13:56:36.197 MQT: stat/tasmota_7A8961/RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":72,"Free":439,"Rules":"on EVENT#ALARMON do gpio12 160 endon on EVENT#ALARMOFF do gpio12 0 endon"}}
13:56:36.422 SRC: Backlog
13:56:36.424 CMD: Grp 0, Cmnd 'RULE', Idx 2, Len 0, Data ''
13:56:36.428 RUL: Rule2 on BME280#temperature>%var1% do Backlog var1 %value%; wifi 1; publish tele/%topic%/SENSOR {"Time":"%timestamp%","BME280":{"Temperature":%value%},"TempUnit":"C"}; var2 %value%; add1 1; sub2 1 endon on BME280#temperature<%var2% do Backlog var2 %value%; wifi 1; publish tele/%topic%/SENSOR {"Time":"%timestamp%","BME280":{"Temperature":%value%},"TempUnit":"C"}; var1 %value%; add1 1; sub2 1 endon on BME280#humidity>%var3% do Backlog var3 %value%; wifi 1; publish tele/%topic%/SENSOR
13:56:36.465 RUL: Rule2 +{"Time":"%timestamp%","BME280":{"Humidity":%value%},"TempUnit":"C"}; var4 %value%; add3 2; sub4 2 endon on BME280#humidity<%var4% do Backlog var4 %value%; wifi 1; publish tele/%topic%/SENSOR {"Time":"%timestamp%","BME280":{"Humidity":%value%},"TempUnit":"C"}; var3 %value%; add3 2; sub4 2 endon
13:56:36.499 MQT: stat/tasmota_7A8961/RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":775,"Free":253,"Rules":"on BME280#temperature>%var1% do Backlog var1 %value%; wifi 1; publish tele/%topic%/SENSOR {\"Time\":\"%timestamp%\",\"BME280\":{\"Temperature\":%value%},\"TempUnit\":\"C\"}; var2 %value%; add1 1; sub2 1 endon on BME280#temperature<%var2% do Backlog var2 %value%; wifi 1; publish tele/%topic%/SENSOR {\"Time\":\"%timestamp%\",\"BME280\":{\"Temperature\":%value%},\"TempUnit\":\"C\"}; var1 %value%; add1 1; sub2 1 endon on BME280#humidity>%var3% do Backlog var3 %value%; wifi 1; publish tele/%topic%/SENSOR {\"Time\":\"%timestamp%\",\"BME280\":{\"Humidity\":%value%},\"TempUnit\":\"C\"}; var4 %value%; add3 2; sub4 2 endon on BME280#humidity<%var4% do Backlog var4 %value%; wifi 1; publish tele/%topic%/SENSOR {\"Time\":\"%timestamp%\",\"BME280\":{\"Humidity\":%value%},\"TempUnit\":\"C\"}; var3 %value%; add3 2; sub4 2 endon"}}
13:56:36.735 SRC: Backlog
13:56:36.737 CMD: Grp 0, Cmnd 'RULE', Idx 3, Len 0, Data ''
13:56:36.744 MQT: stat/tasmota_7A8961/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [X] Provide the output of this command: `Status 0`:
```lua
13:59:39.357 CMD: Status 0
13:59:39.359 SRC: WebConsole from 192.168.1.94
13:59:39.361 CMD: Grp 0, Cmnd 'STATUS', Idx 1, Len 1, Data '0'
13:59:39.368 MQT: stat/tasmota_7A8961/STATUS = {"Status":{"Module":18,"DeviceName":"Multisensor1","FriendlyName":["Multisensor1"],"Topic":"tasmota_7A8961","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"pir","SwitchMode":[1,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
13:59:39.399 MQT: stat/tasmota_7A8961/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota-sensors.bin","RestartReason":"Exception","Uptime":"0T00:04:32","StartupUTC":"2021-07-28T11:55:07","Sleep":150,"CfgHolder":4617,"BootCount":1,"BCResetTime":"2021-07-27T19:57:10","SaveCount":245,"SaveAddress":"FB000"}}
13:59:39.433 MQT: stat/tasmota_7A8961/STATUS2 = {"StatusFWR":{"Version":"9.5.0.1(sensors)","BuildDateTime":"2021-06-29T08:48:39","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"358/699"}}
13:59:39.456 MQT: stat/tasmota_7A8961/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":3,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["nodes_land",""],"TelePeriod":0,"Resolution":"558180C0","SetOption":["000A8009","2805C8000400060000005A0A000000000000","00000280","00006000","00000001"]}}
13:59:39.495 MQT: stat/tasmota_7A8961/STATUS4 = {"StatusMEM":{"ProgramSize":662,"Free":340,"Heap":17,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640D8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDA8787","0415A005","B7F7BFCF","05DA9BC4","64367CC7","00084052","20000000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,62,64,66,67,74"}}
13:59:39.537 MQT: stat/tasmota_7A8961/STATUS5 = {"StatusNET":{"Hostname":"tasmota_7A8961-2401","IPAddress":"192.168.1.92","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"8C:AA:B5:7A:89:61","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
13:59:39.560 MQT: stat/tasmota_7A8961/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.100","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_7A8961","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
13:59:39.583 MQT: stat/tasmota_7A8961/STATUS7 = {"StatusTIM":{"UTC":"2021-07-28T11:59:39","Local":"2021-07-28T13:59:39","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":99,"Sunrise":"06:18","Sunset":"21:34"}}
13:59:39.606 MQT: stat/tasmota_7A8961/STATUS10 = {"StatusSNS":{"Time":"2021-07-28T13:59:39","Switch1":"OFF","BME280":{"Temperature":28.0,"Humidity":38.8,"DewPoint":12.7,"Pressure":1004.5},"BH1750":{"Illuminance":22},"PressureUnit":"hPa","TempUnit":"C"}}
13:59:39.625 MQT: stat/tasmota_7A8961/STATUS11 = {"StatusSTS":{"Time":"2021-07-28T13:59:39","Uptime":"0T00:04:32","UptimeSec":272,"Heap":17,"SleepMode":"Dynamic","Sleep":150,"LoadAvg":6,"MqttCount":1,"Wifi":{"AP":1,"SSId":"nodes_land","BSSId":"F6:9F:C2:3A:CD:0A","Channel":7,"Mode":"11n","RSSI":54,"Signal":-73,"LinkCount":1,"Downtime":"0T00:00:03"}}}
13:59:39.664 MQT: stat/tasmota_7A8961/STATUS12 = {"StatusSTK":{"Exception":28,"Reason":"Exception","EPC":["4024fbf8","00000000","00000000"],"EXCVADDR":"0000068c","DEPC":"00000000","CallChain":["40254bcc","4024ff64","40253c94","4024ff8d","402127b0","40103bb2","40297e19","40253da0","40232f63","40253c4b","40101e27","40253cd0","40233385","4028ba04","402338b7","40233993","4028ba04","402153b0","402153ee","40215415","40230bb6","40231848","40232848","40100064","40101e27","4029b75c","402515ca","40251971","4025b25d","40101b9f","402328a7"]}}

```
[... cut ...]
	Backlog var3 %value%; wifi 1; publish tele/%topic%/SENSOR {"Time":"%timestamp%","BME280":{"Humidity":%value%},"TempUnit":"C"}; var4 %value%; add3 2; sub4 2
endon 
on BME280#humidity<%var4% do
	Backlog var4 %value%; wifi 1; publish tele/%topic%/SENSOR {"Time":"%timestamp%","BME280":{"Humidity":%value%},"TempUnit":"C"}; var3 %value%; add3 2; sub4 2
endon

```

### EXPECTED BEHAVIOUR
The rule should be executed without any issues.

### SCREENSHOTS
nothing

### ADDITIONAL CONTEXT
nothing

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2021-07-30:
What is the goal of the rules? Sorry it is too hot here for my brain ;-)

Gifford47, 2021-07-30:
i want to update only the relevant sensors, that has changed values. not every i.e. 1 hour like teleperiod does

arendst, 2021-07-30:
Using below rule 
```
on event#theo1>%var1% do Backlog var1 %value%; wifi 1; publish tele/%topic%/SENSOR {"Time":"%timestamp%","BME280":{"Temperature":%value%},"TempUnit":"C"}; var2 %value%; add1 1; sub2 1 endon on event#theo1<%var2% do Backlog var2 %value%; wifi 1; publish tele/%topic%/SENSOR {"Time":"%timestamp%","BME280":{"Temperature":%value%},"TempUnit":"C"}; var1 %value%; add1 1; sub2 1 endon 
```
and executing command like ``event#theo1=10``, ``event#theo1=11``, ``event#theo1=12`` etc I cannot reproduce the exception 28. This exception is related to providing a number where a string is expected. The rules are supposed to handle numbers and strings accordingly.

You'll have to do some logging to see what input causes this exception.

Jason2866, 2021-08-05:
@Gifford47  Did you try the suggestion?

Gifford47, 2021-08-06:
Yes, bit nothing changed. Yesterday I changed all gpios to 'none' and after that I re-enabled all gpios (also bme sensor). After the reboot I also got the exception 28.
Maybe it's something wrong with the bme sensor values. Perhaps at startup? How can I debug the raw bme sensor values over a longer time period?

Jason2866, 2021-08-06:
There is nothing wrong with the BME280 driver. Many people use that sensor without any problem (i have 5 of them too). There is no issue with.
Probably your hardware is defect or you have a bad power supply.
Replace the parts and see if your problems go away.

Gifford47, 2021-08-16:
ok,i don't know exactly where the error came from. but it had something to do with SetOption19 1 and SetOption114 1.
i did a complete reset and configured without SetOptions. now everything runs ...

closed 2021-08-16 by Gifford47