arendst/Tasmota #12789: KingArt touch Dimmer

cyber-junkie9, 2021-07-30:
### PROBLEM DESCRIPTION
Tasmota shows devices is ON after power up but its not ON

### REQUESTED INFORMATION

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): KingArts touch Dimmer
- [ ] Tasmota binary firmware version number used: 9.5.0.4
  - [x] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: Tasmotizer
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
15:00:43.869 MQT: stat/bedroom_fan/RESULT = {"NAME":"PS-16-DZ","GPIO":[0,3200,0,3232,0,0,0,0,0,288,0,0,0,0],"FLAG":0,"BASE":58}
15:00:44.079 MQT: stat/bedroom_fan/RESULT = {"Module":{"0":"PS-16-DZ"}}
15:00:44.294 MQT: stat/bedroom_fan/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"3200":"Serial Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3232":"Serial Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"288":"Led1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
14:49:24.076 MQT: stat/bedroom_fan/STATUS = {"Status":{"Module":0,"DeviceName":"Fan","FriendlyName":["Fan"],"Topic":"bedroom_fan","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":0,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
14:49:24.081 MQT: stat/bedroom_fan/STATUS1 = {"StatusPRM":{"Baudrate":19200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:10:42","StartupUTC":"2021-07-30T09:08:42","Sleep":50,"CfgHolder":4617,"BootCount":222,"BCResetTime":"2021-01-07T12:11:23","SaveCount":3412,"SaveAddress":"FA000"}}
14:49:24.089 MQT: stat/bedroom_fan/STATUS2 = {"StatusFWR":{"Version":"9.5.0.4(tasmota)","BuildDateTime":"2021-07-29T18:40:27","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"391/699"}}
14:49:24.095 MQT: stat/bedroom_fan/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["ssid123",""],"TelePeriod":300,"Resolution":"55C180C0","SetOption":["00028009","2805C8000100060000005A0A000000000000","10000280","00006000","00000000"]}}
14:49:24.108 MQT: stat/bedroom_fan/STATUS4 = {"StatusMEM":{"ProgramSize":609,"Free":392,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
14:49:24.116 MQT: stat/bedroom_fan/STATUS5 = {"StatusNET":{"Hostname":"fan","IPAddress":"192.168.1.106","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"60:01:94:B2:B4:D1","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
14:49:24.122 MQT: stat/bedroom_fan/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.118","MqttPort":1883,"MqttClientMask":"kingart_dimmer","MqttClient":"kingart_dimmer","MqttUser":"mqttuser","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
14:49:24.127 MQT: stat/bedroom_fan/STATUS7 = {"StatusTIM":{"UTC":"2021-07-30T09:19:24","Local":"2021-07-30T14:49:24","StartDST":"2021-10-31T02:00:00","EndDST":"2021-10-31T02:00:00","Timezone":99,"Sunrise":"09:51","Sunset":"01:01"}}
14:49:24.134 MQT: stat/bedroom_fan/STATUS10 = {"StatusSNS":{"Time":"2021-07-30T14:49:24"}}
14:49:24.141 MQT: stat/bedroom_fan/STATUS11 = {"StatusSTS":{"Time":"2021-07-30T14:49:24","Uptime":"0T00:10:42","UptimeSec":642,"Heap":23,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":33,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"ssid123","BSSId":"xx:xx:xx:xx:xx:xx","Channel":11,"Mode":"11n","RSSI":60,"Signal":-70,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
00:00:49.264 UPP: Multicast (re)joined
00:00:50.252 MQT: Attempting connection...
00:00:50.462 MQT: Connect failed to 192.168.1.118:1883, rc -2. Retry in 10 sec
00:00:51.258 UPP: Multicast (re)joined
00:01:01.259 MQT: Attempting connection...
00:01:01.274 MQT: Connect failed to 192.168.1.118:1883, rc -2. Retry in 20 sec
00:01:02.257 UPP: Multicast (re)joined
00:01:46.193 MQT: Attempting connection...
00:01:46.205 MQT: Connected
00:01:46.208 MQT: tele/bedroom_fan/LWT = Online (retained)
00:01:46.210 MQT: cmnd/bedroom_fan/POWER = 
00:01:46.224 MQT: tele/bedroom_fan/INFO1 = {"Info1":{"Module":"PS-16-DZ","Version":"9.4.0.2(tasmota)","FallbackTopic":"cmnd/kingart_dimmer_fb/","GroupTopic":"cmnd/tasmotas/"}}
00:01:46.228 MQT: tele/bedroom_fan/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"fan","IPAddress":"192.168.1.103"}}
00:01:46.233 MQT: tele/bedroom_fan/INFO3 = {"Info3":{"RestartReason":"External System"}}
00:01:46.237 MQT: stat/bedroom_fan/RESULT = {"POWER":"ON"}
00:01:46.240 MQT: stat/bedroom_fan/POWER = ON
00:01:47.193 UPP: Multicast (re)joined
00:01:50.451 MQT: tele/bedroom_fan/STATE = {"Time":"1970-01-01T00:01:50","Uptime":"0T00:01:29","UptimeSec":89,"Heap":27,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":50,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"ssid123","BSSId":"xx:xx:xx:xx:xx:xx","Channel":11,"RSSI":62,"Signal":-69,"LinkCount":1,"Downtime":"0T00:00:50"}}
07:49:50.469 CMD: WebLogWeb
07:49:50.477 MQT: stat/bedroom_fan/RESULT = {"Command":"Unknown"}
07:50:01.094 CMD: WebLog 4
07:50:01.101 MQT: stat/bedroom_fan/RESULT = {"WebLog":4}
07:50:01.800 CFG: Saved to flash at F8, Count 1984, Bytes 4096
07:50:11.943 WIF: Checking connection...
[... 17 lines cut ...]
07:51:03.480 SRC: WebGui from 192.168.1.101
07:51:03.555 MQT: tele/bedroom_fan/STATE = {"Time":"2021-05-04T07:51:03","Uptime":"0T00:03:23","UptimeSec":203,"Heap":25,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Dimmer":50,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"ssid123","BSSId":"xx:xx:xx:xx:xx:xx","Channel":11,"RSSI":62,"Signal":-69,"LinkCount":1,"Downtime":"0T00:00:50"}}
07:51:03.563 MQT: stat/bedroom_fan/RESULT = {"POWER":"ON"}
07:51:03.570 MQT: stat/bedroom_fan/POWER = ON
07:51:03.799 CFG: Saved to flash at F6, Count 1986, Bytes 4096
07:51:06.315 HTP: Console

```

### TO REPRODUCE
Keep dimmer on and turn off power supply , on power resume it will show on but it's not actually on
I have to send ON again or send OFF then ON again to actually turn dimmer ON

### EXPECTED BEHAVIOUR
To actually turn on dimmer not from just tasmota side

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
I guess this can be fixed by adding some delay to command sent to tuya mcu(N76E003) . it seems that AT command is sent from ESP(ESP8266) before tuya mcu is ready to receive the command.

PS - I am using this dimmer as FAN dimmer

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2021-07-30:
I see nowhere a MCU communication. Shouldnt this not appear in `weblog 4` ?

cyber-junkie9, 2021-07-30:
> I see nowhere

It should but i am using web console so can't provide previous data . is there any other way to dump log ?
Sorry can't connect via serial at the moment