arendst/Tasmota #12796: Timers deleted, PowerOnStage changes to default, TimeZone deleted, somo other config (weblog,etc) after a webui reset or power outage.

Srjosep, 2021-07-30:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Timers are lost and PowerOnStage goes to default after a power outage or device is reset. Some configuration is lost.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): _____
- [x ] Tasmota binary firmware version number used: _____9.5.0.2
  - [x] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
- [ ] 

21:47:21.836 APP: Serial logging disabled
21:47:36.813 MQT: controlador_acuario_marino/tele/STATE = {"Time":"2021-07-30T21:47:36","Uptime":"0T00:10:15","UptimeSec":615,"Heap":186,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"ON","POWER2":"ON","POWER3":"ON","POWER4":"ON","POWER5":"OFF","POWER6":"ON","POWER7":"ON","POWER8":"OFF","Wifi":{"AP":1,"SSId":"Jc10254ASJ","BSSId":"50:63:21:18:70:9D","Channel":11,"Mode":"11n","RSSI":40,"Signal":-80,"LinkCount":1,"Downtime":"0T00:00:09"}}
21:47:36.832 MQT: controlador_acuario_marino/tele/SENSOR = {"Time":"2021-07-30T21:47:36","ESP32":{"Temperature":63.9},"TempUnit":"C"}
21:52:23.745 CMD: Backlog Template; Module; GPIO 255
21:52:23.816 MQT: controlador_acuario_marino/stat/RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
21:52:24.043 MQT: controlador_acuario_marino/stat/RESULT = {"Module":{"1":"ESP32-DevKit"}}
21:52:24.306 MQT: controlador_acuario_marino/stat/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"225":"Relay2"},"GPIO14":{"226":"Relay3"},"GPIO15":{"227":"Relay4"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"228":"Relay5"},"GPIO19":{"229":"Relay6"},"GPIO20":{"0":"None"},"GPIO21":{"230":"Relay7"},"GPIO22":{"231":"Relay8"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO28":{"0":"None"},"GPIO29":{"0":"None"},"GPIO30":{"0":"None"},"GPIO31":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}
21:52:36.875 MQT: controlador_acuario_marino/tele/STATE = {"Time":"2021-07-30T21:52:36","Uptime":"0T00:15:15","UptimeSec":915,"Heap":189,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"ON","POWER2":"ON","POWER3":"ON","POWER4":"ON","POWER5":"OFF","POWER6":"ON","POWER7":"ON","POWER8":"OFF","Wifi":{"AP":1,"SSId":"Jc10254ASJ","BSSId":"50:63:21:18:70:9D","Channel":11,"Mode":"11n","RSSI":42,"Signal":-79,"LinkCount":1,"Downtime":"0T00:00:09"}}
21:52:36.896 MQT: controlador_acuario_marino/tele/SENSOR = {"Time":"2021-07-30T21:52:36","ESP32":{"Temperature":63.3},"TempUnit":"C"}
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
21:57:59.652 CMD: Status 0
21:57:59.667 MQT: controlador_acuario_marino/stat/STATUS = {"Status":{"Module":1,"DeviceName":"Controlador_acuario_marino","FriendlyName":["Controlador del acuario marino","","","","","","",""],"Topic":"controlador_acuario_marino","ButtonTopic":"0","Power":111,"PowerOnState":1,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
21:57:59.685 MQT: controlador_acuario_marino/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Blocked Loop","Uptime":"0T00:20:38","StartupUTC":"2021-07-30T19:37:21","Sleep":50,"CfgHolder":4617,"BootCount":8,"BCResetTime":"2021-07-05T23:56:04","SaveCount":26}}
21:57:59.703 MQT: controlador_acuario_marino/stat/STATUS2 = {"StatusFWR":{"Version":"9.5.0.2(tasmota)","BuildDateTime":"2021-07-05T17:11:58","Core":"1_0_7_1","SDK":"v3.3.5-138-g714f27f56","CpuFrequency":80,"Hardware":"ESP32-D0WDQ6","CR":"447/699"}}
21:57:59.720 MQT: controlador_acuario_marino/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Jc10254ASJ",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000080","00006000","00000000"]}}
21:57:59.750 MQT: controlador_acuario_marino/stat/STATUS4 = {"StatusMEM":{"ProgramSize":1240,"Free":1856,"Heap":185,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":3,"Features":["00000809","9FDAC787","00152001","B7F7BFCF","05DA9BC0","60360DC7","000840D2","20200000","0000002D"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,17,20,21,24,26,27,29,34,38,50,52","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,87"}}
21:57:59.773 MQT: controlador_acuario_marino/stat/STATUS5 = {"StatusNET":{"Hostname":"controlador_acuario_marino-5144","IPAddress":"192.168.1.15","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"212.166.211.4","Mac":"7C:9E:BD:48:D4:18","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
21:57:59.792 MQT: controlador_acuario_marino/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.10","MqttPort":1883,"MqttClientMask":"Controlador acuario Marino","MqttClient":"Controlador acuario Marino","MqttUser":"Mos7867yLGHGo","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
21:57:59.807 MQT: controlador_acuario_marino/stat/STATUS7 = {"StatusTIM":{"UTC":"2021-07-30T19:57:59","Local":"2021-07-30T21:57:59","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+02:00","Sunrise":"06:21","Sunset":"21:31"}}
21:57:59.822 MQT: controlador_acuario_marino/stat/STATUS10 = {"StatusSNS":{"Time":"2021-07-30T21:57:59","ESP32":{"Temperature":63.3},"TempUnit":"C"}}
21:57:59.837 MQT: controlador_acuario_marino/stat/STATUS11 = {"StatusSTS":{"Time":"2021-07-30T21:57:59","Uptime":"0T00:20:38","UptimeSec":1238,"Heap":185,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"ON","POWER2":"ON","POWER3":"ON","POWER4":"ON","POWER5":"OFF","POWER6":"ON","POWER7":"ON","POWER8":"OFF","Wifi":{"AP":1,"SSId":"Jc10254ASJ","BSSId":"50:63:21:18:70:9D","Channel":11,"Mode":"11n","RSSI":44,"Signal":-78,"LinkCount":1,"Downtime":"0T00:00:09"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

The error (lost of some config data, like timers, timezone, weblog, poweronstage, etc)  is on power outage or reset. And Weblog turns to 2 every time this happen, so I can't do it.
```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Just unplugg or reset the device by webui.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
I need to fix the timers, timezone, and poweronstage issue.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2021-07-30:
Hi,

Sorry, not a bug. It is a feature. Please check https://tasmota.github.io/docs/Device-Recovery/#fast-power-cycle-device-recovery

Closing this issue as it is **duplicated**. Sorry.

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2021-07-30 by ascillato2

Srjosep, 2021-07-30:
Hi, Thanks for the answer, but SetOption 65 was set to 1, and happen even
for a short one second (and just one) unplug-plugg, and If I restart the
device from the webui. And isn't completely reset, Iit still have GPios
user config, wifi and mqtt join data, and some more.
It's different...

Regards

ascillato, 2021-07-30:
Please, check SetOption36 (Boot Loop protection)

Srjosep, 2021-07-30:
SetOption36 set to 0,, I unplug and plug it again (In less than 2 seconds,
only one time, and tried with WebUI restart button too) and the issue is
the same...
This is the console output, noy sure what is powering all the relays off
when the ESP32 restart... PowerOnState was set to 1 , and after the restart
turns to 3. Anyway, relays were on before the restart....
Timers deleted too, but Wifi, MQTT, GPIOs config, and data (user, pass,
SSID) etc, remains.

"""22:43:53.113 MQT: controlador_acuario_marino/tele/INFO3 =
{"Info3":{"RestartReason":"RTC Watch dog reset digital core and rtc
module"}}
22:43:53.124 MQT: controlador_acuario_marino/stat/RESULT = {"POWER1":"OFF"}
22:43:53.130 MQT: controlador_acuario_marino/stat/POWER1 = OFF
22:43:53.136 MQT: controlador_acuario_marino/stat/RESULT = {"POWER2":"OFF"}
22:43:53.141 MQT: controlador_acuario_marino/stat/POWER2 = OFF
22:43:53.149 MQT: controlador_acuario_marino/stat/RESULT = {"POWER3":"OFF"}
22:43:53.159 MQT: controlador_acuario_marino/stat/POWER3 = OFF
22:43:53.165 MQT: controlador_acuario_marino/stat/RESULT = {"POWER4":"OFF"}
22:43:53.176 MQT: controlador_acuario_marino/stat/POWER4 = OFF
22:43:53.188 MQT: controlador_acuario_marino/stat/RESULT = {"POWER5":"OFF"}
22:43:53.197 MQT: controlador_acuario_marino/stat/POWER5 = OFF
22:43:53.211 MQT: controlador_acuario_marino/stat/RESULT = {"POWER6":"OFF"}
22:43:53.222 MQT: controlador_acuario_marino/stat/POWER6 = OFF
22:43:53.229 MQT: controlador_acuario_marino/stat/RESULT = {"POWER7":"OFF"}
22:43:53.240 MQT: controlador_acuario_marino/stat/POWER7 = OFF
22:43:53.247 MQT: controlador_acuario_marino/stat/RESULT = {"POWER8":"OFF"}
22:43:53.254 MQT: controlador_acuario_marino/stat/POWER8 = OFF""

It is critical to set PowerOnState to 1 because this ESP32- wroom controls
the hardware of an aquarium, and the life inside depend on it turning on
again in case of a power outage.
Timers are not so critical because I can manage the relays by MQTT, but for
security reasons (MQTT broker faillure) will be better in timers will
remain too....

THANKS for your help