arendst/Tasmota #12812: Use of deprecated i2s_write_bytes function call for ESP32

sillyfrog, 2021-08-02:
### PROBLEM DESCRIPTION

Currently the files `AudioOutputI2S.cpp` and `AudioOutputI2SNoDAC.cpp` use the function `i2s_write_bytes`, which has been deprecated, and is removed in the arudino-esp32 v2 release.

Looking at the header file, it suggests to _Use 'i2s_write' instead_.

I'm assuming this was included from an upstream library, so some support maybe required for them to update.

Looking at the [recent commit](https://github.com/arendst/Tasmota/commit/14c4be9d51b962c5be81d6318b77858f2aadf947#diff-57602c72fee2c7bade0c7205cc390a51a8794828979066c5217eeacf16664265L98) that caused this issue, it looks like there was a comment already about this problem: `// Deprecated. Use i2s_write` in the previous version about the function.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): Complie time
- [X] Tasmota binary firmware version number used: Development
  - [ ] Pre-compiled
  - [X] Self-compiled
- [ ] Flashing tools used: N/A
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE

To replicate with the latest release, use the included `platformio_tasmota_cenv_sample.ini`, and compile for the `tasmota32idf4` target, eg:
```
[platformio]
default_envs =
            tasmota32idf4
```

### EXPECTED BEHAVIOUR
Compile

### SCREENSHOTS

### ADDITIONAL CONTEXT

**(Please, remember to close the issue when the problem has been addressed)**

sillyfrog, 2021-08-02:
@s-hadinger I'm tagging you as [this](https://github.com/arendst/Tasmota/commit/14c4be9d51b962c5be81d6318b77858f2aadf947) was the commit that cause the issue.

Hopefully it's an easy fix :)

sillyfrog, 2021-08-02:
This maybe the upstream issue: https://github.com/earlephilhower/ESP8266Audio/issues/401

s-hadinger, 2021-08-02:
Indeed that's an upstream issue. However the fix looks very simple. I will not be able to do it for the next two weeks though.

arendst, 2021-08-02:
I'll fix it (again).

s-hadinger, 2021-08-02:
Thanks. Let's put this fix in the description of the lib to not forget it next time

sillyfrog, 2021-08-02:
Thank you both!

That's compiling as expected now.

closed 2021-08-02 by sillyfrog