arendst/Tasmota #12815: Shelly Dimmer 2 - Power Meters do not increment

shuntfield, 2021-08-02:
### PROBLEM DESCRIPTION
The Shelly Dimmer 2 - Power Metering does not function, it does not increment when there is a load on the device. The Electrical characteristics are obtained such as current voltage etc, but the power total, day, and yesterday does not increment up when the device is on.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Shelly Dimmer 2
- [X] Tasmota binary firmware version number used: 9.5.0.5 
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: MgOS to Tasmota
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
15:08:50.781 MQT: stat/kitchen_canopy/RESULT = {"NAME":"Shelly Dimmer 2","GPIO":[0,3200,0,3232,5568,5600,0,0,193,0,192,0,320,4736],"FLAG":0,"BASE":18}
15:08:50.989 MQT: stat/kitchen_canopy/RESULT = {"Module":{"0":"Shelly Dimmer 2"}}
15:08:51.204 MQT: stat/kitchen_canopy/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"3200":"Serial Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3232":"Serial Rx"},"GPIO4":{"5568":"SHD Boot 0"},"GPIO5":{"5600":"SHD Reset"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"193":"Switch_n2"},"GPIO13":{"0":"None"},"GPIO14":{"192":"Switch_n1"},"GPIO15":{"0":"None"},"GPIO16":{"320":"Led_i1"},"GPIO17":{"4736":"ADC Temp"}}

```
- [X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
15:18:37.421 MQT: stat/kitchen_canopy/RESULT = {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":118,"Free":393,"Rules":"ON Power1#State=1 DO Backlog delay 15; teleperiod 30 ENDON ON Power1#State=0 DO Backlog delay 15; teleperiod 300 ENDON"}}
15:18:37.622 SRC: Backlog
15:18:37.625 CMD: Grp 0, Cmnd 'RULE', Idx 2, Len 0, Data ''
15:18:37.632 MQT: stat/kitchen_canopy/RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
15:18:37.835 SRC: Backlog
15:18:37.837 CMD: Grp 0, Cmnd 'RULE', Idx 3, Len 0, Data ''
15:18:37.844 MQT: stat/kitchen_canopy/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [X] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
15:09:24.022 MQT: stat/kitchen_canopy/STATUS = {"Status":{"Module":0,"DeviceName":"Kitchen Canopy","FriendlyName":["Kitchen Canopy"],"Topic":"kitchen_canopy","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[11,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
15:09:24.039 MQT: stat/kitchen_canopy/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:04:51","StartupUTC":"2021-08-02T14:04:33","Sleep":50,"CfgHolder":4617,"BootCount":24,"BCResetTime":"2021-06-29T14:19:21","SaveCount":390,"SaveAddress":"F9000"}}
15:09:24.052 MQT: stat/kitchen_canopy/STATUS2 = {"StatusFWR":{"Version":"9.5.0.5(tasmota)","BuildDateTime":"2021-08-02T09:35:46","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"428/699"}}
15:09:24.066 MQT: stat/kitchen_canopy/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["DD-Superfast 2.4GHz",""],"TelePeriod":30,"Resolution":"558180C0","SetOption":["00008009","0A05C8000100060000005A0A000000000000","00000080","00006000","00000000"]}}
15:09:24.085 MQT: stat/kitchen_canopy/STATUS4 = {"StatusMEM":{"ProgramSize":609,"Free":392,"Heap":27,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"15701C","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
15:09:24.102 MQT: stat/kitchen_canopy/STATUS5 = {"StatusNET":{"Hostname":"kitchen-canopy","IPAddress":"192.168.0.97","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.0.100","DNSServer2":"0.0.0.0","Mac":"E8:DB:84:D7:BC:15","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
15:09:24.115 MQT: stat/kitchen_canopy/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.110","MqttPort":1883,"MqttClientMask":"kitchen_canopy","MqttClient":"kitchen_canopy","MqttUser":"shuntfield","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":50,"SOCKET_TIMEOUT":20}}
15:09:24.131 MQT: stat/kitchen_canopy/STATUS7 = {"StatusTIM":{"UTC":"2021-08-02T14:09:24","Local":"2021-08-02T15:09:24","StartDST":"2021-03-28T01:00:00","EndDST":"2021-10-31T02:00:00","Timezone":99,"Sunrise":"05:36","Sunset":"20:57"}}
15:09:24.144 MQT: stat/kitchen_canopy/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
15:09:24.161 MQT: stat/kitchen_canopy/STATUS10 = {"StatusSNS":{"Time":"2021-08-02T15:09:24","Switch1":"OFF","Switch2":"OFF","ANALOG":{"Temperature":44.2},"ENERGY":{"TotalStartTime":"2021-07-15T16:26:10","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":4,"ApparentPower":4,"ReactivePower":0,"Factor":1.00,"Voltage":235,"Current":0.016},"Global":{"Temperature":44.2},"TempUnit":"C"}}
15:09:24.177 MQT: stat/kitchen_canopy/STATUS11 = {"StatusSTS":{"Time":"2021-08-02T15:09:24","Uptime":"0T00:04:51","UptimeSec":291,"Heap":27,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":100,"Fade":"ON","Speed":5,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"DD-Superfast 2.4GHz","BSSId":"4C:ED:FB:C3:20:F9","Channel":9,"Mode":"11n","RSSI":66,"Signal":-67,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [X] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
15:09:58.170 CMD: weblog 4
15:09:58.176 MQT: stat/kitchen_canopy/RESULT = {"WebLog":4}
15:09:58.788 CFG: Saved to flash at F8, Count 391, Bytes 4096
15:10:08.167 HTP: Main Menu
15:10:13.325 HTP: Console
15:10:14.389 MQT: tele/kitchen_canopy/STATE = {"Time":"2021-08-02T15:10:14","Uptime":"0T00:05:41","UptimeSec":341,"Heap":27,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":100,"Fade":"ON","Speed":5,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"DD-Superfast 2.4GHz","BSSId":"4C:ED:FB:C3:20:F9","Channel":9,"Mode":"11n","RSSI":64,"Signal":-68,"LinkCount":1,"Downtime":"0T00:00:03"}}
15:10:14.402 MQT: tele/kitchen_canopy/SENSOR = {"Time":"2021-08-02T15:10:14","Switch1":"OFF","Switch2":"OFF","ANALOG":{"Temperature":44.2},"ENERGY":{"TotalStartTime":"2021-07-15T16:26:10","Total":0.000,"Yesterday":0.000,"Today":0.000,"Period":0,"Power":4,"ApparentPower":4,"ReactivePower":0,"Factor":1.00,"Voltage":235,"Current":0.016},"Global":{"Temperature":44.2},"TempUnit":"C"}
15:10:17.002 WIF: Checking connection...
15:10:37.005 WIF: Checking connection...
15:10:44.397 MQT: tele/kitchen_canopy/STATE = {"Time":"2021-08-02T15:10:44","Uptime":"0T00:06:11","UptimeSec":371,"Heap":27,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":100,"Fade":"ON","Speed":5,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"DD-Superfast 2.4GHz","BSSId":"4C:ED:FB:C3:20:F9","Channel":9,"Mode":"11n","RSSI":68,"Signal":-66,"LinkCount":1,"Downtime":"0T00:00:03"}}
15:10:44.412 MQT: tele/kitchen_canopy/SENSOR = {"Time":"2021-08-02T15:10:44","Switch1":"OFF","Switch2":"OFF","ANALOG":{"Temperature":44.3},"ENERGY":{"TotalStartTime":"2021-07-15T16:26:10","Total":0.000,"Yesterday":0.000,"Today":0.000,"Period":0,"Power":4,"ApparentPower":4,"ReactivePower":0,"Factor":1.00,"Voltage":235,"Current":0.016},"Global":{"Temperature":44.3},"TempUnit":"C"}

```

### TO REPRODUCE
Install Tasmota on Shelly Dimmer 2 - I have two of these and neither of them will increment the power metering when they are turned on.
When the the connected load is switched on or dimmed etc, the load meters do not register. 

### EXPECTED BEHAVIOUR
Expect power metering to show power usage today, total and yesterday.

### SCREENSHOTS
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2021-08-25:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you.

Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

shuntfield, 2021-08-25:
Nah, can't really be bothered with the Discord Chat.
I'll leave it, maybe it will get fixed maybe it won't, its not exactly a show stopper anyway, the rest of it works.

timota, 2021-08-30:
same here. Both devices on the same firmware as stated above. Uptime more than 2 days. Energy not updated.

ascillato2, 2021-08-30:
Reopening

arendst, 2021-08-30:
Pls try latest dev branch and report back.

arendst, 2021-08-30:
Still doesn't work. I'll test locally. Hold on.

shuntfield, 2021-08-30:
@arendst I can load a test on mine now

shuntfield, 2021-08-30:
> Still doesn't work. I'll test locally. Hold on.

Yup, confirm 9.5.0.7 - does not increment here.

Thanks 😁

arendst, 2021-08-30:
Should be fixed now.

shuntfield, 2021-08-30:
@arendst what version number is the fix in?
I have just tried 9.5.0.7, is that the latest?

arendst, 2021-08-30:
It's the same version but it is available from the ota website after 20 minutes from now.

shuntfield, 2021-08-30:
@arendst fantastic stuff great work.
I'll try it in an hour, and close when tested thanks again!

shuntfield, 2021-08-30:
Is there a reset that needs to be done? Or a specific version needing to be installed? 9.5.0.7 just installed it OTA and the energy isn't incrementing.

Thanks

shuntfield, 2021-08-30:
All working 👍🏻😁

closed 2021-08-30 by shuntfield

daxy01, 2021-10-01:
Minor correction: This issue resolves Energy usage for both Dimmer 1 and Dimmer 2, not just Dimmer 2.