arendst/Tasmota #12819: MQTT Password Length

eddee54455, 2021-08-04:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Set password to less than 5 characters.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

Allow short passwords.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2021-08-04:
Hi, It is designed with a minimum of 5. Less is too easy to crack. Sorry.

closed 2021-08-04 by ascillato2

eddee54455, 2021-08-05:
Fair comment, but that SHOULD be at user's discretion - In fact, last check on the a later version allowed no password for MQTT...

In my environment, being in a semi rural area, the securities I need are mostly against myself, ie loading firmware and not allowing a unit to log onto the broker until it is fully configured etc...

Please reconsider the proposal to allow any length of password, thereby placing full responsibility in the hands of the person deploying the system.

I am now at a point where I would like to upgrade the firmware, or roll out new units to add to the system, but have the problem of having to reconfigure not only the broker, but then a further 40 plus units to get them to talk... No mean feat considering that some of the units are non Tasmota and are "hard coded" with their wifi and MQTT log in credentials. These have to be physically brought in from the field, some of which are in (to put it mildly) inacessible areas....

TIA
E

ascillato, 2021-08-05:
You don't need to change previous devices. Just add a new user/password in your broker for newer devices.

eddee54455, 2021-08-05:
Agreed, this is a workaround.... But then more credentials on my little network to keep track of... I can barely remember my name as it is!!

Security is not a big issue for this site - I have been trying to keep it as simple as I can as far as possible.... Individual user/pass per device is totally overkill in my scenario and one that I can assure you I definitely would not implement... As to why the minimum password length was implemented, I honestly cannot find a reason barring that some @#$ just decided it would be a good idea without thinking it through... For sure, the shorter the password, the easier it is to crack... But really? .... If you want stronger security, then by all means, employ stronger and longer passwords, but, bear in mind, not everybody "needs" a higher level of security than they have chosen, sometimes the level of security actually suits the environment it is deployed in....

E

ascillato, 2021-08-05:
I totally understand your case and your frustration, and I have tried to give you another solution that is adding a new user/pass in your MQTT broker for your newer devices, that is a very easy thing to do. I'm trying to tell you as polite as I can that your request to lower security is not going to be added. Sorry. Anyway, you already have access to Tasmota code and you can do the modifications you need for achieving your use-case if you want to.

eddee54455, 2021-08-05:
Done as recommended....

excalq, 2022-06-10:
Encountered this as well. A simple work around is, using a desktop browser, right click, "inspect", and change `minlength=`5`` to something lower. Fill in your password, and submit. This gets around the JavaScript validation.

BobKircher, 2023-08-13:
+1 for shorter passwords 
(because the 5 digit limit is set arbitrarily)

andreas-bulling, 2024-01-13:
+1 for shorter passwords - the added security from a 5-char password is minimal

BCsabaEngine, 2024-10-21:
You can set up with a console command: https://tasmota.github.io/docs/MQTT/#configure-mqtt-using-backlog