arendst/Tasmota #12820: Livolo new switches lose pair with Sonoff ZBbridge

systemiii, 2021-08-04:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

I have a Sonoff ZBbridge with Tasmota firmware and I have successfully paired 1 switch 2 gangs (VL-FC2SZ-2WP) and 1 reset switch 2 gangs (VL-FC2HNZ-2WP) of Livolo manifacturer.
I used this configuration in Sono ZB Bridge:
{"Channel":26,"PanID":"0x1A64","ExtPanID":"0x21758D19004B1200","KeyL":"0x9CA8CC84EE7740A1","KeyH":"0x9CA8CC84EE7740A1","TxRadio":20}
The problem is when I turn the switches off and on again, they lose the pairing status and i have to pair them again.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x ] Device used (e.g., Sonoff Basic): Sonoff ZB Bridge
- [ x ] Tasmota binary firmware version number used: 9.5.0.5
  - [ x ] Pre-compiled
  - [ ] Self-compiled
- [ x ] Flashing tools used: file upload by tasmota UI
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
{"GPIO0":{"320":"Led_i1"},"GPIO1":{"3552":"Zigbee Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3584":"Zigbee Rx"},"GPIO4":{"5312":"Zigbee Rst1"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"640":"I2C SDA"},"GPIO13":{"576":"LedLink_i"},"GPIO14":{"608":"I2C SCL"},"GPIO15":{"0":"None"},"GPIO16":{"32":"Button1"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
Grp 0, Cmnd 'STATUS', Idx 1, Len 1, Data '0'
15:53:38.849 RSL: STATUS = {"Status":{"Module":75,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_9CEE67","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
15:53:38.853 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Power On","Uptime":"0T00:02:01","StartupUTC":"2021-08-04T14:51:37","Sleep":50,"CfgHolder":4617,"BootCount":5,"BCResetTime":"2021-08-03T20:54:18","SaveCount":14,"SaveAddress":"1FB000"}}
15:53:38.855 RSL: STATUS2 = {"StatusFWR":{"Version":"9.5.0.5(zbbridge)","BuildDateTime":"2021-08-03T13:38:19","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":160,"Hardware":"ESP8266EX","CR":"351/699"}}
15:53:38.858 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":3,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["wifi-titty",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000080","00006000","00000000"]}}
15:53:38.863 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":641,"Free":1148,"Heap":27,"ProgramFlashSize":2048,"FlashSize":2048,"FlashChipId":"1540A1","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0F1007C4","04400001","00000002","00000000","00000000","00020080","00200000","00000000"],"Drivers":"1,2,7,9,10,12,20,23,38,41,50","Sensors":""}}
15:53:38.866 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-9CEE67-3687","IPAddress":"192.168.1.28","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"84:CC:A8:9C:EE:67","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
15:53:38.869 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_9CEE67","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
15:53:38.873 RSL: STATUS7 = {"StatusTIM":{"UTC":"2021-08-04T14:53:38","Local":"2021-08-04T15:53:38","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"05:28","Sunset":"20:24"}}
15:53:38.876 RSL: STATUS10 = {"StatusSNS":{"Time":"2021-08-04T15:53:38"}}
15:53:38.880 RSL: STATUS11 = {"StatusSTS":{"Time":"2021-08-04T15:53:38","Uptime":"0T00:02:01","UptimeSec":121,"Vcc":3.472,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"wifi-titty","BSSId":"74:DA:88:37:72:A7","Channel":2,"Mode":"11n","RSSI":100,"Signal":-41,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [ x ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
15:54:18.293 CMD: weblog 4
15:54:18.294 SRC: WebConsole from 192.168.1.170
15:54:18.296 CMD: Grp 0, Cmnd 'WEBLOG', Idx 1, Len 1, Data '4'
15:54:18.298 RSL: RESULT = {"WebLog":4}
15:54:18.669 CFG: Saved to flash at FD, Count 15, Bytes 4096
15:54:22.784 WIF: Checking connection...
15:54:26.313 HTP: Consoles
15:54:27.242 HTP: Console
15:54:28.223 HTP: Consoles
15:54:28.899 HTP: Main Menu
15:54:30.175 SRC: WebGui from 192.168.1.170
15:54:30.177 CMD: Grp 0, Cmnd 'ZBPERMITJOIN', Idx 1, Len 0, Data ''
15:54:30.179 ZIG: ZbEZSPSend 22003C
15:54:30.180 ZIG: adding packet to_send, to_ack:4, to_send:4, to_end:4
15:54:30.181 ZIG: ZbEZSPSend 3600FCFF00003600000000000000011E0103013C01
15:54:30.182 ZIG: adding packet to_send, to_ack:4, to_send:4, to_end:5
15:54:30.184 RSL: RESULT = {"ZbState":{"Status":21,"Message":"Pairing mode enabled"}}
15:54:30.185 RSL: RESULT = {"ZbPermitJoin":"Done"}
15:54:30.246 ZIG: Something to_send, to_ack:4, to_send:4, to_end:6
[... 214 lines cut ...]
15:55:42.759 WIF: Checking connection...

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

- Send this configuration to Sonoff ZBbridge {"Channel":26,"PanID":"0x1A64","ExtPanID":"0x21758D19004B1200","KeyL":"0x9CA8CC84EE7740A1","KeyH":"0x9CA8CC84EE7740A1","TxRadio":20}
- Put Sonoff ZBbridge in Permit Join status
- Hold switch button 5 seconds until you hear a beep
- Wait for pairing complete
- Power off switch and then power on

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

I expect no need to repair switch every power cycle of switch

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2021-08-04:
Looking at the logs, those devices send non standard zcl frames. The device seem to repeatedly send a proprietary frame from cluster 0x0001 with an undocumented Frame Control value. 

Unless yiu find some documentation about it, I have no clue what those devices are expecting as a response

s-hadinger, 2021-08-04:
I found it here https://github.com/Koenkk/zigbee-herdsman-converters/blob/1a5489a827e28eaa5300cfe0d429cc64af466113/converters/fromZigbee.js#L2360

This is ugly...

s-hadinger, 2021-08-04:
Final comment. I looked at the protocol and it's so specific and so proprietary that I doubt it will ever be supported in Tasmota. Standards exist for a reason but some vendors don't care.

ascillato2, 2021-08-04:
Closing this issue as it has been answered. Sorry.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2021-08-04 by ascillato2