arendst/Tasmota #12837: Error building tasmota32idf4 with latest development

sillyfrog, 2021-08-08:
### PROBLEM DESCRIPTION

With the latest development release (https://github.com/arendst/Tasmota/commit/0b28019544b922a2cd15d882f818f60a1a51366c), using the `platformio_tasmota_cenv_sample.ini`, you can no longer build `tasmota32idf4`.

From my testing, this PR is the cause of the issue https://github.com/arendst/Tasmota/pull/12830/ (ping @Jason2866 ).

Commenting out this line allows me to build `tasmota32idf4` again:
```diff
--- a/pio-tools/gzip-firmware.py
+++ b/pio-tools/gzip-firmware.py
@@ -27,7 +27,7 @@ def map_gzip(source, target, env):
     # remove map file
     if os.path.isfile(bin_file): os.remove(bin_file)
 
-env.AddPostAction("$BUILD_DIR/${PROGNAME}.bin", [map_gzip])
+# env.AddPostAction("$BUILD_DIR/${PROGNAME}.bin", [map_gzip])
 
 # gzip only for ESP8266
 if env["PIOPLATFORM"] != "espressif32":
```

I have also tried using the latest release at https://github.com/Jason2866/esp32-arduino-lib-builder/releases (just in case - I'm not claiming to understand why it's an issue just for `tasmota32idf4`)

Please let me know if I can do anything to help.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2021-08-08:
I have a look.

closed 2021-08-08 by Jason2866

sillyfrog, 2021-08-09:
Thanks @Jason2866 

I'm testing this now and coming up with something... weird...

I couldn't get it going on my main laptop (MacOS), so have been doing this testing on a clean Ubuntu 20 (docker container `ubuntu:latest`), running on a Linux two different hosts. 

The setup is as follows:
```
apt-get update && apt-get install -y python3-pip git
pip install -U pip && pip install platformio
git clone -b development --single-branch https://github.com/arendst/Tasmota.git
cd Tasmota/
cp platformio_override_sample.ini platformio_override.ini ; cp platformio_tasmota_cenv_sample.ini platformio_tasmota_cenv.ini
```

Doing the following, things worked - once at least - but then I couldn't replicate:
```
rm -rf ~/.platformio
pio run -e tasmota32
pio run -e tasmota32idf4
```

I then moved to doing everything as part of a `docker build` (to make the most of docker caching), and neither of the attached `Dockerfile`s worked. I tried a bunch of things including adding a `-t clean` first (for both `tasmota32` and `tasmota32idf4`), all ended up with the map error. What's super strange is I got it working once - but I can't explain how, nor how to replicate.

I'm not even sure what I'm looking for at this point, it's all very very strange to me.

First `Dockerfile` (no `cp`):
```
FROM ubuntu
RUN apt-get update && apt-get install -y python3-pip git
RUN pip install -U pip && pip install platformio
RUN git clone -b development --single-branch https://github.com/arendst/Tasmota.git tasmota
WORKDIR tasmota
RUN pio run -e tasmota32
RUN pio run -e tasmota32idf4
```

Second `Dockerfile` (the `cp` the `.ini` files, didn't change anything):
```
FROM ubuntu
RUN apt-get update && apt-get install -y python3-pip git
RUN pip install -U pip && pip install platformio
RUN git clone -b development --single-branch https://github.com/arendst/Tasmota.git tasmota
WORKDIR tasmota
RUN cp platformio_override_sample.ini platformio_override.ini && cp platformio_tasmota_cenv_sample.ini platformio_tasmota_cenv.ini
#RUN pio run -e tasmota32
#RUN pio run -e tasmota32idf4 -t clean
#RUN pio run -e tasmota32 -t clean
RUN pio run -e tasmota32idf4
```
The commented out lines made no difference (despite my earlier test on the command line).

Please let me know if there's a better way to demonstrate the issue - I'm way out of my depth on this one!

(I do know Python very well, but not the interactions with platformio. What's the minimum version of Python that needs to be supported? Is v2.7 still in the mix for building?).

Thanks!

arendst, 2021-08-09:
Currently I see every ``platformio_tasmota_cenv.ini`` fails with the OPs error message. Also S2 and C3:

```Linking C:\.pioenvs\tasmota32c3\firmware.elf
Retrieving maximum program size C:\.pioenvs\tasmota32c3\firmware.elf
Checking size C:\.pioenvs\tasmota32c3\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  13.4% (used 44068 bytes from 327680 bytes)
Flash: [=======   ]  69.1% (used 1313542 bytes from 1900544 bytes)
Building C:\.pioenvs\tasmota32c3\firmware.bin
esptool.py v3.1
Merged 2 ELF sections
bin_map_copy(["C:\.pioenvs\tasmota32c3\firmware.bin"], ["C:\.pioenvs\tasmota32c3\firmware.elf"])
map_gzip(["C:\.pioenvs\tasmota32c3\firmware.bin"], ["C:\.pioenvs\tasmota32c3\firmware.elf"])
*** [C:\.pioenvs\tasmota32c3\firmware.bin] build_output\map\tasmota32c3.map: No such file or directory```

As the map files are now being gzipped I guess there's the problem. Strange thing is that envs from ``platformio_tasmota_env32.ini`` run fine.

Jason2866, 2021-08-09:
Strange indeed, since IDF4 generates a map file too. BUT now it is in the folder where the generated firmware is. I tested the change https://github.com/arendst/Tasmota/commit/0ce68b1b6ba1966d33017976e792e1bd23049825 with Windows and Gitpod (Ubuntu) and it generated the gzip map file successfully

arendst, 2021-08-09:
I see you're using a supposed path for the ``map_new_loc`` variable. It may well be that it's not there in our situation.

In my case I have the source files on google drive but the pio build_dir and cache_dir are on my c: drive so that these changes are not be propagated to google wasting precious drive google drive space and network traffic.

Jason2866, 2021-08-09:
Ahh, nice trap. The path generating should work. It is adopted from the way it is done for the firmware. I assumed that the map file has the name `Tasmota` (from src folder name of the project).
Have to find a way to get the folder name which is used for the map file

arendst, 2021-08-09:
Try a wildcard. There is probably only one .map file in the folder.

sillyfrog, 2021-08-09:
@Jason2866 If you're working with path manipulation in Python, may I suggest you check out the (builtin) `pathlib`. I find it super helpful when working with files/paths, has builtin methods to check if files exist, add directories / files together etc, all in a platform agnostic way.

PlatformIO requires Python 3.6, so you're safe with assuming support (it came in with Python 3.4).

I'm also happy to have a look (once you've got to the bottom of this issue :) )

Thank you again both!

Jason2866, 2021-08-09:
@sillyfrog Take a look at https://github.com/arendst/Tasmota/commit/18442c7e57ddd53a5485f67ee9e1f4bbce4368fc
It does work for me. I made a suggestion in the commit what should be done different. A easy change for a python guy (i am not).

Jason2866, 2021-08-09:
Found the part which is generating the map name in the platformio builder: 
`        '-Wl,-Map="%s"' % join("$BUILD_DIR", basename(env.subst("${PROJECT_DIR}.map")))`
The latest commit https://github.com/arendst/Tasmota/commit/4209f7d1cbb5e67b8a61a45829a18a44118b3706 should finally nail it.

sillyfrog, 2021-08-09:
Thank you all! My (quick) testing shows this working now.

When I get a chance I'll try to "Python-ify" the code a bit more. I have just figured out that PlatformIO is using [SCons](https://scons.org/), which may provide me with a few additional way to do what we need. I think I understand what's going on as well. :)

@arendst Would you mind sharing how you change your `build_dir` and `cache_dir`? Do you use an environment variable, or do you modify your `_cenv.ini`? I want to test as many cases as possible.

Thanks!

Jason2866, 2021-08-10:
@sillyfrog  easy take a look here https://docs.platformio.org/en/latest/projectconf/section_platformio.html?highlight=src_dir

arendst, 2021-08-10:
Indeed.

This is the top op of my ``platform_override.ini`` file:

```
[platformio]
build_dir = c:\.pioenvs
build_cache_dir = c:\.cache
; For best Gitpod performance remove the ";" in the next line. Needed Platformio files are cached and installed at first run
;core_dir = .platformio
extra_configs           = platformio_tasmota_cenv.ini
```