arendst/Tasmota #12838: Tasmota dont work if inet-connection is lost

mpt99, 2021-08-08:
### PROBLEM DESCRIPTION
Tasmota will not work if there is no internet connection when the module is booting.
If the connection is lost during operation, the functionality is retained.
This means that if the module starts up during an internet failure, the module will fail! 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): _____Sonoff S20 + Esp Whitty Cloud + Sonoff Basic
- [x ] Tasmota binary firmware version number used: _____9.5.0.5
  - [ x] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____tasmotizer-1.2
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
no input / output possible in the event of a problem

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
no rules

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
no input / output possible in the event of a problem

08:37:54.458 RSL: STATUS = {"Status":{"Module":32,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_00B0D8","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
08:37:54.490 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"External System","Uptime":"0T00:00:15","StartupUTC":"2021-08-08T07:37:39","Sleep":50,"CfgHolder":4617,"BootCount":9,"BCResetTime":"2021-08-08T07:56:20","SaveCount":35,"SaveAddress":"F9000"}}
08:37:54.522 RSL: STATUS2 = {"StatusFWR":{"Version":"9.5.0.5(tasmota)","BuildDateTime":"2021-08-06T19:00:14","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"348/699"}}
08:37:54.542 RSL: STATUS3 = {"StatusLOG":{"SerialLog":4,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["DST63",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000080","00006000","00000000"]}}
08:37:54.571 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":609,"Free":392,"Heap":27,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640E0","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
08:37:54.601 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-00B0D8-4312","IPAddress":"10.50.50.184","Gateway":"10.50.50.1","Subnetmask":"255.255.255.0","DNSServer1":"10.90.10.1","DNSServer2":"10.50.50.1","Mac":"5C:CF:7F:00:B0:D8","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
08:37:54.624 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_00B0D8","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
08:37:54.646 RSL: STATUS7 = {"StatusTIM":{"UTC":"2021-08-08T07:37:54","Local":"2021-08-08T08:37:54","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"05:33","Sunset":"20:17"}}
08:37:54.664 RSL: STATUS10 = {"StatusSNS":{"Time":"2021-08-08T08:37:54"}}
08:37:54.673 RSL: STATUS11 = {"StatusSTS":{"Time":"2021-08-08T08:37:54","Uptime":"0T00:00:15","UptimeSec":15,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Dimmer":10,"Color":"1A1A1A","HSBColor":"0,0,10","Channel":[10,10,10],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"DST63","BSSId":"00:22:B0:A3:14:E8","Channel":7,"Mode":"11n","RSSI":100,"Signal":-38,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

{ll▒▒|▒d▒|
▒
d▒
b|▒▒▒s▒b▒
#▒▒no▒$gn▒▒▒cp▒▒ds$sdp▒g▒▒l▒▒
c
n▒|▒l▒
▒c▒▒g'▒l▒▒d`▒'
l`or▒▒▒n
cd`{▒▒obl ▒rd▒▒|▒
dl▒l▒▒'▒
00:00:00.001 HDW: ESP8266EX
00:00:00.049 CFG: Loaded from flash at FB, Count 33
00:00:00.055 QPC: Count 1
00:00:00.057 CFG: CR 348/699, Busy 0
00:00:00.060 ROT: Mode 1
00:00:00.062 SRC: Restart
00:00:00.065 Project tasmota - Tasmota Version 9.5.0.5(tasmota)-2_7_4_9(2021-08-06T19:00:14)
00:00:00.217 WIF: Checking connection...
[... 369 lines cut ...]
00:00:57.463 RSL: INFO3 = {"Info3":{"RestartReason":{"Exception":29,"Reason":"Exception","EPC":["4000df64","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["4010149c","402659cb","402659b8","4026595f","40264ab8","40264ad5","4026254c","401038fe","4025e8b9","4025ac60","40263243","40000f58","40262cb6","4026c40f","4026bccf","40000f49","40000f49","40000e19","40105a7d","40105a83","4010000d","4026abb4","4026ab65","402502c2","402502c2","402502c2","402502c2","402502c2","402501f8","40250d24","402502c2"]}}}
00:00:57.501 RSL: RESULT = {"POWER":"OFF"}
00:00:57.504 RSL: POWER = OFF
00:00:57.552 QPC: Reset
00:01:12.424 NTP: No reply

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Disconnect from the internet + reboot the module

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
start the module as it worked up to 9.1

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
-

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
-

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2021-08-08:
Tasmota needs a real time source for full operation, that includes accessing a NTP server via wifi. You can have a local NTP server too.

mpt99, 2021-08-08:
sorry, but the module has to start, even if the ntp is missing. As a result, the complete functionality of the module is lost. Then I can leave the original firmware on it right away

s-hadinger, 2021-08-08:
The crash in your logs is definitely a bug that needs to be fixed. We'll need to understand what is happening here and how to reproduce it. 

Keep in mind that most features in Tasmota expect an accurate clock (date and time) that is provisioned by NTP or by a RTC module or any other means (zigbee for ex)

arendst, 2021-08-08:
You're running out of heap space for an unknown reason causing an exception on memcpy during a malloc.

Pls report output of command ``ntpserver``.

mpt99, 2021-08-08:
webserver doesn't work; via serial:
00:01:11.681 CMD: ntpserver
00:01:11.686 RSL: RESULT = {"NtpServer1":"pool.ntp.org","NtpServer2":"nl.pool.ntp.org","NtpServer3":"0.nl.pool.ntp.org"}
00:01:11.709 RSL: INFO1 = {"Info1":{"Module":"Witty Cloud","Version":"9.5.0.5(tasmota)","FallbackTopic":"cmnd/DVES_00B0D8_fb/","GroupTopic":"cmnd/tasmotas/"}}
00:01:11.714 RSL: INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota-00B0D8-4312","IPAddress":"10.50.50.184"}}
00:01:11.724 RSL: INFO3 = {"Info3":{"RestartReason":"Software/System restart"}}

mpt99, 2021-08-08:
Correction:
webserver is running.
but extremely slowly

arendst, 2021-08-08:
I cannot reproduce the loss of heap space and the exception.

What might happen in your case is that the NTP code is too busy resolving ntpservers which it can't by lack of DNS connection.

I changed the code to act on long DNS resolve timeouts (which take 10 seconds on an ESP8266 or 14 seconds on an ESP32) stalling any tasmota activity. You notice this in your webresponses.

With three ntpservers configured by internet name (as you have) it takes 27 seconds trying to resolve these names every minute leading to response hangs.

I see no way to fix these response hangs as the code needs to poll an ntpserver using your defined internet names. 

Workarounds are:
- NOT using internet names but fixed ip addresses. This eliminates the DNS lookups. NTP lookup on ip addresses is almost instantaneous so no response hangs are detected.
- Set you gateway ap address as an NTP server (should work but I cannot test as my Experia box refuses to deliver NTP responses).
- Install a local DNS server (using dnsmasq) and NTP server on your local Home Automation server (as I do).
- Not using NTP by setting a fixed initial time using the ``time`` command. This command could also be provided by your local MQTT server on a regular basis like a NTP server.

mpt99, 2021-08-08:
Hello,
fixed ntp-ip-adresses works fine for me.
So I can use the modul in other locations too.
The gateway as NTP don't work.
But, why has it worked until 9.1?
best regards
mpt

ps:
excuse my bad English

arendst, 2021-08-08:
> But, why has it worked until 9.1?

See CHANGELOG.md

Especially v9.1.0.1 - NTP fallback change.

closed 2021-08-08 by mpt99