arendst/Tasmota #12840: Enhancement - Sensors for power plugs should use "long-term statistics"-sensor to allow energy monitoring in Home Assistant

titiviking, 2021-08-08:
### PROBLEM DESCRIPTION

HA 2021.8 introduces Energy monitoring. This is based on sensors that are setup for long-term statistics.
Today the Tasmota sensors are not setup as such & thererfore cannot be added to the HA Energy Monitoring.

https://developers.home-assistant.io/docs/core/entity/sensor#long-term-statistics.
https://www.home-assistant.io/blog/2021/08/04/home-energy-management/

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2021-08-08:
What changes do you envision done in Tasmota for this? Skimming the first link you posted makes it look like a question of what configuration is done in Home Assistant, only.

ascillato, 2021-08-08:
In Home Assistant, you should go to CONFIGURATION, then ENERGY and there add your devices that you want to use in the new ENERGY TAB of Home Assistant.

From Tasmota side, there is nothing extra that has to be done AFAIK.

ascillato2, 2021-08-08:
Moving to Discussions

closed 2021-08-08 by ascillato2