arendst/Tasmota #12846: No events are fired for hold CLEAR (state=6) when using a button instead of a switch

glennmurley15, 2021-08-08:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

No events are fired for hold CLEAR (state=6) when using a button instead of a switch

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): Sonoff T2
- [ x] Tasmota binary firmware version number used: 9.5.0(tasmota)-2_7_4_9
  - [ X] Pre-compiled
  - [ ] Self-compiled
- [X ] Flashing tools used: USB FTDI
- [x ] Provide the output of command: `Backlog Template; Module; GPIO 255`: 
```lua
  Configuration output here:
22:07:33.475 MQT: stat/livingroom/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
22:07:33.686 MQT: stat/livingroom/RESULT = {"Module":{"29":"Sonoff T1 2CH"}}
22:07:33.892 MQT: stat/livingroom/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"225":"Relay2"},"GPIO9":{"33":"Button2"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
22:08:48.032 MQT: stat/livingroom/RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
22:08:48.259 MQT: stat/livingroom/RESULT = {"Rule2":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":71,"Free":440,"Rules":"on button2#state do Publish stat/togglelivingroomfanlight %value% endon"}}
22:08:48.509 MQT: stat/livingroom/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

22:09:36.181 MQT: stat/livingroom/STATUS = {"Status":{"Module":29,"DeviceName":"LivingRoomLight","FriendlyName":["Fan","Light"],"Topic":"livingroom","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
22:09:36.190 MQT: stat/livingroom/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:52:05","StartupUTC":"2021-08-08T20:17:31","Sleep":50,"CfgHolder":4617,"BootCount":4,"BCResetTime":"2021-08-08T21:16:23","SaveCount":25,"SaveAddress":"FB000"}}
22:09:36.198 MQT: stat/livingroom/STATUS2 = {"StatusFWR":{"Version":"9.5.0(tasmota)","BuildDateTime":"2021-06-17T08:26:35","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"351/699"}}
22:09:36.205 MQT: stat/livingroom/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["DrayTek",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000800B","0305C8000100060000005A0A000000000000","00000080","00006000","00000000"]}}
22:09:36.218 MQT: stat/livingroom/STATUS4 = {"StatusMEM":{"ProgramSize":607,"Free":396,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
22:09:36.227 MQT: stat/livingroom/STATUS5 = {"StatusNET":{"Hostname":"livingroom-0905","IPAddress":"192.168.2.103","Gateway":"192.168.2.254","Subnetmask":"255.255.0.0","DNSServer":"192.168.2.254","Mac":"60:01:94:B2:C3:89","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
22:09:36.234 MQT: stat/livingroom/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.200","MqttPort":1883,"MqttClientMask":"LivingRoomLight","MqttClient":"LivingRoomLight","MqttUser":"sonoff","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
22:09:36.242 MQT: stat/livingroom/STATUS7 = {"StatusTIM":{"UTC":"2021-08-08T21:09:36","Local":"2021-08-08T22:09:36","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"05:33","Sunset":"20:17"}}
22:09:36.248 MQT: stat/livingroom/STATUS10 = {"StatusSNS":{"Time":"2021-08-08T22:09:36"}}
22:09:36.256 MQT: stat/livingroom/STATUS11 = {"StatusSTS":{"Time":"2021-08-08T22:09:36","Uptime":"0T00:52:05","UptimeSec":3125,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"DrayTek","BSSId":"30:B1:B5:08:4F:56","Channel":6,"Mode":"11n","RSSI":100,"Signal":-50,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Use rule to detect button hold AND release

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

Rule is fired when released, eg state=6 (CLEAR)

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2021-08-08:
I don't see why you are expecting "clear" as a button state. According to the trigger [docs](https://tasmota.github.io/docs/Rules/#examples-of-available-triggers), only state 0-3 (off/on/toggle/hold) are available. Clear is only included in switch state.

glennmurley15, 2021-08-08:
I have (since posting this issue) configured the T2 to have button2 as switch2 instead
Used setoption114 to decouple and send MQTT message
Also used same rule (switch2#state) to detect CLEAR

Does not happen
Tried exact rule with switch2#state=6 instead of just firing on all states, and it still doesnt fire

The capacitive touch button works fine, the config is fine after swapping it from button to switch, but still no CLEAR event fired after hold

glennmurley15, 2021-08-08:
I didnt realise I could set a capacitive button to actually be a switch (and im not a tasmota n00b either) - but even changing the template doesnt make the event fire

sfromis, 2021-08-08:
Docs only mention clear being available for a switch config if you use `SwitchMode 11`

glennmurley15, 2021-08-08:
Halfway through asking where in the docs this may be, the comment was edited with a link

After looking where the link goes, I see this
```
Releasing the button also starts an internal timer (time is set in SetOption32). When released for the time set in SetOption32, Tasmota will send a 'delayed' CLEAR command (use Switch<x>#state=6 in rules).
```

Yes, it states that when using the dimmer option of SwitchMode11 the event is available

It does NOT state that this is ONLY available when using this option.

I didn't just jump straight into creating an issue, I've been scouring tasmota docs/wiki/issues/discussions for many hours, with many "Reset 6" commands to start fresh

I do believe this event should be fired in ANY circumstances where a HOLD event happens

ascillato2, 2021-08-08:
> I do believe this event should be fired in ANY circumstances where a HOLD event happens

Nice idea. Moving to feature requests.

closed 2021-08-08 by ascillato2