arendst/Tasmota #12849: Shutter WDT Exception

guidotillema, 2021-08-09:
### PROBLEM DESCRIPTION
I flashed my Wemos D1 Mini with Tasmota to automate my curtains. For this, I used the shutter support (setoption80). It all works perfectly, however, when I send multiple shuttercommands I receive a Soft WDT Reset. So for example, when the curtain is closed and I send a `ShutterOpen` command, while it is opening I send a `ShutterClose` and then while closing I send a `ShutterOpen`, it crashes. This also occurs with the `ShutterPosition` command. So for example, if you continuously move the position slider in Home Assistant, it will also crash. 

### REQUESTED INFORMATION
- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Wemos D1 Mini
- [x] Tasmota binary firmware version number used: 9.4.0 & 9.5.0
  - [x] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: Tasmotizer 1.2
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
{"NAME":"Generic","GPIO":[0,0,0,162,160,161,0,0,163,256,416,352,225,0],"FLAG":0,"BASE":18}
{"Module":{"0":"Generic"}}
{"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"162":"Switch3"},"GPIO4":{"160":"Switch1"},"GPIO5":{"161":"Switch2"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"163":"Switch4"},"GPIO13":{"256":"Relay_i1"},"GPIO14":{"416":"PWM1"},"GPIO15":{"352":"Counter1"},"GPIO16":{"225":"Relay2"},"GPIO17":{"0":"None"}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
None

```
- [x] Provide the output of this command: `Status 0`:
```lua
STATUS = {"Status":{"Module":0,"DeviceName":"Office Curtains","FriendlyName":["Office Curtains","Tasmota2","Tasmota3"],"Topic":"curtains","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[2,2,2,2,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software Watchdog","Uptime":"0T00:15:23","StartupUTC":"2021-08-09T08:20:49","Sleep":50,"CfgHolder":4617,"BootCount":6,"BCResetTime":"2021-08-09T09:13:31","SaveCount":62,"SaveAddress":"F6000"}}
STATUS2 = {"StatusFWR":{"Version":"9.5.0(tasmota)","BuildDateTime":"2021-06-17T08:26:35","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"420/699"}}
STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Guido & Eunbyul",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000A009","2805C8000100060000005A0A000000000000","40000080","00006000","00000001"]}}
STATUS4 = {"StatusMEM":{"ProgramSize":607,"Free":396,"Heap":24,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640EF","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
STATUS5 = {"StatusNET":{"Hostname":"curtains-2128","IPAddress":"192.168.0.137","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"18:FE:34:F3:08:50","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.100","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_F30850","MqttUser":"HomeControlMQTT","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
STATUS7 = {"StatusTIM":{"UTC":"2021-08-09T08:36:12","Local":"2021-08-09T10:36:12","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":99,"Sunrise":"06:17","Sunset":"21:17"}}
STATUS10 = {"StatusSNS":{"Time":"2021-08-09T10:36:12","Switch1":"OFF","Switch2":"OFF","Switch3":"OFF","Switch4":"OFF","COUNTER":{"C1":29999},"Shutter1":{"Position":0,"Direction":0,"Target":0}}}
STATUS11 = {"StatusSTS":{"Time":"2021-08-09T10:36:12","Uptime":"0T00:15:23","UptimeSec":923,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","Dimmer":10,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"Guido & Eunbyul","BSSId":"84:D8:1B:5F:C2:B6","Channel":3,"Mode":"11n","RSSI":94,"Signal":-53,"LinkCount":1,"Downtime":"0T00:00:05"}}}
STATUS12 = {"StatusSTK":{"Exception":4,"Reason":"Software Watchdog","EPC":["402150b4","00000000","402150b7"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["4021502c","4022ebcd","40208162","4024f544","4028e7f0","4023b025","4027f2bb","4023b30c","4021dafc","4027f2a8","4027f234","40280e3d","4023bd42","40280e04","40280ddc","40216ab9","40228811","40245b35","401018eb","40101b73","40208162","40228909","4025a720","40255dfa","4023dbca","4023dc9a","40244c8c","40244ddd","4024c424","40245230","401011da"]}}
STATUS13 = {"StatusSHT":{"SHT0":{"Relay1":1,"Relay2":2,"Open":100,"Close":100,"50perc":50,"Delay":10,"Opt":"0000","Calib":[300,500,700,900,1000],"Mode":"4"}}}

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
**10:45:08.788 CMD: shutteropen**
10:45:08.790 SRC: WebConsole from 192.168.0.124
10:45:08.793 CMD: Grp 0, Cmnd 'SHUTTEROPEN', Idx 1, Len 0, Data ''
10:45:08.796 SHT: Pos. in: payload  (0), payload 100, idx 1, src 7
10:45:08.798 SHT: lastsource 7:, real 0, target 200000, payload 100
10:45:08.800 SHT: Delay Start. var1 <99>=<>, max10s?
10:45:08.802 SHT: Delay Start. Done
10:45:08.803 SRC: Shutter
10:45:08.805 SHT: Switched relay 0 by Shutter
10:45:08.807 SHT: Shtr1, Source Shutter, Powerstate 0, RelayMask 0, ManualChange 0
10:45:08.817 MQT: home/office/curtains/stat/RESULT = {"POWER2":"OFF"}
10:45:08.822 MQT: home/office/curtains/stat/POWER2 = OFF
10:45:08.824 SRC: Shutter
10:45:08.826 SHT: Switched relay 1 by Shutter
10:45:08.828 SHT: Shtr1, Source Shutter, Powerstate 1, RelayMask 1, ManualChange 0
10:45:08.833 MQT: home/office/curtains/stat/RESULT = {"POWER1":"ON"}
10:45:08.839 MQT: home/office/curtains/stat/POWER1 = ON
10:45:08.843 MQT: home/office/curtains/stat/RESULT = {"ShutterOpen1":100}
10:45:09.947 SHT: Time 23, cStop 5500, cVelo 3000, mVelo 3000, aVelo 0, mRun 500, aPos 17480, nStop 22980, Trgt 200000, mVelo 300, Dir 1
10:45:09.997 SHT: Time 24, cStop 5500, cVelo 3000, mVelo 3000, aVelo 0, mRun 500, aPos 18480, nStop 23980, Trgt 200000, mVelo 300, Dir 1
10:45:10.046 SHT: Time 25, cStop 5500, cVelo 3000, mVelo 3000, aVelo 0, mRun 500, aPos 19500, nStop 25000, Trgt 200000, mVelo 300, Dir 1
10:45:10.096 SHT: Time 26, cStop 5500, cVelo 3000, mVelo 3000, aVelo 0, mRun 500, aPos 20480, nStop 25980, Trgt 200000, mVelo 300, Dir 1
10:45:10.146 SHT: Time 27, cStop 5500, cVelo 3000, mVelo 3000, aVelo 0, mRun 500, aPos 21493, nStop 26993, Trgt 200000, mVelo 300, Dir 1
10:45:10.196 SHT: Time 28, cStop 5500, cVelo 3000, mVelo 3000, aVelo 0, mRun 500, aPos 22480, nStop 27980, Trgt 200000, mVelo 300, Dir 1
10:45:10.246 SHT: Time 29, cStop 5500, cVelo 3000, mVelo 3000, aVelo 0, mRun 500, aPos 23486, nStop 28986, Trgt 200000, mVelo 300, Dir 1
10:45:10.295 SHT: Time 30, cStop 5500, cVelo 3000, mVelo 3000, aVelo 0, mRun 500, aPos 24500, nStop 30000, Trgt 200000, mVelo 300, Dir 1
[... 190 lines cut ...]
3ffffe70:  3fff3d74 3fff3d74 3fff3f24 4021e534  
3ffffe80:  00736300 3fff3d74 00ff3d34 4024848d  
3ffffe90:  3fff3f24 3fff3d74 3fff3d34 402255ea  
3ffffea0:  0073632f 9b0627c0 03ff1400 74736f00  
3ffffeb0:  00230000 00006e61 000e4b48 3fff1440  
3ffffec0:  3fff3d74 00000000 4bc6a7f0 00000001  
3ffffed0:  00000001 00000001 401011da c9374bc6  
3ffffee0:  00000000 3fff49ac 3fff3d34 3fff1440  
3ffffef0:  00000001 3fff3d58 3fff3d34 40225704  
3fffff00:  4024d568 00000000 00001388 3ffef988  
3fffff10:  00000000 3fff49ac 000e4b48 3fff1440  
3fffff20:  00000005 00000000 00000005 402258b6  
3fffff30:  00000000 4bc6a7f0 000e4b48 40225905  
3fffff40:  00000000 00000000 4bc6a7f0 00000000  
3fffff50:  00000005 00000000 00000005 40216ab9  
3fffff60:  3fffdad0 00000000 000e4b48 4023a9e5  
3fffff70:  3fffdad0 00000000 000e4b48 4023aaa8  
3fffff80:  00000000 00000000 00000001 40100a50  
3fffff90:  3fffdad0 00000000 3fff1400 3fff1440  
3fffffa0:  3fffdad0 00000000 3fff1400 40249638  
3fffffb0:  feefeffe feefeffe 3ffe86cc 40101e69  
<<<stack<<<
```

**(Please, remember to close the issue when the problem has been addressed)**

stefanbode, 2021-08-09:
Thanks for bringing this up. I can see that you use stepper motors or servos with acceleration functionality. Can you tell which of those you use. I can agree that there should be no reset. The most critical time for a reset is during the ramp phase. Anyhow this is intensively tested. Can you ensure that the problem does not exist if the final speed is already reached. This makes search easier

guidotillema, 2021-08-09:
Dear stefanbode, thank you for your reply. I indeed forgot to mention, I use stepper motors for my curtains. Furthermore, I used the `ShutterMotorDelay 0.5`. I just tested, with `ShutterMotorDelay 0` and the exact same problem was still observed. Therefore, I suppose the problem does not come from the ramp phase. If there is any additional information required, please let me know. If desired, I can also upload a recording.

stefanbode, 2021-08-11:
I am this week on Holiday and can Check Next week. It worked definitely before. As I can see the last command even does not make it to the log files. 

Currently I assume a change in the core that now interfere. Anyhow this needs investigation in the latest build. It will not be easy to find; I assume.

guidotillema, 2021-08-16:
Dear stefanbode, enjoy your holiday ;) You are right, it did not make it to the log files. Meanwhile, I will test a little bit with some settings and different builds to see if I can find anything out of the ordinary. 

If I find anything, I will post it.

stefanbode, 2021-08-22:
Hi, I'm back from holiday and I'm able to reproduce the WDT reset. Looks like a print command with invalid content

stefanbode, 2021-08-22:
I found the bug and hopefully made the smartest change to get it fixed without breaking something new. Please check when the above pull request is merged and compile your new version with the new development branch. The issue was indeed in the decelerate procedure for steppers. While it worked perfectly for planned stops it was a bit planless on emergency stops that happen if you change direction during movement.

closed 2021-08-22 by ascillato2

guidotillema, 2021-09-04:
Hey Stefanbode, I was away for a couple of weeks but I am finally able to test the code. The code does not crash anymore which is very nice, so it resolved the WDT Exception. However, I think there are some strange artefacts happening currently. Where I suspect that the position is updated incorrectly. Therefore I included the following snapshot:

```
20:30:29.237 MQT: home/office/curtains/stat/RESULT = {"POWER2":"ON"}
20:30:29.241 MQT: home/office/curtains/stat/POWER2 = ON
20:30:29.248 MQT: home/office/curtains/stat/RESULT = {"POWER1":"ON"}
20:30:29.256 MQT: home/office/curtains/stat/POWER1 = ON
20:30:29.261 MQT: home/office/curtains/stat/RESULT = {"ShutterClose1":0}
20:30:29.268 MQT: home/office/curtains/stat/RESULT = {"Shutter1":{"Position":100,"Direction":-1,"Target":0}}
20:30:29.414 MQT: home/office/curtains/stat/RESULT = {"Shutter1":{"Position":100,"Direction":-1,"Target":0}}
20:30:30.411 MQT: home/office/curtains/stat/RESULT = {"Shutter1":{"Position":91,"Direction":-1,"Target":0}}
20:30:31.412 MQT: home/office/curtains/stat/RESULT = {"Shutter1":{"Position":81,"Direction":-1,"Target":0}}
20:30:32.418 MQT: home/office/curtains/stat/RESULT = {"Shutter1":{"Position":71,"Direction":-1,"Target":0}}
20:30:33.412 MQT: home/office/curtains/stat/RESULT = {"Shutter1":{"Position":61,"Direction":-1,"Target":0}}
20:30:34.417 MQT: home/office/curtains/stat/RESULT = {"Shutter1":{"Position":51,"Direction":-1,"Target":0}}
20:30:35.413 MQT: home/office/curtains/stat/RESULT = {"Shutter1":{"Position":41,"Direction":-1,"Target":0}}
20:30:36.664 MQT: home/office/curtains/stat/RESULT = {"POWER1":"OFF"}
20:30:36.667 MQT: home/office/curtains/stat/POWER1 = OFF
20:30:36.675 MQT: home/office/curtains/stat/RESULT = {"POWER2":"OFF"}
20:30:36.679 MQT: home/office/curtains/stat/POWER2 = OFF
20:30:36.689 MQT: home/office/curtains/stat/RESULT = {"POWER2":"OFF"}
20:30:36.692 MQT: home/office/curtains/stat/POWER2 = OFF
20:30:36.700 MQT: home/office/curtains/stat/RESULT = {"POWER1":"ON"}
20:30:36.709 MQT: home/office/curtains/stat/POWER1 = ON
20:30:36.713 MQT: home/office/curtains/stat/RESULT = {"ShutterOpen1":100}
20:30:36.720 MQT: home/office/curtains/stat/RESULT = {"Shutter1":{"Position":30,"Direction":1,"Target":100}}
20:30:36.731 MQT: home/office/curtains/stat/RESULT = {"Shutter1":{"Position":30,"Direction":1,"Target":100}}
20:30:37.408 MQT: home/office/curtains/stat/RESULT = {"Shutter1":{"Position":34,"Direction":1,"Target":100}}
20:30:38.421 MQT: home/office/curtains/stat/RESULT = {"Shutter1":{"Position":44,"Direction":1,"Target":100}}
20:30:39.419 MQT: home/office/curtains/stat/RESULT = {"Shutter1":{"Position":54,"Direction":1,"Target":100}}
20:30:40.313 MQT: home/office/curtains/stat/RESULT = {"POWER1":"OFF"}
20:30:40.317 MQT: home/office/curtains/stat/POWER1 = OFF
20:30:40.325 MQT: home/office/curtains/stat/RESULT = {"POWER2":"ON"}
20:30:40.329 MQT: home/office/curtains/stat/POWER2 = ON
20:30:40.335 MQT: home/office/curtains/stat/RESULT = {"POWER1":"ON"}
20:30:40.340 MQT: home/office/curtains/stat/POWER1 = ON
20:30:40.344 MQT: home/office/curtains/stat/RESULT = {"ShutterClose1":0}
20:30:40.355 MQT: home/office/curtains/stat/RESULT = {"Shutter1":{"Position":32,"Direction":-1,"Target":0}}
20:30:40.427 MQT: home/office/curtains/stat/RESULT = {"Shutter1":{"Position":32,"Direction":-1,"Target":0}}
20:30:41.415 MQT: home/office/curtains/stat/RESULT = {"Shutter1":{"Position":24,"Direction":-1,"Target":0}}
20:30:42.417 MQT: home/office/curtains/stat/RESULT = {"Shutter1":{"Position":14,"Direction":-1,"Target":0}}
20:30:43.417 MQT: home/office/curtains/stat/RESULT = {"Shutter1":{"Position":4,"Direction":-1,"Target":0}}
20:30:44.004 MQT: home/office/curtains/stat/RESULT = {"POWER1":"OFF"}
20:30:44.009 MQT: home/office/curtains/stat/POWER1 = OFF
20:30:44.015 MQT: home/office/curtains/stat/RESULT = {"POWER2":"OFF"}
20:30:44.021 MQT: home/office/curtains/stat/POWER2 = OFF
20:30:44.026 MQT: home/office/curtains/stat/SHUTTER1 = 0
20:30:44.031 MQT: home/office/curtains/stat/RESULT = {"Shutter1":{"Position":0,"Direction":0,"Target":0}}
```

Here I start with the curtains open (position 100), then I give the command to close. This all works fine, when I reach about 30% I command the curtain to open again (at 20:30:36.713). This also works perfectly. However, when the curtains are about 54% (at 20:30:39.419) I command it to close again, but then it takes the position of 32 as can be seen at 20:30:40.355. I suspect that the position is not updating correctly after the first time, or am I mistaken?

stefanbode, 2021-09-04:
Very interesting. And agree that this should not happen. I will have some time tomorrow to reproduce and solve.Again thanks for bringing it up. Only with your feedback we can make it better. 

Sent from remote phone

guidotillema, 2021-09-05:
Hey Stefanbode, if you need any additional information for reproducing this issue just let me know. I forgot to mention, but I was using firmware 9.5.0.8

stefanbode, 2021-09-05:
I just submitted the small fix. As soon as arendst merged it into developemt you can try. I'm 100% sure this use-case is now working fine. If you want to compile on your own just change the one line in the stepper driver
#1088
```
OLD:        Shutter[index].start_position = Shutter[index].target_position;
NEW:       Shutter[index].target_position = Shutter[index].start_position;
```

guidotillema, 2021-09-05:
Hey Stefanbode, thank you for your fix. It took me a while but I got it working in PlatformIO and was able to compile the code myself. You are completely right, the position is now working correctly. However, I think this fix introduces a small issue with the shutter target value. I included a snap to explain what I mean:

```
19:30:33.058 MQT: home/office/curtains/stat/RESULT = {"POWER2":"OFF"}
19:30:33.062 MQT: home/office/curtains/stat/POWER2 = OFF
19:30:33.067 MQT: home/office/curtains/stat/RESULT = {"POWER1":"ON"}
19:30:33.071 MQT: home/office/curtains/stat/POWER1 = ON
19:30:33.074 MQT: home/office/curtains/stat/RESULT = {"ShutterOpen1":100}
19:30:33.083 MQT: home/office/curtains/stat/RESULT = {"Shutter1":{"Position":0,"Direction":1,"Target":100}}
19:30:33.607 MQT: home/office/curtains/stat/RESULT = {"Shutter1":{"Position":3,"Direction":1,"Target":100}}
19:30:34.606 MQT: home/office/curtains/stat/RESULT = {"Shutter1":{"Position":13,"Direction":1,"Target":100}}
19:30:35.613 MQT: home/office/curtains/stat/RESULT = {"Shutter1":{"Position":23,"Direction":1,"Target":100}}
19:30:36.603 MQT: home/office/curtains/stat/RESULT = {"Shutter1":{"Position":33,"Direction":1,"Target":100}}
19:30:37.763 MQT: home/office/curtains/stat/RESULT = {"POWER1":"OFF"}
19:30:37.767 MQT: home/office/curtains/stat/POWER1 = OFF
19:30:37.775 MQT: home/office/curtains/stat/RESULT = {"POWER2":"ON"}
19:30:37.779 MQT: home/office/curtains/stat/POWER2 = ON
19:30:37.786 MQT: home/office/curtains/stat/RESULT = {"POWER1":"ON"}
19:30:37.790 MQT: home/office/curtains/stat/POWER1 = ON
19:30:37.793 MQT: home/office/curtains/stat/RESULT = {"ShutterClose1":0}
19:30:37.799 MQT: home/office/curtains/stat/RESULT = {"Shutter1":{"Position":43,"Direction":-1,"Target":0}}
19:30:37.878 MQT: home/office/curtains/stat/RESULT = {"Shutter1":{"Position":43,"Direction":-1,"Target":0}}
19:30:38.608 MQT: home/office/curtains/stat/RESULT = {"Shutter1":{"Position":38,"Direction":-1,"Target":0}}
19:30:39.612 MQT: home/office/curtains/stat/RESULT = {"Shutter1":{"Position":28,"Direction":-1,"Target":0}}
19:30:40.603 MQT: home/office/curtains/stat/RESULT = {"Shutter1":{"Position":18,"Direction":-1,"Target":0}}
19:30:41.013 MQT: home/office/curtains/stat/RESULT = {"POWER1":"OFF"}
19:30:41.018 MQT: home/office/curtains/stat/POWER1 = OFF
19:30:41.023 MQT: home/office/curtains/stat/RESULT = {"POWER2":"OFF"}
19:30:41.030 MQT: home/office/curtains/stat/POWER2 = OFF
19:30:41.037 MQT: home/office/curtains/stat/RESULT = {"POWER2":"OFF"}
19:30:41.041 MQT: home/office/curtains/stat/POWER2 = OFF
19:30:41.051 MQT: home/office/curtains/stat/RESULT = {"POWER1":"ON"}
19:30:41.055 MQT: home/office/curtains/stat/POWER1 = ON
19:30:41.059 MQT: home/office/curtains/stat/RESULT = {"ShutterOpen1":100}
19:30:41.066 MQT: home/office/curtains/stat/RESULT = {"Shutter1":{"Position":15,"Direction":1,"Target":43}}
19:30:41.600 MQT: home/office/curtains/stat/RESULT = {"Shutter1":{"Position":18,"Direction":1,"Target":43}}
19:30:42.606 MQT: home/office/curtains/stat/RESULT = {"Shutter1":{"Position":28,"Direction":1,"Target":43}}
19:30:43.606 MQT: home/office/curtains/stat/RESULT = {"Shutter1":{"Position":38,"Direction":1,"Target":43}}
19:30:44.390 MQT: home/office/curtains/stat/RESULT = {"POWER1":"OFF"}
19:30:44.394 MQT: home/office/curtains/stat/POWER1 = OFF
19:30:44.398 MQT: home/office/curtains/stat/SHUTTER1 = 43
19:30:44.402 MQT: home/office/curtains/stat/RESULT = {"Shutter1":{"Position":43,"Direction":0,"Target":43}}
```

I again started with the curtains close, then I give a shutteropen command (at 19:30:33.074) which gives a target of 100, while it is opening I give a shutter close command (at 19:30:37.793) which gives a target of 0. While it is closing I give a shutteropen command (at 19:30:41.059), which gives a target of 43. However, here I would expect that the target would be 100 instead of 43. The result is that the shutter stops at 43% even though I gave a shutter open command.

stefanbode, 2021-09-05:
Indeed the problem is more complex than initially thought. But not unsolvable.

stefanbode, 2021-09-05:
I now reverted back not to mess with the start and target position anymore. This was good before. Now used a different method to avoid the WDT reset.

guidotillema, 2021-09-06:
Hey Stefanbode, I just tested the code. Everything seems to be working correctly ;) Very nice. Thanks for the effort.

PR #13072 (2021-09-06): Update xdrv_27_shutter.ino
#12849 Additional side effect correction

## Description:

**Related issue (if applicable):** fixes #12849 

## Checklist:
  - [ x] The pull request is done against the latest development branch
  - [x ] Only relevant files were touched
  - [x ] Only one feature/fix was added per PR and the code change compiles without warnings
  - [ x] The code change is tested and works with Tasmota core ESP8266 V.2.7.4.9
  - [x ] The code change is tested and works with Tasmota core ESP32 V.1.0.7.3
  - [x ] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CON [... cut]