arendst/Tasmota #12851: Tuya switch reports correctly but wont toggle over web interface

nickg90, 2021-08-09:
### PROBLEM DESCRIPTION
WIFI Tuya switch status can be read over web interface but cannot write using toggle button on web interface. when operated through RF wallswitch, it behaves as expected

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Quinetic QURW03 (smarthome based)
- [x] Tasmota binary firmware version number used: 9.5.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: TuyaConvert OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
13:09:48.856 CMD: Backlog Template; Module; GPIO 255
13:09:48.859 SRC: WebConsole from 192.168.0.50
13:09:48.861 CMD: Grp 0, Cmnd 'BACKLOG', Idx 1, Len 26, Data 'Template; Module; GPIO 255'
13:09:48.946 SRC: Backlog
13:09:48.948 CMD: Grp 0, Cmnd 'TEMPLATE', Idx 1, Len 0, Data ''
13:09:48.955 MQT: stat/tasmota_012B80/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
13:09:49.158 SRC: Backlog
13:09:49.160 CMD: Grp 0, Cmnd 'MODULE', Idx 1, Len 0, Data ''
13:09:49.165 MQT: stat/tasmota_012B80/RESULT = {"Module":{"54":"Tuya MCU"}}
13:09:49.409 SRC: Backlog
13:09:49.411 CMD: Grp 0, Cmnd 'GPIO', Idx 1, Len 3, Data '255'
13:09:49.421 MQT: stat/tasmota_012B80/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"2272":"Tuya Tx"},"GPIO2":{"0":"None"},"GPIO3":{"2304":"Tuya Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
13:10:40.342 CMD: Grp 0, Cmnd 'STATUS', Idx 1, Len 1, Data '0'
13:10:40.349 MQT: stat/tasmota_012B80/STATUS = {"Status":{"Module":54,"DeviceName":"Kitchen","FriendlyName":["Kitchen"],"Topic":"tasmota_012B80","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
13:10:40.355 MQT: stat/tasmota_012B80/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:28:58","StartupUTC":"2021-08-09T11:41:42","Sleep":50,"CfgHolder":4617,"BootCount":14,"BCResetTime":"2021-08-09T12:16:53","SaveCount":62,"SaveAddress":"F6000"}}
13:10:40.362 MQT: stat/tasmota_012B80/STATUS2 = {"StatusFWR":{"Version":"9.5.0(tasmota)","BuildDateTime":"2021-06-17T08:26:35","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"337/699"}}
13:10:40.367 MQT: stat/tasmota_012B80/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Searching...",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000080","00006000","00000000"]}}
13:10:40.380 MQT: stat/tasmota_012B80/STATUS4 = {"StatusMEM":{"ProgramSize":607,"Free":396,"Heap":24,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
13:10:40.386 MQT: stat/tasmota_012B80/STATUS5 = {"StatusNET":{"Hostname":"Kitchen","IPAddress":"192.168.0.207","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"24:62:AB:01:2B:80","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
13:10:40.391 MQT: stat/tasmota_012B80/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.100","MqttPort":1883,"MqttClientMask":"Kitchen","MqttClient":"Kitchen","MqttUser":"hiomqtt2","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
13:10:40.398 MQT: stat/tasmota_012B80/STATUS7 = {"StatusTIM":{"UTC":"2021-08-09T12:10:40","Local":"2021-08-09T13:10:40","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"05:35","Sunset":"20:15"}}
13:10:40.406 MQT: stat/tasmota_012B80/STATUS10 = {"StatusSNS":{"Time":"2021-08-09T13:10:40"}}
13:10:40.413 MQT: stat/tasmota_012B80/STATUS11 = {"StatusSTS":{"Time":"2021-08-09T13:10:40","Uptime":"0T00:28:58","UptimeSec":1738,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Searching...","BSSId":"84:D8:1B:E2:04:CF","Channel":9,"Mode":"11n","RSSI":62,"Signal":-69,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
13:11:49.475 QPC: Reset
13:11:51.236 HTP: Main Menu
13:11:51.412 APP: Boot Count 15
13:11:51.593 CFG: Saved to flash at F5, Count 63, Bytes 4096
13:11:52.427 MQT: tele/tasmota_012B80/STATE = {"Time":"2021-08-09T13:11:52","Uptime":"0T00:00:11","UptimeSec":11,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":21,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Searching...","BSSId":"84:D8:1B:E2:04:CF","Channel":9,"Mode":"11n","RSSI":60,"Signal":-70,"LinkCount":1,"Downtime":"0T00:00:05"}}
13:11:52.438 TYA: Send "55aa00000000ff"
13:11:52.978 HTP: Console
13:11:57.420 MQT: tasmota/discovery/2462AB012B80/config = {"ip":"192.168.0.207","dn":"Kitchen","fn":["Kitchen",null,null,null,null,null,null,null],"hn":"Kitchen","mac":"2462AB012B80","md":"Tuya MCU","ty":1,"if":0,"ofln":"Offline","onln":"Online","state":["OFF","ON","TOGGLE","HOLD"],"sw":"9.5.0","t":"tasmota_012B80","ft":"%prefix%/%topic%/","tp":["cmnd","stat","tele"],"rl":[1,0,0,0,0,0,0,0],"swc":[-1,-1,-1,-1,-1,-1,-1,-1],"swn":[null,null,null,null,null,null,null,null],"btn":[0,0,0,0,0,0,0,0],"so":{"4":0,"11":0,"13":0,"17":0,"20":0,"30":0,"68":0,"73":0,"82":0,"114":0,"117":0},"lk":1,"lt_st":0,"sho":[0,0,0,0],"ver":1} (retained)
13:11:57.425 MQT: tasmota/discovery/2462AB012B80/sensors = {"sn":{"Time":"2021-08-09T13:11:57"},"ver":1} (retained)
13:12:00.394 CMD: weblog 4
13:12:00.396 SRC: WebConsole from 192.168.0.50
13:12:00.399 CMD: Grp 0, Cmnd 'WEBLOG', Idx 1, Len 1, Data '4'
13:12:00.404 MQT: stat/tasmota_012B80/RESULT = {"WebLog":4}
[... 36 lines cut ...]
13:12:20.955 SRC: WebGui from 192.168.0.50
13:12:20.970 TYA: Send "55aa0006000501010001000d"
13:12:20.994 MQT: stat/tasmota_012B80/RESULT = {"POWER":"OFF"}
13:12:20.997 MQT: stat/tasmota_012B80/POWER = OFF
13:12:21.590 CFG: Saved to flash at F7, Count 69, Bytes 4096
13:12:23.107 HTP: Console

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
in the above I toggled the light on the web interface twice (no response from switch). I actuated the manual wallswitch to on then off (which did operate the switch) and then I repeated toggling the light on web interface (again no response on the switch)

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
I expected the light to switch on and off 3 times but only saw it happen once. when I do operate the wall switch, I do see the value change in the web interface

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
I have attempted changing the Tuya TX pin to all other possibilities with no success. I have tried using tuyasend1 103,1 and tuyasend1 1,1 but this has no result

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2021-11-01:
The root problem of this issue is the same as https://github.com/arendst/Tasmota/issues/11349

Let's hope that **TheEebb** can fix it.

Thanks for reporting.

closed 2021-11-01 by ascillato2

nickg90, 2021-11-11:
not sure if anything has changed by my switch has started responding again though I havnt changed the firmware in a while and it wasnt working last time I tried. now on 9.1.0

lorenzoconforti, 2022-10-30:
@nickg90, I too have a Quinetic QURW03. Have you manged to get it working reliably now? Thanks

nickg90, 2022-10-30:
> @nickg90, I too have a Quinetic QURW03. Have you manged to get it working reliably now? Thanks

Yep, all been working fine since my last post :)

lorenzoconforti, 2022-10-30:
@nickg90 great. Which Tasmota firmware version are you using?