arendst/Tasmota #12860: PWMDimmerPWMs setting prevents PWM dimmer brightness LEDs from working properly

Tediore, 2021-08-11:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
When the `PWMDimmerPWMs` option is set to anything other than 1, the brightness LEDs on the PWM dimmer do not behave properly. When the `dimmer` value is changed, all brightness LEDs except the bottom one turn off and then a few seconds later the brightness LEDs change to correctly reflect the brightness level. They also do not fade to the correct brightness level even if `fade 1` is set.

The brightness LEDs work as expected when `PWMDimmerPWMs` is set to 1.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Martin Jerry SD01
- [x] Tasmota binary firmware version number used: Tasmota 9.5.0.5 (latest dev build when this issue was opened)
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: Tuya-convert
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
22:34:28.649 MQT: stat/office_lights/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
22:34:28.854 MQT: stat/office_lights/RESULT = {"Module":{"73":"PWM Dimmer"}}
22:34:29.060 MQT: stat/office_lights/RESULT = {"GPIO0":{"34":"Button3"},"GPIO1":{"33":"Button2"},"GPIO2":{"0":"None"},"GPIO3":{"323":"Led_i4"},"GPIO4":{"576":"LedLink_i"},"GPIO5":{"322":"Led_i3"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"321":"Led_i2"},"GPIO13":{"416":"PWM1"},"GPIO14":{"320":"Led_i1"},"GPIO15":{"96":"Button_i1"},"GPIO16":{"256":"Relay_i1"},"GPIO17":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
N/A

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
22:35:12.849 MQT: stat/office_lights/STATUS = {"Status":{"Module":73,"DeviceName":"Office lights dimmer","FriendlyName":["Office"],"Topic":"office_lights","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":0,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
22:35:12.856 MQT: stat/office_lights/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:06:51","StartupUTC":"2021-08-11T03:28:21","Sleep":50,"CfgHolder":4617,"BootCount":30,"BCResetTime":"2021-07-16T01:43:11","SaveCount":569,"SaveAddress":"FA000"}}
22:35:12.860 MQT: stat/office_lights/STATUS2 = {"StatusFWR":{"Version":"9.5.0.5(tasmota)","BuildDateTime":"2021-08-09T16:34:59","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"382/699"}}
22:35:12.865 MQT: stat/office_lights/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Tasmota",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00128009","2805C8000100060000005A0A000000000000","000002C0","00006008","00000000"]}}
22:35:12.876 MQT: stat/office_lights/STATUS4 = {"StatusMEM":{"ProgramSize":609,"Free":392,"Heap":22,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
22:35:12.884 MQT: stat/office_lights/STATUS5 = {"StatusNET":{"Hostname":"office-lights-3159","IPAddress":"10.0.4.207","Gateway":"10.0.4.1","Subnetmask":"255.255.255.0","DNSServer1":"10.0.0.1","DNSServer2":"0.0.0.0","Mac":"DC:4F:22:7A:8C:57","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
22:35:12.889 MQT: stat/office_lights/STATUS6 = {"StatusMQT":{"MqttHost":"tediore.xyz","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_7A8C57","MqttUser":"matt","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
22:35:12.895 MQT: stat/office_lights/STATUS7 = {"StatusTIM":{"UTC":"2021-08-11T03:35:12","Local":"2021-08-10T22:35:12","StartDST":"2021-03-14T02:00:00","EndDST":"2021-11-07T02:00:00","Timezone":99,"Sunrise":"05:54","Sunset":"19:56"}}
22:35:12.901 MQT: stat/office_lights/STATUS10 = {"StatusSNS":{"Time":"2021-08-10T22:35:12"}}
22:35:12.909 MQT: stat/office_lights/STATUS11 = {"StatusSTS":{"Time":"2021-08-10T22:35:12","Uptime":"0T00:06:51","UptimeSec":411,"Heap":22,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":30,"Color":"0,0,0,0,77","HSBColor":"0,0,0","White":30,"CT":500,"Channel":[0,0,0,0,30],"Scheme":0,"Fade":"ON","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"Tasmota","BSSId":"1E:E8:29:91:C1:84","Channel":6,"Mode":"11n","RSSI":86,"Signal":-57,"LinkCount":1,"Downtime":"0T00:00:07"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
38:05.685 SRC: WebConsole from 10.0.1.44
22:38:05.688 CMD: Grp 0, Cmnd 'DIMMER', Idx 1, Len 2, Data '20'
22:38:05.699 MQT: tele/office_lights/STATE = {"Time":"2021-08-10T22:38:05","Uptime":"0T00:09:44","UptimeSec":584,"Heap":25,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":20,"Color":"0,0,0,0,51","HSBColor":"0,0,0","White":20,"CT":500,"Channel":[0,0,0,0,20],"Scheme":0,"Fade":"ON","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"Tasmota","BSSId":"1E:E8:29:91:C1:84","Channel":6,"Mode":"11n","RSSI":84,"Signal":-58,"LinkCount":1,"Downtime":"0T00:00:07"}}
22:38:05.706 MQT: stat/office_lights/RESULT = {"POWER":"ON","Dimmer":20,"Color":"0,0,0,0,51","HSBColor":"0,0,0","White":20,"CT":500,"Channel":[0,0,0,0,20]}
22:38:05.723 DGR: Sending office_lights message to network: seq=11, flags=0, 10=51, 5=51
22:38:05.890 DGR: Received office_lights message from 10.0.4.227: seq=11, flags=8
22:38:05.892 DGR: Received office_lights message from 10.0.4.81: seq=11, flags=8
22:38:05.900 DGR: Received office_lights message from 10.0.4.82: seq=11, flags=8
22:38:05.930 DGR: Sending office_lights message to 10.0.4.48: seq=11, flags=0, last ack=10
22:38:05.933 DGR: Sending office_lights message to 10.0.4.72: seq=11, flags=0, last ack=10
22:38:05.980 DGR: Received office_lights message from 10.0.4.48: seq=11, flags=8
22:38:05.982 DGR: Received office_lights message from 10.0.4.72: seq=11, flags=8
22:38:05.990 DGR: Received office_lights message from 10.0.4.48: seq=11, flags=8
22:38:06.002 DGR: Received office_lights message from 10.0.4.72: seq=11, flags=8
22:38:07.511 DGR: Received office_lights message from 10.0.4.82: seq=33, flags=64
22:38:07.626 CFG: Saved to flash at F5, Count 574, Bytes 4096

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
1. Change `PWMDimmerPWMs` to any value other than 1
2. Change the `dimmer` value

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
The brightness LEDs should change directly to the new `dimmer` value

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
Video of the behavior attached (`dimmer` value changed from 30 to 60)

https://user-images.githubusercontent.com/49540618/128967000-bbc65d74-13bf-459a-a6ba-4a2eea47025b.mp4

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
N/A

**(Please, remember to close the issue when the problem has been addressed)**

pcdiem, 2021-08-11:
Fixed by https://github.com/arendst/Tasmota/pull/12862

closed 2021-08-11 by arendst

Tediore, 2021-08-16:
So, unfortunately I've discovered another issue with this. I didn't test it with `ct` values other than the max or min in the `ctrange`. When I set the dimmer to a `ct` value between the min/max (for example, 300), the same behavior shown in video in my original message occurs. Based on a conversation with pcdiem right after I opened the original issue, I understand that the brightness LEDs are set based on the value of the highest PWM, so I could see why color temps between fully warm white and fully cool white cause this issue. Not sure if this is a big enough issue to actually try to fix, but I did just want to make it known.

PR #12862 (2021-08-11): Set LEDs to max PWM value
## Description:

The PWM Dimmer brightness LEDs were being set to the value of the highest-numbered PWM pin in use. When PWMDimmerPWMs is set, we want the LEDs to be set to the highest PWM value of all the virtual PWMs even though there is only one real PWM.

**Related issue (if applicable):** fixes #12860

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works wit [... cut]