arendst/Tasmota #12865: I2C Detect: ADS1115 was detected instead LM75ad

petrkr, 2021-08-11:
On latest stock tasmota32 firmware is collision in i2c devices discovery.

I tried wire-up LM75 temperature sensor, but tasmota detected it as ADS1115.

I goes thrught this docs https://tasmota.github.io/docs/LM75AD/ but seems if there is more drivers enabled (which are in stock firmware) then first driver takes lead.

```
00:00:00.100 Project tasmota Tasmota Version 9.5.0(tasmota)-1_0_6(2021-06-17T08:29:40)
00:00:00.148 I2C: ADS1115 found at 0x49
```

Would be nice somehow disable drivers in runtime (for peoples who are not familiar with build process) or select which driver to use for detected I2C devices.

closed 2021-08-11 by close-issue-app[bot]

ascillato, 2021-08-11:
That is already available.

Please, check for `I2CDriver` command in https://tasmota.github.io/docs/Commands/

More information in the docs at https://tasmota.github.io/docs/I2CDevices/

spcqike, 2021-11-11:
Hi all,

i just want to bring this topic back.
i tried to use a LM75, but its not working.

with the normal tasmota_sensors for esp8266, the LM75 is recognized as DS1624. when i deactivate this sensor via i2cdriver42 0, tasmota does not recognize any sensor. i2cscan shows me 0x4f, which is fine.

when i follow [this ](http://www.esp8266learning.com/esp8266-lm75-temperature-sensor-example.php)tutorial, i can use the sensor, so the hardware works.

may it be possible, that there is a bug somewhere, so that LM75 isn't working right now?

barbudor, 2021-11-11:
https://tasmota.github.io/docs/BUILDS/
LM75 is not included in tasmota-sensors. You have to compile your own build.
Or look at https://github.com/tasmota/install/tree/main/firmware/unofficial

spcqike, 2021-11-11:
isn't LM75 the same as LM75AD, which is included in sensors?

barbudor, 2021-11-11:
Yes but it is not included in tasmota-sensors.bin
It is in tasmota32-sensors.bin
In  `-/x` the signa before the `/` is for ESP8266 and the one after is for ESP32

spcqike, 2021-11-11:
yes, but thats only for the "t" (tasmota) column. in "s" (sensors) is jsut one `x`

see also:
https://tasmota.github.io/docs/LM75AD/
`This feature is included only in tasmota-sensors.bin`

barbudor, 2021-11-11:
Whats is the result of `I2Cdriver`
Is I2Cdriver 20 present and enabled ?

barbudor, 2021-11-11:
Yes all LM75 and TMP175 are interoperable
Just newer design or different manufacturer

spcqike, 2021-11-12:
> 
> 
> Whats is the result of `I2Cdriver` Is I2Cdriver 20 present and enabled ?

i think so.

`11:27:46.242 CMD: i2cdriver
11:27:46.248 RSL: RESULT = {"I2CDriver":"8,9,10,15,20,21,32,36,41,!42,43,45,51,52,60"}`

barbudor, 2021-11-12:
Please disable every drivers except 20
As Tasmota will reboot, you should use a backlog to limit the number of reboot (and time spent):
`backlog i2cdriver8 0; i2cdriver9 0; i2cdriver10 0 .....`

spcqike, 2021-11-15:
still no sensor and values

`07:30:32.123 CMD: i2cdriver`
`07:30:32.128 RSL: RESULT = {"I2CDriver":"!8,!9,!10,!15,20,!21,!32,!36,!41,!42,!43,!45,!51,!52,!60"}`

barbudor, 2021-11-15:
```
00:00:00.065 I2C: LM75AD found at 0x48
...
22:27:16.776 CMD: i2cdriver
22:27:16.782 MQT: stat/nodemcu/I2CDRIVER = {"I2CDriver":"9,10,11,12,14,16,18,20,24,40,45"}
22:27:26.426 CMD: i2cscan
22:27:26.448 MQT: stat/nodemcu/I2CSCAN = {"I2CScan":"Device(s) found at 0x48"}
...
22:27:40.512 MQT: tele/nodemcu/SENSOR = {"Time":"2021-11-15T22:26:40+01:00","LM75AD":{"Temperature":19.5},"TempUnit":"C"} (retained)
```

Based on tasmota latest dev 10.0.0.3
NodeMCU, 
I2C SCL/SDA on D2/D1. No external pull-up.

Let me check if I can change the address from 0x48 to 0x4F

barbudor, 2021-11-15:
```
00:00:00.066 I2C: LM75AD found at 0x4f
...
22:37:54.517 MQT: tele/nodemcu/SENSOR = {"Time":"2021-11-15T22:37:54+01:00","LM75AD":{"Temperature":26.0},"TempUnit":"C"} (retained)
22:37:59.394 CMD: i2cscan
22:37:59.415 MQT: stat/nodemcu/I2CSCAN = {"I2CScan":"Device(s) found at 0x4f"}
```

So all as expected
No idea what is the problem on your side
Do you have external pull-ups ?
I didn't need ones because I'm on very short wires, but still a pair of external 4.7K on SCL and SDA are highly advised.

spcqike, 2021-11-16:
Hi,

yes, i have external pull-up resistors (4.7K). i have a plain LM75 from national semiconductor on a prototype board with pullups to SCL/SDA (pin 4 and 5 on the [feather board](https://www.adafruit.com/product/2821))and capacitor between VCC and GND. 

this LM75 doesnt work with [arduinos library](https://www.arduino.cc/reference/en/libraries/i2c-temperature-sensors-derived-from-the-lm75/) either, but i think thats because of the esp8266, as the esp crashes with some exception when i try to use the included examples.

however, i can use the sensor with the library, that is used [here](http://www.esp8266learning.com/esp8266-lm75-temperature-sensor-example.php).

so i guess i switch to another sensor :) maybe i order a LM75A (with 0.125°C resolution, not the LM75 with 0.5°C that i have now) and try again :)