arendst/Tasmota #12877: AIThinker ESP32 Camera Module very high temperature

redguyz, 2021-08-12:
### PROBLEM DESCRIPTION
AIThinker ESP32 Camera Module overheats. The temperature can reach more than 90 deg Celsius. Any suggestion?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): AI Thinker ESP32 Camera Module
- [ ] Tasmota binary firmware version number used: 9.5.0.5(webcam)
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: esptool.py
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
22:41:53.895 CMD: Backlog Template; Module; GPIO 255
22:41:53.921 RSL: RESULT = {"NAME":"AITHINKER CAM","GPIO":[4992,1,672,1,416,5088,1,1,1,6720,608,640,1,1,5089,5090,0,5091,5184,5152,0,5120,5024,5056,0,0,0,0,4928,576,5094,5095,5092,0,0,5093],"FLAG":0,"BASE":2}
22:41:54.119 RSL: RESULT = {"Module":{"0":"AITHINKER CAM"}}
22:41:54.331 RSL: RESULT = {"GPIO0":{"4992":"CAM_XCLK"},"GPIO1":{"0":"None"},"GPIO2":{"672":"SPI MISO1"},"GPIO3":{"0":"None"},"GPIO4":{"416":"PWM1"},"GPIO5":{"5088":"CAM_DATA1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"6720":"SDCard CS"},"GPIO14":{"608":"I2C SCL1"},"GPIO15":{"640":"I2C SDA1"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"5089":"CAM_DATA2"},"GPIO19":{"5090":"CAM_DATA3"},"GPIO20":{"0":"None"},"GPIO21":{"5091":"CAM_DATA4"},"GPIO22":{"5184":"CAM_PCLK"},"GPIO23":{"5152":"CAM_HREF"},"GPIO24":{"0":"None"},"GPIO25":{"5120":"CAM_VSYNC"},"GPIO26":{"5024":"CAM_SIOD"},"GPIO27":{"5056":"CAM_SIOC"},"GPIO28":{"0":"None"},"GPIO29":{"0":"None"},"GPIO30":{"0":"None"},"GPIO31":{"0":"None"},"GPIO32":{"4928":"CAM_PWDN"},"GPIO33":{"576":"LedLink_i"},"GPIO34":{"5094":"CAM_DATA7"},"GPIO35":{"5095":"CAM_DATA8"},"GPIO36":{"5092":"CAM_DATA5"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"5093":"CAM_DATA6"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
22:58:38.007 CMD: Grp 0, Cmnd 'STATUS', Idx 1, Len 1, Data '0'
22:58:38.010 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_B19D28","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
22:58:38.012 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/tasmota32-webcam.bin","RestartReason":"Software reset CPU","Uptime":"0T01:01:14","StartupUTC":"2021-08-12T20:57:24","Sleep":0,"CfgHolder":4617,"BootCount":24,"BCResetTime":"2021-08-10T19:06:57","SaveCount":81}}
22:58:38.015 RSL: STATUS2 = {"StatusFWR":{"Version":"9.5.0.5(webcam)","BuildDateTime":"2021-08-12T18:42:36","Core":"1_0_7_3","SDK":"v3.3.5-138-g714f27f56","CpuFrequency":240,"Hardware":"ESP32-D0WD","CR":"377/699"}}
22:58:38.017 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["LSB2",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008001","2805C8000100060000005A0A000000000000","00000080","00006000","00000000"]}}
22:58:38.025 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":1156,"Free":1856,"Heap":127,"PsrMax":4095,"PsrFree":4035,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC78F","00140001","000000CF","010013C0","40000981","80004080","00200000","00000024"],"Drivers":"1,2,3,4,5,7,8,9,10,12,20,21,24,26,27,29,38,50,81","Sensors":"1,2,3,5,6,127"}}
22:58:38.030 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-B19D28-7464","IPAddress":"192.168.1.123","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"95.87.194.5","Mac":"B8:F0:09:B1:9D:28","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
22:58:38.036 RSL: STATUS7 = {"StatusTIM":{"UTC":"2021-08-12T21:58:38","Local":"2021-08-12T22:58:38","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"05:39","Sunset":"20:10"}}
22:58:38.041 RSL: STATUS10 = {"StatusSNS":{"Time":"2021-08-12T22:58:38","ESP32":{"Temperature":72.8},"Global":{"Temperature":72.8},"TempUnit":"C"}}
22:58:38.046 RSL: STATUS11 = {"StatusSTS":{"Time":"2021-08-12T22:58:38","Uptime":"0T01:01:14","UptimeSec":3674,"Heap":127,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"MqttCount":0,"POWER":"OFF","Dimmer":50,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"LSB2","BSSId":"08:55:31:CD:E3:69","Channel":11,"Mode":"11n","RSSI":42,"Signal":-79,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
22:48:51.313 RSL: STATE = {"Time":"2021-08-12T22:48:51","Uptime":"0T00:51:27","UptimeSec":3087,"Heap":129,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"MqttCount":0,"POWER":"OFF","Dimmer":54,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"LSB2","BSSId":"08:55:31:CD:E3:69","Channel":11,"Mode":"11n","RSSI":50,"Signal":-75,"LinkCount":1,"Downtime":"0T00:00:04"}}
22:48:51.317 RSL: SENSOR = {"Time":"2021-08-12T22:48:51","ESP32":{"Temperature":76.1},"Global":{"Temperature":76.1},"TempUnit":"C"}
22:48:52.717 HTP: Main Menu
22:48:52.863 CAM: Handle camserver
22:48:52.864 CAM: Create client
22:48:52.865 CAM: Start stream
22:48:56.867 CAM: Frame fail
22:48:56.868 CAM: Stream exit
22:48:56.881 HTP: Consoles
22:48:58.813 HTP: Console

```

### TO REPRODUCE
Erased with flash_download_tool_3.8.5.exe. Flashed with esptool.py
[image]
.

### EXPECTED BEHAVIOUR
Lower working temperature.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2021-08-12:
Hi, I have the same device and the temperature is normal. Seems to be hardware issue. You should ask for a replacement to your seller before it burns.

redguyz, 2021-08-12:
Hi. Thanks for answering. Actually I use 3 identical modules. Previously they worked normal with ESP32 Cam Webserver. Now they are flashed with Tasmota. Two of them show high temperature. The third one is OK. Strangely, the mentioned two were firstly erased with ESP download tool. The good working one was flashed directly from ESP32 Cam Webserver to Tasmota.

Jason2866, 2021-08-21:
Please try latest version with PR #12936

redguyz, 2021-08-24:
Unfortunately no change with 9.5.0.7.  Web interface is a bit more responsive. The temperature is still very high. Currently the readings are around 90 deg C.

ascillato, 2021-08-24:
Please, try to erase all flash and flash again

redguyz, 2021-08-25:
Well, flashed the module after full erase. The result is not good. Still have readings around 90+ deg C...

ascillato, 2021-08-25:
And if you touch it, it feels too hot or just warm?

redguyz, 2021-08-25:
It feels hot. But not too much. Maybe around 40-50C. I don't believe the chip inside to be 90C. That's why I'm saying readings. Is it possible to be a program error for calculating the temperature? Or maybe another type / faulty thermal sensor?

Wolvverine, 2022-01-24:
Have the same issue with temperature.