arendst/Tasmota #12894: ZBStatus3 to device MQTT Topic

gonzzzales, 2021-08-16:
Hi, thanks a lot for your produkt, it is realy great!

I have ZbBridge  with some ZigBee sensors. I configure MQTT topics, so every sensor publish his state in own topic 
tele/tasmota/456F/SENSOR = {"ZbReceived":{"0x456F":{"Device":"0x456F","Occupancy":1,"Endpoint":1,"LinkQuality":110}}}, but it happens only when a change of sensor values occurs. I want to force it, but dont understand, how can i make it.
Command ZbStatus3 0x456F published in topic stat/tasmota/RESULT, so i cant determine without parsing, from which sensor is it.

Is it possible to force ZbBridge to send the state of sensor in his own topic?
Regards,
Alex

closed 2021-08-16 by close-issue-app[bot]

s-hadinger, 2021-08-16:
Sorry but you can't. This is not the way Zigbee works. The device sends periodically values. Tasmota does memorize some, but if you need to know the current state you need to track it at gateway level.

gonzzzales, 2021-08-16:
ok, thanks for the answer