arendst/Tasmota #12911: WebUI slow over Internet

sebsebseb1982, 2021-08-17:
### PROBLEM DESCRIPTION
Web UI is working perfectly on LAN
Web UI is very slow through Internet. UI load fast but led control, CPU temperature and video feed is very slow to show.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): [Ai-Thinker](https://fr.aliexpress.com/item/32999991632.html?spm=a2g0s.9042311.0.0.27426c37vMB03G)
- [x] Tasmota binary firmware version number used: [tasmota32-webcam.bin](http://ota.tasmota.com/tasmota32/release/tasmota32-webcam.bin) 9.5.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: esptool.py
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
18:24:12.603 CMD: Backlog Template; Module; GPIO 255
18:24:12.605 SRC: WebConsole from 192.168.1.12
18:24:12.606 CMD: Grp 0, Cmnd 'BACKLOG', Idx 1, Len 26, Data 'Template; Module; GPIO 255'
18:24:12.875 SRC: Backlog
18:24:12.878 CMD: Grp 0, Cmnd 'TEMPLATE', Idx 1, Len 0, Data ''
18:24:12.886 RSL: RESULT = {"NAME":"AITHINKER CAM","GPIO":[4992,1,672,1,416,5088,1,1,1,6720,736,704,1,1,5089,5090,0,5091,5184,5152,0,5120,5024,5056,0,0,0,0,4928,576,5094,5095,5092,0,0,5093],"FLAG":0,"BASE":2}
18:24:13.113 SRC: Backlog
18:24:13.115 CMD: Grp 0, Cmnd 'MODULE', Idx 1, Len 0, Data ''
18:24:13.118 RSL: RESULT = {"Module":{"0":"AITHINKER CAM"}}
18:24:13.363 SRC: Backlog
18:24:13.365 CMD: Grp 0, Cmnd 'GPIO', Idx 1, Len 3, Data '255'
18:24:13.384 RSL: RESULT = {"GPIO0":{"4992":"CAM_XCLK"},"GPIO1":{"0":"None"},"GPIO2":{"672":"SPI MISO1"},"GPIO3":{"0":"None"},"GPIO4":{"416":"PWM1"},"GPIO5":{"5088":"CAM_DATA1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"6720":"SDCard CS"},"GPIO14":{"736":"SPI CLK1"},"GPIO15":{"704":"SPI MOSI1"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"5089":"CAM_DATA2"},"GPIO19":{"5090":"CAM_DATA3"},"GPIO20":{"0":"None"},"GPIO21":{"5091":"CAM_DATA4"},"GPIO22":{"5184":"CAM_PCLK"},"GPIO23":{"5152":"CAM_HREF"},"GPIO24":{"0":"None"},"GPIO25":{"5120":"CAM_VSYNC"},"GPIO26":{"5024":"CAM_SIOD"},"GPIO27":{"5056":"CAM_SIOC"},"GPIO28":{"0":"None"},"GPIO29":{"0":"None"},"GPIO30":{"0":"None"},"GPIO31":{"0":"None"},"GPIO32":{"4928":"CAM_PWDN"},"GPIO33":{"576":"LedLink_i"},"GPIO34":{"5094":"CAM_DATA7"},"GPIO35":{"5095":"CAM_DATA8"},"GPIO36":{"5092":"CAM_DATA5"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"5093":"CAM_DATA6"}}
18:24:22.633 RSL: STATE = {"Time":"2021-08-17T18:24:22","Uptime":"0T00:15:43","UptimeSec":943,"Heap":120,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Dimmer":100,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"Sebastien","BSSId":"CC:2D:E0:30:76:8F","Channel":1,"Mode":"11n","RSSI":100,"Signal":-40,"LinkCount":1,"Downtime":"0T00:00:06"}}
18:24:22.638 RSL: SENSOR = {"Time":"2021-08-17T18:24:22","ESP32":{"Temperature":53.3},"TempUnit":"C"}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
18:15:51.052 SRC: WebConsole from 192.168.1.12
18:15:51.053 CMD: Grp 0, Cmnd 'STATUS', Idx 1, Len 1, Data '0'
18:15:51.057 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_984050","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
18:15:51.085 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T00:07:12","StartupUTC":"2021-08-17T17:08:39","Sleep":50,"CfgHolder":4617,"BootCount":10,"BCResetTime":"2021-08-16T21:24:38","SaveCount":60}}
18:15:51.119 RSL: STATUS2 = {"StatusFWR":{"Version":"9.5.0(webcam)","BuildDateTime":"2021-06-17T08:31:13","Core":"1_0_6","SDK":"v3.3.5-1-g85c43024c","CpuFrequency":240,"Hardware":"ESP32-D0WD","CR":"362/699"}}
18:15:51.135 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":4,"LogHost":"","LogPort":514,"SSId":["Sebastien",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008001","2805C80001000600001E5A0A000000000000","00000080","00006000","00000000"]}}
18:15:51.164 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":1391,"Free":1856,"Heap":120,"PsrMax":4095,"PsrFree":3345,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC78F","00140001","000000CF","010013C0","40000981","80004000","00200000","00000024"],"Drivers":"1,2,3,4,5,7,8,9,10,12,20,21,24,26,27,29,50,81","Sensors":"1,2,3,5,6,87"}}
18:15:51.194 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota_984050-0080","IPAddress":"192.168.1.234","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"F4:CF:A2:98:40:50","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
18:15:51.217 RSL: STATUS7 = {"StatusTIM":{"UTC":"2021-08-17T17:15:51","Local":"2021-08-17T18:15:51","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"05:46","Sunset":"20:01"}}
18:15:51.237 RSL: STATUS10 = {"StatusSNS":{"Time":"2021-08-17T18:15:51","ESP32":{"Temperature":53.3},"TempUnit":"C"}}
18:15:51.248 RSL: STATUS11 = {"StatusSTS":{"Time":"2021-08-17T18:15:51","Uptime":"0T00:07:12","UptimeSec":432,"Heap":120,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":23,"MqttCount":0,"POWER":"OFF","Dimmer":100,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"Sebastien","BSSId":"CC:2D:E0:30:76:8F","Channel":1,"Mode":"11n","RSSI":100,"Signal":-42,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
No interesting log.

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Flash http://ota.tasmota.com/tasmota32/release/tasmota32-webcam.bin 9.5.0 on [Ai-Thinker camera](https://fr.aliexpress.com/item/32999991632.html?spm=a2g0s.9042311.0.0.27426c37vMB03G).
Set-up Wifi.
On your router, forward port xxxxx to esp32's port 80.
Connect through lan : no problem.
Connect over internet : very very slow.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Web UI should be a little less fast over FTTH

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2021-08-17:
First, you shouldn't do this. Tasmota uses HTTP with no encryption. It shouldn't leave your local network. You are going through the internet unencrypted...

Tasmota is very limited in RAM so it sends HTTP responses in 'chunked mode'. This means that small portion of response are sent through the network to make sure we don't buffer too much. This means a lot of round-trips between Tasmota and the web client. On a LAN it's not a problem, on a high latency network it is.

However you have maybe another problem on top of this one.

There's nothing we can do. Tasmota is optimized for MQTT.

sebsebseb1982, 2021-08-17:
> First, you shouldn't do this. Tasmota uses HTTP with no encryption. It shouldn't leave your local network. You are going through the internet unencrypted...

Yup, I know but there is nothing critical shown by this camera.

> On a LAN it's not a problem, on a high latency network it is.
Even over FTTH ?

s-hadinger, 2021-08-17:
Then you may have another problem. Using the developer tools of your browser might give you a hint about the problem

sebsebseb1982, 2021-08-17:
> Using the developer tools of your browser might give you a hint about the problem

Good idea ! I think that I've found the problem ;

Web page try to load /stream from LAN IP even througt Internet through external DNS/IP

`html
<center>[image]</center>
`
Could img src be relative and not absolute ?

Jason2866, 2021-08-17:
Maybe interesting to read https://github.com/arendst/Tasmota/discussions/12523

sebsebseb1982, 2021-08-17:
My assumption is correct. WebUI works well through VPN over Internet and local IP.

> Maybe interesting to read #12523

Thank you for this. 
If stream must be served on 81 port, it would be great (as said earlier) if stream hostname could be overriden

ascillato2, 2021-08-17:
Moving to Discussions

closed 2021-08-17 by ascillato2