arendst/Tasmota #12929: Compilation Error of Tasmota 9.5 in Arduino IDE

smhmaqsood, 2021-08-20:
I've recently downloaded the Tasmota 9.5.0 and it is not compiling. 
It shows the error that Ping_t was not declared.
I updated all the libraries and yet it is not compiling.

closed 2021-08-20 by close-issue-app[bot]

Jason2866, 2021-08-20:
Arduino IDE is not supported anymore

smhmaqsood, 2021-08-23:
Which IDE is supported.
I'm tring to compile it with VS Code & PlatformIO But I'm getting this error

AttributeError: 'SConsEnvironment' object has no attribute 'ElfToBin':
  File "C:\Users\mhassan.hw\.platformio\penv\Lib\site-packages\platformio\builder\main.py", line 177:    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\mhassan.hw\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 591:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\mhassan.hw\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\mhassan.hw\.platformio\platforms\espressif8266@2.6.3\builder\main.py", line 247:    
    target_firm = env.ElfToBin(
==================================== [FAILED] Took 21.18 seconds ====================================
Environment    Status    Duration
-------------  --------  ------------
tasmota        FAILED    00:00:21.180
=============================== 1 failed, 0 succeeded in 00:00:21.180 ===============================The terminal process "C:\Users\mhassan.hw\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

Jason2866, 2021-08-23:
VSC with PlatformIO plugin is the recommended way. Your setup seems screwed.

smhmaqsood, 2021-08-24:
What are the recommended settings ?

ascillato, 2021-08-24:
> What are the recommended settings ?

The default settings are the recommended ones. VSCode + PIO are the supported IDE.

Latest Tasmota is compiling fine. Try to delete your Platformio and reinstall again. Also, download Tasmota again. Please, start from scratch. Seems that your PIO is broken or your setup is missing files.

If you need further assistance, please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md)

Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)