arendst/Tasmota #12931: Energy Today with SDM120 keeps reseting

gajotnt, 2021-08-20:
### PROBLEM DESCRIPTION
Energy today keeps reseting during the day, tried several different sonoffs, always same result.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Sonoff TH10
- [X] Tasmota binary firmware version number used: 9.5.0 (sensors)
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: Tasmotize
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
12:11:01.832 CMD: Backlog Template; Module; GPIO 255
12:11:01.903 MQT: stat/PowCasaTH/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
12:11:02.114 MQT: stat/PowCasaTH/RESULT = {"Module":{"4":"Sonoff TH"}}
12:11:02.370 MQT: stat/PowCasaTH/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"1952":"SDMx20 Rx"},"GPIO2":{"0":"None"},"GPIO3":{"1920":"SDMx20 Tx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [X] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
12:11:36.587 CMD: Status 0
12:11:36.595 MQT: stat/PowCasaTH/STATUS = {"Status":{"Module":4,"DeviceName":"PowCasaTH","FriendlyName":["PowCasaTH"],"Topic":"PowCasaTH","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
12:11:36.600 MQT: stat/PowCasaTH/STATUS1 = {"StatusPRM":{"Baudrate":2400,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T10:19:42","StartupUTC":"2021-08-20T00:51:54","Sleep":50,"CfgHolder":4617,"BootCount":17,"BCResetTime":"2021-08-20T01:15:07","SaveCount":51,"SaveAddress":"F9000"}}
12:11:36.607 MQT: stat/PowCasaTH/STATUS2 = {"StatusFWR":{"Version":"9.5.0(sensors)","BuildDateTime":"2021-06-17T08:26:50","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"340/699"}}
12:11:36.612 MQT: stat/PowCasaTH/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["TNT-iot",""],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00008009","2805C8000100068000005A0A000000000000","00400080","00006000","00000000"]}}
12:11:36.631 MQT: stat/PowCasaTH/STATUS4 = {"StatusMEM":{"ProgramSize":661,"Free":340,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440E0","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDA8787","0415A005","B7F7BFCF","05DA9BC4","64367CC7","00084052","20000000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,62,64,66,67,74"}}
12:11:36.640 MQT: stat/PowCasaTH/STATUS5 = {"StatusNET":{"Hostname":"PowCasaTH","IPAddress":"192.168.20.39","Gateway":"192.168.20.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.20.1","Mac":"60:01:94:17:2D:86","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
12:11:36.646 MQT: stat/PowCasaTH/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.149","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_172D86","MqttUser":"DVES_USER","MqttCount":2,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
12:11:36.653 MQT: stat/PowCasaTH/STATUS7 = {"StatusTIM":{"UTC":"2021-08-20T11:11:36","Local":"2021-08-20T12:11:36","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"05:50","Sunset":"19:56"}}
12:11:36.657 MQT: stat/PowCasaTH/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
12:11:36.666 MQT: stat/PowCasaTH/STATUS10 = {"StatusSNS":{"Time":"2021-08-20T12:11:36","ENERGY":{"TotalStartTime":"2021-08-20T00:00:00","Total":15117.239,"Yesterday":0.000,"Today":0.053,"ExportActive":304.273,"Power":710,"ApparentPower":725,"ReactivePower":-124,"Factor":0.99,"Frequency":50,"Voltage":237,"Current":3.027,"ImportActive":15117.239,"ImportReactive":5975.281,"ExportReactive":781.424,"PhaseAngle":0.00}}}
12:11:36.673 MQT: stat/PowCasaTH/STATUS11 = {"StatusSTS":{"Time":"2021-08-20T12:11:36","Uptime":"0T10:19:42","UptimeSec":37182,"Heap":19,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":2,"POWER":"OFF","Wifi":{"AP":1,"SSId":"TNT-iot","BSSId":"1A:E8:29:C1:8F:0D","Channel":1,"Mode":"11n","RSSI":74,"Signal":-63,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [X] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
12:27:54.589 MQT: domoticz/in = {"idx":102,"nvalue":0,"svalue":"666;15117431.0","Battery":100,"RSSI":7}
12:27:54.594 MQT: tele/PowCasaTH/SENSOR = {"Time":"2021-08-20T12:27:54","ENERGY":{"TotalStartTime":"2021-08-20T00:00:00","Total":15117.431,"Yesterday":0.000,"Today":0.102,"ExportActive":304.273,"Period":12,"Power":667,"ApparentPower":701,"ReactivePower":-118,"Factor":0.99,"Frequency":50,"Voltage":236,"Current":2.785,"ImportActive":15117.432,"ImportReactive":5975.312,"ExportReactive":781.426,"PhaseAngle":0.00}}
12:27:54.806 DMP: 01 04 04 44 29 33 24 2A 57
12:27:55.058 DMP: 01 04 04 C2 EA CC CD 73 5D
12:27:55.313 DMP: 01 04 04 3F 7C 1D 47 7F 2A
12:27:55.510 DMP: 01 04 04 42 48 00 00 6F EA
12:27:55.763 DMP: 01 04 04 46 70 F6 D3 E9 2A
12:27:56.019 DMP: 01 04 04 46 6C 35 BB 79 F2
12:27:56.291 DMP: 01 04 04 43 98 22 F2 F6 CA
12:27:56.540 DMP: 01 04 04 45 BA BA 7F FD ED
12:27:56.791 DMP: 01 04 04 44 43 5B 44 25 A3
12:27:57.042 DMP: 01 04 04 00 00 00 00 FB 84
12:27:57.298 DMP: 01 04 04 43 6B 66 66 34 56
12:27:57.547 DMP: 01 04 04 40 49 FB E7 3C E8
12:27:57.798 DMP: 01 04 04 44 2E 59 9A 35 46
12:27:58.017 DMP: 01 04 04 44 33 C4 F8 4C 39
12:27:58.277 DMP: 01 04 04 C2 F8 00 00 47 CD
12:27:58.526 DMP: 01 04 04 3F 7C 28 B1 E8 3C
12:28:00.272 DMP: 01 04 04 42 48 00 00 6F EA
12:28:00.528 DMP: 01 04 04 46 70 F6 D4 A8 E8
[... 54 lines cut ...]
12:28:30.304 DMP: 01 04 04 46 6C 35 C3 79 D0
12:28:30.551 DMP: 01 04 04 43 98 22 F2 F6 CA
12:28:30.763 DMP: 01 04 04 45 BA BA 81 7C 6D
12:28:30.869 HTP: Console

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Just a SDM120 with a RS485TTL to give me the info i need on powerusage.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Correct counting of the power usage.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

Maybe my RS485 TLL isnt working right? 
I noticed the "12:28:12.542 NRG: Energy reset by invalid data", but that happens several times and doesnt reset the today counter everytime it happens.

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2021-08-20:
You are using software serial. Change to hardware serial as shown here https://github.com/jarcikw/SDM120_Homeassistant/wiki/SDM120-&-nodemcu-&-RS485-wiring

gajotnt, 2021-08-20:
Thanks for the reply.
I have the wires connected to the programing pins of the Sonoff TH (3.3V, GND, TX, RX).
Either way, i put it on "Generic" module instead of "Sonoff TH" to see if anything changes.
[image]

It still resets, i will order a new TTL module, its probably not working right. Or do i have to put the TTL RX on the RX of the sonoff?

```lua
15:36:53.396 MQT: tele/PowCasaTH/STATE = {"Time":"2021-08-20T15:36:53","Uptime":"0T00:12:28","UptimeSec":748,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"TNT-iot","BSSId":"1A:E8:29:C1:8F:0D","Channel":1,"Mode":"11n","RSSI":74,"Signal":-63,"LinkCount":1,"Downtime":"0T00:00:03"}}
15:36:53.407 MQT: domoticz/in = {"idx":102,"nvalue":0,"svalue":"588;15119278.0","Battery":100,"RSSI":7}
15:36:53.413 MQT: tele/PowCasaTH/SENSOR = {"Time":"2021-08-20T15:36:53","ENERGY":{"TotalStartTime":"2021-08-20T00:00:00","Total":15119.278,"Yesterday":0.000,"Today":0.081,"ExportActive":304.391,"Period":12,"Power":588,"ApparentPower":594,"ReactivePower":-12,"Factor":1.00,"Frequency":50,"Voltage":234,"Current":2.763,"ImportActive":15119.278,"ImportReactive":5975.939,"ExportReactive":781.485,"PhaseAngle":0.00}}
15:36:54.518 DMP: 01 04 04 C1 9A 66 66 4D DD
15:36:56.257 DMP: 01 04 04 3F 7F E8 DB C8 13
15:36:58.046 DMP: 01 04 04 42 47 CC CD CB 7C
15:36:58.296 DMP: 01 04 04 46 70 FE B0 AE C3
15:36:58.509 DMP: 01 04 04 46 6C 3D 20 3F 99
15:36:58.773 DMP: 01 04 04 43 98 32 0C 7A 8A
15:36:59.023 DMP: 01 04 04 45 BA BF 83 FE FC
15:37:00.784 DMP: 01 04 04 44 43 5F 0A A7 57
15:37:02.511 DMP: 01 04 04 00 00 00 00 FB 84
15:37:05.798 DMP: 01 04 04 43 6A 19 9A 45 E7
15:37:06.050 DMP: 01 04 04 40 1F 4B C7 A8 E0
15:37:06.299 DMP: 01 04 04 44 15 A6 66 05 3A
15:37:08.017 DMP: 01 04 04 44 28 69 05 81 2F
15:37:08.280 DMP: 01 04 04 C1 9F 33 33 A2 B3
15:37:08.547 DMP: 01 04 04 3F 7F E3 26 0E A2
15:37:09.047 WIF: Checking connection...
15:37:10.259 DMP: 01 04 04 42 48 28 F6 F1 AC
15:37:10.508 DMP: 01 04 04 46 70 FE B2 2F 02
15:37:12.256 DMP: 01 04 04 46 6C 3D 22 BE 58
15:37:15.522 DMP: 01 04 04 43 98 32 0C 7A 8A
15:37:15.779 DMP: 01 04 04 45 BA BF 83 FE FC
15:37:16.025 DMP: 01 04 04 44 43 5F 0A A7 57
15:37:16.278 DMP: 01 04 04 00 00 00 00 FB 84
15:37:16.531 DMP: 01 04 04 43 6A 00 00 CE 1C
15:37:18.253 DMP: 01 04 04 40 39 78 D5 DD D6
15:37:18.511 DMP: 01 04 04 44 1B C6 66 4C F9
15:37:18.769 DMP: 01 04 04 44 1B E3 C0 D6 13
15:37:19.030 DMP: 01 04 04 C1 B8 00 00 46 5D
15:37:19.282 DMP: 01 04 04 3F 7F D3 31 5A AC
15:37:21.031 DMP: 01 04 04 42 48 00 00 6F EA
15:37:22.799 DMP: 01 04 04 46 70 FE B4 AF 00
15:37:26.044 DMP: 01 04 04 46 6C 3D 25 FF 9A
15:37:27.757 DMP: 01 04 04 43 98 32 0C 7A 8A
15:37:28.005 DMP: 01 04 04 45 BA BF 85 7E FE
15:37:29.047 WIF: Checking connection...
15:37:29.802 DMP: 01 04 04 44 43 5F 0A A7 57
15:37:30.047 DMP: 01 04 04 00 00 00 00 FB 84
15:37:30.301 DMP: 01 04 04 43 6A 00 00 CE 1C
15:37:30.502 DMP: 01 04 04 40 32 C0 83 5E 2A
15:37:30.760 DMP: 01 04 04 44 1D 73 33 1A 57
15:37:31.010 DMP: 01 04 04 44 21 1C E8 B7 F0
15:37:31.262 DMP: 01 04 04 C1 9D 99 9A BD AD
15:37:33.039 DMP: 01 04 04 3F 7F DC BE 1E F8
15:37:33.297 DMP: 01 04 04 42 48 00 00 6F EA
15:37:33.502 DMP: 01 04 04 46 70 FE B6 2E C1
15:37:33.761 DMP: 01 04 04 46 6C 3D 26 BF 9B
15:37:35.506 DMP: 01 04 04 43 98 32 0C 7A 8A
15:37:35.762 DMP: 01 04 04 45 BA BF 85 7E FE
15:37:36.009 DMP: 01 04 04 44 43 5F 0A A7 57
15:37:36.262 DMP: 01 04 04 00 00 00 00 FB 84
15:37:36.521 DMP: 01 04 04 43 6A 33 33 9A F9
15:37:36.776 DMP: 01 04 04 40 30 72 B0 CB 5F
15:37:37.024 DMP: 01 04 04 44 18 C6 66 BC F9
15:37:38.781 DMP: 01 04 04 44 1E 33 53 DB BF
15:37:39.028 DMP: 01 04 04 C1 92 66 66 CC 1F
15:37:39.289 DMP: 01 04 04 3F 7F E4 71 4D 6C
15:37:39.539 DMP: 01 04 04 42 47 CC CD CB 7C
15:37:39.803 DMP: 01 04 04 46 70 FE B7 EF 01
15:37:40.024 DMP: 01 04 04 46 6C 3D 27 7E 5B
[... 73 lines cut ...]
15:38:34.295 DMP: 01 04 04 43 98 32 0C 7A 8A
15:38:34.546 DMP: 01 04 04 45 BA BF 85 7E FE
15:38:34.797 DMP: 01 04 04 44 43 5F 0A A7 57
15:38:35.057 DMP: 01 04 04 00 00 00 00 FB 84
15:38:35.256 DMP: 01 04 04 43 6A 00 00 CE 1C
15:38:35.506 DMP: 01 04 04 40 42 E1 48 06 36
15:38:37.255 DMP: 01 04 04 44 35 66 66 54 F0
15:38:37.506 DMP: 01 04 04 44 35 85 D6 1C 74
15:38:37.760 DMP: 01 04 04 C1 B5 99 9A 3D A5
15:38:39.509 DMP: 01 04 04 3F 7F DD 1D 5F 11
15:38:39.768 DMP: 01 04 04 42 48 00 00 6F EA
15:38:43.018 DMP: 01 04 04 46 70 FE C4 AE E4
15:38:44.779 DMP: 01 04 04 46 6C 3D 33 7E 54
15:38:48.047 DMP: 01 04 04 43 98 32 0C 7A 8A
15:38:49.283 WIF: Checking connection...
15:38:49.797 DMP: 01 04 04 45 BA BF 85 7E FE
15:38:50.054 DMP: 01 04 04 44 43 5F 0A A7 57
15:38:51.774 DMP: 01 04 04 00 00 00 00 FB 84
15:38:52.078 CFG: Saved to flash at F9, Count 91, Bytes 4096
15:38:52.081 DMP: 01 04 04 43 6A 00 00 CE 1C
15:38:53.429 MQT: tele/PowCasaTH/STATE = {"Time":"2021-08-20T15:38:53","Uptime":"0T00:14:28","UptimeSec":868,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"TNT-iot","BSSId":"1A:E8:29:C1:8F:0D","Channel":1,"Mode":"11n","RSSI":76,"Signal":-62,"LinkCount":1,"Downtime":"0T00:00:03"}}
15:38:53.439 MQT: domoticz/in = {"idx":102,"nvalue":0,"svalue":"725;15119299.0","Battery":100,"RSSI":7}
15:38:53.444 MQT: tele/PowCasaTH/SENSOR = {"Time":"2021-08-20T15:38:53","ENERGY":{"TotalStartTime":"2021-08-20T00:00:00","Total":15119.299,"Yesterday":0.000,"Today":0.052,"ExportActive":304.391,"Period":42949632,"Power":726,"ApparentPower":726,"ReactivePower":-23,"Factor":1.00,"Frequency":50,"Voltage":234,"Current":3.045,"ImportActive":15119.300,"ImportReactive":5975.940,"ExportReactive":781.485,"PhaseAngle":0.00}}
15:38:53.799 DMP: 01 04 04 40 52 4D D3 3B 58
```

Jason2866, 2021-08-20:
Still software serial. You have to use on `GPIO1 SDMx20 TX` and `GPIO3 SDMx20 RX`

gajotnt, 2021-08-20:
Changed it, working like a charm. Thanks :) 🙏

closed 2021-08-20 by gajotnt