arendst/Tasmota #12935: Multiple DS18B20 sensors not supported in script

HMvdZijden, 2021-08-20:
### PROBLEM DESCRIPTION
In tasmota script on wemos D1 mini with 1 DS18B20 the section >T with temp=DS18B20#Temperature works, but when 2 sensors are connected they are visible on the main page as DS18B20-1 and DS18B20-2, but in the script  section >T with temp=DS18B20-1#Temperature nor temp2=DS18B20-2#Temperature do not work. no values come in vars they stay like initialized in the >D section..

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): Wemos D1 mini
- [x ] Tasmota binary firmware version number used: 9.5.0
  - [ ] Pre-compiled
  - [X ] Self-compiled
- [x ] Flashing tools used: tasmotizer
- [x ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
17:36:02.409 CMD: Backlog Template; Module; GPIO 255
17:36:02.515 MQT: TEST/radiator/radiator-keuken/stat/RESULT = {"NAME":"Radiator","GPIO":[1312,0,0,0,256,0,0,0,225,226,0,544,0,0],"FLAG":0,"BASE":18}
17:36:02.759 MQT: TEST/radiator/radiator-keuken/stat/RESULT = {"Module":{"0":"Radiator"}}
17:36:03.016 MQT: TEST/radiator/radiator-keuken/stat/RESULT = {"GPIO0":{"1312":"DS18x20"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"256":"Relay_i1"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"225":"Relay2"},"GPIO13":{"226":"Relay3"},"GPIO14":{"0":"None"},"GPIO15":{"544":"LedLink"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
no rules, script is to be used.
current script
>D
aanvoer=10

>S
if upsecs%60==0
then
print aanvoer %aanvoer%
endif

>T
aanvoer=DS18B20-1#Temperature

```
- [x ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
17:37:15.716 MQT: TEST/radiator/radiator-keuken/stat/STATUS = {"Status":{"Module":0,"DeviceName":"radiator-keuken","FriendlyName":["radiator-keuken","Tasmota2","Tasmota3"],"Topic":"radiator/radiator-keuken","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
17:37:15.753 MQT: TEST/radiator/radiator-keuken/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:20:09","StartupUTC":"2021-08-20T15:17:06","Sleep":50,"CfgHolder":4617,"BootCount":7,"BCResetTime":"2021-08-20T16:01:01","SaveCount":29,"SaveAddress":"FD000"}}
17:37:15.790 MQT: TEST/radiator/radiator-keuken/stat/STATUS2 = {"StatusFWR":{"Version":"9.5.0(tasmota)","BuildDateTime":"2021-08-20T14:01:11","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"447/699"}}
17:37:15.815 MQT: TEST/radiator/radiator-keuken/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["ZDNHA","ZDNPowerWifi"],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000080","00006000","00000000"]}}
17:37:15.847 MQT: TEST/radiator/radiator-keuken/stat/STATUS4 = {"StatusMEM":{"ProgramSize":656,"Free":1388,"Heap":15,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640E0","FlashFrequency":40,"FlashMode":3,"Features":["00000809","87DAC787","043E8001","000000CF","010013C0","C000F981","00004004","00201000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,50","Sensors":"1,2,3,4,5,6"}}
17:37:15.883 MQT: TEST/radiator/radiator-keuken/stat/STATUS5 = {"StatusNET":{"Hostname":"radiator-keuken","IPAddress":"192.168.2.224","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"5C:CF:7F:2B:B2:1C","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
17:37:15.906 MQT: TEST/radiator/radiator-keuken/stat/STATUS6 = {"StatusMQT":{"MqttHost":"zdnnas","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_2BB21C","MqttUser":"Herman","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
17:37:15.932 MQT: TEST/radiator/radiator-keuken/stat/STATUS7 = {"StatusTIM":{"UTC":"2021-08-20T15:37:15","Local":"2021-08-20T17:37:15","StartDST":"2021-03-28T03:00:00","EndDST":"2021-10-31T03:00:00","Timezone":99,"Sunrise":"06:23","Sunset":"20:45"}}
17:37:15.953 MQT: TEST/radiator/radiator-keuken/stat/STATUS10 = {"StatusSNS":{"Time":"2021-08-20T17:37:15","DS18B20-1":{"Id":"02161B9B71EE","Temperature":22.3},"DS18B20-2":{"Id":"041674C1FDFF","Temperature":22.1},"TempUnit":"C"}}
17:37:15.973 MQT: TEST/radiator/radiator-keuken/stat/STATUS11 = {"StatusSTS":{"Time":"2021-08-20T17:37:15","Uptime":"0T00:20:09","UptimeSec":1209,"Heap":15,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","Wifi":{"AP":2,"SSId":"ZDNPowerWifi","BSSId":"3C:A6:2F:45:74:ED","Channel":6,"Mode":"11n","RSSI":98,"Signal":-51,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

17:39:06.424 aanvoer 10.00
```

### TO REPRODUCE
_Steps to reproduce the behavior:_
na

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
the var aanvoer should get the temperature value from the sensor

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
na

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
So 1 sensor ok, multiple sensors do not work.

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2021-08-20:
You should have read the docs about scripting. To quote:

> Remark: json variable names (like all others) may not contain math operators like - , you should set SetOption64 1 to replace - (dash) with _ (underscore).

ascillato2, 2021-08-20:
Closing this issue as it has been answered.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2021-08-20 by ascillato2

HMvdZijden, 2021-08-20:
> Closing this issue as it has been answered.
> 
> ### Support Information
> See [Docs](https://tasmota.github.io/docs/) for more information.
> See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
> See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

> You should have read the docs about scripting. To quote:
> 
> > Remark: json variable names (like all others) may not contain math operators like - , you should set SetOption64 1 to replace - (dash) with _ (underscore).

I just found out, been searching all day, so easyly solved.
Thanks, closing the issue.