arendst/Tasmota #12938: IEM3000 Series wattmeter - invalid readings

kurkav, 2021-08-21:
### PROBLEM DESCRIPTION
I've connected the Tasmota to modbus of IEM3155 wattmeter and found several issues:
- I've got CRC error issue in log
- Power is showing kilowatts instead of watts (I see 2 watts instead of 2000 watts it is supposed to show)
- Total power consupmtion is wrong.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Wemos D1 Mini
- [x] Tasmota binary firmware version number used: Tasmota 9.5.0.6
  - [ ] Pre-compiled
  - [x] Self-compiled
- [y] Flashing tools used: Platformio and OTA upgrade
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
20:21:10.148 CMD: Backlog Template; Module; GPIO 255
20:21:10.150 SRC: WebConsole from 10.0.0.60
20:21:10.153 CMD: Grp 0, Cmnd 'BACKLOG', Idx 1, Len 26, Data 'Template; Module; GPIO 255'
20:21:10.165 SRC: Backlog
20:21:10.167 CMD: Grp 0, Cmnd 'TEMPLATE', Idx 1, Len 0, Data ''
20:21:10.174 MQT: stat/sensor/home/home/mainPowerMeter/RESULT = {"NAME":"Generic","GPIO":[0,1,1,1,1,1,1,1,1,1,1,5248,5280,1],"FLAG":0,"BASE":18}
20:21:10.392 SRC: Backlog
20:21:10.394 CMD: Grp 0, Cmnd 'MODULE', Idx 1, Len 0, Data ''
20:21:10.399 MQT: stat/sensor/home/home/mainPowerMeter/RESULT = {"Module":{"0":"Generic"}}
20:21:10.643 SRC: Backlog
20:21:10.646 CMD: Grp 0, Cmnd 'GPIO', Idx 1, Len 3, Data '255'
20:21:10.657 MQT: stat/sensor/home/home/mainPowerMeter/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"5248":"iEM3000 TX"},"GPIO5":{"5280":"iEM3000 RX"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"5248":"iEM3000 TX"},"GPIO16":{"5280":"iEM3000 RX"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

20:22:09.297 CMD: Backlog Rule1; Rule2; Rule3
20:22:09.300 SRC: WebConsole from 10.0.0.60
20:22:09.302 CMD: Grp 0, Cmnd 'BACKLOG', Idx 1, Len 19, Data 'Rule1; Rule2; Rule3'
20:22:09.315 SRC: Backlog
20:22:09.318 CMD: Grp 0, Cmnd 'RULE', Idx 1, Len 0, Data ''
20:22:09.325 MQT: stat/sensor/home/home/mainPowerMeter/RESULT = {"Command":"Unknown"}
20:22:09.540 SRC: Backlog
20:22:09.543 CMD: Grp 0, Cmnd 'RULE', Idx 2, Len 0, Data ''
20:22:09.548 MQT: stat/sensor/home/home/mainPowerMeter/RESULT = {"Command":"Unknown"}
20:22:09.791 SRC: Backlog
20:22:09.794 CMD: Grp 0, Cmnd 'RULE', Idx 3, Len 0, Data ''
20:22:09.800 MQT: stat/sensor/home/home/mainPowerMeter/RESULT = {"Command":"Unknown"}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Use Schneider IEM3155 wattmeter

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
No CRC error in output. Watts should be as watts. Total enegry should be calculated properly.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

kurkav, 2021-08-21:
I already have working code solving this issue.
To solve the CRC input buffer had to be expanded.
To solve the Power it had to be multiplied by 1000.
To solve the Total consumption it had to be read as float32 and not int64. (The float is also specified in documentation.)

ascillato, 2021-08-21:
> I already have working code solving this issue.

So, as you have stated the issue and the solution, please provide a pull request to incorporate that into Tasmota. You may want to add this support in a separated key (in my_user_config.h) so as to not brake backwards compatibility. Thanks.

ascillato2, 2021-08-21:
Moving this to discussions.

closed 2021-08-21 by ascillato2