arendst/Tasmota #12949: Permanent vars not right in script

HMvdZijden, 2021-08-23:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
When I define a var with p: in a script it gets unpredictable value
p:iv=30
>S
print iv %iv%

on console
11:46:49.975 iv -2779700264960.00

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): wemos d1 mini
- [x ] Tasmota binary firmware version number used: 9.5.0
  - [ ] Pre-compiled
  - [x ] Self-compiled
- [x ] Flashing tools used: tasmotizer
- [ x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here: 12:00:08.035 CMD: Backlog Template; Module; GPIO 255
12:00:08.101 MQT: TEST/radiator/radiator-keuken/stat/RESULT = {"NAME":"Radiator","GPIO":[1312,0,0,0,256,0,0,0,225,226,0,544,0,0],"FLAG":0,"BASE":18}
12:00:08.312 MQT: TEST/radiator/radiator-keuken/stat/RESULT = {"Module":{"0":"Radiator"}}
12:00:08.566 MQT: TEST/radiator/radiator-keuken/stat/RESULT = {"GPIO0":{"1312":"DS18x20"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"256":"Relay_i1"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"225":"Relay2"},"GPIO13":{"226":"Relay3"},"GPIO14":{"0":"None"},"GPIO15":{"544":"LedLink"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
12:01:05.122 CMD: Status 0
12:01:05.130 MQT: TEST/radiator/radiator-keuken/stat/STATUS = {"Status":{"Module":0,"DeviceName":"radiator-keuken","FriendlyName":["radiator-keuken","Tasmota2","Tasmota3"],"Topic":"radiator/radiator-keuken","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
12:01:05.145 MQT: TEST/radiator/radiator-keuken/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"External System","Uptime":"2T15:27:46","StartupUTC":"2021-08-20T18:33:19","Sleep":50,"CfgHolder":4617,"BootCount":6,"BCResetTime":"2021-08-20T17:12:13","SaveCount":87,"SaveAddress":"3FB000"}}
12:01:05.157 MQT: TEST/radiator/radiator-keuken/stat/STATUS2 = {"StatusFWR":{"Version":"9.5.0(tasmota)","BuildDateTime":"2021-08-20T18:26:34","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"447/699"}}
12:01:05.165 MQT: TEST/radiator/radiator-keuken/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["ZDNHA","ZDNPowerWifi"],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00004080","00006000","00000000"]}}
12:01:05.180 MQT: TEST/radiator/radiator-keuken/stat/STATUS4 = {"StatusMEM":{"ProgramSize":656,"Free":1388,"Heap":15,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640E0","FlashFrequency":40,"FlashMode":3,"Features":["00000809","87DAC787","043E8001","000000CF","010013C0","C000F981","00004004","00201000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,50","Sensors":"1,2,3,4,5,6"}}
12:01:05.190 MQT: TEST/radiator/radiator-keuken/stat/STATUS5 = {"StatusNET":{"Hostname":"radiator-keuken","IPAddress":"192.168.2.224","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"5C:CF:7F:2B:B2:1C","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
12:01:05.198 MQT: TEST/radiator/radiator-keuken/stat/STATUS6 = {"StatusMQT":{"MqttHost":"zdnnas","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_2BB21C","MqttUser":"Herman","MqttCount":2,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
12:01:05.209 MQT: TEST/radiator/radiator-keuken/stat/STATUS7 = {"StatusTIM":{"UTC":"2021-08-23T10:01:05","Local":"2021-08-23T12:01:05","StartDST":"2021-03-28T03:00:00","EndDST":"2021-10-31T03:00:00","Timezone":99,"Sunrise":"06:28","Sunset":"20:39"}}
12:01:05.219 MQT: TEST/radiator/radiator-keuken/stat/STATUS10 = {"StatusSNS":{"Time":"2021-08-23T12:01:05","DS18B20_9B71EE":{"Id":"02161B9B71EE","Temperature":23.4},"DS18B20_C1FDFF":{"Id":"041674C1FDFF","Temperature":22.8},"TempUnit":"C"}}
12:01:05.229 MQT: TEST/radiator/radiator-keuken/stat/STATUS11 = {"StatusSTS":{"Time":"2021-08-23T12:01:05","Uptime":"2T15:27:46","UptimeSec":228466,"Heap":15,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":2,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","Wifi":{"AP":1,"SSId":"ZDNHA","BSSId":"F4:F2:6D:6A:B0:3A","Channel":11,"Mode":"11n","RSSI":94,"Signal":-53,"LinkCount":2,"Downtime":"0T00:00:04"}}}

```
- [x ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
12:03:30.990 iv 195.00
12:03:31.959 iv 195.00

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Define one of the vars as p:

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
I expected a var with the right value which cannot be changed by the script.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
script:
>D
p:iv=30

>S
print iv %iv%

**(Please, remember to close the issue when the problem has been addressed)**

gemu2015, 2021-08-24:
permanent vars may no be preset in the >D section.
they are immediately overwritten by permanent values from flash.
they are meant to be set under program control and keep their values after subsequent reboots.
simply try to set a p var in the >B section, then remove the line and see that the value ist kept at the next reboot.

HMvdZijden, 2021-08-24:
So they cannot be initialized in the >D section, but can be initialized in
the >B section which runs every time the script is started.
I just wanted to be sure that no programming by script or otherwise could
change these values. But the >B and probably the >BS section too can?

Is it possible to define a var in >D without a value? Or do I have to, but
the >B section or the start of the script overwrites them?

It would be nice to mention that in the documentation.
even in the example they are used wrong.

Scripting Cookbook~
<https://tasmota.github.io/docs/Scripting-Language/#scripting-cookbook>Scripting
Language Example~
<https://tasmota.github.io/docs/Scripting-Language/#scripting-language-example>

a valid script must start with >D in the first line!
some samples still contain comment lines before >D. This is no longer valid!

**Actually this code is too large**. This is only meant to show some
of the possibilities

>D
; define all vars here
p:mintmp=10  (p:means permanent)
p:maxtmp=30

They are not initialized in the >B section, but also not used anywhere in
the script.
But thank you for the fast response.

gemu2015, 2021-08-24:
yes the docs do contain some erroneous infos. feel free to edit the docs and make it better. everyone can edit them.

simply setting a pvar in the >B section is not what you want since it shall hold some vars that are altered by the program and not always preset to default. 

my suggestion is to create a subroutine e.g. called #reset to set vars and call that if needed.

\>D
; declare pvars, zero is sufficient 
p:iv=0

\>B

; Subroutine to set pvars
#reset
print resetting vars
iv=30
...

\>S

and then call in console
script >=#reset

ascillato2, 2021-08-25:
Closing this issue as it has been answered.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2021-08-25 by ascillato2

Erickrk, 2022-11-09:
> 
Hey @gemu2015 you also mean that pvars can be initialized as zero at >D and then have their value updated at >B?

My issue now is that I want a variable to be changed remotly, e.g. via an MQTT event, while it keeps its value after reboot. I'm not sure if the value will be kept since I'm not updating it at boot. I'm going to give it a try.

gemu2015, 2022-11-15:
any p variable will keep its value after reboot. you may update at any time via MQTT or console cmd

Erickrk, 2022-11-15:
Yeah, I had the opportunity to test that. It isn't safe against power cutting though. It might be a good practice to use the 'svars' command once in a while inside the code.