arendst/Tasmota #12970: USE_ADC_VCC activates bogus energy monitoring?

AndreKR, 2021-08-24:
### PROBLEM DESCRIPTION
I have two Shelly 1 devices, both with Tasmota 9.5.0, but one is compiled **with** `USE_ADC_VCC` and uses this template:
```
{"NAME":"Shelly 1","GPIO":[0,0,0,0,224,192,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":46}
```
and the other one is compiled **without** `USE_ADC_VCC` and uses this template:
```
{"NAME":"Shelly 1","GPIO":[0,0,0,0,224,192,0,0,0,0,0,0,0,4704],"FLAG":0,"BASE":46}
```

The one without `USE_ADC_VCC` shows the ADC input, as expected:

[image]

But the other one shows energy measurements, which of course are all 0 because the Shelly 1 does not have energy monitoring nor is it enabled:

[image]

It is also discovered as an energy monitoring device in Home Assistant.

I think the only difference between the two devices is the value of `USE_ADC_VCC`.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Shelly 1
- [X] Tasmota binary firmware version number used: 9.5.0(TasmoCompiler-esp8266generic)
  - [ ] Pre-compiled
  - [X] Self-compiled
- [X] Flashing tools used: OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
23:58:52.495 MQT: stat/tasmota_D80FCE/RESULT = {"NAME":"Shelly 1","GPIO":[0,0,0,0,224,192,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":46}
23:58:52.705 MQT: stat/tasmota_D80FCE/RESULT = {"Module":{"0":"Shelly 1"}}
23:58:52.960 MQT: stat/tasmota_D80FCE/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"224":"Relay1"},"GPIO5":{"192":"Switch_n1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [X] Provide the output of this command: `Status 0`:
```lua
23:59:34.110 MQT: stat/tasmota_D80FCE/STATUS0 = {"Status":{"Module":0,"DeviceName":"Amplifier","FriendlyName":["Amplifier"],"Topic":"tasmota_D80FCE","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0},"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:12:19","StartupUTC":"2021-08-24T22:47:15","Sleep":50,"CfgHolder":4617,"BootCount":15,"BCResetTime":"2021-08-24T03:40:06","SaveCount":30,"SaveAddress":"F9000"},"StatusFWR":{"Version":"9.5.0(TasmoCompiler-esp8266generic)","BuildDateTime":"2021-08-22T01:09:54","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"389/699"},"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Fotzelschnitte",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000080","00006000","00000000"]},"StatusMEM":{"ProgramSize":590,"Free":412,"Heap":26,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"15701C","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FC04587","04268001","1400000F","410013C0","C800EDC1","08244004","00001000","00000020"],"Drivers":"1,2,3,4,6,8,9,10,12,16,18,19,22,24,26,27,29,30,35,37,45","Sensors":"1,3"},"StatusNET":{"Hostname":"tasmota_D80FCE-4046","IPAddress":"192.168.42.117","Gateway":"192.168.42.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.42.1","Mac":"E8:DB:84:D8:0F:CE","Webserver":2,"WifiConfig":4,"WifiPower":17.0},"StatusMQT":{"MqttHost":"192.168.42.5","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_D80FCE","MqttUser":"tasmota","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4},"StatusTIM":{"UTC":"2021-08-24T22:59:34","Local":"2021-08-24T23:59:34","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"05:56","Sunset":"19:48"},"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0},"StatusSNS":{"Time":"2021-08-24T23:59:34","Switch1":"OFF","ENERGY":{"TotalStartTime":"2021-08-24T03:40:06","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":0,"Current":0.000},"TIC":{}},"StatusSTS":{"Time":"2021-08-24T23:59:34","Uptime":"0T00:12:19","UptimeSec":739,"Vcc":3.499,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":27,"MqttCount":1,"POWER":"ON","Wifi":[redacted]}}```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
Compile Tasmota with `USE_ADC_VCC` and apply the Shelly 1 template. I don't know if it is necessary to do this on an actual Shelly 1 device ( I don't know what triggers the energy monitoring functionality to activate).

### EXPECTED BEHAVIOUR
The device home page only shows the Device type, its name, the on/off state and the buttons. No energy monitoring output.

### SCREENSHOTS
See above.

### ADDITIONAL CONTEXT
-

AndreKR, 2021-08-24:
It turns out `USE_ADC_VCC` is **not** the culprit.

I have just flashed the affected device with a firmware that doesn't have `USE_ADC_VCC` enabled and now the analog input is shown as expected, but the bogus energy monitoring is *still* shown:

[image]

[image]

Not sure what is going on.

ascillato, 2021-08-25:
Please, do a `reset 6` on that device.

AndreKR, 2021-08-25:
[image]

ascillato, 2021-08-25:
Please, try to update that device to latest Tasmota of the development branch:

In the console type:
```lua
OTAurl http://ota.tasmota.com/tasmota/tasmota.bin.gz
Upgrade 1
```

ascillato, 2021-08-25:
Also, after all that, please, set weblog to 4 and then restart the device. Please, copy the content of the console. Thanks.

AndreKR, 2021-08-25:
The problem doesn't happen with the official development image. It does not happen with the official release image (9.5.0) either.

It only happens with my image from TasmoCompiler. Here's my `user_config_override.h`:
<details><summary>user_config_override.h</summary>

```c
#ifndef _USER_CONFIG_OVERRIDE_H_
#define _USER_CONFIG_OVERRIDE_H_

#ifdef CODE_IMAGE_STR
  #undef CODE_IMAGE_STR
#endif
#define CODE_IMAGE_STR "TasmoCompiler-esp8266generic"

#ifdef USE_MHZ19
  #undef USE_MHZ19
#endif

#ifdef USE_SENSEAIR
  #undef USE_SENSEAIR
#endif

#ifdef USE_PMS5003
  #undef USE_PMS5003
#endif

#ifdef USE_MGS
  #undef USE_MGS
#endif

#ifdef USE_NOVA_SDS
  #undef USE_NOVA_SDS
#endif

#ifdef USE_SGP30
  #undef USE_SGP30
#endif

#ifdef USE_CCS811
  #undef USE_CCS811
#endif

#ifdef USE_SCD30
  #undef USE_SCD30
#endif

#ifdef USE_SPS30
  #undef USE_SPS30
#endif

#ifdef USE_HPMA
  #undef USE_HPMA
#endif

#ifdef USE_IAQ
  #undef USE_IAQ
#endif

#ifdef USE_EMULATION
  #undef USE_EMULATION
#endif

#ifdef USE_EMULATION_HUE
  #undef USE_EMULATION_HUE
#endif

#ifdef USE_EMULATION_WEMO
  #undef USE_EMULATION_WEMO
#endif

[... 455 lines cut ...]
02:39:54.053 MQT: Subscribe to cmnd/tasmota_D80FCE/#
02:39:54.054 MQT: Subscribe to cmnd/tasmotas/#
02:39:54.057 MQT: Subscribe to cmnd/DVES_D80FCE_fb/#
02:39:54.059 MQT: Unsubscribe from homeassistant/status
02:39:54.064 MQT: tele/tasmota_D80FCE/INFO1 = {"Info1":{"Module":"Shelly 1","Version":"9.5.0(TasmoCompiler-esp8266generic)","FallbackTopic":"cmnd/DVES_D80FCE_fb/","GroupTopic":"cmnd/tasmotas/"}}
02:39:54.075 MQT: tele/tasmota_D80FCE/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota_D80FCE-4046","IPAddress":"192.168.42.117"}}
02:39:54.089 MQT: tele/tasmota_D80FCE/INFO3 = {"Info3":{"RestartReason":"Software/System restart"}}
02:39:54.096 MQT: stat/tasmota_D80FCE/RESULT = {"POWER":"ON"}
02:39:54.100 MQT: stat/tasmota_D80FCE/POWER = ON
02:39:55.376 SRC: MQTT
02:39:55.378 CMD: Grp 0, Cmnd 'STATE', Idx 1, Len 0, Data ''
02:39:55.385 MQT: stat/tasmota_D80FCE/RESULT = {"Time":"2021-08-25T02:39:55","Uptime":"0T00:00:19","UptimeSec":19,"Heap":29,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":33,"MqttCount":1,"POWER":"ON","Wifi":[redacted]}
02:39:55.451 SRC: MQTT
02:39:55.453 CMD: Grp 0, Cmnd 'STATUS', Idx 1, Len 2, Data '10'
02:39:55.498 MQT: stat/tasmota_D80FCE/STATUS10 = {"StatusSNS":{"Time":"2021-08-25T02:39:55","Switch1":"OFF","ANALOG":{"A0":45},"ENERGY":{"TotalStartTime":"2021-08-25T01:27:43","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":0,"Current":0.000},"TIC":{}}}
02:39:58.381 MQT: tele/tasmota_D80FCE/STATE = {"Time":"2021-08-25T02:39:58","Uptime":"0T00:00:22","UptimeSec":22,"Heap":29,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":52,"MqttCount":1,"POWER":"ON","Wifi":[redacted]}
02:39:58.445 MQT: tele/tasmota_D80FCE/SENSOR = {"Time":"2021-08-25T02:39:58","Switch1":"OFF","ANALOG":{"A0":47},"ENERGY":{"TotalStartTime":"2021-08-25T01:27:43","Total":0.000,"Yesterday":0.000,"Today":0.000,"Period":0,"Power":0,"Current":0.000},"TIC":{}}
02:40:01.781 WIF: Checking connection...
02:40:03.391 MQT: tasmota/discovery/E8DB84D80FCE/config = {"ip":"192.168.42.117","dn":"Amplifier","fn":["Amplifier",null,null,null,null,null,null,null],"hn":"tasmota_D80FCE-4046","mac":"E8DB84D80FCE","md":"Shelly 1","ty":0,"if":0,"ofln":"Offline","onln":"Online","state":["OFF","ON","TOGGLE","HOLD"],"sw":"9.5.0","t":"tasmota_D80FCE","ft":"%prefix%/%topic%/","tp":["cmnd","stat","tele"],"rl":[1,0,0,0,0,0,0,0],"swc":[-1,-1,-1,-1,-1,-1,-1,-1],"swn":[null,null,null,null,null,null,null,null],"btn":[0,0,0,0,0,0,0,0],"so":{"4":0,"11":0,"13":0,"17":0,"20":0,"30":0,"68":0,"73":0,"82":0,"114":0,"117":0},"lk":1,"lt_st":0,"sho":[0,0,0,0],"ver":1} (retained)
02:40:03.442 MQT: tasmota/discovery/E8DB84D80FCE/sensors = {"sn":{"Time":"2021-08-25T02:40:03","Switch1":"OFF","ANALOG":{"A0":42},"ENERGY":{"TotalStartTime":"2021-08-25T01:27:43","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":0,"Current":0.000},"TIC":{}},"ver":1} (retained)
02:40:22.036 WIF: Checking connection...
02:40:42.298 WIF: Checking connection...
02:41:02.274 WIF: Checking connection...
```
</details>

ascillato, 2021-08-25:
Ok

In your custom build you are enabling:

#define USE_ENERGY_DUMMY

So that is why you have that.

Don't define that and you won't get the dummy energy consumption.

ascillato2, 2021-08-25:
Closing this issue as it has been answered.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2021-08-25 by ascillato2

AndreKR, 2021-08-25:
I don't think I have control over that in TasmoCompiler, I have opened https://github.com/benzino77/tasmocompiler/issues/219.

ascillato, 2021-08-25:
I have tested on Tasmocompiler with selecting use_vcc and don't selecting energy_monitoring and the resulting firmware is ok. It doesn't have energy dummy sensor. 

For this device, you should compile without adding energy monitoring due to your device don't have that and you don't want that neither.

ascillato, 2021-08-25:
When selecting ENERGY MEASUREMENTS, Tasmocompiler adds also ENERGY DUMMY SENSOR. This is expected and correct.
If you don't need and you don't want energy measurements (neither with a power measurement chip or by a dummy sensor) you should not select energy measurements in Tasmocompiler.

Energy Dummy will count power consumption using a fixed voltage value + a fixed current value and the time the relay is ON.
Those values are configured in the same way as the calibration of the power measurement chip (please, see the docs for that).

So, this is neither a Tasmota bug nor a Tasmocompiler bug. Just disable what you don't need. Thanks.

AndreKR, 2021-08-25:
I see. My intention was to have a firmware that *can* do everything I need, but it will only actually do it when it is configured to do so.

ascillato, 2021-08-25:
In that case, In Tasmocompiler select ENERGY MONITORING and then add the following in the CUSTOM PARAMETER section:

#undef USE_ENERGY_DUMMY