arendst/Tasmota #13005: TLS MQTT Frequent reconnects (9.3.1, 9.5.0, 9.5.0.7)

mrkeuz, 2021-08-28:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): esp32doit-devkit-v1
- [x] Tasmota binary firmware version number used: 9.3.1, 9.5.0, 9.5.0.7 (latest `development`)
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: pio
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:

```lua
  Configuration output here:
{"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"640":"I2C SDA1"},"GPIO22":{"608":"I2C SCL1"},"GPIO23":{"269":"Relay_i14"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO28":{"0":"None"},"GPIO29":{"0":"None"},"GPIO30":{"0":"None"},"GPIO31":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
21:13:56.568 CMD: Backlog Rule1; Rule2; Rule3
21:13:56.569 SRC: WebConsole from 192.168.1.249
21:13:56.570 CMD: Grp 0, Cmnd 'BACKLOG', Idx 1, Len 19, Data 'Rule1; Rule2; Rule3'
21:13:56.589 SRC: Backlog
21:13:56.590 CMD: Grp 0, Cmnd 'RULE', Idx 1, Len 0, Data ''
21:13:56.601 MQT: stat/mpu_242284/RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
21:13:56.813 SRC: Backlog
21:13:56.815 CMD: Grp 0, Cmnd 'RULE', Idx 2, Len 0, Data ''
21:13:56.826 MQT: stat/mpu_242284/RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
21:13:57.062 SRC: Backlog
21:13:57.064 CMD: Grp 0, Cmnd 'RULE', Idx 3, Len 0, Data ''
21:13:57.076 MQT: stat/mpu_242284/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
WebConsole from 192.168.1.249
21:12:17.947 CMD: Grp 0, Cmnd 'STATUS', Idx 1, Len 1, Data '0'
21:12:17.956 MQT: stat/mpu_242284/STATUS = {"Status":{"Module":0,"DeviceName":"mpu","FriendlyName":["Tasmota"],"Topic":"mpu_242284","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
21:12:17.968 MQT: stat/mpu_242284/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T02:50:15","StartupUTC":"2021-08-28T17:22:02","Sleep":50,"CfgHolder":4617,"BootCount":2,"BCResetTime":"2021-08-28T18:22:08","SaveCount":19}}
21:12:17.980 MQT: stat/mpu_242284/STATUS2 = {"StatusFWR":{"Version":"9.5.0.7(tasmota)","BuildDateTime":"2021-08-28T20:20:32","Core":"1_0_7_3","SDK":"v3.3.5-138-g714f27f56","CpuFrequency":240,"Hardware":"ESP32-D0WDQ6","CR":"411/699"}}
21:12:17.988 MQT: stat/mpu_242284/STATUS3 = {"StatusLOG":{"SerialLog":4,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["<MY_SSID>",""],"TelePeriod":60,"Resolution":"5581C0C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000080","00206000","00000000"]}}
21:12:18.002 MQT: stat/mpu_242284/STATUS4 = {"StatusMEM":{"ProgramSize":1312,"Free":1856,"Heap":160,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":3,"Features":["00000809","9FDAC7C7","00152001","B7F7BFCF","05DA9BC8","60360DC7","000840D2","20200000","0000002D"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,17,20,21,24,26,27,29,34,38,50,52","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,32,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,127"}}
21:12:18.014 MQT: stat/mpu_242284/STATUS5 = {"StatusNET":{"Hostname":"mpu-242284-0644","IPAddress":"192.168.1.98","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"1.1.1.1","Mac":"<MAC_HERE>","Webserver":2,"HTTP_API":0,"WifiConfig":4,"WifiPower":4.8}}
21:12:18.023 MQT: stat/mpu_242284/STATUS6 = {"StatusMQT":{"MqttHost":"ha.islife.ru","MqttPort":8883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_242284","MqttUser":"curious_fox","MqttCount":1562,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
21:12:18.037 MQT: stat/mpu_242284/STATUS7 = {"StatusTIM":{"UTC":"2021-08-28T20:12:18","Local":"2021-08-28T21:12:18","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"06:01","Sunset":"19:40"}}
21:12:18.067 MQT: stat/mpu_242284/STATUS10 = {"StatusSNS":{"Time":"2021-08-28T21:12:18","MPU6050":{"Temperature":24.6,"AccelXAxis":492.00,"AccelYAxis":206.00,"AccelZAxis":154.00,"GyroXAxis":-38.00,"GyroYAxis":56.00,"GyroZAxis":122.00,"Yaw":-38.79,"Pitch":-61.38,"Roll":-22.31},"ESP32":{"Temperature":43.3},"Global":{"Temperature":24.6},"TempUnit":"C"}}
21:12:18.070 BRY: GC from 5347 to 2539 bytes (in 1 ms)
21:12:18.083 MQT: stat/mpu_242284/STATUS11 = {"StatusSTS":{"Time":"2021-08-28T21:12:18","Uptime":"0T02:50:16","UptimeSec":10216,"Heap":165,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1562,"POWER":"ON","Wifi":{"AP":1,"SSId":"<MY_SSID>","BSSId":"<MY_MAC>","Channel":9,"Mode":"11n","RSSI":52,"Signal":-74,"LinkCount":1,"Downtime":"0T00:00:05"}}}
21:12:22.123 MQT: tele/mpu_242284/STATE = {"Time":"2021-08-28T21:12:22","Uptime":"0T02:50:20","UptimeSec":10220,"Heap":168,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1562,"POWER":"ON","Wifi":{"AP":1,"SSId":"7STAGE","BSSId":"B8:69:F4:3C:1A:A5","Channel":9,"Mode":"11n","RSSI":58,"Signal":-71,"LinkCount":1,"Downtime":"0T00:00:05"}}
21:12:22.149 MQT: tele/mpu_242284/SENSOR = {"Time":"2021-08-28T21:12:22","MPU6050":{"Temperature":24.6,"AccelXAxis":501.00,"AccelYAxis":221.00,"AccelZAxis":160.00,"GyroXAxis":-38.00,"GyroYAxis":56.00,"GyroZAxis":122.00,"Yaw":-38.86,"Pitch":-61.40,"Roll":-22.31},"ESP32":{"Temperature":43.3},"Global":{"Temperature":24.6},"TempUnit":"C"}

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
...
14:53:13.819 SRC: MQTT
14:53:13.821 CMD: Grp 0, Cmnd 'STATE', Idx 1, Len 0, Data ''
14:53:13.833 MQT: Message too large 
...

```
[... 65 lines cut ...]
### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

Stable MQTT connect. 

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
As mentions @s-hadinger in (#12992) as well as it affect 9.3.1 I have created this issue.

Mosquitto 2.0.11 behind proxy (via traefik) with let's enrypt cert.

Mosquitto log:
```
...
1630159174: Client DVES_242284 disconnected: No error information.
1630159175: New connection from 172.20.2.3:57604 on port 1883.
1630159175: New client connected from 172.20.2.3:57604 as DVES_242284 (p2, c1, k30, u'<user>').
...
```

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2021-08-28:
Thanks. The error message comes from here https://github.com/arendst/Tasmota/blob/debef0641326bf717218e0f99dd5639f14044b73/tasmota/xdrv_02_9_mqtt.ino#L506

The bytes sent do not match the message size. The message is misleading - not necessarily too large. 

What are the Tasmota logs before this message?

mrkeuz, 2021-08-28:
Yes, I found too. I try to log what is message size is  and message was 300 bytes. Not so big i think.

Logs are next:
```
22:49:00.126 MQT: MFLN not supported by TLS server
22:49:00.127 MQT: Connected
22:49:00.144 MQT: tele/mpu_242284/LWT = Online (retained)
22:49:00.162 MQT: cmnd/mpu_242284/POWER = 
22:49:00.164 MQT: Subscribe to cmnd/mpu_242284/#
22:49:00.176 MQT: Subscribe to cmnd/tasmotas/#
22:49:00.194 MQT: Subscribe to cmnd/DVES_242284_fb/#
22:49:00.196 MQT: Unsubscribe from homeassistant/status
22:49:00.210 RSL: config = {"ip":"192.168.1.98","dn":"mpu","fn":["Tasmota",null,null,null,null,null,null,null],"hn":"mpu-242284-0644","mac":"246F28242284","md":"MPU6050_Config","ty":0,"if":0,"ofln":"Offline","onln":"Online","state":["OFF","ON","TOGGLE","HOLD"],"sw":"9.5.0.7","t":"mpu_242284","ft":"%prefix%/%topic%/","tp":["cmnd","stat","tele"],"rl":[1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"swc":[-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1],"swn":[null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null],"btn":[0,0,0,0,0,0,0,0],"so":{"4":0,"11":0,"13":0,"17":0,"20":0,"30":0,"68":0,"73":0,"82":0,"114":0,"117":0},"lk":0,"lt_st":0,"sho":[0,0,0,0],"ver":1}
22:49:00.232 RSL: sensors = {"sn":{"Time":"2021-08-28T22:49:00","MPU6050":{"Temperature":24.6,"AccelXAxis":513.00,"AccelYAxis":221.00,"AccelZAxis":167.00,"GyroXAxis":100.00,"GyroYAxis":34.00,"GyroZAxis":-111.00,"Yaw":100.34,"Pitch":-61.01,"Roll":-22.00},"ESP32":{"Temperature":43.9},"Global":{"Temperature":24.6},"TempUnit":"C"},"ver":1}
22:49:00.850 MQT: Attempting connection...
22:49:01.331 MQT: TLS connected in 479 ms, stack low mark 2472
22:49:01.332 MQT: MFLN not supported by TLS server
22:49:01.332 MQT: Connected
22:49:01.348 MQT: tele/mpu_242284/LWT = Online (retained)
22:49:01.367 MQT: cmnd/mpu_242284/POWER = 
22:49:01.368 MQT: Subscribe to cmnd/mpu_242284/#
22:49:01.380 MQT: Subscribe to cmnd/tasmotas/#
22:49:01.392 MQT: Subscribe to cmnd/DVES_242284_fb/#
22:49:01.404 MQT: Unsubscribe from homeassistant/status
22:49:01.447 MQT: tasmota/discovery/246F28242284/config = {"ip":"192.168.1.98","dn":"mpu","fn":["Tasmota",null,null,null,null,null,null,null],"hn":"mpu-242284-0644","mac":"246F28242284","md":"MPU6050_Config","ty":0,"if":0,"ofln":"Offline","onln":"Online","state":["OFF","ON","TOGGLE","HOLD"],"sw":"9.5.0.7","t":"mpu_242284","ft":"%prefix%/%topic%/","tp":["cmnd","stat","tele"],"rl":[1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"swc":[-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1],"swn":[null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null],"btn":[0,0,0,0,0,0,0,0],"so":{"4":0,"11":0,"13":0,"17":0,"20":0,"30":0,"68":0,"73":0,"82":0,"114":0,"117":0},"lk":0,"lt_st":0,"sho":[0,0,0,0],"ver":1} (retained)
22:49:01.483 MQT: tasmota/discovery/246F28242284/sensors = {"sn":{"Time":"2021-08-28T22:49:01","MPU6050":{"Temperature":24.6,"AccelXAxis":519.00,"AccelYAxis":215.00,"AccelZAxis":184.00,"GyroXAxis":100.00,"GyroYAxis":34.00,"GyroZAxis":-111.00,"Yaw":100.32,"Pitch":-61.00,"Roll":-21.99},"ESP32":{"Temperature":43.9},"Global":{"Temperature":24.6},"TempUnit":"C"},"ver":1} (retained)
22:49:18.852 WIF: Checking connection...
22:49:32.103 MQT: Attempting connection...
22:49:32.581 MQT: TLS connected in 475 ms, stack low mark 2472
22:49:32.582 MQT: MFLN not supported by TLS server
22:49:32.582 MQT: Connected
22:49:32.598 MQT: tele/mpu_242284/LWT = Online (retained)
22:49:32.617 MQT: cmnd/mpu_242284/POWER = 
22:49:32.618 MQT: Subscribe to cmnd/mpu_242284/#
22:49:32.630 MQT: Subscribe to cmnd/tasmotas/#
22:49:32.643 MQT: Subscribe to cmnd/DVES_242284_fb/#
22:49:32.654 MQT: Unsubscribe from homeassistant/status
22:49:32.950 MQT: tasmota/discovery/246F28242284/config = {"ip":"192.168.1.98","dn":"mpu","fn":["Tasmota",null,null,null,null,null,null,null],"hn":"mpu-242284-0644","mac":"246F28242284","md":"MPU6050_Config","ty":0,"if":0,"ofln":"Offline","onln":"Online","state":["OFF","ON","TOGGLE","HOLD"],"sw":"9.5.0.7","t":"mpu_242284","ft":"%prefix%/%topic%/","tp":["cmnd","stat","tele"],"rl":[1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"swc":[-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1],"swn":[null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null],"btn":[0,0,0,0,0,0,0,0],"so":{"4":0,"11":0,"13":0,"17":0,"20":0,"30":0,"68":0,"73":0,"82":0,"114":0,"117":0},"lk":0,"lt_st":0,"sho":[0,0,0,0],"ver":1} (retained)
22:49:32.987 MQT: tasmota/discovery/246F28242284/sensors = {"sn":{"Time":"2021-08-28T22:49:32","MPU6050":{"Temperature":24.7,"AccelXAxis":508.00,"AccelYAxis":208.00,"AccelZAxis":164.00,"GyroXAxis":100.00,"GyroYAxis":34.00,"GyroZAxis":-112.00,"Yaw":100.20,"Pitch":-61.00,"Roll":-21.97},"ESP32":{"Temperature":43.9},"Global":{"Temperature":24.7},"TempUnit":"C"},"ver":1} (retained)
22:49:33.795 SRC: MQTT
22:49:33.797 CMD: Grp 0, Cmnd 'STATE', Idx 1, Len 0, Data ''
22:49:33.810 MQT: Message too large
22:49:33.811 RSL: RESULT = {"Time":"2021-08-28T22:49:33","Uptime":"0T00:14:04","UptimeSec":844,"Heap":167,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":36,"MqttCount":202,"POWER":"OFF","Wifi":{"AP":1,"SSId":"7STAGE","BSSId":"<mac>","Channel":9,"Mode":"11n","RSSI":46,"Signal":-77,"LinkCount":1,"Downtime":"0T00:00:05"}}
22:49:33.813 BRY: GC from 4562 to 2213 bytes (in 0 ms)
22:49:34.343 MQT: Attempting connection...
````

mrkeuz, 2021-08-28:
Also, sometimes "Message too large" not appear (see logs above). So maybe it reconnect not only on write packet?

mrkeuz, 2021-08-28:
I can add insert some debug messages where you want for more information.

mrkeuz, 2021-08-29:
What I found at this time. If I understand things correctly.

While `MqttCheck()` that happens every several milliseconds (if I understand by schedule), `MqttCheck` checks underling `BearSSL` (`_client.connected()`) and `BearSSL` reports that not connected by some reason. After several tries (`Mqtt.retry_counter`) MQTT doing reconnect. 

Also, I enable `DEBUG_TLS`, `DEBUG_ESP_SSL` and see  `_run_until: Not connected` in serial on every reconnect.

s-hadinger, 2021-08-29:
I suspect a buffer overflow on the response from the Mqtt server. The current buffer is 1024 bytes. Unless the server supports MFLN (which it does not according to logs) the TLS connection will drop. 
You can change it here https://github.com/arendst/Tasmota/blob/debef0641326bf717218e0f99dd5639f14044b73/tasmota/xdrv_02_9_mqtt.ino#L38
It is the second parameter that you can change to 2048

mrkeuz, 2021-08-29:
I don't use USE_MQTT_AZURE_IOT flag. So executes exactly from `ESP32` define block:
https://github.com/arendst/Tasmota/blob/03b26adf07b6ee5f42a9511eaae6fece3948cb0d/tasmota/xdrv_02_9_mqtt.ino#L214
```c
...
 if (Mqtt.mqtt_tls) {
#ifdef ESP32
    tlsClient = new BearSSL::WiFiClientSecure_light(2048,2048);
#else // ESP32 - ESP8266
    tlsClient = new BearSSL::WiFiClientSecure_light(1024,1024);
#endif
...
```
I debug now it calls constructor exactly with 2048. 

ALSO
I debug code every night :) and found next thing. 

From debug prints I see chaotically call a `BearSsl.stop()` and `BearSsl._freeSSL()` methods. For most these calls this calls I found cause:

Here logs using `ArduinoTrace` lib (lines can not match with `development`) but logic calls you can understand:

```
PubSubClient.cpp:818: boolean PubSubClient::connected()
WiFiClientSecureLightBearSSL.cpp:292: virtual void BearSSL::WiFiClientSecure_light::stop()
WiFiClientSecureLightBearSSL.cpp:333: void BearSSL::WiFiClientSecure_light::_freeSSL()
```
```
xdrv_02_9_mqtt.ino:1018: void MqttReconnect()
WiFiClientSecureLightBearSSL.cpp:292: virtual void BearSSL::WiFiClientSecure_light::stop()
WiFiClientSecureLightBearSSL.cpp:333: void BearSSL::WiFiClientSecure_light::_freeSSL()
```
This ok for now, But for next call I CANNOT found the caller, I insert debug prints in before every `stop()` call that I found, (even in `#define` statement), but not luck, log is next:
```
09:47:33.094 MQT: Connected
> Heap _write = 2472
> Heap _write = 2472
09:47:33.104 MQT: tele/mpu_242284/LWT = Online (retained)
> Heap _write = 2472
09:47:33.108 MQT: cmnd/mpu_242284/POWER = 
09:47:33.112 MQT: Subscribe to cmnd/mpu_242284/#
> Heap _write = 2472
09:47:33.119 MQT: Subscribe to cmnd/tasmotas/#
> Heap _write = 2472
09:47:33.124 MQT: Subscribe to cmnd/DVES_242284_fb/#
> Heap _write = 2472
09:47:33.130 MQT: Unsubscribe from homeassistant/status
> Heap _write = 2472
> Heap _write = 2472
WiFiClientSecureLightBearSSL.cpp:292: virtual void BearSSL::WiFiClientSecure_light::stop()
WiFiClientSecureLightBearSSL.cpp:333: void BearSSL::WiFiClientSecure_light::_freeSSL()
...

So, as you can see, is last example that NOT function that who call `stop()`. I don't know how in `arduino` show stack trace easily for investigate this.

Maybe it helps for analysis.

s-hadinger, 2021-08-29:
Hmmm. One option would be to run debug mode with a jtag probe using ft233h. 
Or generate a crash, but that could not be the one you are looking for. 
Maybe there is a simple way to dump stacktrace with Arduino calls.

mrkeuz, 2021-08-29:
Well... I'm almost porting this solution from `esp-idf` from this answer
https://www.esp32.com/viewtopic.php?t=18964

But it prints only memory pointers (as I understand) like this, but unfortunately without func names:

Stack trace for 5 depth:
```
Backtrace:
0x401221C5:0x3FFB1C30 
0x40106CF9:0x3FFB1C50 
0x40106E60:0x3FFB1D30 
0x40114B98:0x3FFB1D50 
0x40114DE7:0x3FFB1F70 
0x40114E2F:0x3FFB1F90  |<-CONTINUES

```
Will try to find method for decode it to function names. I don't know how to do this yet. I found '.map' file in `pio` build folder for my firmware, but simple search not give results.

s-hadinger, 2021-08-29:
Great. I just typically look in the map file in what function the address is. Just do a search a remove the last digits one by one until you get a match and then scan through the addresses visually

joba-1, 2021-08-29:
…and then you go on reading the compiled code in hex… :D

s-hadinger, 2021-08-29:
Well. Not that much. But you can disassemble the firmware if you like :)

joba-1, 2021-08-29:
comes with time...

but until then, I found this might help (https://esp32.com/viewtopic.php?t=5579#p24263)

I found the addr2line easy to use and helpful in the past

```
Re: Function call backtracing with esp-idf
Postby markwj » Thu May 03, 2018 1:16 am

What you want is not trivial. But, here are some notes which may help:

1. There are no symbols in the .bin file in flash. But, if your running App outputs something that looks like an address, the 'make monitor' part of esp idf will convert it to a symbol and show it.

2. As an alternative to [1], the xtensa tools have a utility to convert an address to a symbol. It is not very accurate on the line number, but close enough:
CODE: SELECT ALL

xtensa-esp32-elf-addr2line -pfiaC -e build/<MYAPP>.elf 0x<ADDRESS>
3. See panic.c doBacktrace() for how to do a backtrace.

4. If you can trigger the issue yourself, then gdb is your friend. Make menuconfig and set the gdbstub option. Use abort() in your code, and you'll be in gdb with a full backtrace and examination capability.

There is also jtag...
```

mrkeuz, 2021-08-29:
Wow! Thanks for advices!

I also found python script (that works over `xtensa`)
https://github.com/janLo/EspArduinoExceptionDecoder

Tomorrow will continue debugging. I learn so many new things in short time. :)
Thanks for support!

If there are any ideas what could be the problem I would be very grateful. 
I can provide some additional information or try any recipes.

s-hadinger, 2021-09-01:
I've got some good news. I installed Mosquitto 2.0.11 and was able to reproduce the issue. It really smells like a race condition.

I followed your track, and here is my stacktrace with corresponding functions.

```
Backtrace:
0x40102A3B:0x3FFB1940   _ZN7BearSSL22WiFiClientSecure_lightD2Ev             BearSSL::WiFiClientSecure_light::~WiFiClientSecure_light()
0x400F95FD:0x3FFB1960   _ZN10WiFiClient5writeEPKhj                          WiFiClient::flush()
0x40102F89:0x3FFB1980   _ZN7BearSSL22WiFiClientSecure_light5flushEv         BearSSL::WiFiClientSecure_light::flush()
0x40102F41:0x3FFB1A90   _ZN7BearSSL22WiFiClientSecure_light6_writeEPKhjb    BearSSL::WiFiClientSecure_light::_write(unsigned char const*, unsigned int, bool)
0x40102F6E:0x3FFB19D0   _ZN7BearSSL22WiFiClientSecure_light5writeEPKhj      BearSSL::WiFiClientSecure_light::write(unsigned char const*, unsigned int)
0x4011A3DA:0x3FFB19F0   _ZN12PubSubClient5writeEhPht                        PubSubClient::write(unsigned char, unsigned char*, unsigned short)
0x4011A5DD:0x3FFB1A10   _ZN12PubSubClient9subscribeEPKc                     PubSubClient::subscribe(char const*)
0x4011A5ED:0x3FFB1A30   _ZN12PubSubClient9subscribeEPKc                     PubSubClient::subscribe(char const*)
0x400DD460:0x3FFB1A50   _Z16MqttSubscribeLibPKc                             MqttSubscribeLib(char const*)
0x400DD495:0x3FFB1A70   _Z13MqttSubscribePKc                                MqttSubscribe(char const*)
0x400F2F6E:0x3FFB1A90   _Z13MqttConnectedv                                  MqttConnected()
0x400F3472:0x3FFB1C10   _Z9MqttCheckv                                       MqttCheck()
0x400F3513:0x3FFB1D30   _Z9MqttCheckv                                       MqttCheck()
0x400F8BE9:0x3FFB1D50   _Z16Every250mSecondsv                               Every250mSeconds()
0x400F8E53:0x3FFB1F70   _Z9Schedulerv                                       Scheduler()
0x400F8E9B:0x3FFB1F90   _Z4loopv                                            loop()
0x4011EAA1:0x3FFB1FB0   _Z8loopTaskPv                                       loopTask(void*)
0x4008DFCA:0x3FFB1FD0   vPortTaskWrapper                                    vPortTaskWrapper
```

The call to `stop()` came from `WiFiClient::flush()`. It appears that `flush()` also flushes the receive buffers and first calls `available()` to know how many bytes are waiting to arrive, then read all bytes. But due to inheritance problems, `WiFiClient::flush()` actually called `BearSSL::WiFiClientSecure_light::available()` instead of `WiFiClient::available()` and got the value of the TLS receive buffer.
`Flush()` then tried to read bytes that weren't in the wifi buffer, and caused an error 11 `no more processes` (don't know what it means) and stopped the wifi connection.

I'm trying the fix now, but actually `BearSSL::WiFiClientSecure_light` shouldn't call `flush()` at all because it does not want the input buffer to be emptied.

The patch is so simple... Don't call `WiFiClient::flush()` in the first place. I'm testing now.

closed 2021-09-01 by s-hadinger

mrkeuz, 2021-09-01:
@s-hadinger Already build and setup. Seems all work fine!!! ))) 

Yesterday I also found this `WiFiClient::flush()` call. But I gigged to debugging and spend many hours to try to understand things under the hood (I'm not c/c++ developer). Insistently wanted to grasp and understand. Now I can breathe out calmly.

Thank you very much! You made my day!!! 🚀🚀🚀

PR #13033 (2021-09-01): Fix ESP32 TLS reconnects
## Description:

See here for details: https://github.com/arendst/Tasmota/issues/13005#issuecomment-910507479

`WiFiClient::flush()` has a different behavior on ESP32 compared to ESP8266. On ESP8266 it makes sure all outgoing packets have been acknowledged. On ESP32, it empties the input buffer, creating MQTT protocol errors. On top of that the call to `WiFiClient::available()` is actually sent to `BearSSL::WiFiClientSecure_light` and causes internal error and disconnects the TCP connection.

Simple fix: don't call `WiFiClient::flush()` on ESP32

**Related issue (if ap [... cut]