arendst/Tasmota #13008: [AC Frequency Detection Switch] Relays outputs toggles erratically

Youpimatin, 2021-08-30:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Relays outputs toggles erratically when using "AC Frequency Detection Switch"

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4) (No discord account)
- [x] Device used (e.g., Sonoff Basic): Modified LoraTap SC500W
- [x] Tasmota binary firmware version number used: 9.5.0.7
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: OTA update (Tasmotizer on first time)
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
10:08:54.734 RSL: RESULT = {"NAME":"Test","GPIO":[0,0,0,576,160,161,0,0,224,32,225,0,0,0],"FLAG":0,"BASE":18}
10:08:54.940 RSL: RESULT = {"Module":{"0":"Test"}}
10:08:55.196 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"576":"LedLink_i"},"GPIO4":{"160":"Switch1"},"GPIO5":{"161":"Switch2"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"32":"Button1"},"GPIO14":{"225":"Relay2"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
(No rules)

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
10:12:59.712 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota",""],"Topic":"tasmota_F526B7","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
10:12:59.718 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:21:56","StartupUTC":"2021-08-30T08:51:03","Sleep":50,"CfgHolder":4617,"BootCount":80,"BCResetTime":"2021-08-19T04:11:20","SaveCount":604,"SaveAddress":"F9000"}}
10:12:59.724 RSL: STATUS2 = {"StatusFWR":{"Version":"9.5.0.7(tasmota)","BuildDateTime":"2021-08-29T14:59:09","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"357/699"}}
10:12:59.729 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Home","Test"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000088","00006000","00004000"]}}
10:12:59.739 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":610,"Free":392,"Heap":26,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
10:12:59.744 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-F526B7-1719","IPAddress":"192.168.1.245","Gateway":"192.168.1.49","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"C4:4F:33:F5:57:E7","Webserver":2,"HTTP_API":1,"WifiConfig":2,"WifiPower":17.0}}
10:12:59.748 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_F526B7","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
10:12:59.753 RSL: STATUS7 = {"StatusTIM":{"UTC":"2021-08-30T09:12:59","Local":"2021-08-30T10:12:59","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"06:04","Sunset":"19:36"}}
10:12:59.758 RSL: STATUS10 = {"StatusSNS":{"Time":"2021-08-30T10:12:59","Switch1":"OFF","Switch2":"OFF"}}
10:12:59.764 RSL: STATUS11 = {"StatusSTS":{"Time":"2021-08-30T10:12:59","Uptime":"0T00:21:56","UptimeSec":1316,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":2,"SSId":"IRIS-R&D","BSSId":"12:5E:A2:00:AF:4C","Channel":11,"Mode":"11n","RSSI":64,"Signal":-68,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
00:00:00.001 HDW: ESP8285
00:00:00.050 CFG: Loaded from flash at F7, Count 606
00:00:00.056 QPC: Count 1
00:00:00.058 CFG: CR 357/699, Busy 0
00:00:00.061 ROT: Mode 1
00:00:00.063 SRC: Restart
00:00:00.065 Project tasmota - Tasmota Version 9.5.0.7(tasmota)-2_7_4_9(2021-08-29T14:59:09)
00:00:00.217 WIF: Checking connection...
00:00:00.217 WIF: Attempting connection...
00:00:00.555 WIF: Connecting to AP2 Test Channel 11 BSSId 12:5E:A2:00:AF:4C in mode 11n as tasmota-F526B7-1719...
00:00:01.418 WIF: Checking connection...
00:00:01.418 WIF: Attempting connection...
00:00:02.551 WIF: Checking connection...
00:00:02.551 WIF: Attempting connection...
00:00:03.501 WIF: Checking connection...
00:00:03.502 WIF: Connected
00:00:03.752 HTP: Web server active on tasmota-F526B7-1719 with IP address 192.168.1.245
00:00:04.328 NTP: Synch time...
00:00:04.403 RTC: UTC 2021-08-30T09:19:38, DST 2021-03-28T02:00:00, STD 2021-10-31T03:00:00
10:19:38.000 NTP: Synched
10:19:38.354 RSL: INFO1 = {"Info1":{"Module":"Test","Version":"9.5.0.7(tasmota)","FallbackTopic":"cmnd/DVES_F526B7_fb/","GroupTopic":"cmnd/tasmotas/"}}
10:19:38.359 RSL: INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota-F526B7-1719","IPAddress":"192.168.1.245"}}
10:19:38.369 RSL: INFO3 = {"Info3":{"RestartReason":"External System"}}
10:19:38.376 RSL: RESULT = {"POWER1":"OFF"}
10:19:38.379 RSL: POWER1 = OFF
10:19:38.382 RSL: RESULT = {"POWER2":"OFF"}
10:19:38.386 RSL: POWER2 = OFF
10:19:40.378 QPC: Reset
10:19:40.384 SRC: Switch
10:19:40.387 RSL: RESULT = {"POWER1":"ON"}
10:19:40.389 RSL: POWER1 = ON
10:19:40.433 SRC: Switch
10:19:40.435 RSL: RESULT = {"POWER1":"OFF"}
10:19:40.437 RSL: POWER1 = OFF
10:19:42.334 APP: Boot Count 82
10:19:43.372 CFG: Saved to flash at F6, Count 607, Bytes 4096
10:19:43.377 RSL: STATE = {"Time":"2021-08-30T10:19:43","Uptime":"0T00:00:11","UptimeSec":11,"Heap":29,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":2,"SSId":"IRIS-R&D","BSSId":"20:E5:2A:00:FA:C4","Channel":11,"Mode":"11n","RSSI":60,"Signal":-70,"LinkCount":1,"Downtime":"0T00:00:05"}}
10:19:43.401 RSL: SENSOR = {"Time":"2021-08-30T10:19:43","Switch1":"ON","Switch2":"OFF"}
10:19:43.412 SRC: Switch
10:19:43.415 RSL: RESULT = {"POWER1":"ON"}
10:19:43.417 RSL: POWER1 = ON
10:19:43.511 SRC: Switch
10:19:43.514 RSL: RESULT = {"POWER1":"OFF"}
10:19:43.516 RSL: POWER1 = OFF

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
- Use a hardware with "AC Frequency Detection Switch"
- Set the physical mains switch to ON (0-3.3V square 50Hz pulses goes to the switch input of the Tasmotized hardware)
- Power ON the Tasmotized hardware (or restart it from the web menu)
It connects to the Wifi, and then the relays outputs toggles 2 or 3 times
(Relays outputs toggles also AFTER mains switch goes from OFF to ON and corresponding relay went ON)

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Relays outputs should not toggle multiple times at power up
Relays outputs should be stable after a state change

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
I cheched with an isolated oscilloscope if the 50Hz at the input is continuous : it is, there is no drop at all, so it's really a software issue.
Maybe it's in the Switchdebounce section of the switch code, when writing in flash or when the wifi chip is communicating and Tasmota gets longer periods without "CPU time" than the switchdebounce time.
But if I use "AC Frequency Detection Switch", it's for the expected faster response time compared to the use of a resistor bridge and capacitor filtering to the logic input (which lags too much for me)

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2021-08-30:
Have you tried different `SwitchDebounce` settings? See similar device https://templates.blakadder.com/moes-MS-104B.html

Youpimatin, 2021-08-30:
I've tried 49 (40ms) to 209 (200ms) SwitchDebounce settings.
It seems to work when it is set to 140 ms and up (but then it annihilates my attempt to reduce switching lag)

I also tried different Sleep values, from 12 ms and up with no result for this problem (but the response seems faster then).

Here is my hardware for the reference :

Jason2866, 2021-08-30:
So the issue head line is not correct. FR: faster response.

Youpimatin, 2021-08-30:
Hum, in https://tasmota.github.io/docs/Commands/#switchdebounce it's stated that "If the AC frequency is 50 Hz, SwitchDebounce 69 will turn on the switch after three pulses and off after three missing one"
and "For example: SwitchDebounce 69 will turn the switch on after three pulses and turn it off after three missing ones (3 * 20 msec is 60 and the last digit must be 9 to activate the AC detection). You will probably have to experiment with the values depending on your AC frequency and the devices frequency detection implementation."

But whith the stated value of 60ms, Tasmota behaves erratically...

Youpimatin, 2021-08-30:
More information :
I used "SaveData 60" to delay the flash write after startup, here is the startup log :

```lua
00:00:02.004 HTP: Web server active on tasmota-F526B7-1719 with IP address 192.168.1.245
00:00:02.402 NTP: Synch time...
00:00:02.674 RTC: UTC 2021-08-30T15:34:46, DST 2021-03-28T02:00:00, STD 2021-10-31T03:00:00
16:34:46.001 NTP: Synched
16:34:47.029 RSL: INFO1 = {"Info1":{"Module":"Test","Version":"9.5.0.7(tasmota)","FallbackTopic":"cmnd/DVES_F526B7_fb/","GroupTopic":"cmnd/tasmotas/"}}
16:34:47.034 RSL: INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota-F526B7-1719","IPAddress":"192.168.1.245"}}
16:34:47.044 RSL: INFO3 = {"Info3":{"RestartReason":"Software/System restart"}}
16:34:47.051 RSL: RESULT = {"POWER1":"OFF"}
16:34:47.055 RSL: POWER1 = OFF
16:34:47.058 RSL: RESULT = {"POWER2":"OFF"}
16:34:47.061 RSL: POWER2 = OFF
16:34:47.982 HTP: Main Menu
16:34:49.264 HTP: Console
```
Nothing wrong until there, where there is something that makes the software switch input to toggle 2 times rapidly :
```lua
16:34:50.460 QPC: Reset
16:34:50.466 SRC: Switch
16:34:50.469 RSL: RESULT = {"POWER2":"ON"}
16:34:50.471 RSL: POWER2 = ON
16:34:50.515 SRC: Switch
16:34:50.518 RSL: RESULT = {"POWER2":"OFF"}
16:34:50.519 RSL: POWER2 = OFF
16:34:51.419 RSL: STATE = {"Time":"2021-08-30T16:34:51","Uptime":"0T00:00:09","UptimeSec":9,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":26,"MqttCount":0,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":2,"SSId":"IRIS-R&D","BSSId":"20:E5:2A:00:FA:C4","Channel":11,"Mode":"11n","RSSI":80,"Signal":-60,"LinkCount":1,"Downtime":"0T00:00:03"}}
16:34:51.442 RSL: SENSOR = {"Time":"2021-08-30T16:34:51","Switch1":"OFF","Switch2":"ON"}
16:34:52.390 APP: Boot Count 96
16:35:06.761 WIF: Checking connection...
16:35:27.027 WIF: Checking connection...
```
And there, the Flash write makes the software switch input to toggle 2 times rapidly again :
```lua
16:35:45.202 CFG: Saved to flash at F6, Count 743, Bytes 4096
16:35:45.211 SRC: Switch
16:35:45.214 RSL: RESULT = {"POWER2":"ON"}
16:35:45.216 RSL: POWER2 = ON
16:35:45.260 SRC: Switch
16:35:45.263 RSL: RESULT = {"POWER2":"OFF"}
16:35:45.265 RSL: POWER2 = OFF
16:35:47.265 WIF: Checking connection...
```

my setup was "Sleep 50" and "SwitchDebounce 89" (80ms)

It seems to me that when the Tasmota loop routine can't execute for a longer time than "SwitchDebounce",  it produces these glitches (only when there is AC Frequency at the input pin)

stefanbode, 2021-09-06:
From my experience with fast signals, I do not see a chance to get this avoided all the time. The 50ms reaction time is something the firmware cannot guarantee by far. Works in most cases but as you mentioned not all the time. WIFI, Saving and Network transmissions (especially when the network is weak) give you losses. I had my fun with the shutter control. In the end, I had to accept that in rare situations I get a small overrun. I assume the debounce has the same problem. I did not get time to do the work and when there is far too much time passed by.... I would assume that the 140ms is more or less the minimum where you get back control. You can play with the CPU frequency and should see that the problem between 80MHz and 160MHz should double. Alternatively change to ESP32. I would assume with 2 cores and more power this will get the job done in-time.

Youpimatin, 2021-09-07:
> 
> 
> The 50ms reaction time is something the firmware cannot guarantee by far. Works in most cases but as you mentioned not all the time. 

It's not the problem, here.
The problem is that the relays toggles without a reason (MAYBE because there is no measure for a certain time)
Even if there is no measure, and the reaction time is longer sometimes, the relays should not toggle rapidly !
Port polling should resume and take the time needed to be sure of the input state, not presume that the input in in a certain state, when the state is uncertain.

stefanbode, 2021-09-07:
If you use Pulses to keep the light on. And then these pulses are not kept because the device was busy the light switches off. And immediate after 3 pulses on again. There is no count of the off stages. It is just a time compare after it get idle time. 

Again. No chance to fix it in my mind

closed 2021-09-21 by ascillato2