arendst/Tasmota #13012: VINDRIKTNING does not show correct values

ikarisan, 2021-08-30:
### PROBLEM DESCRIPTION
I compiled Tasmota from the development branch to use my VINDRIKTNING sensor (03b26adf07b6ee5f42a9511eaae6fece3948cb0d) but it seems that the values are read the wrong way. 

I used a Saleae clone (8CH, 24MHz) logic analyzer and I got:

16 11 0B 00 00 00 18 00 00 02 E8 00 00 00 17 01 00 00 00 B4

That should result in a PM2.5 value of 24µg/m³. 
But Tasmota reports 1174µg/m³ for PM1.0 and 0µg/m³ for PM2.5 and PM10!

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Ikea VINDRIKTNING
- [X] Tasmota binary firmware version number used: 9.5.0.7
  - [ ] Pre-compiled
  - [X] Self-compiled
- [X] Flashing tools used: Platform.IO + VS Code 2019
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
Compile Tasmota from origin/development branch (28. August, 03b26adf07b6ee5f42a9511eaae6fece3948cb0d) and flash it to an Wemos D1 device. Connect the Wemos D1 to the VINDRIKTNING and watch the read out values. 

### EXPECTED BEHAVIOUR
Tasmote should show 24µg/m³ for PM2.5 and not 1174µg/m³ for PM1.0 and 0 for PM2.5 and PM10

### SCREENSHOTS
Captured UART values:
[image]

What Tasmota reports:
[image]

### ADDITIONAL CONTEXT
...

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2021-08-30:
What are correct values? From a user in Discord. It seems working
[image]

blakadder, 2021-08-30:
[image]
perhaps

linuxrrze, 2021-08-30:
> What are correct values? From a user in Discord.
> Same values for 2.5 and 10. Coincedence?

Values are very close - but differ on rare occasions.

As I have no idea what realistic values are, I just used the PM1006 specs for reference (and yes, the mentioned "Send string" differs on the IKEA version, so maybe other values also vary):

https://pdf.directindustry.com/pdf/cubic-sensor-instrument-co-ltd/pm1006k-led-particle-sensor-module/54752-927719.html

arendst, 2021-08-30:
I've seen other implementations using an average of 5 measurements. They also provide only one value as the other two do not seem to be valid values.

blakadder, 2021-08-30:
There's a lenghty discussion on the validity of the other two measurements and additional confusion with multiple datasheets of PM1006 sensors depending on where they're sold

Jason2866, 2021-08-31:
As Theo already wrote the sensor is only for PM 2.5 rated. https://en.gassensor.com.cn/ParticulateMatterSensor/info_itemid_105.html

arendst, 2021-08-31:
From what I receive

[image]

It's safe to say the VINDIKTNING hardware only supports PM2.5. The other values just don't make sense (way too high or almost equal to 2.5). I propose to delete them from the output.

Regarding received data, enabling ``weblog 4`` will dump the received data. No logic analyzer needed.

15:41:52.442 DMP: 16 11 0B 00 00 00 10 00 00 04 8D 00 00 00 10 01 00 00 00 1C
15:41:54.457 DMP: 16 11 0B 00 00 00 10 00 00 04 8C 00 00 00 10 01 00 00 00 1D
15:41:56.458 DMP: 16 11 0B 00 00 00 10 00 00 04 8B 00 00 00 10 01 00 00 00 1E
15:41:58.415 DMP: 16 11 0B 00 00 00 0F 00 00 04 89 00 00 00 0F 01 00 00 00 22
15:42:00.423 DMP: 16 11 0B 00 00 00 0F 00 00 04 88 00 00 00 0F 01 00 00 00 23

From the data, the possible datasheet and other programmers it seems the relevant data is in byte 5 and 6 (counted from 0). In this case it's around 16.

So for now I'll remove the invalid values for PM1.0 and PM10.

Youpimatin, 2021-08-31:
Ikea product page only talks about PM 2.5 indeed.

I got one from IKEA, I intend to run it along with Tasmota, with a few improvements :
- 3 LEDs (drived in PWM) to RC filters to determine which one is lit up.
- Use Front light sensor (PD1) voltage to the analog input to get a measure of ambiant light
- Reconnect the Fan- pin to a PWM output to enable Fan speed variation (between 1.7V and 5V... or maybe between 0V and 3.3V)

I can't wait for the PM sensor readings to be debugged... tell me if I can help.

ikarisan, 2021-08-31:
That's because you datasheet references the PM1006K sensor and not the used PM1006. :) 

> As I have no idea what realistic values are, I just used the PM1006 specs for reference (and yes, the mentioned "Send string" differs on the IKEA version, so maybe other values also vary):

ikarisan, 2021-08-31:
> so there are more than one use for certain pins... and maybe pins used as inputs AND as outputs, I'll see...

Someome reverse engineered the pcb: https://threadreaderapp.com/thread/1415291684569632768.html

Jason2866, 2021-08-31:
Why checking the leds color? The values are known when the different colors lights up.
PWM mod is easy since you just need to cut a PCB trace.
[image]

Youpimatin, 2021-09-01:
Thanks for the reverse engineered source... appreciated (I saw it on the late, but it confims my analysis)

- Yes LEDs color is a function of the measured value, but I think there is a slight hysteresys, it was to get no mismatch between HA and the physical device (if needed)... maybe overthinked.
- I tried PWM to drive the fan yesterday night, but as the PWM is running at about 1kHz, it causes an audible hum, and it is worse than simply drive the fan from the 3.3V rail, so I will keep it simple (maybe drive the fan with an ON/OFF "Relay" output)
- A wire from the photodiode, with a 200kOhm in series (to feed 5V, not 3.3V max) through the ADC pin works like a charm.
(there is also a little hysteresys from the light sensor to the bright/night LEDs mode)

Jason2866, 2021-09-01:
Pwm frequency can be changed with `PwmFrequency <freq>`

closed 2021-09-01 by Jason2866

joba-1, 2021-09-01:
but max pwm frequency is 1k for an esp8266 and lower usually means even more audible

Jason2866, 2021-09-01:
@joba-1 should be 4khz https://github.com/arendst/Tasmota/blob/development/tasmota/tasmota.h#L141

joba-1, 2021-09-01:
maybe, have not tested it. But then the espressif datasheet lies (happens…)

12.1.4. Parameter Specification
…
• PWM period: 1000 μs (1 KHz) ~ 10000 μs (100 Hz)

Jason2866, 2021-09-01:
Funny, since PWM is done in software with the ESP8266 (the big disadvantage of the chip), so the datasheet is...
The truth is here https://github.com/tasmota/Arduino/blob/2.7.4.9/cores/esp8266/core_esp8266_wiring_pwm.cpp#L36-L44