arendst/Tasmota #13042: Trigger condition in rule is wrong when using one-shot feature

Tidle999, 2021-09-02:
### PROBLEM DESCRIPTION
When using the example ruleto set PowerDelta regarding to power consumption its not working when enable once.

For Example:

`Backlog PowerDelta 0; Rule1 0; Rule1 5

Rule1
  ON ENERGY#Power>=35 DO Backlog PowerDelta 10; Status 8 BREAK 
  ON ENERGY#Power>=15 DO Backlog PowerDelta 25; Status 8 BREAK 
  ON ENERGY#Power>5 DO Backlog PowerDelta 35; Status 8 BREAK 
  ON ENERGY#Power<=5 DO PowerDelta 100 ENDON

Rule1 1`

Creating this rule is NOT setting the PowerDelta to 10, although powerconsumption is higher than 35w.
But when disabling once (Rule1 4) its working and PowerDelta will be set correct.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ X] Device used (e.g., Sonoff Basic): Blitzwolf SHP6
- [ X] Tasmota binary firmware version number used: 9.5
  - [X ] Pre-compiled
  - [ ] Self-compiled
- [ X] Flashing tools used: already flashed
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
`13:04:36.738 MQT: stat/tasmota_610E24/RESULT = {"PowerDelta1":5}
13:04:36.990 MQT: stat/tasmota_610E24/STATUS8 = {"StatusSNS":{"Time":"2021-09-02T13:04:36","ENERGY":{"TotalStartTime":"2021-08-16T08:53:32","Total":28.987,"Yesterday":2.082,"Today":1.122,"Power":121,"ApparentPower":146,"ReactivePower":83,"Factor":0.82,"Voltage":234,"Current":0.626}}}
13:04:37.284 RUL: ENERGY#POWER>=100 performs "Backlog PowerDelta 5; Status 8"
13:04:37.328 MQT: stat/tasmota_610E24/RESULT = {"PowerDelta1":5}
13:04:37.580 MQT: stat/tasmota_610E24/STATUS8 = {"StatusSNS":{"Time":"2021-09-02T13:04:37","ENERGY":{"TotalStartTime":"2021-08-16T08:53:32","Total":28.987,"Yesterday":2.082,"Today":1.122,"Power":120,"ApparentPower":146,"ReactivePower":84,"Factor":0.82,"Voltage":234,"Current":0.626}}}
13:04:37.886 RUL: ENERGY#POWER>=100 performs "Backlog PowerDelta 5; Status 8"
13:04:37.928 MQT: stat/tasmota_610E24/RESULT = {"PowerDelta1":5}
13:04:37.976 CMD: Rule1 5
13:04:37.984 MQT: stat/tasmota_610E24/RESULT = {"Rule1":{"State":"ON","Once":"ON","StopOnError":"OFF","Length":296,"Free":215,"Rules":"ON ENERGY#Power>=100 DO Backlog PowerDelta 5; Status 8 BREAK    ON ENERGY#Power>=35 DO Backlog PowerDelta 10; Status 8 BREAK    ON ENERGY#Power>=15 DO Backlog PowerDelta 20; Status 8 BREAK    ON ENERGY#Power>8 DO Backlog PowerDelta 25; Status 8 BREAK    ON ENERGY#Power<=8 DO PowerDelta 102 ENDON"}}
13:04:38.444 MQT: stat/tasmota_610E24/STATUS8 = {"StatusSNS":{"Time":"2021-09-02T13:04:38","ENERGY":{"TotalStartTime":"2021-08-16T08:53:32","Total":28.987,"Yesterday":2.082,"Today":1.122,"Power":121,"ApparentPower":146,"ReactivePower":83,"Factor":0.82,"Voltage":234,"Current":0.625}}}
13:04:39.037 RUL: ENERGY#POWER>=100 performs "Backlog PowerDelta 5; Status 8"
13:04:39.078 MQT: stat/tasmota_610E24/RESULT = {"PowerDelta1":5}
13:04:39.283 MQT: stat/tasmota_610E24/STATUS8 = {"StatusSNS":{"Time":"2021-09-02T13:04:39","ENERGY":{"TotalStartTime":"2021-08-16T08:53:32","Total":28.987,"Yesterday":2.082,"Today":1.122,"Power":120,"ApparentPower":146,"ReactivePower":83,"Factor":0.82,"Voltage":234,"Current":0.625}}}
13:04:39.642 RUL: ENERGY#POWER>=35 performs "Backlog PowerDelta 10; Status 8"
13:04:39.681 MQT: stat/tasmota_610E24/RESULT = {"PowerDelta1":10}
13:04:39.933 MQT: stat/tasmota_610E24/STATUS8 = {"StatusSNS":{"Time":"2021-09-02T13:04:39","ENERGY":{"TotalStartTime":"2021-08-16T08:53:32","Total":28.987,"Yesterday":2.082,"Today":1.122,"Power":121,"ApparentPower":146,"ReactivePower":82,"Factor":0.83,"Voltage":234,"Current":0.624}}}
13:04:40.245 RUL: ENERGY#POWER>=15 performs "Backlog PowerDelta 20; Status 8"
13:04:40.280 MQT: stat/tasmota_610E24/RESULT = {"PowerDelta1":20}
13:04:40.535 MQT: stat/tasmota_610E24/STATUS8 = {"StatusSNS":{"Time":"2021-09-02T13:04:40","ENERGY":{"TotalStartTime":"2021-08-16T08:53:32","Total":28.987,"Yesterday":2.082,"Today":1.122,"Power":120,"ApparentPower":146,"ReactivePower":82,"Factor":0.83,"Voltage":234,"Current":0.624}}}
13:04:40.819 RUL: ENERGY#POWER>8 performs "Backlog PowerDelta 25; Status 8"
13:04:40.850 MQT: stat/tasmota_610E24/RESULT = {"PowerDelta1":25}
13:04:41.298 MQT: stat/tasmota_610E24/STATUS8 = {"StatusSNS":{"Time":"2021-09-02T13:04:41","ENERGY":{"TotalStartTime":"2021-08-16T08:53:32","Total":28.987,"Yesterday":2.082,"Today":1.122,"Power":120,"ApparentPower":145,"ReactivePower":81,"Factor":0.83,"Voltage":234,"Current":0.620}}}
13:04:59.823 CMD: Backlog Template; Module; GPIO 255
13:05:00.115 MQT: stat/tasmota_610E24/RESULT = {"NAME":"BlitzWolf SHP6","GPIO":[320,0,321,0,2656,2720,0,0,2624,32,0,224,0,0],"FLAG":0,"BASE":45}
13:05:00.339 MQT: stat/tasmota_610E24/RESULT = {"Module":{"0":"BlitzWolf SHP6"}}
13:05:00.596 MQT: stat/tasmota_610E24/RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"0":"None"},"GPIO2":{"321":"Led_i2"},"GPIO3":{"0":"None"},"GPIO4":{"2656":"HLWBL CF1"},"GPIO5":{"2720":"BL0937 CF"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"2624":"HLWBL SEL_i"},"GPIO13":{"32":"Button1"},"GPIO14":{"0":"None"},"GPIO15":{"224":"Relay1"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}`

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
`13:06:22.176 MQT: stat/tasmota_610E24/RESULT = {"Rule1":{"State":"ON","Once":"ON","StopOnError":"OFF","Length":296,"Free":215,"Rules":"ON ENERGY#Power>=100 DO Backlog PowerDelta 5; Status 8 BREAK    ON ENERGY#Power>=35 DO Backlog PowerDelta 10; Status 8 BREAK    ON ENERGY#Power>=15 DO Backlog PowerDelta 20; Status 8 BREAK    ON ENERGY#Power>8 DO Backlog PowerDelta 25; Status 8 BREAK    ON ENERGY#Power<=8 DO PowerDelta 102 ENDON"}}
13:06:22.406 MQT: stat/tasmota_610E24/RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
13:06:22.655 MQT: stat/tasmota_610E24/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}`

```
[... 10 lines cut ...]
13:06:56.346 MQT: stat/tasmota_610E24/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.21","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_610E24","MqttUser":"DVES_USER","MqttCount":14,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
13:06:56.370 MQT: stat/tasmota_610E24/STATUS7 = {"StatusTIM":{"UTC":"2021-09-02T12:06:56","Local":"2021-09-02T13:06:56","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"06:08","Sunset":"19:30"}}
13:06:56.392 MQT: stat/tasmota_610E24/STATUS9 = {"StatusPTH":{"PowerDelta":[25,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
13:06:56.418 MQT: stat/tasmota_610E24/STATUS10 = {"StatusSNS":{"Time":"2021-09-02T13:06:56","ENERGY":{"TotalStartTime":"2021-08-16T08:53:32","Total":28.992,"Yesterday":2.082,"Today":1.126,"Power":120,"ApparentPower":145,"ReactivePower":82,"Factor":0.82,"Voltage":233,"Current":0.622}}}
13:06:56.442 MQT: stat/tasmota_610E24/STATUS11 = {"StatusSTS":{"Time":"2021-09-02T13:06:56","Uptime":"12T19:59:34","UptimeSec":1108774,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":21,"MqttCount":14,"POWER":"ON","Wifi":{"AP":1,"SSId":"@home","BSSId":"00:11:32:72:07:F7","Channel":8,"Mode":"11n","RSSI":54,"Signal":-73,"LinkCount":1,"Downtime":"0T00:00:03"}}}`
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Tidle999, 2021-09-02:
i guess rule example has to be changed from

``Backlog PowerDelta 0; Rule1 0; Rule1 5

Rule1
ON ENERGY#Power>=35 DO Backlog PowerDelta 10; Status 8 BREAK
ON ENERGY#Power>=15 DO Backlog PowerDelta 25; Status 8 BREAK
ON ENERGY#Power>5 DO Backlog PowerDelta 35; Status 8 BREAK
ON ENERGY#Power<=5 DO PowerDelta 100 ENDON

Rule1 1``

to

`Backlog PowerDelta 0; Rule1 0; Rule1 5

Rule1

ON ENERGY#Power<5 DO PowerDelta 100; Status 8 BREAK 
ON ENERGY#Power>=5 DO Backlog PowerDelta 35; Status 8 BREAK
ON ENERGY#Power>=15 DO Backlog PowerDelta 25; Status 8 BREAK
ON ENERGY#Power>=35 DO Backlog PowerDelta 10 ENDON

Rule1 1`

meingraham, 2021-09-02:
@Tidle999 

I think your proposed rule change would result in the wrong outcome. In your sequence, the second trigger would fire once when the `Power` is 5 or greater and set `PowerDelta` to 35. The rule would then not trigger again until the `Power` drops below 5. Your third & fourth trigger would never be triggered.

sfromis, 2021-09-02:
Looks like your expectation for one-shot is for it to cover the whole rule set, which it is not. Instead, it is for each individual condition, and the energy triggers will be checked about every second. This means that when the first condition has been eliminated by one-shot, the next condition in the rule set will get the opportunity. Thus, with a stable power value of 40, the first 3 triggers will fire, one every cycle. Not what you envisioned.

What I'd recommend is to simply not use such rules, but instead something like `PowerDelta 105` which will result in reporting if the energy consumption level changes by 5W or more. For most purposes, this works better than trying to change the percentage change by consumption level.

Tidle999, 2021-09-02:
You are right. I tried it for a few hours now and it’s not working like expected. But to set powerdelta to a fixed value of 102 or whatever is also no solution for me. 
when power consumption is below 10w a change of 1-2w might be interesting. But if it’s above 100 or even more i don’t Care about 1-2w. Instead it should be at least 5 or even more. 
the rule in the example does not work. Mine is also not working. Do you have another idea how to change the ruleset so it meets my expectations?

Jason2866, 2021-09-03:
Closing since it is answered

closed 2021-09-03 by Jason2866