arendst/Tasmota #13079: Sonoff ZHABridge becomes unresponsive after around 1h

stic82, 2021-09-06:
Hi,

i was using the Sonoff Bridge very successfully over the last year. 
But yesterday the configuration was lost.

Then i made a new configuration.
Installed Tasmota 9.5.0  with ncp-uart-sw_6.7.8_115200.ota.

Everything is running again but only for 1 hour. Then i need to restart the zigbee Bridge and it is working fine.
It is loosing MQTT connection and reconnecting fails.
At same time it starts to scan for Wifi Networks and did not connect.

closed 2021-09-06 by close-issue-app[bot]

MattWestb, 2021-09-12:
SonOff ZBB shall ruing on EZSP 6.7.9.0 then its have some critical fixes that the chip is needing.

AdrinkBeer, 2021-09-30:
Hi @stic82 , still having this issue ? 
I think I'm experiencing the same

stic82, 2021-10-01:
hi, i have downgraded to Tasmota 9.2.0 and ncp-uart-sw_6.5.5_115200.ota. no issues anymore.